networkupstools/nut #2467: Trying to upgrade on RPi

Hotcoffeeman, 2024-06-10:
I've read the docs but can't understand how to upgrade, i've tried downloading the tar file and extracting it and running the commands it mentions here - https://networkupstools.org/docs/user-manual.chunked/Upgrading_notes.html - but i get command not found....

> pi@PI:/tmp/nut-2.8.2$ ./configure --with-all
> -bash: ./configure: Permission denied
pi@PI:/tmp/nut-2.8.2$ sudo ./configure --with-all
sudo: ./configure: command not found
pi@PI:/tmp/nut-2.8.2$ sudo ./configure
sudo: ./configure: command not 

then i tried https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests and got this error....

> pi@PI:/tmp/nut-2.8.2$ cd /tmp
pi@PI:/tmp$ git clone https://github.com/networkupstools/nut
Cloning into 'nut'...
remote: Enumerating objects: 88765, done.
remote: Counting objects: 100% (6611/6611), done.
remote: Compressing objects: 100% (1926/1926), done.
remote: Total 88765 (delta 4939), reused 6274 (delta 4679), pack-reused 82154
Receiving objects: 100% (88765/88765), 84.05 MiB | 3.79 MiB/s, done.
Resolving deltas: 100% (66154/66154), done.
pi@PI:/tmp$ cd nut
pi@PI:/tmp/nut$ ./autogen.sh
Regenerating Augeas ups.conf lens with 'python3'...
Regenerating the USB helper files...
Calling autoreconf...
./autogen.sh: 176: autoreconf: not found
FAILED: did not generate an executable configure script!

> Note: on some systems "autoreconf", "automake" et al are dispatcher
 scripts, and need you to explicitly say which version you want, e.g.
    export AUTOCONF_VERSION=2.65 AUTOMAKE_VERSION=1.13
 If you get issues with AC_DISABLE_STATIC make sure you have libtool.

> If it complains about "too few" or "excess" "arguments to builtin ifdef",
 check the configure.ac line it refers to and un-comment (or comment away)
 the third argument for AM_SILENT_RULES check, or comment away the whole
 "ifdef" block if your autotools still would not grok it.

Please help thanks

jimklimov, 2024-06-10:
From the same wiki article, there is a reference to
`docs/config-prereqs.txt` and packages you would need on a few typical
OSes/distros.

Not sure OTOH why `./configure` complains, maybe the exec bit is not set? :\

jimklimov, 2024-06-11:
For the first part, with permission denied etc. - can you elaborate what you did there? I've just re-checked that the tarball has proper permissions in the stored script:
````
jim:/dev/shm$ wget https://www.networkupstools.org/source/2.8/nut-2.8.2.tar.gz
--2024-06-11 09:30:29--  https://www.networkupstools.org/source/2.8/nut-2.8.2.tar.gz
Resolving www.networkupstools.org (www.networkupstools.org)... 185.199.108.153, 185.199.109.153, 185.199.110.153, ...
Connecting to www.networkupstools.org (www.networkupstools.org)|185.199.108.153|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://networkupstools.org/source/2.8/nut-2.8.2.tar.gz [following]
--2024-06-11 09:30:29--  https://networkupstools.org/source/2.8/nut-2.8.2.tar.gz
Resolving networkupstools.org (networkupstools.org)... 185.199.111.153, 185.199.110.153, 185.199.109.153, ...
Connecting to networkupstools.org (networkupstools.org)|185.199.111.153|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 5599935 (5.3M) [application/gzip]
Saving to: ‘nut-2.8.2.tar.gz’

nut-2.8.2.tar.gz                                        100%[=============================================================================================================================>]   5.34M  14.1MB/s    in 0.4s

2024-06-11 09:30:29 (14.1 MB/s) - ‘nut-2.8.2.tar.gz’ saved [5599935/5599935]

jim:/dev/shm$ tar xzf nut-2.8.2.tar.gz

jim:/dev/shm$ cd nut-2.8.2/

jim:/dev/shm/nut-2.8.2$ ./configure --with-all
checking for CONFIG_FLAGS... --with-all
Network UPS Tools version 2.8.2
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... x86_64-pc-linux-gnu
...
````

jimklimov, 2024-06-11:
One odd idea that comes to mind: on a Pi, would you happen to have a FAT32/vFAT/exFAT/... formatted SD card under your `/tmp` location? The FAT filesystem does not have a notion of POSIX permissions bits, so that could be one explanation why it fails.

Hotcoffeeman, 2024-06-11:
> One odd idea that comes to mind: on a Pi, would you happen to have a FAT32/vFAT/exFAT/... formatted SD card under your `/tmp` location? The FAT filesystem does not have a notion of POSIX permissions bits, so that could be one explanation why it fails.
> 
> Although the later `./autogen.sh` running successfully kind of dispells that idea.
> 
> Another similar problem that rings a bell can be with end of line characters (e.g. the shebang in `configure` written and saved as text on some Windows system having extra chars when Linux sees it, in a viewer this would be seen like `#! /bin/sh^M` - but again, seems to not be the problem here.

The permissions seem to be there I extracted it on the device itself, how can I check that and if its formatted FAT?

This is a RPi 2 could that be the issue?

Why is it that apt install doesn't install the latest version?

Thanks for your help

jimklimov, 2024-06-11:
For filesystem types, `mount` with no arguments should list all active logical filesystems and their types, mount options, etc.

I don't think RPi2 per se is an issue (probably needs a 32-bit ARM build, but that's about it).

Regarding `apt` - a question to the distro, what and why they package and when they would update the recipe :) And at that, with Debian-like projects, it would likely first be updated in a testing/unstable repo, then may tickle down to the newest (or next expected) release, and not likely to impact older release numbers (e.g. if Debian 12 is current, base version of whatever is in Debian 11 or older would remain as it was, with only some bugfix patches applied).

Hotcoffeeman, 2024-06-14:
Mount outputs a lot of lines, I searched for fat in it and its not found, one line mentions ext4...

/dev/mmcblk0p2 on / type ext4 (rw,relatime)

Btw its running Ubuntu 22.04.4 LTS (GNU/Linux 5.15.0-1055-raspi armv7l). Should this be ok?

jimklimov, 2024-07-15:
Hm, stumbled on this issue in a review again, and one more thing comes to mind: you did not clarify if you have installed all the prerequisites for the platform as listed in `docs/config-prereqs.txt`, originally or at a later date. The message you posted first may indicate that you do not have `autoconf` or `automake` packages installed. You then likely lack many other things a NUT build would need, too.

Hotcoffeeman, 2024-07-15:
Ah good point. I started reviewing that file but tbh i'm very unexperiecned with Linux so was quite confused. 

So for RPi should i follow the steps under the Debian 10/11/12 heading? Even though its not Debian? It does say "Related common operating systems include Ubuntu"

jimklimov, 2024-07-15:
Yes, at least until proven otherwise and we need a new chapter there :)

Hotcoffeeman, 2024-07-15:
Oh yes sorry completely missed that, was searching for RPi for some reason.

Will give it a go

Hotcoffeeman, 2024-07-15:
`pi@PI2:~$ sudo apt-get install \
    ccache time \
    git python perl curl \
    make autoconf automake libltdl-dev libtool \
    valgrind \
    cppcheck \
    pkg-config \
    gcc g++ clang
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  2to3 python2-minimal python2 dh-python python-is-python3

E: Package 'python' has no installation candidate`

Should I install one of those other python packages?

jimklimov, 2024-07-15:
As a comment just below goes, probably yes. Currently either python 2.7 (maybe 2.6+) should be function, but EOL'ed at upstream, or 3.5+ roughly (earlier 3.x were not tested for NUT, I think).

If your distro is recent, it installation likely includes python3 for whatever tools of its own already; then NUT `./autogen.sh && ./configure ...` part should find it.

jimklimov, 2024-07-15:
Posted a doc update to make this bit clearer, thanks for the feedbacks :)

Hotcoffeeman, 2024-07-15:
I think i've got all the pre reqs installed now.

I installed originally using sudo apt install nut nut-client nut-server so which option from https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests is best to do the upgrade?

jimklimov, 2024-07-15:
I am not sure about packaging recipes used by different distros, but can guess `nut` would be an umbrella package that pulls in everything (or most of it), in particular the many sub-driver packages (with whatever each one depends on).

I'd suggest going with the default "in-place" build first, to see if it configures at least the drivers you need for the device you have (and inherits other options like the user/group, etc.)