networkupstools/nut #2470: Supported device: GreenCell UPS07 (800VA)

yannick-Lapierre, 2024-06-12:
Hi,
I managed to set up NUT for GreenCell UPS07 (800VA).
Configuration:
``` YAML
- name: myups
  driver: nutdrv_qx
  port: auto
  config: []
```

Entities (sorry, it's French):
[image]

newkind, 2025-02-28:
Bump! Please add support for this device, its quite popular in Poland (Green Cell is a Polish company, probably that's why)

newkind, 2025-02-28:
@jimklimov Thank you very much for your detailed response. If I only had the skills I'd definitely try to tackle that, unfortunately its out of my league. The truth is that I got this UPS and I tried to connect it to my Ugreen NAS which is powered by Ugos Pro (a Debian based) OS similar to Synologys DSM. The UPS gets dectected and even looks like its connecting, but then it fails, which made me digg further until I came upon this issue. I assume that it connects, then tries to get the readings of values and then it fails. I SSH'ed into the NAS and I can see they're utilising the NUT, hence my post above ;) My hope is that once (if at all) the support for this UPS lands into the NUT, then Ugreen team will update the NUT package with one of their OS updates (or i'll do it manually via apt-get) and have it fully working. That's the whole story.

`apt list --installed` reports this version:

```
nut-client/stable,now 2.8.0-7 amd64 [installed,automatic]
nut-server/now 2.8.0-7 amd64 [installed,upgradable to: 2.8.0-7]
nut-snmp/stable,now 2.8.0-7 amd64 [installed]
```