networkupstools/nut #2482: Salicru SPS 2000 ADV T not showing

DonSalty, 2024-06-18:
Hello, 

I'm trying to configure NUT on proxmox using the following link (https://www.kreaweb.be/diy-home-server-2021-software-proxmox-ups/) but when i use the command "nut-scanner -U" it doesnt report any info about the UPS,´, but when i passthrough the connection to my unraid server running on the proxmox host , and use the nut tools addon it recognizes automatically the UPS.

The autoconfiguration files report the following:

"-----------------------------------------------------
INFO: The NUT Scanner is also available from the terminal with more advanced settings.
INFO: You can use 'nut-scanner -h' to see all available options for searching UPS devices.
-----------------------------------------------------

Making sure all existing NUT services are stopped (before UPS scanning)...
Reading the existing NUT configuration...

NUT Scanner is now searching for UPS devices...
#####################################################

[nutdev-usb1]
driver = "usbhid-ups"
port = "auto"
vendorid = "2E66"
productid = "0302"
product = "2000"
serial = "000000000000"
vendor = "1"
# bus = "001"
# device = "008"
# busport = "002"

#####################################################

Adjusting the NUT configuration for the found UPS...
Setting the manual configuration mode ('for UPS Driver') in GUI...
Writing the modified NUT configuration to the respective locations...

NUT Scanner has finished configuring NUT - please attempt to start your NUT services now."

i cant understand the issue i'm facing
Can someone help ?

I've attached the .dev file generated by the nut addon on unraid
[nut-ups.dev.txt](https://github.com/user-attachments/files/15892963/nut-ups.dev.txt)

jimklimov, 2024-06-18:
Just in case, which NUT versions are on proxmox and on unraid?

DonSalty, 2024-06-18:
Hello @jimklimov 

Proxmox is running - 2.8.0-7
Unraid is running - nut-2.8.2

But when trying to update in proxmox it gives me the following message:
"nut is already the newest version (2.8.0-7)"

jimklimov, 2024-06-18:
You can build your own for proxmox then, check wiki. Or poke the distro (Debian) to update their recipe.

DonSalty, 2024-06-19:
Finally got it working , but now the Output VAC showed is diferent from the Output showed on the UPS itself.

On the UPS shows - 239 
On the Network UPS Tools shows -  9.1 

[image]

Have i done something incorrectly during the manual build ?

jimklimov, 2024-06-19:
Probably something is misinterpreted by the driver, or even wrongly published by the device (many vendors get USB HID encoding nuances wrong, so we have some fixes around that).

I'm on the road currently, but please grep the settings for an option about disabling descriptor fix (to check if our fix-up of a common issue breaks stuff here).

jimklimov, 2024-06-24:
Can you run the driver with a higher debug verbosity from the command line (after stopping the normal monitoring service)? e.g. collect one cycle of data dump like this:
````
:; /lib/nut/usbhid-ups -DDDDDD -d1 -a nut-ups
````
(assuming a Debian-like packaged path above; you can also run it from your build workspace as `./drivers/usbhid-ups` if it is on that same system)

Afterwards you can restart the usual monitoring.

I hope the log would show the bytes it received for one of those `output.voltage` readings:
````
$ grep output.voltage drivers/salicru-hid.c
  { "output.voltage", 0, 0, "UPS.Output.Voltage", NULL, "%.1f", 0, NULL },
  { "output.voltage.nominal", 0, 0, "UPS.Output.ConfigVoltage", NULL, "%.0f", 0, NULL },
  { "output.voltage", 0, 0, "UPS.PowerConverter.Output.Voltage", NULL, "%.1f", 0, NULL },
  { "output.voltage.nominal", 0, 0, "UPS.PowerSummary.ConfigVoltage", NULL, "%.0f", 0, NULL },
````

Note in your original post, you had such wrong numbers from unraid:
````
output.voltage: 6.4
output.voltage.nominal: 24
````