networkupstools/nut #2489: Temperature not reported for SMT1500i

cristianstoica, 2024-06-24:
Do I need a different firmware or is there a different issue? https://networkupstools.org/ddl/APC/Smart-UPS_1500.html shows battery.temperature variable but I don't see it on the status page. 

"status": {
        "battery.charge": "100",
        "battery.charge.low": "10",
        "battery.charge.warning": "50",
        "battery.runtime": "20100",
        "battery.runtime.low": "120",
        "battery.type": "PbAc",
        "battery.voltage": "27.4",
        "battery.voltage.nominal": "24.0",
        "device.mfr": "American Power Conversion ",
        "device.model": "Smart-UPS 1500",
        "device.serial": "AS1239220678  ",
        "device.type": "ups",
        "driver.name": "usbhid-ups",
        "driver.parameter.pollfreq": "30",
        "driver.parameter.pollinterval": "2",
        "driver.parameter.port": "auto",
        "driver.parameter.synchronous": "auto",
        "driver.parameter.vendorid": "051d*",
        "driver.version": "2.8.0",
        "driver.version.data": "APC HID 0.98",
        "driver.version.internal": "0.47",
        "driver.version.usb": "libusb-1.0.26 (API: 0x1000109)",
        "ups.beeper.status": "enabled",
        "ups.delay.shutdown": "20",
        "ups.firmware": "UPS 08.3 / ID=18",
        "ups.mfr": "American Power Conversion ",
        "ups.mfr.date": "2012/09/24",
        "ups.model": "Smart-UPS 1500",
        "ups.productid": "0003",
        "ups.serial": "AS1239220678  ",
        "ups.status": "OL",
        "ups.timer.reboot": "-1",
        "ups.timer.shutdown": "-1",
        "ups.vendorid": "051d"
      },

cristianstoica, 2024-06-24:
Indeed, using HA NUT add on https://github.com/hassio-addons/addon-nut (based on 2.8.0) version. However, battery temperature seems to have been available since (at least) 2.7.1.

jimklimov, 2024-06-24:
In that case https://github.com/networkupstools/nut/wiki/Building-NUT-integration-for-Home-Assistant from community members could also help with a custom build.