networkupstools/nut #2492: Eaton 9E Model (bypass and Eco) Low/High Voltage Transfer is missing however present when usbhid-usb debug Driver

masterwishx, 2024-06-26:
[lsusb.txt](https://github.com/user-attachments/files/15984635/lsusb.txt)
[usbhid-ups.txt](https://github.com/user-attachments/files/15984636/usbhid-ups.txt)

masterwishx, 2024-06-26:
using Unraid NUT plugin v2.8.2 , posted from old version from https://github.com/networkupstools/nut/issues/1925 if need new will post again .

values present in usbhid debug but not in UPS variables:  

```
usbhid-ups

  ECO Transfer

   4.830228	[D5] hid_lookup_path: 00840004 -> UPS
   4.830232	[D5] hid_lookup_path: 00840016 -> PowerConverter
   4.830236	[D5] hid_lookup_path: 0084001c -> Output
   4.830240	[D5] hid_lookup_path: ffff005c -> FrequencyRangeEcoTransfer
   4.830245	[D1] Path: UPS.PowerConverter.Output.FrequencyRangeEcoTransfer, Type: Feature, ReportID: 0x67, Offset: 0, Size: 8, Value: 5
   4.830249	[D4] Entering libusb_get_report
   4.850325	[D3] Report[get]: (3 bytes) => 46 08 01
   4.850338	[D5] PhyMax = 0, PhyMin = 0, LogMax = 65535, LogMin = 0
   4.850351	[D5] Unit = 00f0d121, UnitExp = 7
   4.850355	[D5] Exponent = 0

4.870484	[D5] hid_lookup_path: 00840004 -> UPS
   4.870487	[D5] hid_lookup_path: 00840016 -> PowerConverter
   4.870490	[D5] hid_lookup_path: 0084001c -> Output
   4.870493	[D5] hid_lookup_path: ffff005b -> HighVoltageEcoTransfer
   4.870498	[D1] Path: UPS.PowerConverter.Output.HighVoltageEcoTransfer, Type: Feature, ReportID: 0x68, Offset: 0, Size: 16, Value: 241
   4.870501	[D4] Entering libusb_get_report
   4.892707	[D3] Report[get]: (5 bytes) => 51 2c 01 64 00
   4.892719	[D5] PhyMax = 0, PhyMin = 0, LogMax = 65535, LogMin = 0
   4.892723	[D5] Unit = 00f0d121, UnitExp = 7
   4.892725	[D5] Exponent = 0

4.952657	[D5] hid_lookup_path: 00840004 -> UPS
   4.952662	[D5] hid_lookup_path: 00840016 -> PowerConverter
   4.952668	[D5] hid_lookup_path: 0084001c -> Output
   4.952674	[D5] hid_lookup_path: ffff005a -> LowVoltageEcoTransfer
   4.952682	[D1] Path: UPS.PowerConverter.Output.LowVoltageEcoTransfer, Type: Feature, ReportID: 0x69, Offset: 0, Size: 16, Value: 218
   4.952690	[D3] Report[buf]: (5 bytes) => 51 2c 01 64 00
   4.952695	[D5] PhyMax = 0, PhyMin = 0, LogMax = 65535, LogMin = 0
   4.952700	[D5] Unit = 00f0d121, UnitExp = 7
   4.952704	[D5] Exponent = 0

Bypass Transfer

 4.811105	[D5] hid_lookup_path: 00840004 -> UPS
   4.811108	[D5] hid_lookup_path: 00840016 -> PowerConverter
   4.811111	[D5] hid_lookup_path: 0084001c -> Output
   4.811113	[D5] hid_lookup_path: ffff0059 -> FrequencyRangeBypassTransfer
   4.811118	[D1] Path: UPS.PowerConverter.Output.FrequencyRangeBypassTransfer, Type: Feature, ReportID: 0x44, Offset: 0, Size: 8, Value: 10
   4.811120	[D4] Entering libusb_get_report
   4.830200	[D3] Report[get]: (2 bytes) => 67 05
   4.830215	[D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
   4.830219	[D5] Unit = 00000000, UnitExp = 0
   4.830223	[D5] Exponent = 0

   4.850360	[D5] hid_lookup_path: 00840004 -> UPS
   4.850365	[D5] hid_lookup_path: 00840016 -> PowerConverter
   4.850369	[D5] hid_lookup_path: 0084001c -> Output
   4.850373	[D5] hid_lookup_path: ffff0058 -> HighVoltageBypassTransfer
   4.850381	[D1] Path: UPS.PowerConverter.Output.HighVoltageBypassTransfer, Type: Feature, ReportID: 0x46, Offset: 0, Size: 16, Value: 264
   4.850385	[D4] Entering libusb_get_report
   4.870463	[D3] Report[get]: (3 bytes) => 68 f1 00
   4.870475	[D5] PhyMax = 0, PhyMin = 0, LogMax = 65535, LogMin = 0
   4.870478	[D5] Unit = 00f0d121, UnitExp = 7
   4.870480	[D5] Exponent = 0

    4.932351	[D5] hid_lookup_path: 00840004 -> UPS
   4.932356	[D5] hid_lookup_path: 00840016 -> PowerConverter
   4.932360	[D5] hid_lookup_path: 0084001c -> Output
   4.932365	[D5] hid_lookup_path: ffff0057 -> LowVoltageBypassTransfer
   4.932372	[D1] Path: UPS.PowerConverter.Output.LowVoltageBypassTransfer, Type: Feature, ReportID: 0x45, Offset: 0, Size: 16, Value: 184
   4.932377	[D4] Entering libusb_get_report
   4.952621	[D3] Report[get]: (3 bytes) => 69 da 00
   4.952640	[D5] PhyMax = 0, PhyMin = 0, LogMax = 65535, LogMin = 0
   4.952646	[D5] Unit = 00f0d121, UnitExp = 7
   4.952650	[D5] Exponent = 0
```

[image]

arnaudquette-eaton, 2024-06-26:
to me, it's just another variation of the missing model name.
As long as this issue is not fixed, all these data will continue to be missing

masterwishx, 2024-06-26:
> As long as this issue is not fixed, all these data will continue to be missing

Thanks for answer, so @jimklimov should add the model 9E or add all those vars of model to fix it  ?

masterwishx, 2024-06-26:
@jimklimov can you please check it , it seems same other vars also is missing ...

jimklimov, 2024-06-27:
To keep things cross-linked and context visible, this post (and the bit on model naming improvements) is related to at least:
* #1925
* #2380
* #2442

masterwishx, 2024-09-06:
> to me, it's just another variation of the missing model name. As long as this issue is not fixed, all these data will continue to be missing

Now as the iProduct and etc was fixed by #2604 those (and more) values still is missing ...
@jimklimov @tormodvolden if this need more code in usblib or in mge-hid or its blocked by firmware ?

tormodvolden, 2024-09-06:
Unless you provide libusb debug output, I cannot help. Do the manufacturer/product strings show up in xusb?

masterwishx, 2024-09-06:
> Unless you provide libusb debug output, I cannot help. Do the manufacturer/product strings show up in xusb?

I will post new libusb debug here tomorrow , the old one
Is here above in this page, the first comment and here https://github.com/networkupstools/nut/issues/2492#issuecomment-2191108335 
Do you mean xusb for Eaton 9E `0463/ffff`? 
Nont sure if I posted when compiled xusb befor, but can check it...

masterwishx, 2024-09-07:
Here is the new log after the fix #2604 : 
[nut debug log.txt](https://github.com/user-attachments/files/16917136/nut.debug.log.txt)

masterwishx, 2024-09-07:
libusb debug from xusb (with code to check langid) is same as here : https://github.com/libusb/libusb/issues/1548#issuecomment-2308879614

[libusb 1.0.27 debug log.txt](https://github.com/user-attachments/files/16917260/libusb.1.0.27.debug.log.txt)

tormodvolden, 2024-09-07:
I see in your nut_debug_log.txt that the device is identified with Manufacturer: EATON and Product: Eaton 9E. I also see HighVoltageTransfer with value 300 and LowVoltageTransfer with value 100. So I guess there is nothing wrong with the USB communication and nothing I can help you with.

masterwishx, 2024-09-07:
> I see in your nut_debug_log.txt that the device is identified with Manufacturer: EATON and Product: Eaton 9E

Sure ,you already make amazing work to fix `langid` from `unknown` to `Eaton 2000i` in #2604.

> HighVoltageTransfer with value 300 and LowVoltageTransfer with value 100

Thouse values already was avalibe in UPS variables befor the fix , as `input.transfer.low = 100` and `input.transfer.high = 300`

but the question is about other values that avalibe in debug but not in variables like  :  

`FrequencyRangeBypassTransfer, FrequencyRangeEcoTransfer, HighVoltageBypassTransfer, HighVoltageEcoTransfer` and etc ...

[image]

masterwishx, 2024-09-07:
and here is nut details : 
```
battery.capacity: 7.00
battery.charge: 100
battery.charge.low: 20
battery.charge.restart: 0
battery.protection: yes
battery.runtime: 3444
battery.runtime.low: 180
battery.type: PbAc
battery.voltage: 83.0
battery.voltage.nominal: 72
device.mfr: EATON
device.model: 9E2000i
device.serial:  xxxxxxxxxxxx                   #REDACTED by me # !!! serial is shown now !!!
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.state: updateinfo
driver.version: 2.8.2.1
driver.version.data: MGE HID 1.48
driver.version.internal: 0.56
driver.version.usb: libusb-1.0.26 (API: 0x1000108)
input.bypass.frequency: 49.9
input.bypass.voltage: 232.0
input.frequency: 49.9
input.frequency.nominal: 50
input.transfer.high: 300
input.transfer.low: 100
input.voltage: 232.0
input.voltage.nominal: 231
outlet.1.status: on
outlet.desc: Main Outlet
outlet.id: 0
outlet.switchable: no
output.current: 1.60
output.frequency: 49.9
output.frequency.nominal: 50
output.voltage: 229.0
output.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: 00.02.000
ups.load: 19
ups.load.high: 105
ups.mfr: EATON
ups.model: 9E2000i
ups.power: 393
ups.power.nominal: 2000
ups.productid: ffff
ups.realpower: 261
ups.realpower.nominal: 1600
ups.serial:  xxxxxxxxxxxx                   #REDACTED by me # !!! serial is shown now !!!
ups.shutdown: enabled
ups.start.auto: yes
ups.start.battery: yes
ups.start.reboot: yes
ups.status: OL
ups.temperature: 24.9
ups.test.interval: 604800
ups.test.result: Done and passed
ups.timer.shutdown: -1
ups.timer.start: -1
ups.type: online
ups.vendorid: 0463
```

masterwishx, 2024-09-08:
@jimklimov @tormodvolden 

after a little research it seems thouse values are missing in `Data lookup table (HID <-> NUT) ` https://github.com/masterwishx/nut/blob/64435823ac369f3906cefe20b9b19d27bec8c96d/drivers/mge-hid.c#L1251

but present in `/* Eaton / MGE HID usage table */ `
https://github.com/masterwishx/nut/blob/64435823ac369f3906cefe20b9b19d27bec8c96d/drivers/mge-hid.c#L872

masterwishx, 2024-09-09:
So I can try to add thouse values to table, but seems its more complicated to add ECO Mode to NUT in #2495 for Eaton ups

arnaudquette-eaton, 2024-09-09:
indeed @masterwishx , as you found, not all HID data are mapped to NUT variables.
Some may not to extend NUT namespace...

masterwishx, 2024-09-09:
> indeed @masterwishx , as you found, not all HID data are mapped to NUT variables. Some may not to extend NUT namespace, but I should have paved the road enough to make it easy...

I can try to add thouse missing values but still need some way to activate bypass or eco mode in code...

masterwishx, 2024-09-09:
> but I should have paved the road enough to make it easy...

Thanks for amazing work for we can use NUT and upgrade it :)

masterwishx, 2024-09-09:
> indeed @masterwishx , as you found, not all HID data are mapped to NUT variables. Some may not to extend NUT namespace, but I should have paved the road enough to make it easy...

`input.transfer.high` already avaliable but why it's not `output.transfer.high`? 

```
{ "input.transfer.high", ST_FLAG_RW | ST_FLAG_STRING, 5, "UPS.PowerConverter.Output.HighVoltageTransfer", NULL, "%.0f", HU_FLAG_SEMI_STATIC, NULL },```

masterwishx, 2024-09-09:
> `input.transfer.high` already avaliable but why it's not `output.transfer.high`?
>
> ```
> { "input.transfer.high", ST_FLAG_RW | ST_FLAG_STRING, 5, "UPS.PowerConverter.Output.HighVoltageTransfer", NULL, [...]

Oh or it's mean like input to ups from wall socket power

closed 2024-09-13 by jimklimov