networkupstools/nut #2493: Info about unsupported device: MicroUPS 300

mickeyreg, 2024-06-26:
Hi,

I have the "no name" device, called "MicroUPS 3000": https://voltpolska.pl/zasilanie-awaryjne/micro-ups-3000-4x9ah-1800-3000w-komputerowy-zasilacz-awaryjny.html

According to product information it is supported by ViewPower software, so it is possible, that it should be supported by nutdrv_qx. But I have an info "Device not supported!" :( 

Maybe somebody can take a look  in below logs to check if the support is possible.

After testing in NUT from pkg on FreeBSD 14.0 I downloaded the latest mater from GitHub, compiled and tested:
**git clone -b master https://github.com/networkupstools/nut**

Scanner shows:
```
# tools/nut-scanner/nut-scanner -U
Scanning USB bus.
[nutdev-usb1]
        driver = "nutdrv_qx"    # alternately: blazer_usb nutdrv_atcl_usb
        port = "auto"
        vendorid = "0001"
        productid = "0000"
        product = "MEC0003"
        vendor = "MEC"
        # bus = "000"
        # device = "003"
        # busport = "011"
```

So in the ups.conf I have:
```
[microups]
  driver = nutdrv_qx
  port = auto
  desc = "MicroUPS 3000"
  vendorid = 0001
  productid = 0000
```

Log from nutdrv_qx:
```
# ./drivers/nutdrv_qx -a microups -DDDDDD -d1
   0.000002     [D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.2-382-gd683bc1da)
USB communication driver (libusb 1.0) 0.48
   0.000173     [D1] upsdrv_makevartable...
   0.000254     [D5] send_to_all: SETINFO driver.version.usb "libusb-1.0.0 (API: 0x1000102)"
   0.000269     [D1] Using USB implementation: libusb-1.0.0 (API: 0x1000102)
   0.000609     [D5] do_upsconf_args: confupsname=(null), var=maxretry, val=3
   0.000621     [D5] do_upsconf_args: call do_global_args()
   0.000632     [D3] do_global_args: var='maxretry' val='3'
   0.001039     [D5] do_upsconf_args: confupsname=smartups, var=driver, val=dummy-ups
   0.001055     [D5] do_upsconf_args: confupsname=smartups, var=port, val=ups@192.168.80.244
   0.001069     [D5] do_upsconf_args: confupsname=smartups, var=desc, val=APC Smart-UPS 1500
   0.001084     [D5] do_upsconf_args: confupsname=microups, var=driver, val=nutdrv_qx
   0.001094     [D5] do_upsconf_args: call main_arg()
   0.001104     [D3] main_arg: var='driver' val='nutdrv_qx'
   0.001114     [D5] do_upsconf_args: not a main_arg()
   0.001124     [D5] do_upsconf_args: this is a 'driver' setting, may we proceed?
   0.001136     [D6] testval_reloadable: var=driver, oldval=nutdrv_qx, newval=nutdrv_qx, reloadable=0, reload_flag=0
   0.001147     [D6] testval_reloadable: verdict for (re)loading var=driver value: -1
   0.001157     [D5] do_upsconf_args: 'driver' setting already applied with this value
   0.001175     [D5] do_upsconf_args: confupsname=microups, var=port, val=auto
   0.001184     [D5] do_upsconf_args: call main_arg()
   0.001194     [D3] main_arg: var='port' val='auto'
   0.001205     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.001215     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.001228     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.001242     [D5] do_upsconf_args: confupsname=microups, var=desc, val=MicroUPS 3000
   0.001251     [D5] do_upsconf_args: call main_arg()
   0.001261     [D3] main_arg: var='desc' val='MicroUPS 3000'
   0.001274     [D5] do_upsconf_args: confupsname=microups, var=vendorid, val=0001
[... 639 lines cut ...]
 frame[1] READ 4 bytes
12:42:59.300980 usbus0.3 DONE-CTRL-EP=00000080,SPD=LOW,NFR=1,SLEN=0,IVAL=0,ERR=IOERROR
 frame[0] WRITE 0 bytes
12:42:59.367135 usbus0.3 SUBM-CTRL-EP=00000080,SPD=LOW,NFR=2,SLEN=8,IVAL=0
 frame[0] WRITE 8 bytes
 0000  80 06 00 03 00 00 FF 00  -- -- -- -- -- -- -- --  |........        |
 frame[1] READ 255 bytes
12:42:59.367947 usbus0.3 DONE-CTRL-EP=00000080,SPD=LOW,NFR=1,SLEN=0,IVAL=0,ERR=IOERROR
 frame[0] WRITE 0 bytes
12:42:59.434234 usbus0.3 SUBM-CTRL-EP=00000080,SPD=LOW,NFR=2,SLEN=8,IVAL=0
 frame[0] WRITE 8 bytes
 0000  80 06 00 03 00 00 04 00  -- -- -- -- -- -- -- --  |........        |
 frame[1] READ 4 bytes
12:42:59.434959 usbus0.3 DONE-CTRL-EP=00000080,SPD=LOW,NFR=1,SLEN=0,IVAL=0,ERR=IOERROR
 frame[0] WRITE 0 bytes
12:42:59.501159 usbus0.3 SUBM-CTRL-EP=00000080,SPD=LOW,NFR=2,SLEN=8,IVAL=0
 frame[0] WRITE 8 bytes
 0000  80 06 00 03 00 00 FF 00  -- -- -- -- -- -- -- --  |........        |
 frame[1] READ 255 bytes
12:42:59.501948 usbus0.3 DONE-CTRL-EP=00000080,SPD=LOW,NFR=1,SLEN=0,IVAL=0,ERR=IOERROR
 frame[0] WRITE 0 bytes
```

Regards,
Mickey

jimklimov, 2024-07-01:
Hello, just in case: did you check also with `nutdrv_atcl_usb` and for kicks with `usbhid-ups` (probably need to do some guesswork about `subdriver` option there)?

mickeyreg, 2024-07-02:
I tested all subdrivers from `nutdrv_qx` changind options like **norating** and **novendor** if applicable. I tested also: `blazer_usb`, `usbhid-ups` and `nutdrv_atcl_usb` - all without success.

As I [found](https://github.com/networkupstools/nut/issues?q=is%3Aissue+is%3Aopen+label%3A%22USB+VID%3D0001+PID%3D0000+%28Fry%27s+Electronics%2FMEC0003%29%22) there is a lot 0001/0000 UPSes, which theoretically should work with NUT, but does not work :(

In the NUTDRV_QX documentation can be read:

**norating**
    _Some UPSes will lock up if you attempt to read rating information from them. Setting this flag will make the driver skip this step._
**novendor**
    _Some UPSes will lock up if you attempt to read vendor information from them. Setting this flag will make the driver skip this step._

What means "UPSes will lock up"? Do I have to disconnect/reconnect USB to unlock?

jimklimov, 2024-07-02:
Well, with vendors not even caring to identify their devices properly (so you get a proverbial "cat in a sack"), the "theoretically should work with NUT" part is a big leap of faith :\

jimklimov, 2024-07-02:
I believe "UPSes will lock up" means that unexpected input can wedge their on-board controller. Whether it un-freezes after some time or requires some replug or even UPS power-cycling to clear this up, wholly depends on the firmware.

mickeyreg, 2024-07-08:
> Well, with vendors not even caring to identify their devices properly (so you get a proverbial "cat in a sack"), the "theoretically should work with NUT" part is a big leap of faith :\

After reading above I think I understood the nature of the problem. We have a lot of devices, which identifies itself as 0001/0000 and uses unidentified protocol. VID & PID is not enough to recognize the type of the protocol :(

Some of these 0001/0000 devices have software for Windows or Linux. It is possible to record USB communication between this software and the device? On Linux? On Windows? Can such a recording be useful in extending the capabilities of the specific driver in NUT?

jimklimov, 2024-07-08:
Yes, some platforms have native tools like `usbdump`, others have modules for WireShark and relatives. I suppose that some drivers were written by reverse-engineering, where sniffing the protocol of vendor's software could play a part.

Unfortunately, this would likely need a programmer ready to dig into the USB protocol (or serial-over-USB) and make sense of it, which I myself am not. I'll post a query on the mailing list, perhaps someone would have at least suggestions about this...

jimklimov, 2024-07-08:
Just in case, did you check https://github.com/networkupstools/nut/issues?q=label%3AViewPower for other devices supported by ViewPower - if those discussions led to a working driver?