networkupstools/nut #2494: suppress alarm = "UPS is in ECO Mode."

spacelama, 2024-06-27:
Any chance of optionally suppressing `alarm = "UPS is in ECO Mode."` in nutdrv_qx_voltronic.c?  In my particular situation, I don't want to be constantly alarmed about what's a normal situation for my UPS:

````
> upsc powershield-centurion@pve1 | grep -i alarm
ups.alarm: UPS is in ECO Mode.
ups.status: ALARM OL
````

jimklimov, 2024-06-27:
Well, looking at stock sources - currently no, it is treated in same grouping as a fault or shutting-down or "converter" mode (whatever that is, seems different from bypass; maybe boost/buck?) - see the `voltronic_mode()` method. It should not be hard to make a custom build with that line changed into e.g. `status = "ECO";` to see if that fares better for you.

There generally seems to be a mess in that subdriver with statuses like `!OL` instead of `OB` which many clients would not recognize if they parse it.

Another thing to note is that in `docs/new-drivers.txt` section about status values, we do not currently have a standard one for ECO/HE (High Efficiency). I'll raise some discussion on mailing list on that.

spacelama, 2024-10-07:
Oh, and a clarification on what converter mode is: For my PSCE1000 Powershield Centurion, the manual states that "converter" is frequency converter.  Since it's a double conversion unit, it's free to invent whatever frequency is most useful for the user (so long as your preferred frequencies are either 50 or 60Hz).  Again, probably not an alarm situation (except that you're meant to derate the power rating of the UPS to 80% of its normal level when in frequency converter or indeed when the output voltage is adjusted to 200 or 208VAC).

jimklimov, 2025-07-18:
It seems the PR #2850 (shipped with NUT v2.8.3 as part of work for issue #2708) took a shot at addressing this area; although upon further review - not quite correctly (mixed up capability report vs. current-status report).

@spacelama : do you still have the UPS? Would you be able and willing to test a custom build of a new PR following the https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests ? You'd need to install tools and other prerequisites listed there, build NUT, run new driver - can do from the build workspace without changing the packaged setup until you like the new build better.

spacelama, 2025-07-18:
Yep, I still have the UPS.  It was on my TODO list to work out how to build nut in a (Debian/testing) container, and I guess you just accelerated it up to this weekend's list.