networkupstools/nut #2500: Xmart by Integra SUPRA 901-230-S* works wit blazer_usb but not with nutdrv_qx

tokafondo, 2024-06-30:
Just a report, not an issue at all:

BRAND: Xmart by Integra
MODEL: SUPRA 901-230-S*

When nutdrv_qx is used, upscmd commands return ERR_FAILED

When blazer_usb is used, upscmd returns SUCCESS

jimklimov, 2024-07-01:
Thank you for the report. Can you please clarify which NUT version you have, and what the actual driver debug log says when you try to handle the device with one or the other driver? => https://github.com/networkupstools/nut/wiki/Changing-NUT-daemon-debug-verbosity

The `upscmd` is pretty high up in the stack (libusb => driver => upsd => clients like upscmd) so is likely not the best place to drill down to discrepancies of `blazer*` drivers and `nutdrv_qx` which is supposed to replace them and some others eventually.

tokafondo, 2024-07-03:
```Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D1] debug level is '3'
   0.000278     [D1] Saving PID 20596 into /run/nut/nutdrv_qx-supra.pid
   0.000315     [D1] upsdrv_initups...
   0.003058     [D2] Checking device 1 of 8 (0557/2212)
   0.003090     [D1] Failed to open device (0557/2212), skipping: Access denied (insufficient permissions)
   0.003099     [D2] Checking device 2 of 8 (0557/8021)
   0.003113     [D1] Failed to open device (0557/8021), skipping: Access denied (insufficient permissions)
   0.003122     [D2] Checking device 3 of 8 (0665/5161)
   0.006614     [D2] - VendorID: 0665
   0.006635     [D2] - ProductID: 5161
   0.006646     [D2] - Manufacturer: INNO TECH
   0.006656     [D2] - Product: USB to Serial
   0.006666     [D2] - Serial Number: unknown
   0.006676     [D2] - Bus: 002
   0.006686     [D2] - Device: unknown
   0.006700     [D2] - Device release number: 0002
   0.006706     [D2] Trying to match device
   0.006717     [D3] match_function_regex: matching a device...
   0.006777     [D2] Device matches
   0.006787     [D2] Reading first configuration descriptor
   0.006800     [D3] libusb_kernel_driver_active() returned 0
   0.006821     [D2] Claimed interface 0 successfully
   0.006831     [D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.007589     [D3] send: QGS
   0.150759     [D3] read: QGS
   0.150782     [D2] qx_process_answer: short reply (input.voltage)
   0.151628     [D3] send: QGS
   0.295766     [D3] read: QGS
   0.295789     [D2] qx_process_answer: short reply (input.voltage)
   0.296386     [D3] send: QGS
   0.440769     [D3] read: QGS
   0.440791     [D2] qx_process_answer: short reply (input.voltage)
   0.441628     [D3] send: M
   0.565773     [D3] read: V
   0.566510     [D3] send: QS
   0.882717     [D3] read: (226.5 226.5 226.5 000 49.8 13.5 --.- 00001001
   0.882752     Using protocol: Voltronic-QS 0.07
   0.882774     [D1] upsdrv_initinfo...
   0.883329     [D3] send: QS
   1.199791     [D3] read: (226.5 226.5 226.5 000 50.0 13.5 --.- 00001001
   1.199855     [D2] ups_infoval_set: non numerical value [ups.temperature: --.-]
   1.200659     [D3] send: F
   1.407892     [D3] read: #230.0 003 12.00 50.0
   1.407973     No values for battery high/low voltages
   1.407994     Using 'guesstimation' (low: 10.400000, high: 13.000000)!
   1.408014     Battery runtime will not be calculated (runtimecal not set)
   1.408035     [D1] upsdrv_updateinfo...
   1.408051     [D1] Quick update...
   1.408784     [D3] send: QS
   1.724805     [D3] read: (226.5 226.5 226.5 000 50.0 13.5 --.- 00001001
   1.724897     [D2] dstate_init: sock /run/nut/nutdrv_qx-supra open on fd 9```

tokafondo, 2024-07-03:
```Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Megatec/Q1 protocol USB driver 0.14 (2.8.0)
   0.000000     [D1] debug level is '3'
   0.000506     [D1] Saving PID 20783 into /run/nut/blazer_usb-supra.pid
   0.003316     [D2] Checking device 1 of 8 (0557/2212)
   0.003347     [D1] Failed to open device (0557/2212), skipping: Access denied (insufficient permissions)
   0.003356     [D2] Checking device 2 of 8 (0557/8021)
   0.003369     [D1] Failed to open device (0557/8021), skipping: Access denied (insufficient permissions)
   0.003378     [D2] Checking device 3 of 8 (0665/5161)
   0.003922     [D1] nut_libusb_open get iManufacturer failed, retrying...
   0.007416     [D2] - VendorID: 0665
   0.007439     [D2] - ProductID: 5161
   0.007449     [D2] - Manufacturer: INNO TECH
   0.007460     [D2] - Product: USB to Serial
   0.007470     [D2] - Serial Number: unknown
   0.007480     [D2] - Bus: 002
   0.007490     [D2] - Device: unknown
   0.007500     [D2] - Device release number: 0002
   0.007512     [D2] Trying to match device
   0.007524     [D3] match_function_regex: matching a device...
   0.007599     [D2] Device matches
   0.007608     [D2] Reading first configuration descriptor
   0.007620     [D3] libusb_kernel_driver_active() returned 0
   0.007640     [D2] Claimed interface 0 successfully
   0.007649     [D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.007666     Please note that this driver is deprecated and will not receive
new development. If it works for managing your devices - fine,
but if you are running it to try setting up a new device, please
consider the newer nutdrv_qx instead, which should handle all 'Qx'
protocol variants for NUT. (Please also report if your device works
with this driver, but nutdrv_qx would not actually support it with
any subdriver!)

   0.007681     [D2] Trying megatec protocol...
   0.008522     [D3] send: Q1
   0.142465     [D3] read: Q1
   0.142489     [D2] blazer_status: short reply
   0.142496     [D2] Status read 1 failed
   0.143302     [D3] send: Q1
   0.277475     [D3] read: Q1
   0.277499     [D2] blazer_status: short reply
   0.277511     [D2] Status read 2 failed
   0.278338     [D3] send: Q1
   0.412472     [D3] read: Q1
   0.412494     [D2] blazer_status: short reply
   0.412505     [D2] Status read 3 failed
   0.412515     [D2] Trying mustek protocol...
   0.413346     [D3] send: QS
   0.729608     [D3] read: (226.5 226.5 226.5 000 50.0 13.5 --.- 00001001
   0.729677     [D2] blazer_status: non numerical value [--.-]
   0.729698     [D2] Status read in 1 tries
   0.729711     Supported UPS detected with mustek protocol
   0.730467     [D3] send: F
   0.937610     [D3] read: #230.0 003 12.00 50.0
   0.937668     [D2] Ratings read in 1 tries
   0.938470     [D3] send: I
   1.062490     [D3] read: I
   1.062512     [D2] blazer_vendor: short reply
   1.062523     [D1] Vendor information read 1 failed
   1.063326     [D3] send: I
   1.187494     [D3] read: I
   1.187516     [D2] blazer_vendor: short reply
   1.187527     [D1] Vendor information read 2 failed
   1.188365     [D3] send: I
   1.312541     [D3] read: I
   1.312562     [D2] blazer_vendor: short reply
   1.312574     [D1] Vendor information read 3 failed
   1.312583     Vendor information unavailable
   1.312594     No values provided for battery high/low voltages in ups.conf

   1.312618     Using 'guestimation' (low: 10.400000, high: 13.000000)!
   1.312629     Battery runtime will not be calculated (runtimecal not set)
   1.313362     [D3] send: QS
   1.629509     [D3] read: (226.5 226.5 226.5 000 50.0 13.5 --.- 00001001
   1.629555     [D2] blazer_status: non numerical value [--.-]
   1.629631     [D2] dstate_init: sock /run/nut/blazer_usb-supra open on fd 9```