networkupstools/nut #2515: NUT Server stopped working with Powerwalker UPS

4integration, 2024-07-07:
Hi,

A week ago I setup NUT Server on a Raspberry Pi 3 connected with USB to **Powerwalker VFI 1500 RMG PF1** UPS.
`nut-scanner` detected it like

```
[powerwalker]
        driver = "nutdrv_qx"
        port = "auto"
        vendorid = "0665"
        productid = "5161"
        product = "OLHV1K5"
        serial = "83222307500617"
        bus = "001"
```

Everything was fine and connected 3 different clients on 2 different hosts.
Then suddenly it stopped working and `sudo nut-scanner -U` reports nothing anymore

When I run
```
sudo service nut-server restart
sudo service nut-client restart
sudo systemctl restart nut-monitor
sudo upsdrvctl stop
sudo upsdrvctl start
```

I get:

```
Network UPS Tools - UPS driver controller 2.8.0
Can't open /run/nut/nutdrv_qx-powerwalker.pid: No such file or directory
Can't open /run/nut/nutdrv_qx-auto.pid either: No such file or directory
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
libusb1: Could not open any HID devices: insufficient permissions on everything
No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 nutdrv_qx).

Driver failed to start (exit status=1)

Broadcast message from root@upspi (somewhere) (Sun Jul  7 10:30:09 2024):

Communications with UPS powerwalker@localhost lost

Broadcast message from root@upspi (somewhere) (Sun Jul  7 10:30:09 2024):

Communications with UPS powerwalker@localhost lost

Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
libusb1: Could not open any HID devices: insufficient permissions on everything
No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 nutdrv_qx).

Driver failed to start (exit status=1)

Broadcast message from root@upspi (somewhere) (Sun Jul  7 10:30:14 2024):

UPS powerwalker@localhost is unavailable

Broadcast message from root@upspi (somewhere) (Sun Jul  7 10:30:14 2024):

UPS powerwalker@localhost is unavailable

Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
libusb1: Could not open any HID devices: insufficient permissions on everything
No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 nutdrv_qx).

Driver failed to start (exit status=1)
```

`lsusb` reports
```
Bus 001 Device 004: ID 0424:7800 Microchip Technology, Inc. (formerly SMSC)
Bus 001 Device 003: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Any idea what went wrong?

4integration, 2024-07-08:
I have tried another port, rebooted etc etc
But I agree, USB power of RPi3 I have experienced with this device before using Z-wave stick. Switching to RPi4 solved that.
I have a "large" server in the same location, I will try that out....

Thanks for eye-opener :)

4integration, 2024-07-08:
Made a quick try on the server but no...

```
$ sudo nut-scanner -U
SNMP library not found. SNMP search disabled.
Neon library not found. XML search disabled.
IPMI library not found. IPMI search disabled.
Scanning USB bus.
```

Need to dig deeper...

Mic-Pod, 2024-10-23:
Hi,
I have a similar issue with **VI 3000 CW FR**. Connected via USB to Intel NUC with Ubuntu Server 22.04 (nut 2.7.4). It works fine for about few days and than the communication is breaking up (battery level is reporting 0% and 100% back and forth).
The only thing that solved (temporarily) this issue for me was restarting the whole UPS. But the issue returned...

VI 3000 CW FR is detected as Cypress Semiconductor USB to Serial.
**ups.conf:**
```
[qnapups]
    driver = nutdrv_qx
    protocol = voltronic
    subdriver = cypress
    port = auto
    vendorid = "0665"
    productid = "5161"
    product = "LIHVX3K0"
    bus = "005"
    desc = "PowerWalker"
```

jimklimov, 2024-10-23:
Just in case, check if increasing `pollfreq` would help - thinking if it might be a situation similar to some (CPS) UPS controllers going into power-saving mode if not poked frequently enough, and then taking time to wake up again.

Generally, check also:
* https://github.com/networkupstools/nut/wiki/Troubleshooting-eventual-disconnections-(Data-stale)
* https://github.com/networkupstools/nut/wiki/CyberPower-Systems-(CPS)-know-how

Mic-Pod, 2024-10-23:
@jimklimov Thanks! I will reboot the UPS and correct the setting.

jimklimov, 2024-10-23:
It should not be needed to reboot it, the setting should be in the NUT driver (if it was not only for `usbhid-ups`...) so just update `ups.conf` and reload/restart the driver.

Jim

Mic-Pod, 2024-10-24:
I have to reboot the UPS because the USB port is dead. `lsusb` is no longer detecting the UPS (I've tried connecting UPS to other machines with other OSes but there is no communication over USB). Maybe it is a problem with UPS itself that after some time the USB port is not responding anymore untill a UPS reboot.

jimklimov, 2024-10-24:
Aha... well, there's not too much NUT can do about HW issues :\

Mic-Pod, 2024-10-24:
Yeah... I know :( But maybe it is possible to configure a driver in a better way so that the UPS and its USB port will not freak out and just die...
For now I'm experimenting with drivers blazer_usb and nutdrv_qx (with different protocols and subdrivers).

4integration, 2024-10-26:
@Mic-Pod do you get anything from `nutscanner`?

At first I got a proper "response" but after it "died" it doesn't respond anything, I will try to reboot UPS also

Mic-Pod, 2024-10-27:
@4integration When the USB port died, both `lsusb` and `nut-scanner` were not detecting the device at all.
In order to bring the USB port back online on the UPS I had to disconnect everything from it (unplug all the load and the USB and input cable). After that You just press and hold the OFF button until the LCD goes off. Than just wait a few moments and connect everything back and power on the UPS. The `lsusb` and `nut-scanner` should show the device. Of course this is the procedure that worked in my case - VI 3000 CW.

Mic-Pod, 2024-10-27:
I'm trying different driver configurations, but still have not found a good enough solution. A problem reappeared today:
```
# upsc qnapups
Init SSL without certificate database
Error: Data stale

# nut-scanner
Neon library not found. XML search disabled.
IPMI library not found. IPMI search disabled.
Scanning USB bus.
No start IP, skipping SNMP
No start IP, skipping NUT bus (old connect method)
[nutdev1]
        driver = "blazer_usb"
        port = "auto"
        vendorid = "0665"
        productid = "5161"
        bus = "005"

# upsdrvctl stop
Network UPS Tools - UPS driver controller 2.7.4

# upsdrvctl start
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic Q* USB/Serial driver 0.28 (2.7.4)
USB communication driver 0.33
Device not supported!
Device not supported!
Driver failed to start (exit status=1)

# lsusb
Bus 004 Device 002: ID 04f2:b335 Chicony Electronics Co., Ltd HD WebCam
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port / Mobile Action MA-8910P
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 0665:5161 Cypress Semiconductor USB to Serial
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```
So for now the device is being detected but there is no communication...

4integration, 2024-10-27:
Thanks @Mic-Pod , I will give that a try when it is a good time.

I would like to see some firmware upgrades from PowerWalker but haven't seen any :(

Mic-Pod, 2024-10-27:
I doubt they will do anything about it (sadly...). I hope there is a way to tweak the driver so the USB controller on the UPS does not freak out. Maybe the driver is "talking" too much and some kind of buffer on the UPS side is overflowing...
Unfortunately I don't have that kind of knowledge, but I would love to help tshoot this further to make the PowerWalkers more compatible with nut.