networkupstools/nut #2524: Add new templates options

mikewhippsHarvest, 2024-07-09:
I'd like an option to select a template at runtime when running UPSSTATS.CGI. 

Something like a query string element 
    `upsstats.cgi?template=mytemplate.html`, 
or possibly even 
    `upsstats.cgi?template=/path/to/mytemplate.html`

It would be good to allow the use of `json` or `xml` as an extension instead of `html`, as in
    `upsstats.cgi?template=mytemplate.json`, 
 
The specific use case is to emit data in JSON format to import into PRTG for monitoring while leaving the friendlier HTML version alone for those wanting to query NUT directly in a browser.

jimklimov, 2024-07-10:
This is a really interesting point, would you endeavour to make a PR?

For the time being, especially regarding json, check the third-party projects list, IIRC some emitters were there.

But having some out of the box was backlogged for a while...

mikewhippsHarvest, 2024-07-11:
By PR do you mean Pull Request? My skill level won't run to that, I'm afraid :( I last wrote C (not even C++) about 1984

I have a working JSON template set up as upsstats-single.html. The Content-type: header is wrong, but PRTG doesn't seem to care.

We're still evaluating so this would be nice in the future. I imagine others would find it useful, though.

neg-dean, 2024-09-20:
I'm keen to understand the real timeline to this feature being added to the NUT package. Being able to retrieve the data shown on either the table or graph view of the cgi interface as JSON would be a major advantage allowing us to make a PRTG sensor to bring the data & monitoring into PRTG

jimklimov, 2024-09-20:
Timeline? Rather, milestones: Somebody writes the code, posts a PR, it passes CI and works in fact and does not add regressions and errors / predictable flaws, it gets merged and eventually published as part of a release...

jimklimov, 2026-02-06:
Some goals of this issue (JSON emission from CGI and `upsc`) were addressed in PRs below:
* #3171 
* #3178

closed 2026-02-10 by jimklimov