networkupstools/nut #2525: Variable for STATUS includes spurious HTML

mikewhippsHarvest, 2024-07-09:
While editing the `upsstats-single.html` template to produce JSON output I discovered that the @STATUS@ template variable returns a spurious `<br>` element, as in `ONLINE<br>`

I haven't tested the output of other values for this variable.

This is running on a Raspberry Pi. Here's the driver info block output from `upsc`

driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: ffff
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0463
driver.version: 2.8.0
driver.version.data: MGE HID 1.46
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)

jimklimov, 2024-07-11:
Looking at `git blame`, this `<br>` seems deliberate and is there for a long time, per https://github.com/networkupstools/nut/blame/0d61b63182aedc3be2388126bb05c9f804b352b0/clients/upsstats.c#L170-L173 => https://github.com/networkupstools/nut/commit/8ae7be7c0676c2c3ab3a3984f06572760bfaf05c

I'd say nowadays it should be more properly a `<br/>` tag (opened+closed), and for your question - only added before second+ status token. Maybe a space could be better, but likely less predictable for visual width/wrapping etc.
* Is your question about the `br` tag here due to #2524 - are you trying to parse info from HTML markup for now?

FWIW there is another break tag at https://github.com/networkupstools/nut/blame/0d61b63182aedc3be2388126bb05c9f804b352b0/clients/upsstats.c#L925 and I struggle to see why: it is a fixed end of a table cell... @aquette : any memories here? some crafty browsers liked it so? :)

mikewhippsHarvest, 2024-07-11:
I've created a template to output the data I want in JSON. Apart from limitations on the use of custom templates (for which I opened another issue) this is working well, but...

PRTG has limited capabilities for processing data it imports this way. I can import a single custom text message along with properly formatted numeric data. If I want to import a message like, for example, `UPS is ONLINE` the presence of the `<br>` tag messes up the formatting in the PRTG display, if it will import it at all. 

PRTG provides no facility to manipulate imported text, or substitute from a lookup table, so I have to use the value from the UPS (just `OL` or `OB`, etc.), or use `ONLINE<br>` as is.

If I were editing the template to produce HTML I'd expect a substituted variable to return just the value. I wouldn't want to have to work around an unwanted `<br>`. If I need a `<br>` in my HTML then I can add it in the template.

I wonder if this is left over from some pre-template version of code.

We're still evaluating the software. None of this is a showstopper for us, but it seemed odd, so I raised the issue. I could write a Powershell script to retrieve the data and manipulate it, but that's overkill for one unwanted tag at present.

jimklimov, 2024-07-12:
Fair enough. Since the single STATUS entry in the template is constructed from several tokens active at the time, they should be somehow separated - spaces, "br" tokens, marked up as a JSON or YAML list of quoted strings, etc.

And it is the precompiled program that should add those marks.

Maybe we should add optional extensions to the tags, e.g. `@STATUS:JSON@ or `@STATUS:SPACES@` to cause emission of certain markup.

jimklimov, 2026-02-12:
PR #3318 happens to address this same issue.