networkupstools/nut #2534: VOLT Micro works with blazer_usb but not with nutdrv_qx

mwatney-mars, 2024-07-15:
VOLT POLSKA MICRO 1000VA 600W 1x9ah 

Using "nutdrv_qx" I get the following output and **NO battery percentage** when starting the agent:

```
Using protocol: Mustek 0.07
Can't autodetect number of battery packs [-1/13.20]
Battery runtime will not be calculated (runtimecal not set)
```

When using "blazer_usb" I get battery percentage and the following message when starting the agent: 

```
Supported UPS detected with megatec protocol
Vendor information unavailable
No values provided for battery high/low voltages in ups.conf

Using 'guestimation' (low: 10.400000, high: 13.000000)!
Battery runtime will not be calculated (runtimecal not set)
```

Any plans to add this UPS to the compatibility list anytime soon?

jimklimov, 2024-07-15:
Just in case, which NUT release version?

> Any plans to add this UPS to the compatibility list anytime soon?

Are you ready to work on making it happen? :)

mwatney-mars, 2024-07-15:
Version "2.8.0-7" I guess, checked the package version with "dpkg -l"

#Are you ready to work on making it happen? 

Of course, how can I help?

jimklimov, 2024-07-15:
First of all, gotta find a flag in the sand, marked "Do not approach", and cut a hole in the reactor... :D

On a more serious note, a good first step is to check if the current NUT codebase has not "by chance" already figured out that issue. According to https://github.com/networkupstools/nut/blob/master/NEWS.adoc there were a few changes about battery charge handling in these drivers for NUT releases 2.8.1 and later.

Notes at https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests can help you get the needed prerequisites, build NUT from sources and test the new driver programs (can do right from the build workspace, do not have to overwrite the packaged installation until you like the custom build better).

Still, given that there were a few recent reports like this, about `blazer*` drivers being better in some regards than `nutdrv_qx` supposed to replace them eventually (though none from most recent master/releases, IIRC - distros tend to package and ship older versions), there is a chance that your issue is not yet solved. Then it would be needed to get the programmer's hat on and compare the two logical code paths about how the devices initialize, and in particular how they might differ in the early "guesstimation" part, and thanks to the prepared build/test environment - experiment about making those behaviors the same.

mwatney-mars, 2024-07-15:
I just upgraded to 2.8.1-3.1 and it behaves exactly the same way, so I think this has to be the hard way

dmarczydlo, 2025-04-10:
@mwatney-mars how did you configure the NUT with VOLT UPS? I have some RACK UPS connected to the proxmox by USB.. The `nut-scanner -U` say:

```
	driver = "nutdrv_qx"
	port = "auto"
	vendorid = "0665"
	productid = "5161"
	vendor = "WCHCH544"
	bus = "001"

```

But I cannot establish the connection. The `systemctl status nut-server` alway say:
```
  Loaded: loaded (/lib/systemd/system/nut-server.service; enabled; preset: enabled)
     Active: active (running) since Thu 2025-04-10 21:28:52 CEST; 9min ago
    Process: 230689 ExecReload=/sbin/upsd -c reload -P $MAINPID (code=exited, status=0/SUCCESS)
   Main PID: 230226 (upsd)
      Tasks: 1 (limit: 18852)
     Memory: 624.0K
        CPU: 32ms
     CGroup: /system.slice/nut-server.service
             └─230226 /lib/nut/upsd -F

Apr 10 21:28:52 proxmox upsd[230226]: Running as foreground process, not saving a PID file
Apr 10 21:29:16 proxmox systemd[1]: Reloading nut-server.service - Network UPS Tools - power devices information >
Apr 10 21:29:16 proxmox nut-server[230689]: Network UPS Tools upsd 2.8.0
Apr 10 21:29:16 proxmox upsd[230226]: mainloop: Interrupted system call
Apr 10 21:29:16 proxmox nut-server[230226]: mainloop: Interrupted system call
Apr 10 21:29:16 proxmox nut-server[230226]: SIGHUP: reloading configuration
Apr 10 21:29:16 proxmox systemd[1]: Reloaded nut-server.service - Network UPS Tools - power devices information s>
Apr 10 21:29:16 proxmox upsd[230226]: SIGHUP: reloading configuration
Apr 10 21:33:52 proxmox nut-server[230226]: Can't connect to UPS [ups] (nutdrv_qx-ups): No such file or directory
Apr 10 21:33:52 proxmox upsd[230226]: Can't connect to UPS [ups] (nutdrv_qx-ups): No such file or directory
```

jimklimov, 2025-04-10:
What does your `systemctl status nut-driver@ups` say? Did such service instance appear? if not, please check docs or wiki about nut-driver-enumerator script and service.