networkupstools/nut #2543: something wrong with char

phlnil, 2024-07-20:
Thank you for this gem of sw!
I am getting an error as prompted to my command :systemctl status nut-client on proxmox node (slave).
[image]

Why ? 
config issue ? or something wrong with the reply from ups ?

jimklimov, 2024-07-20:
Never saw that one. A wild guess: is your UPS (or server host name) non-ASCII? Gotta check if byte values over 127 are allowed in such strings.

phlnil, 2024-07-20:
Thank you, I also checked all fields... but no strange chars at all ([A..z]). 
upsc comand from the client retrieve all info flawless.
That chars where could be ?

jimklimov, 2024-07-20:
The message would come from some security hardening per commit https://github.com/networkupstools/nut/commit/2d6b47296c2b9345a603054681a4ecfc902d9ebb with current version seen here:
* https://github.com/networkupstools/nut/blob/5530760f6b050d8d9343b2e197e12ecfefe1629a/common/parseconf.c#L184-L189

One more thing to check for is UTF-8 encoding, your repeating sequence `0xe2 0x80 0xNN` looks like a symptom.
`...0x9c` and `...0x9d` in particular are pretty quotes (left/right), and `...0x93` is a dash - probably you've pasted sample configs from a web page or word/pdf/... document that was rendered friendlier to humans than to computers?

For that matter, check if your `ups.conf` or similar NUT configs were not last saved in an Unicode-capable editor which might embed also the first 3 bytes as a BOM (Byte order mark) `0xEF 0xBB 0xBF`.

phlnil, 2024-07-20:
Thank you ! anyway I pasted using always using _nano_ I will inspect any other file  ups.conf is yet untouched.
Also on the server side could be checked ? (I am just playing (my bad)  for the client/slave side)

jimklimov, 2024-07-20:
Well, here I think the most relevant file to check (first) would be `upsmon.conf` since this is the program which complains.

phlnil, 2024-07-20:
maybe it's solved ! THANK YOU
[image]
maybe the " « » ‘ ’ “ ” was not the right one !

PR #2548 (2024-07-23): Refactor and fix check for process name by PID; update docs for ASCII in config files
Follows-up from #2463

Closes: #2543
* On this one, thought about adding some backlog of known bytes to report particularly UTF-8 sequence (suspicion), but that was too intrusive of a change
closed 2024-07-23 by jimklimov