networkupstools/nut #2547: weird problem on raspbian

ilker-aktuna, 2024-07-22:
Hi,

I have a weird problem and Github issues might not be the correct place to write this. Please forgive me and let me know where to write it in that case.

The problem:

I have a working NUT setup on a raspberry pi 3B with raspbian bullseye.
Now I need to move it to another RPI 3B, but the OS is older on this one (buster)
In fact, I also have a working NUT setup on this RPI (the one that I am moving to) but it is using a different UPS.
So , I have 2 RPI ; lets call them RPI-a and RPI-b
Both RPI has working NUT setup but with different UPS. RPI-a is connected to UPS-a , RPI-b is connected to UPS-b

Now  I want to connect RPI-b to UPS-a

When I connect the UPS with its USB cable to RPI-b, it does not recognize it as a device.
Does not show up in "lsusb" , and all I see in "dmesg" is :

root@raspbx:/boot# dmesg  | grep usb
[  841.494024] usb 1-1.4: new low-speed USB device number 8 using dwc_otg
[  842.044035] usb 1-1.4: new low-speed USB device number 9 using dwc_otg
[  842.494267] usb 1-1-port4: attempt power cycle
[  843.154035] usb 1-1.4: new low-speed USB device number 10 using dwc_otg
[  843.694036] usb 1-1.4: new low-speed USB device number 11 using dwc_otg

so I collected these information below from each RPI when UPS-a is connected with a USB cable.

Working PI:
```
root@proxy:~# uname -a
Linux proxy 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

root@proxy:~# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

root@proxy:~# lsusb
Bus 001 Device 004: ID 0001:0000 Fry's Electronics Mroot@proxy:~# dmesg  | grep usbEC0003
Bus 001 Device 005: ID 0424:7800 Microchip Technology, Inc. (formerly SMSC) 
Bus 001 Device 003: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hubroot@proxy:~# dmesg  | grep usb

root@proxy:~# dmesg  | grep usb
[    4.704890] usb 1-1.1.1: new high-speed USB device number 5 using dwc_otg
[    4.805777] usb 1-1.1.1: New USB device found, idVendor=0424, idProduct=7800, bcdDevice= 3.00
[    4.805824] usb 1-1.1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0

root@proxy:~# dpkg -l  | grep libusb
ii  libgusb2:arm64                       0.3.5-1                          arm64        GLib wrapper around libusb1
ii  libusb-1.0-0:arm64                   2:1.0.24-3                       arm64        userspace USB programming library
ii  libusbmuxd6:arm64                    2.0.2-3                          arm64        USB multiplexor daemon for iPhone and iPod Touch devices - library
```

not working PI:
```
root@raspbx:/boot# uname -a
Linux raspbx 5.10.63-v7+ #1496 SMP Wed Dec 1 15:58:11 GMT 2021 armv7l GNU/Linux

root@raspbx:/boot# cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

root@raspbx:/boot# lsusb 
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

root@raspbx:/boot# dmesg  | grep usb
[  841.494024] usb 1-1.4: new low-speed USB device number 8 using dwc_otg
[  842.044035] usb 1-1.4: new low-speed USB device number 9 using dwc_otg
[  842.494267] usb 1-1-port4: attempt power cycle
[  843.154035] usb 1-1.4: new low-speed USB device number 10 using dwc_otg
[  843.694036] usb 1-1.4: new low-speed USB device number 11 using dwc_otg

root@raspbx:/boot# dpkg -l  | grep libusb
ii  libgusb2:armhf                                                   0.3.0-1                                 armhf        GLib wrapper around libusb1
ii  libusb-0.1-4:armhf                                               2:0.1.12-32                             armhf        userspace USB programming library
ii  libusb-1.0-0:armhf                                               2:1.0.22-2                              armhf        userspace USB programming library
ii  libusb-1.0-0-dev:armhf                                           2:1.0.22-2                              armhf        userspace USB programming library development files
ii  libusb-1.0-doc                                                   2:1.0.22-2                              all          documentation for userspace USB programming
ii  libusbmuxd4:armhf                                                1.1.0~git20181007.07a493a-1             armhf        USB multiplexor daemon for iPhone and iPod Touch devices - library
```

I don't know where to focus so I could only try:
- different cables
- different USB ports on RPI
- removing all other USB devices 
- tried to update libusb , libusb2 and libusb-dev packages but raspbian reports that already the latest version is installed.

Could you please tell me where to focus ?

note:
1. both UPS-a and UPS-b are identified as "Fry's Electronics" on RPI-a
2. UPS-b is identified as "Fry's Electronics" on RPI-b

jimklimov, 2024-07-22:
I'd first focus on physical aspects - try different ports, cables, check if the same cable and UPS that work on one RPi misbehave on the other. Maybe even switch SD cards and boot with the "successful" OS on the misbehaving one. It may be possible that it is just broken (some soldering went loose, GPIO overlap, firmware wrong, etc.)

We had a fair amount of reports of issues with Raspberries and allegedly RPi3 and RPi5 had problems with USB, RPi4 was better. But I still can't say if that says more about popularity of the platform (so we see many reports with them due to frequent use, reporter's bias) or really about its quality.

ilker-aktuna, 2024-07-23:
thanks. As I wrote , I have already tried diffrent cables. I also tried all ports.
There is no physical issue on the PI. Other USB devices work fine.
Also other UPS is recognized well.

" switch SD cards and boot with the "successful" OS on the misbehaving one" --> that's something I did not try. I will try and report back.

is there a USB OTG driver that I should update on the Raspbian ?

jimklimov, 2024-07-23:
Can't really say, don't have much experience with Raspberries. I know some HATs require pre-loaded files (DTB, kernel settings args, or whatnot) to function.

ilker-aktuna, 2024-07-23:
what else can I check ?

jimklimov, 2024-07-23:
Looking more at your original post, I see that both devices claim to be Raspberry Pi 3B (or so you say), but have different USB Hub identifiers. Probably different board revisions and maybe variation due to that. Also the not-working one seems to have an Ethernet adapter on the USB bus - is that the built-in port or a separate dongle (can you try seeing the UPS with this port unplugged then)?

One more thought is about USB subsystem perhaps not having enough electrical power to establish a stable connection (e.g. the Ethernet port eats some away?) Can you try feeding this RPi from a more powerful source, to confirm or rule out this possibility?

ilker-aktuna, 2024-07-25:
I just had time to try different combinations.
First, I tried with 2 other USB power adapters. Both of them are able to provide up to 45W but the result did not change. RPI3B with Raspbian Buster can not detect what's plugged on USB.

Then as you asked, I checked the board revisions. 
The working setup's PI is:
Raspberry Pi 3 Model B Plus Rev 1.3
Not working setup has:
Raspberry Pi 3 Model B Rev 1.2

So they are both RPI3B but one is B+

` Also the not-working one seems to have an Ethernet adapter on the USB bus - is that the built-in port or a separate dongle (can you try seeing the UPS with this port unplugged then)?`

there is nothing else on the external USB ports of the PI. I assume the ethernet adapter is an internal connection from USB hub.

I tried the working OS SD card on the not-working PI as you suggested.
It works !
So the working setup's SD card is successful on the not working RPI.

In conclusion, the problem is not about HW but the OS , kernel or drivers.
I have to focus on the software. But what can I try ?
I can't upgrade the buster to bullseye. Maybe a kernel upgrade ?
Is there a possibility of smooth kernel upgrade on the OS ?
Or can I install a kernel module (driver) to support this ?

jimklimov, 2024-07-25:
That's more of an OS question. I gather `apt dist-upgrade` is an option to try; you may have to set `/etc/apt...` sources to archived package repo URLs, if the distribution is no longer served as active.

If you have spare SD cards, you might want to `dd` a clone of the older system in case you would need to roll back.

ilker-aktuna, 2024-07-29:
"apt dist-upgrade" didn't change the release.
It updated all packages including kernel
kernel came to 5.10.103 from 5.10.63
release is still buster

another "dist-upgrade" does not do anything.
is there a way to upgrade to bullseye ?

ilker-aktuna, 2024-07-29:
ok, now following :
```

1. update all the packages on your current version. apt dist-upgrade
2. edit /etc/apt/souces.list Don't delete any lines just copy them all and chage buster to bullseye save the file.
3. run apt update make sure there are no missing repos or mistyped lines. It should offer you 1000+ packages to upgrade and a few hundred to remove. Check your applications do not need to rely on things like that.
4. Once you are happy with the changes run apt upgrade and wait for everything to upgrade. A newer kernel will be installed so once it is all done, and assuming there are no errors, attempt a reboot and see if it all comes up again.
5. Assuming it all works. Once you are happy it is working as expected, you can run apt autoremove to remove any unnecessary packages. Over time all your updates will come from bullseye and when that happens you can comment out or remove the lines in /etc/apt/sources.list with buster in them

```

let's see...

ilker-aktuna, 2024-07-29:
with that method I get this error:
```
1442 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://repos.influxdata.com/debian buster InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY D8FF8E1F7DF8B07E
W: Failed to fetch https://repos.influxdata.com/debian/dists/buster/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY D8FF8E1F7DF8B07E
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

I think I can solve the signature issue and try upgrade again.
But in any case, changing the kernel (5.10.63 to 5.10.103) did not solve the problem.
So ;
1. this is not HW issue
2. changing kernel did not resolve

I believe, I need a better USB driver but I know nothing about drivers on linux. 
I'd appreciate if anyone could jump in and guide me about this USB driver.
maybe just change the USB hub driver ???

ilker-aktuna, 2024-07-31:
UPDATE:
I've successfully updated kernel to 6.6
It was easy with the "rpi-update" command

But the USB device is still not detected:

[quote]root@raspbx:~# uname -a
Linux raspbx 6.6.42-v7+ #1787 SMP Thu Jul 25 17:42:40 BST 2024 armv7l GNU/Linux
root@raspbx:~# lsusb 
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@raspbx:~#
root@raspbx:~# dmesg | grep usb
[   62.823249] usb 1-1.4: new low-speed USB device number 4 using dwc_otg
[   63.363184] usb 1-1.4: new low-speed USB device number 5 using dwc_otg
[   63.823396] usb 1-1-port4: attempt power cycle
[   64.483159] usb 1-1.4: new low-speed USB device number 6 using dwc_otg
[   65.033209] usb 1-1.4: new low-speed USB device number 7 using dwc_otg[/quote]

isn't that interesting / weird ?

If I put the bullseye SD card to the same RPI 3B, it can detect the USB device.
But when I upgrade my old buster to bullseye , and then even update the kernel, it can not detect the same USB device.

this seems like a missing or old usb device driver (kernel mod) to me.
Any ideas ?

jimklimov, 2024-07-31:
Weird indeed. Try to compare `lsmod` outputs from the two boot-up cards?

ilker-aktuna, 2024-07-31:
well, I could not find anything about usb  in lsmod output.

this is the working one:
```
root@proxy:~# lsmod 
Module                  Size  Used by
xt_conntrack           16384  1
rfcomm                 53248  4
nft_chain_nat          16384  3
xt_MASQUERADE          16384  1
nf_nat                 49152  2 nft_chain_nat,xt_MASQUERADE
nf_conntrack_netlink    49152  0
nf_conntrack          147456  4 xt_conntrack,nf_nat,nf_conntrack_netlink,xt_MASQUERADE
nf_defrag_ipv6         20480  1 nf_conntrack
nf_defrag_ipv4         16384  1 nf_conntrack
xfrm_user              49152  1
xfrm_algo              20480  1 xfrm_user
xt_addrtype            16384  2
nft_compat             20480  4
nf_tables             237568  57 nft_compat,nft_chain_nat
nfnetlink              20480  4 nft_compat,nf_conntrack_netlink,nf_tables
br_netfilter           32768  0
bridge                204800  1 br_netfilter
cmac                   16384  3
algif_hash             16384  1
aes_arm64              16384  3
aes_generic            36864  1 aes_arm64
algif_skcipher         16384  1
af_alg                 28672  6 algif_hash,algif_skcipher
bnep                   24576  2
hci_uart               53248  1
btbcm                  24576  1 hci_uart
bluetooth             565248  31 hci_uart,btbcm,bnep,rfcomm
ecdh_generic           16384  2 bluetooth
ecc                    36864  1 ecdh_generic
libaes                 16384  3 aes_arm64,bluetooth,aes_generic
overlay               126976  0
8021q                  32768  0
garp                   20480  1 8021q
stp                    16384  2 bridge,garp
llc                    16384  3 bridge,stp,garp
vc4                   331776  5
brcmfmac              331776  0
snd_soc_hdmi_codec     20480  1
drm_display_helper     16384  1 vc4
cec                    53248  1 vc4
drm_dma_helper         20480  1 vc4
brcmutil               24576  1 brcmfmac
drm_kms_helper        204800  3 drm_dma_helper,vc4
snd_soc_core          274432  2 vc4,snd_soc_hdmi_codec
cfg80211              925696  1 brcmfmac
snd_compress           20480  1 snd_soc_core
snd_pcm_dmaengine      20480  1 snd_soc_core
syscopyarea            16384  1 drm_kms_helper
sysfillrect            16384  1 drm_kms_helper
raspberrypi_hwmon      16384  0
sysimgblt              16384  1 drm_kms_helper
fb_sys_fops            16384  1 drm_kms_helper
bcm2835_codec          49152  0
bcm2835_v4l2           45056  0
bcm2835_isp            28672  0
rfkill                 32768  6 bluetooth,cfg80211
bcm2835_mmal_vchiq     36864  3 bcm2835_codec,bcm2835_v4l2,bcm2835_isp
v4l2_mem2mem           45056  1 bcm2835_codec
videobuf2_dma_contig    20480  2 bcm2835_codec,bcm2835_isp
videobuf2_vmalloc      16384  1 bcm2835_v4l2
videobuf2_memops       16384  2 videobuf2_vmalloc,videobuf2_dma_contig
i2c_bcm2835            16384  0
videobuf2_v4l2         32768  4 bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,bcm2835_isp
videobuf2_common       69632  8 bcm2835_codec,videobuf2_vmalloc,videobuf2_dma_contig,videobuf2_v4l2,bcm2835_v4l2,v4l2_mem2mem,videobuf2_memops,bcm2835_isp
[... 91 lines cut ...]

working:
```
ipp-usb/oldstable,now 0.9.17-3+b4 arm64 [installed,automatic]
libgusb2/oldstable,now 0.3.5-1 arm64 [installed,automatic]
libusb-1.0-0/oldstable,now 2:1.0.24-3 arm64 [installed,automatic]
libusbmuxd6/oldstable,now 2.0.2-3 arm64 [installed,automatic]
usb-modeswitch-data/oldstable,oldstable,now 20191128-3 all [installed,automatic]
usb-modeswitch/oldstable,now 2.6.1-1 arm64 [installed]
usb.ids/now 2023.01.16-0+deb11u1 all [installed,upgradable to: 2024.01.20-0+deb11u1]
usbutils/oldstable,now 1:013-3 arm64 [installed]
```
not-working:
```
libgusb2/oldoldstable,now 0.3.0-1 armhf [installed]
libusb-1.0-0-dev/oldoldstable,now 2:1.0.22-2 armhf [installed]
libusb-1.0-0/oldoldstable,now 2:1.0.22-2 armhf [installed]
libusb-1.0-doc/oldoldstable,now 2:1.0.22-2 all [installed,automatic]
libusbmuxd4/oldoldstable,now 1.1.0~git20181007.07a493a-1 armhf [installed,automatic]
usb-modeswitch-data/oldoldstable,now 20170806-2 all [installed,automatic]
usb-modeswitch/oldoldstable,now 2.5.2+repack0-2 armhf [installed]
usb.ids/oldoldstable,now 2019.07.27-0+deb10u1 all [installed,automatic]
usbmuxd/oldoldstable,now 1.1.1~git20181007.f838cf6-1 armhf [installed,automatic]
usbutils/oldoldstable,now 1:010-3 armhf [installed]
```

jimklimov, 2024-07-31:
Well, the module lists do differ significantly. I've sorted the two and passed into `meld`, to see that among other modules that seem less relevant (backlight , sound, drm*, aes*, alg* etc.) some present on "proxy" only are: i2c_bcm2835 i2c_dev rfcomm cec vc4 whatever those are. At least the names look hardware-related.

IIRC Raspberries also have a lot of impact from DTB files and other configuration on the card which passes device information and kernel arguments to the Linux which is eventually loaded. Do those differ in an interesting manner?

ilker-aktuna, 2024-07-31:
> Well, the module lists do differ significantly. I've sorted the two and passed into `meld`, to see that among other modules that seem less relevant (backlight , sound, drm*, aes*, alg* etc.) some [...]

Ok but I can't try every possibility. I need some clue to test 1-2 of missing modules.

> IIRC Raspberries also have a lot of impact from DTB files and other configuration on the card which passes device information and kernel arguments to the Linux which is eventually loaded. Do those differ in an interesting manner?

Where are those files ?

jimklimov, 2024-07-31:
> Where are those files ?

Well, it is your Raspberry. I've only read about them and saw at colleagues' rigs, but did not really get to setting up any myself.

My understanding is that the SD card is usually FAT-formatted for the boot part, and there are text files on it... Can't say more.