networkupstools/nut #2555: nut-scanner unable to find libraries

jacksonattwood, 2024-07-25:
MacOS Ventura | NUT v. 2.8.2 from MacPorts

When trying to run ```sudo nut-scanner -DDDDDD -C > /tmp/nutscan.log 2>&1``` on MacOS Ventura (installed via macports) I get the following output:

```
Cannot load USB library (libusb-1.0.so) : file not found. USB search disabled.
Cannot load XML library (libneon.so) : file not found. XML search disabled.
Cannot load NUT library (libupsclient.so) : file not found. NUT search disabled.
```

FYI I am able to launch upsd and upsmon, both of which are up to date. No matter what, upsd reports back that it cannot connect to the ups:
```
listening on 0.0.0.0 port 3493
Can't connect to UPS [OFFICE] (usbhid-ups-OFFICE): No such file or directory
Found 1 UPS defined in ups.conf
```

After running ```sudo /opt/local/bin/usbhid-ups -DD -a OFFICE```
```
Network UPS Tools - Generic HID driver 0.53 (2.8.2)
USB communication driver (libusb 1.0) 0.47
   0.000000	[D1] upsdrv_makevartable...
   0.000048	[D1] Using USB implementation: libusb-1.0.27 (API: 0x100010a)
   0.000117	[D1] Network UPS Tools version 2.8.2 (release/snapshot of 2.8.2) built with Apple clang version 14.0.3 (clang-1403.0.22.14.1); Target: x86_64-apple-darwin22.6.0; Thread model: posix and configured with flags: --prefix=/opt/local --with-serial=no --with-usb=auto --with-snmp=no --with-hal=auto --with-cgi=no --with-neonxml=no --with-ssl=yes --with-ipv6=no --with-statepath=/opt/local/var/db/ups --with-pidpath=/opt/local/var/run --with-hotplug-dir=/opt/local/etc/nut/hotplug --with-udev-dir=/opt/local/etc/nut/udev --with-user=_nut --with-group=_nut
   0.000123	[D1] debug level is '2'
   0.002064	[D1] Succeeded to become_user(_nut): now UID=504 GID=502
   0.002121	[D1] upsdrv_initups (non-SHUT)...
   0.002130	[D2] Initializing an USB-connected UPS with library libusb-1.0.27 (API: 0x100010a) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.47')
   0.017947	[D2] Checking device 1 of 3 (0764/0501)
   0.046693	[D2] - VendorID: 0764
   0.046707	[D2] - ProductID: 0501
   0.046710	[D2] - Manufacturer: CPS
   0.046712	[D2] - Product: BR1000ELCD
   0.046716	[D2] - Serial Number: unknown
   0.046719	[D2] - Bus: 020
   0.046722	[D2] - Bus Port: 006
   0.046725	[D2] - Device: 006
   0.046727	[D2] - Device release number: 0001
   0.046730	[D2] Trying to match device
   0.046740	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.046756	[D2] Device matches
   0.046759	[D2] Reading configuration descriptor 1 of 1
   0.046889	[D2] successfully set kernel driver auto-detach flag
   0.047940	[D2] failed to claim USB device: Access denied (insufficient permissions)
   0.047973	[D1] failed to detach kernel driver from USB device: Access denied (insufficient permissions)
   0.048771	[D2] failed to claim USB device: Access denied (insufficient permissions)
   0.048799	[D1] failed to detach kernel driver from USB device: Access denied (insufficient permissions)
   0.049709	[D2] failed to claim USB device: Access denied (insufficient permissions)
   0.049742	[D1] failed to detach kernel driver from USB device: Access denied (insufficient permissions)
   0.050485	[D2] failed to claim USB device: Access denied (insufficient permissions)
   0.050512	[D1] failed to detach kernel driver from USB device: Access denied (insufficient permissions)
   0.050518	Can't claim USB device [0764:0501]@0/0/0: Access denied (insufficient permissions)
   0.050527	upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
```

jimklimov, 2024-07-25:
Regarding `nut-scanner` library search and MacOS (two separate problems that happened to intersect), this was recently addressed in a series of PRs merged to `master` branch, so the solution would be part of a next release. For more details, you can trawl the labels assigned to this issue.

If you are able to build NUT, you can give it a shot roughly following the instructions at https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

So far there was little recent feed-back from the MacOS community, so I can only reflect on a limited experience with VMs provided by CI farms, where HomeBrew packaging is used - and their tooling sufficed to get NUT built and pass CI tests.

As for the failed driver start-up, for a quick test try `-u root` on command line (or `user=root` in `ups.conf`), my first guess would be that the run-time user account (`Succeeded to become_user(_nut): now UID=504 GID=502`) does not have permissions to the device filesystem node for the USB entity. NUT sources do generate files with lists of USB vendor and product ID pairs for several platforms that automate permissions assignment (udev, upower, devd, etc.) but I am not sure if any of those is available on MacOS and/or in the delivered packages.

A second guess would be that someone else (kernel or program, possibly virtualization pass-through) got to handle the device first, especially as it is classed as a HID device (so maybe generic kernel handler for HID). The permissions assignment solutions noted above usually cover this part too (which handler to auto-launch, if any).

jacksonattwood, 2024-07-26:
> If you are able to build NUT, you can give it a shot roughly following the instructions at https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

This is the first thing I tried, however as soon as it tries to ```make```, it throws heaps of errors and exits:
```
./ci_build.sh: configure phase complete (0)
Configuration finished, starting make
For parallel builds, '-j 24 -l 96.0' options would be used
Making all in include
NUT_VERSION: "2.8.2-726-g95e2d98c1"
test -f "nut_version.h" || cp "nut_version.h.tmp" "nut_version.h"
cmp -s "nut_version.h.tmp" "nut_version.h" || cp "nut_version.h.tmp" "nut_version.h"
rm -f "nut_version.h.tmp"
make  all-am
make[2]: Nothing to be done for `all-am'.
Making all in common
make  all-am
  CC       parseconf.lo
  CC       libcommon_la-state.lo
  CC       libcommon_la-upsconf.lo
  CC       libcommon_la-str.lo
  CC       libcommon_la-common.lo
  CC       libcommonclient_la-str.lo
  CC       libcommonclient_la-state.lo
  CC       libcommonclient_la-common.lo
  CXX      libnutconf_la-nutconf.lo
  CXX      libnutconf_la-nutwriter.lo
  CXX      libnutconf_la-nutstream.lo
  CXX      libnutconf_la-nutipc.lo
  CC       libcommonstr_la-str.lo
  CC       libcommonstr_la-common.lo
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
2 errors generated.
2 errors generated.
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
make[2]: *** [libcommonstr_la-str.lo] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [libcommon_la-str.lo] Error 1
2 errors generated.
2 errors generated.
make[2]: *** [parseconf.lo] Error 1
2 errors generated.
2 errors generated.
make[2]: *** [libcommonclient_la-str.lo] Error 1
2 errors generated.
[... 66 lines cut ...]
However, when launching ```upsd``` as sudo, I get this:
```
jacksonattwood@motioneye etc % sudo upsd
Network UPS Tools upsd 2.8.2
fopen /opt/local/var/db/ups/upsd.pid: No such file or directory
Could not find PID file '/opt/local/var/db/ups/upsd.pid' to see if previous upsd instance is already running!
/opt/local/etc/upsd.conf is world readable
listening on 0.0.0.0 port 3493
Can't connect to UPS [OFFICE] (usbhid-ups-OFFICE): No such file or directory
Found 1 UPS defined in ups.conf
/opt/local/etc/upsd.users is world readable
jacksonattwood@motioneye etc % 
```
If it helps, here's my ```ups.conf```
```
[OFFICE]
    driver = usbhid-ups
    port = auto
    desc = "Office UPS"
    vendorid = "0764*"
    user = root
```
> A second guess would be that someone else (kernel or program, possibly virtualization pass-through) got to handle the device first, especially as it is classed as a HID device (so maybe generic [...]

Interesting to know, Virtualbox always fails to claim ownership of the device, so I can't passthrough to my Ubuntu VM

jimklimov, 2024-07-26:
Regarding the warnings: did you export a `HOMEBREW_PREFIX` to let `ci_build.sh` fit that build system? For example,
````
eval "$(brew shellenv)"
````
should do it.

At worst, specifically and clumsily:
````
HOMEBREW_PREFIX="`brew config | grep HOMEBREW_PREFIX: | awk '{print $2}'`"
````

Updating the wiki and in-source docs to clarify this bit :)

jacksonattwood, 2024-07-26:
> Regarding the warnings: did you export a `HOMEBREW_PREFIX` to let `ci_build.sh` fit that build system? For example,
> 
> ```
> eval "$(brew shellenv)"
> ```

So that didn't do it :(
 
> At worst, specifically and clumsily:
> 
> ```
> HOMEBREW_PREFIX="`brew config | grep HOMEBREW_PREFIX: | awk '{print $2}'`"
> ```

Mighty unfortunate news is that trying both of those return the same error:
```
Making all in common
make  all-am
  CC       parseconf.lo
  CC       libcommon_la-state.lo
  CC       libcommon_la-str.lo
  CC       libcommon_la-upsconf.lo
  CC       libcommonclient_la-state.lo
  CC       libcommon_la-common.lo
  CC       libcommonclient_la-str.lo
  CC       libcommonclient_la-common.lo
  CXX      libnutconf_la-nutconf.lo
  CXX      libnutconf_la-nutstream.lo
  CXX      libnutconf_la-nutwriter.lo
  CXX      libnutconf_la-nutipc.lo
  CC       libcommonstr_la-str.lo
  CC       libcommonstr_la-common.lo
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
2 errors generated.
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
2 errors generated.
make[2]: *** [libcommon_la-str.lo] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [libcommonstr_la-str.lo] Error 1
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
2 errors generated.
make[2]: *** [libcommonclient_la-str.lo] Error 1
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
2 errors generated.
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
2 errors generated.
make[2]: *** [libcommon_la-state.lo] Error 1
make[2]: *** [libcommon_la-upsconf.lo] Error 1
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
2 errors generated.
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
2 errors generated.
make[2]: *** [parseconf.lo] Error 1
make[2]: *** [libcommonclient_la-state.lo] Error 1
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
2 errors generated.
make[2]: *** [libcommon_la-common.lo] Error 1
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
2 errors generated.
make[2]: *** [libcommonclient_la-common.lo] Error 1
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
2 errors generated.
make[2]: *** [libcommonstr_la-common.lo] Error 1
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
2 errors generated.
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
2 errors generated.
make[2]: *** [libnutconf_la-nutstream.lo] Error 1
make[2]: *** [libnutconf_la-nutipc.lo] Error 1
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
2 errors generated.
make[2]: *** [libnutconf_la-nutwriter.lo] Error 1
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
error: include location '/usr/local/include' is unsafe for cross-compilation [-Werror,-Wpoison-system-directories]
2 errors generated.
make[2]: *** [libnutconf_la-nutconf.lo] Error 1
make[1]: *** [all] Error 2
make: *** [all-recursive] Error 1
```

jimklimov, 2024-07-26:
Did you fully re-run `ci_build.sh` after setting the environment? Try also `rm -f configure Makefile` before such re-running, to make sure brew'ed packages are taken into account at all steps of the way.

jacksonattwood, 2024-07-27:
So what I was doing before was removing the entire repo, then cloning into it again and trying again. 
I've tried:
- ```./ci_build.sh```
- ```sudo ./ci_build.sh```
- ```./ci_build.sh inplace```
- ```sudo ./ci_build.sh inplace```

However, none of these have changed the error at all.

jimklimov, 2024-07-28:
Even stranger, then... Maybe something else should be detected for auto-magic and is not.

You can try to pre-export CFLAGS with the anti-warning values set in the script, at least...

jimklimov, 2024-09-12:
@jacksonattwood : cheers, did you have any luck progressing with this? On the MacOS VMs in the cloud, I did see the error about "poison system" include directories, hence the specific HomeBrew handling in the `ci_build.sh` script, and details posted above earlier. Other than that, I don't know much about the platform...

jacksonattwood, 2024-09-12:
Nah unfortunately not, I ended up passing it through to a Linux VM, however I have to run Virtualbox as sudo via cli for it to work

jimklimov, 2025-03-19:
Note: in recent recipe clean-up, `ci_build.sh` should take care of "poison system include directories" in more cases, making it easier to use for dev/testing and not only CI runs. So maybe the current master branch would be easier to build and test about this issue.

jimklimov, 2025-03-19:
One more point from previous discussions on other platforms - in some cases, packages with libraries deliver a detailed API version in their filename, e.g. `libsomething.so.1.2.3` (prone to change due to updates) and the short-hand symlinks like `libsomething.so.1` (for major API version) may be present in a development package along with header files and other resources.

Current NUT is expected to try and find both (short name, and whatever longer name it could have resolved to on the build system); older versions might have picked one of the choices and in some cases it was not satisfied by the system a NUT package was deployed to.

An administrative workaround however should be possible, to just create the short-hand symbolic links near the existing libraries, pointing to them, e.g.
```
:; ln -s libsomething.so.1.2.3 /usr/lib/libsomething.so.1
```

closed 2025-04-03 by jimklimov