networkupstools/nut #2559: reconciling manual nut build w/ apt

kirsch33, 2024-07-27:
Hi all,

I followed the instructions here:
https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

To in-place upgrade my `nut-client` install to `2.8.2` and it went well. However, `apt` still sees my `nut-client` package as `2.8.0`.

i am certain that if a newer package, say `2.8.1` were to get pushed to the Debian repos, `apt` would overwrite my manual deployment with what it thinks is a newer version.

whats the recommended course of action to reconcile the two?

jimklimov, 2024-07-27:
Apt deals with packaging metadata, which is coincidentally related (or not) to actual installed binaries.

To "reconcile" you can either build your own until packaged version surpasses 2.8.2 (not sure if Debian ships significantly changed packages, other than small patching of whatever version of software like NUT a distro release was initially issued with), or use and adapt their recipes to build a 2.8.2 via `debuild`. Maybe you can tell `apt` to freeze your package version for NUT and so not auto-update it?..

kirsch33, 2024-07-27:
@jimklimov thanks for the reply

for now, I just did `apt-mark hold` on `nut` `nut-client` `nut-monitor` and `nut-server` thinking that will be enough. i also made a note in my server documentation to remove this at such a time that 2.8.3 is released.

if anyone else comes across this and has any better ideas, let me know, but closing for now!

closed 2024-07-27 by kirsch33

jimklimov, 2024-07-28:
I updated the wiki article with ideas from this discussion, thanks!

kirsch33, 2024-07-28:
> I updated the wiki article with ideas from this discussion, thanks!

well, i should have brought this up with my reply earlier but under the "Replacing a systemd deployment" heading the following info might also be good additions:

 i had to do some googling to figure out that i needed to use the `--with-libsystemd` build flag in addition to `--enable-inplace-runtime` to get proper logs to output when running, for example, `systemctl status nut-client` 

before this, i was getting the message:
`upsnotify: failed to notify about state 2: no notification tech defined`

for the `--with-libsystemd` flag to work, `apt install libsystemd-dev` was required.

but all works great now!

jimklimov, 2024-07-28:
I think having the dependency installed was the important part, and that's documented in config-prereqs.