networkupstools/nut #2569: Switched from blazer_usb to nutdrv_qx - Battery voltage repporting wrong value

ajcorrea, 2024-07-30:
Switched from blazer_usb to nutdrv_qx 

I have a Intelbras DNB1500 (Megatec protocol).

With blazer_usb the battery voltage is ok. (41 V) and with nutdrv_qx is repporting 2.28

NUT version 0.13.1 (Home assistant add-on)

```
--- Logs
-----------------------------------------------------------
 Add-on: Network UPS Tools
 Manage battery backup (UPS) devices
-----------------------------------------------------------
 Add-on version: 0.13.1
 You are running the latest version of this add-on.
 System: Debian GNU/Linux 11 (bullseye)  (amd64 / qemux86-64)
 Home Assistant Core: 2024.7.4
 Home Assistant Supervisor: 2024.06.2
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[15:41:10] INFO: Configuring timezone (America/Sao_Paulo)...
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nut.sh
[15:41:11] INFO: Setting mode to netserver...
[15:41:11] INFO: Generating /etc/nut/upsd.users...
[15:41:11] INFO: Configuring user: admin
[15:41:11] INFO: Password is NOT in the Have I Been Pwned database! Nice!
[15:41:12] INFO: Configuring Device named Intelbras...
[15:41:12] INFO: Starting the UPS drivers...
Using protocol: Megatec 0.06
No values for battery high/low voltages
Using 'guesstimation' (low: 31.200000, high: 39.000000)!
Battery runtime will not be calculated (runtimecal not set)
Network UPS Tools - UPS driver controller 2.8.0
cont-init: info: /etc/cont-init.d/nut.sh exited 0
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun upsd (no readiness notification)
services-up: info: copying legacy longrun upsmon (no readiness notification)
s6-rc: info: service legacy-services successfully started
[15:41:15] INFO: Starting the UPS information server...
   0.000000	fopen /run/nut/upsd.pid: No such file or directory
   0.000021	Could not find PID file '/run/nut/upsd.pid' to see if previous upsd instance is already running!

   0.000357	[D1] debug level is '1'
   0.000430	listening on 0.0.0.0 port 3493
   0.000782	Connected to UPS [Intelbras]: nutdrv_qx-Intelbras
   0.001694	Running as foreground process, not saving a PID file
[15:41:16] INFO: Starting the UPS monitor and shutdown controller...
   0.000000	fopen /run/nut/upsmon.pid: No such file or directory
   0.000132	Could not find PID file to see if previous upsmon instance is already running!

   0.000702	Using power down flag file /etc/killpower
   0.001298	UPS: Intelbras@localhost (primary) (power value 1)
   0.001360	[D1] debug level is '1'
   0.001397	[D1] debug level is '1'
   0.001403	Warning: running as one big root process by request (upsmon -p)
   0.001926	[D1] Saving PID 460 into /run/nut/upsmon.pid
Init SSL without certificate database
   0.009611	[D1] Trying to connect to UPS [Intelbras@localhost]
   0.506864	User upsmonmaster@127.0.0.1 logged into UPS [Intelbras]
   0.013303	[D1] Logged into UPS Intelbras@localhost
```

[image]

[image]

jimklimov, 2024-07-30:
Would you be able to show also the start-up log (guesstimation part is of interest) from `blazer_usb` driver?

It seems like the two drivers default to different values for the mix of battery cell voltage, cells per pack, and packs per UPS; possibly different protocol flavours also report them differently (e.g. cell or pack voltage).

Are you sure 41V was correct though? It might mean 3 well-charged 12V (nominal) / ~14V (max) PbAc packs, but small-to-medium sized UPSes usually include 1, 2 or 4 battery packs (easier to balance the weight symmetrically), so 3 puts me off a bit as uncommon.

Later NUT releases included some more smarts to the guesstimation, but some of that learning may help with the older release you have: try setting the floating-point numbers `override.battery.voltage.(low|nominal|high)` values in `ups.conf` (or the HA YAML that translates into that) OR `override.battery.packs = 18` (assuming the driver reports cell voltage as one pack, so you have 18 of those entities summing up to 36-42V range).

ajcorrea, 2024-07-30:
My ups has three internal batteries (3 * 13.8 = 41.4v) .

blazer_usb logs:
```

-----------------------------------------------------------
 Add-on: Network UPS Tools
 Manage battery backup (UPS) devices
-----------------------------------------------------------
 Add-on version: 0.13.1
 You are running the latest version of this add-on.
 System: Debian GNU/Linux 11 (bullseye)  (amd64 / qemux86-64)
 Home Assistant Core: 2024.7.4
 Home Assistant Supervisor: 2024.06.2
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[18:53:21] INFO: Configuring timezone (America/Fortaleza)...
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nut.sh
[18:53:22] INFO: Setting mode to netserver...
[18:53:22] INFO: Generating /etc/nut/upsd.users...
[18:53:22] INFO: Configuring user: admin
[18:53:23] INFO: Password is NOT in the Have I Been Pwned database! Nice!
[18:53:23] INFO: Configuring Device named Intelbras...
[18:53:23] INFO: Starting the UPS drivers...
Please note that this driver is deprecated and will not receive
new development. If it works for managing your devices - fine,
but if you are running it to try setting up a new device, please
consider the newer nutdrv_qx instead, which should handle all 'Qx'
protocol variants for NUT. (Please also report if your device works
with this driver, but nutdrv_qx would not actually support it with
any subdriver!)

Supported UPS detected with megatec protocol
Vendor information read in 1 tries
No values provided for battery high/low voltages in ups.conf

Using 'guestimation' (low: 31.200000, high: 39.000000)!
Battery runtime will not be calculated (runtimecal not set)
Network UPS Tools - UPS driver controller 2.8.0
cont-init: info: /etc/cont-init.d/nut.sh exited 0
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun upsd (no readiness notification)
services-up: info: copying legacy longrun upsmon (no readiness notification)
s6-rc: info: service legacy-services successfully started
[18:53:24] INFO: Starting the UPS information server...
   0.000000	fopen /run/nut/upsd.pid: No such file or directory
   0.000041	Could not find PID file '/run/nut/upsd.pid' to see if previous upsd instance is already running!

   0.000281	[D1] debug level is '1'
   0.000351	listening on 0.0.0.0 port 3493
   0.000523	Connected to UPS [Intelbras]: blazer_usb-Intelbras
   0.000964	Running as foreground process, not saving a PID file
[18:53:25] INFO: Starting the UPS monitor and shutdown controller...
   0.000000	fopen /run/nut/upsmon.pid: No such file or directory
   0.000327	Could not find PID file to see if previous upsmon instance is already running!

   0.000692	Using power down flag file /etc/killpower
   0.001188	UPS: Intelbras@localhost (primary) (power value 1)
   0.001358	[D1] debug level is '1'
   0.001532	[D1] debug level is '1'
   0.001613	Warning: running as one big root process by request (upsmon -p)
   0.001802	[D1] Saving PID 460 into /run/nut/upsmon.pid
Init SSL without certificate database
   0.009580	[D1] Trying to connect to UPS [Intelbras@localhost]
   0.481647	User upsmonmaster@127.0.0.1 logged into UPS [Intelbras]
   0.011241	[D1] Logged into UPS Intelbras@localhost
```

[image]
[image]

I think QX  is repporting the value of each cell´s.. 

assuming each battery has 6 cells, that's 18 cells and 2.28V each, so 18 * 2.28 exactly 41.04 as reported by blazer_usb.

I will switch back to QX and try override.battery.packs = 18

ajcorrea, 2024-07-30:
Configuration seems to be ok but didn´t work:

[image]

[image]

[image]

jimklimov, 2024-08-05:
Hm, I see the reported NUT version is 2.8.0 while the current release is 2.8.2. Maybe the addon build relies on an older (Debian?) packaged version than the one where some fixes around guesstimation were made?

Can you please check if community suggestions at https://github.com/networkupstools/nut/wiki/Building-NUT-integration-for-Home-Assistant would help you custom-build an add-on container with current NUT?

vazquezjm, 2025-05-12:
@ajcorrea were you able to solve this?

jimklimov, 2025-05-12:
Are you guys trying with current NUT (2.8.3 now)?

vazquezjm, 2025-05-12:
> Are you guys trying with current NUT (2.8.3 now)?

I'm using [Network UPS Tools](https://github.com/hassio-addons/addon-nut) for Home Assistant. Log shows 2.8.0

```
[18:21:03] INFO: Starting the UPS drivers...
   0.000000	[D1] Starting UPS: lyonn1200
Using protocol: Voltronic-QS-Hex 0.10
Can't autodetect number of battery packs [-1/27.29]
Battery runtime will not be calculated (runtimecal not set)
Network UPS Tools - UPS driver controller 2.8.0
cont-init: info: /etc/cont-init.d/nut.sh exited 0
```

jimklimov, 2025-05-12:
There's a community article in the wiki on building custom HA plugin, maybe that can help (linked above).

vazquezjm, 2025-05-12:
Thanks! I'm not good at building custom plugins so I reported the issue to the add-on developers.

ajcorrea, 2025-05-13:
> [@ajcorrea](https://github.com/ajcorrea) were you able to solve this?

I didn't try....