networkupstools/nut #2578: SMART1000RMXL2U vendor specific instcmd sequence for individual outlet bank control

ndbrew, 2024-08-02:
The SMART1000RMXL2U has 2 switchable outlet banks which I'm trying to add support for in the `tripplite-hid` driver.
Sniffing the wire for the HID Reports sent from the TrippLite PowerAlert software, it looks like I need to write 2 values in a specific order and within a time limit to the device in order to trigger the desired effect.

To turn off outlet bank 2 I would need to send:
1. Value 0x01 to "UPS.OutletSystem.Outlet.0xffff0098"
2. Value 0x01 to "UPS.OutletSystem.Outlet.0xffff0099"

To turn on outlet bank 2 1 would need to send:
1. Value 0x03 to "UPS.OutletSystem.Outlet.0xffff0098"
2. Value 0x03 to "UPS.OutletSystem.Outlet.0xffff009a"

Failure to write the second value seems to disable all the ports.

I'm still in the process of reverse engineering the protocol, I suspect the value is a mask which controls the state of all outlets together. But there are other features I've observed sniffing the USB wire that also require writing a sequence of data to the UPS which I'm unsure how best to handle.

At the moment it looks like the usbhid library expects instcmd's to only write data to one location. The implementation of instcmd does appear to have some global commands which do seem to execute custom logic but that doesn't feel like the right place for a vender specific protocol.

My thought process at the moment is to implement something which allows instcmd to execute custom logic defined in the driver in a way that other drivers could extend the same functionality. But before I go through the effort, is there a better way to do this?

Also, obviously when dealing with undocumented protocols and firmware anything can happen but how likely am I brick my UPS if I send the wrong value to the wrong place? 😄

jimklimov, 2024-08-02:
Can't really comment at the moment, but CC'ed a couple of people who might, eventually.

acidice333, 2024-10-02:
Any luck @ndbrew ?  I've come into the same problem and I prefer not to use the Power Alert software just to be able to switch two individual banks.