networkupstools/nut #2580: RICHCOMM_USB >> Fatal error: 'vendorid' is not a valid variable name for this driver.

Enzisblog, 2024-08-03:
Daer all,
when i try to setup the USB UPS in my NUT server (proxmox installation), i get an error.

Here my usb scan (_**nut-scanner -U**_)

```
Scanning USB bus.
[nutdev1]
        driver = "richcomm_usb"
        port = "auto"
        vendorid = "0925"
        productid = "1234"
        product = "UPS USB Mon V2.0"
        vendor = "RICHCOMM"
        bus = "003"
```
With these data i updated the **_ups.conf_**
and here the error after to start the  **_upsdrvctl start_**  command

```
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Richcomm dry-contact to USB driver 0.10 (2.8.0)
Warning: This is an experimental driver.
Some features may not function correctly.

Fatal error: 'vendorid' is not a valid variable name for this driver.

Look in the man page or call this driver with -h for a list of
valid variable names and flags.
```

_**'vendorid'**_ seems to be not valid as field in the driver, is there a driver update or patch somewhere or a workaround?

Regards,
Enzo

jimklimov, 2024-08-03:
Looking at sources, this ID combo is built in and is the only one supported. Try commenting away these ID lines. Later releases have it supporting common options, but that older 2.8.0 probably did not yet.

jimklimov, 2024-08-07:
Cross-linking with earlier work:
* #1754
* #1763
* #1768

Per the latter open issue, this driver was among those that remained without the options. So it is (still) arguably a `nut-scanner` bug to offer them.

PR #2589 (2024-08-11): Fix compilation on newer gcc-13.x and clang-18, fix `nut-scanner` offering config values not applicable for some USB drivers
Touches on a number of issues:

#1763
#1764 
#1768 
Closes: #2580
Closes: #2585
Closes: #2588
closed 2024-08-11 by jimklimov