networkupstools/nut #2581: USBHID-UPS Invalid parameter with AMETEK Powervar UPM 2

dvd604, 2024-08-04:
I have an AMETEK Powervar 2 connected via USB to a Home Assistant box running the NUT addon. I have tried NUT running directly on a Ubuntu machine and had the same outcome. At least one other user is having the same issue with NUT and Ametek devices: https://www.mail-archive.com/nut-upsuser@alioth-lists.debian.net/msg02385.html

Apologies for the screengrab and not a copy/paste log - security on HAOS made getting a copy/pastable log very difficult.

[image]

jimklimov, 2024-08-07:
One idea that comes to mind so far is whether these devices use a non-zero "interface" number for data that would be interesting to NUT? Older practice (and de-facto hard-coded expectation) was to use interface 0 but later some composite devices were seen to challenge this assumption.

So my current guess is that the "Invalid parameter" refers to us asking for an entity which is not there at the address we ask for.

Currently (as of NUT v2.8.2 release) these values should be configurable via driver options, but it may well be that some code still passes certain numbers and not the variables to `libusb` calls (so testing is welcome - more so on devices which use values we did not have to expect for two decades prior):
* #2331
* #2149

dvd604, 2024-08-07:
> Also, to clarify: the HA and the Ubuntu machines, are they technologically different
The HA machine is a ProxMox VM - the ProxMox host and Ubuntu machine are both running x86 processors. Neither are RPis.

> this is not a case of trying to monitor several devices from same vendor
Correct, one UPS only.

>  sibling driver instances that may have been started by systemd
No other instances appear to be running.

> Would you have a chance to build current NUT code base and test the built driver from that workspace with such options?

Sure!

Also, after looking at the `lsusb -v` output, I suspect the interface number is indeed 0, although I'm not 100% certain what I'm looking at here:

`Bus 001 Device 003: ID 4234:0002 AMETEK-POWERVAR Corporation UPM UPS (v01.52, Jan 24 2020)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        16
  idVendor           0x4234
  idProduct          0x0002
  bcdDevice            0.02
  iManufacturer           1 AMETEK-POWERVAR Corporation
  iProduct                2 UPM UPS (v01.52, Jan 24 2020)
  iSerial                 4 5114403R-2040100
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          3 UPM CUSPP
    bmAttributes         0xc0
      Self Powered
    MaxPower               36mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
[... 31 lines cut ...]
   0.005149     [D2] - VendorID: 4234
   0.005159     [D2] - ProductID: 0002
   0.005167     [D2] - Manufacturer: AMETEK-POWERVAR Corporation
   0.005171     [D2] - Product: UPM UPS (v01.52, Jan 24 2020)
   0.005175     [D2] - Serial Number: 5xxxxxx-xxxxxx
   0.005178     [D2] - Bus: 001
   0.005181     [D2] - Bus Port: 011
   0.005185     [D2] - Device: 003
   0.005189     [D2] - Device release number: 0002
   0.005193     [D2] Trying to match device
   0.005198     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.005206     [D3] match_function_regex: matching a device...
   0.005248     [D2] Device matches
   0.005254     [D2] Reading configuration descriptor 1 of 1
   0.005263     [D3] libusb_kernel_driver_active() returned 0: Success
   0.005268     [D2] failed to claim USB device: Entity not found
   0.005275     [D1] failed to detach kernel driver from USB device: Invalid parameter
   0.005283     [D2] failed to claim USB device: Entity not found
   0.005291     [D1] failed to detach kernel driver from USB device: Invalid parameter
   0.005295     [D2] failed to claim USB device: Entity not found
   0.005301     [D1] failed to detach kernel driver from USB device: Invalid parameter
   0.005306     [D2] failed to claim USB device: Entity not found
   0.005312     [D1] failed to detach kernel driver from USB device: Invalid parameter
   0.005317     Can't claim USB device [4234:0002]@0/1/0: Invalid parameter
`

dvd604, 2024-08-07:
> Also, to clarify: the HA and the Ubuntu machines, are they technologically different

The HA machine is a ProxMox VM - the ProxMox host and Ubuntu machine are both running x86 processors. Neither are RPis.

> this is not a case of trying to monitor several devices from same vendor

Correct, one UPS only.

>  sibling driver instances that may have been started by systemd

No other instances appear to be running.

> Would you have a chance to build current NUT code base and test the built driver from that workspace with such options?

Sure!

Also, after looking at the `lsusb -v` output, I suspect the interface number is indeed 0, although I'm not 100% certain what I'm looking at here:

```Bus 001 Device 003: ID 4234:0002 AMETEK-POWERVAR Corporation UPM UPS (v01.52, Jan 24 2020)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        16
  idVendor           0x4234
  idProduct          0x0002
  bcdDevice            0.02
  iManufacturer           1 AMETEK-POWERVAR Corporation
  iProduct                2 UPM UPS (v01.52, Jan 24 2020)
  iSerial                 4 5114403R-2040100
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          3 UPM CUSPP
    bmAttributes         0xc0
      Self Powered
    MaxPower               36mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
[... 34 lines cut ...]
   0.005149     [D2] - VendorID: 4234
   0.005159     [D2] - ProductID: 0002
   0.005167     [D2] - Manufacturer: AMETEK-POWERVAR Corporation
   0.005171     [D2] - Product: UPM UPS (v01.52, Jan 24 2020)
   0.005175     [D2] - Serial Number: 5xxxxxx-xxxxxx
   0.005178     [D2] - Bus: 001
   0.005181     [D2] - Bus Port: 011
   0.005185     [D2] - Device: 003
   0.005189     [D2] - Device release number: 0002
   0.005193     [D2] Trying to match device
   0.005198     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.005206     [D3] match_function_regex: matching a device...
   0.005248     [D2] Device matches
   0.005254     [D2] Reading configuration descriptor 1 of 1
   0.005263     [D3] libusb_kernel_driver_active() returned 0: Success
   0.005268     [D2] failed to claim USB device: Entity not found
   0.005275     [D1] failed to detach kernel driver from USB device: Invalid parameter
   0.005283     [D2] failed to claim USB device: Entity not found
   0.005291     [D1] failed to detach kernel driver from USB device: Invalid parameter
   0.005295     [D2] failed to claim USB device: Entity not found
   0.005301     [D1] failed to detach kernel driver from USB device: Invalid parameter
   0.005306     [D2] failed to claim USB device: Entity not found
   0.005312     [D1] failed to detach kernel driver from USB device: Invalid parameter
   0.005317     Can't claim USB device [4234:0002]@0/1/0: Invalid parameter
```

jimklimov, 2025-02-25:
Sorry, this issue fell through the cracks a little. But by now I'm a bit wiser about reading the `lsusb` outputs. Yours says the device only has one logical USB interface (which is common), but that one interface only serves short report descriptors (27 bytes) that are not useful as e.g. USB HID payloads (typically 700-900 bytes long).

Your `usb_hid_ep_in` setting seems to be not fully set (should equal a number like 1), but that does not seem to the the root cause of an issue here.

I wonder if in fact this model uses some other protocol over USB, e.g. if the device presents to your OS also as a serial port converter - is there any `/dev/ttyUSB0` (or other number) that correlates with plug/unplug of this UPS? If so, there's a decent probability that it talks a Megatec Qx protocol dialect, and the `nutdrv_qx` driver could help.

dvd604, 2025-02-25:
No worries, any sort of help is much appreciated.

Wish I had better news - with no other USB devices connected, `/dev/` gains and loses `hidraw0` and `usb` when the UPS is connected/ disconnected. I expect this is normal behaviour for HID devices?

jimklimov, 2025-02-25:
You could also try to connect to `/dev/hidraw0` with `minicom`/`screen`/`nc`/... and try to send/receive bytes, on the odd chance it would respond somehow. But without a layer like serial-over-USB emulation (e.g. `/dev/ttyUSB0`) I doubt it would be useful.

FWIW, I've checked with a local FTDI serial USB dongle, it does not show up as a HID device at all:
````
:; lsusb -vvv -d 0403:6001

Bus 001 Device 004: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x0403 Future Technology Devices International, Ltd
  idProduct          0x6001 FT232 Serial (UART) IC
  bcdDevice            6.00
  iManufacturer           1 FTDI
  iProduct                2 TTL232R-3V3
  iSerial                 3 FTFV4359
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               90mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 TTL232R-3V3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
````