networkupstools/nut #2588: CI: `clang-18` complains about a lot of `switch` clauses without a `default` label

jimklimov, 2024-08-09:
Bumped the OI agent which delivers `clang-18` and `gcc-13` as options now, but the former dislikes `./ci_build.sh` with a few dozen "incomplete" `switch` clauses.

closed 2024-08-11 by jimklimov

biergaizi, 2026-03-10:
Right now the fix is to add `default: break` blindly to all "impossible" conditions. But wouldn't it be more appropriate to abort the program with an assertion failure in the `default` branch (e.g. `fatalx(EXIT_FAILURE...)`) to both suppress the warning and add defensive programming at the same time?

jimklimov, 2026-03-10:
Could be worth some experimentation... I just wonder how many end-users that unilateral switch to `fatalx` would break.

With things where we know all expectable values (*our* enums and equivalents) this is reasonable. With values returned by third-party code (e.g. depending at run-time on shared libraries available in end-user deployments and/or headers at time of build) this is much more questionable, and given that many cases might need actual devices AND platform together to test, it would be adventurous...

Maybe some semi-reasonable way forward would be to use a macro for that default handling - including the `default:` label if macros can do it to issue a warning debug message and a `break` in common builds, or a `fatalx()` in defensive builds, and NO `default:` label in `fightwarn` CI test builds to have them fail due to compiler warnings at least on the library combos we have in the NUT CI farm.

The next layer of complexity I suppose would be the `switch` handling cases that a particular system may never offer, and compiler knows that. Some may be hidden by `#ifdef SOME_ENUMLIKE_VALUE` some maybe not.

PR #2589 (2024-08-11): Fix compilation on newer gcc-13.x and clang-18, fix `nut-scanner` offering config values not applicable for some USB drivers
Touches on a number of issues:

#1763
#1764 
#1768 
Closes: #2580
Closes: #2585
Closes: #2588