networkupstools/nut #2590: [HCL] TED Electric 3100VA 1800W UPS Line Interactive supported by probably Richcomm

pr0fetul, 2024-08-10:
Hi, 
I have recently purchased this TED Electric UPS (TED Electric 3100VA 1800W UPS Line Interactive - https://www.tedelectric.eu/product/3100va-1800w-ups-line-interactive-lcd-display-3-outputs-ted-electric/) and I want to add it to Home Assistant. In the packaging I found a CD pointing to Richcomm for drivers - and I am in the process of finding a CD-ROM to try to get those drivers from the CD and see at least who's the manufacturer. 
From the NUT https://networkupstools.org/stable-hcl.html list, I have tested with almost all drivers I found listed, none of the worked, the only one that showed some promise was indeed the richcomm one, but with the below errors:
```
-----------------------------------------------------------
 Add-on: Network UPS Tools
 Manage battery backup (UPS) devices
-----------------------------------------------------------
 Add-on version: 0.13.1
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.4  (amd64 / qemux86-64)
 Home Assistant Core: 2024.8.0
 Home Assistant Supervisor: 2024.08.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
[14:11:54] INFO: Configuring timezone (Europe/Bucharest)...
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nut.sh
[14:12:04] INFO: Setting mode to netserver...
[14:12:04] INFO: Generating /etc/nut/upsd.users...
[14:12:04] INFO: Configuring user: kiri
[14:12:05] INFO: Password is NOT in the Have I Been Pwned database! Nice!
[14:12:06] INFO: Configuring Device named TE_1800W...
[14:12:06] INFO: Starting the UPS drivers...
USB device [0925:1234] matches, but driver callback failed: Entity not found
Driver failed to start (exit status=1)
Network UPS Tools - Richcomm dry-contact to USB driver 0.10 (2.8.0)
Warning: This is an experimental driver.
Some features may not function correctly.

Network UPS Tools - UPS driver controller 2.8.0
cont-init: info: /etc/cont-init.d/nut.sh exited 1
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
cont-init: warning: some scripts exited nonzero
s6-rc: warning: unable to start service legacy-cont-init: command exited 1
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-timezone: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service base-addon-timezone successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```
I have tested with both the USB cable that came in the box and also with another one, in the VM where the HA is hosted, I have 2 USB devices listed - I have connected them both to the VM: 
[image]
[image]

Can someone please guide as to how I can add this UPS to HomeAssistant? The reasoning is that if I can add and monitor it via HA - then I can write automations to shutdown the environment and then the Synology in case the battery drops under a certain threshold. 

Any help is greatly appreciated. And if I missed any information, please let me know and I will try to find it asap.

jimklimov, 2024-08-10:
Hello, try increasing the debug verbosity for the driver (`debug_min=6` in its settings can help) to see what it complains about to not start. There was a recent report #2580 about auto-detection with `nut-scanner` yielding config options `vendorid`/`productid` that this older driver does not support. If those are in your config too, commenting them away might help.

pr0fetul, 2024-08-10:
Hi @jimklimov - thank you for the reply. 
I have switched the logging to debug, and now the log shows:
```
-----------------------------------------------------------
 Add-on: Network UPS Tools
 Manage battery backup (UPS) devices
-----------------------------------------------------------
 Add-on version: 0.13.1
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.4  (amd64 / qemux86-64)
 Home Assistant Core: 2024.8.0
 Home Assistant Supervisor: 2024.08.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[19:25:27] INFO: Configuring timezone (Europe/Bucharest)...
Log level is set to DEBUG
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nut.sh
[19:25:40] INFO: Setting mode to netserver...
[19:25:40] INFO: Connected USB devices:
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 003: ID 0525:a4a7 Netchip Technology, Inc. Linux-USB Serial Gadget (CDC ACM mode)
Bus 002 Device 002: ID 0925:1234 Lakeview Research UPS USB Mon V2.0
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 0627:0001 Adomax Technology Co., Ltd QEMU Tablet
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
[19:25:40] INFO: Generating /etc/nut/upsd.users...
[19:25:40] INFO: Configuring user: kiri
[19:25:41] DEBUG: Password SHA1: 40D504F096532AFB7D65D495D9AC882E54D6CB7B
[19:25:41] DEBUG: Requested API resource: https://api.pwnedpasswords.com/range/40D50
[19:25:41] DEBUG: API HTTP Response code: 200
[19:25:41] INFO: Password is NOT in the Have I Been Pwned database! Nice!
[19:25:42] INFO: Configuring Device named TE_1800W...
[19:25:42] INFO: Starting the UPS drivers...
   0.000000	[D1] Starting UPS: TE_1800W
USB device [0925:1234] matches, but driver callback failed: Entity not found
Network UPS Tools - Richcomm dry-contact to USB driver 0.10 (2.8.0)
Warning: This is an experimental driver.
Some features may not function correctly.

   0.039468	Driver failed to start (exit status=1)
Network UPS Tools - UPS driver controller 2.8.0
cont-init: info: /etc/cont-init.d/nut.sh exited 1
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
cont-init: warning: some scripts exited nonzero
s6-rc: warning: unable to start service legacy-cont-init: command exited 1
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-timezone: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-timezone successfully stopped
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```
These are the log levels I have in the addon:
[image]

Anything I can try?

pr0fetul, 2024-08-11:
I ran again the addon with Trace as debug level and here is the output:
```
 Add-on: Network UPS Tools
 Manage battery backup (UPS) devices
-----------------------------------------------------------
 Add-on version: 0.13.1
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.4  (amd64 / qemux86-64)
 Home Assistant Core: 2024.8.1
 Home Assistant Supervisor: 2024.08.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[11:57:14] INFO: Configuring timezone (Europe/Bucharest)...
Log level is set to TRACE
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nut.sh
[11:57:27] TRACE: bashio::config: mode
[11:57:27] TRACE: bashio::addon.config
[11:57:27] TRACE: bashio::cache.exists: addons.self.options.config
[11:57:27] TRACE: bashio::fs.file_exists: /tmp/.bashio/addons.self.options.config.cache
[11:57:27] TRACE: bashio::cache.get: addons.self.options.config
[11:57:27] TRACE: bashio::cache.exists: addons.self.options.config
[11:57:27] TRACE: bashio::fs.file_exists: /tmp/.bashio/addons.self.options.config.cache
[11:57:27] TRACE: bashio::jq: {"users":[{"username":"kiri","password":"MumuCucu@44","instcmds":["all"],"actions":[]}],"devices":[{"name":"TE_1800W","driver":"richcomm_usb","port":"auto","config":[]}],"mode":"netserver","shutdown_host":false,"list_usb_devices":true,"log_level":"trace"} if (.mode == null) then
            null
        elif (.mode | type == "string") then
            .mode // empty
        elif (.mode | type == "boolean") then
            .mode // false
        elif (.mode | type == "array") then
            if (.mode == []) then
                empty
            else
                .mode[]
            end
        elif (.mode | type == "object") then
            if (.mode == {}) then
                empty
            else
                .mode
            end
        else
            .mode
        end
[11:57:27] INFO: Setting mode to netserver...
[11:57:27] TRACE: bashio::config.true: list_usb_devices
[11:57:27] TRACE: bashio::config: list_usb_devices
[11:57:27] TRACE: bashio::addon.config
[11:57:27] TRACE: bashio::cache.exists: addons.self.options.config
[11:57:27] TRACE: bashio::fs.file_exists: /tmp/.bashio/addons.self.options.config.cache
[11:57:27] TRACE: bashio::cache.get: addons.self.options.config
[11:57:27] TRACE: bashio::cache.exists: addons.self.options.config
[11:57:27] TRACE: bashio::fs.file_exists: /tmp/.bashio/addons.self.options.config.cache
[11:57:27] TRACE: bashio::jq: {"users":[{"username":"kiri","password":"MumuCucu@44","instcmds":["all"],"actions":[]}],"devices":[{"name":"TE_1800W","driver":"richcomm_usb","port":"auto","config":[]}],"mode":"netserver","shutdown_host":false,"list_usb_devices":true,"log_level":"trace"} if (.list_usb_devices == null) then
            null
        elif (.list_usb_devices | type == "string") then
            .list_usb_devices // empty
        elif (.list_usb_devices | type == "boolean") then
            .list_usb_devices // false
        elif (.list_usb_devices | type == "array") then
            if (.list_usb_devices == []) then
[... 104 lines cut ...]
        elif (.upsmon_deadtime | type == "object") then
            if (.upsmon_deadtime == {}) then
                empty
            else
                .upsmon_deadtime
            end
        else
            .upsmon_deadtime
        end
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
cont-init: warning: some scripts exited nonzero
s6-rc: warning: unable to start service legacy-cont-init: command exited 1
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-timezone: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service base-addon-timezone successfully stopped
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

jimklimov, 2024-08-11:
These options seem to bump the debugging of HA scripts, not NUT programs (driver in particular). They do have a bit of effect, with the driver's logs getting a timestamp and lines' debug level prefix (e.g. `[D1] Starting UPS: TE_1800W`) but I do not see messages e.g. at NUT debug verbosity level 4 seen a lot at https://github.com/networkupstools/nut/blob/v2.8.0/drivers/richcomm_usb.c#L301-L564 and which could carry details about why it gets to `fatalx()` line with the `driver callback failed` message, as well as what successes it had along the way.

Unfortunately I am not a HA user so can't help directly about it, more so when it is a VM, so can only suggest generic ideas at the moment: check that the host system does not have any software attached to the device (e.g. a NUT driver or something else), and that it is not forwarded to any other guests. If you have a chance to rule out more moving parts, by checking if the UPS is seen by NUT on a bare-metal machine (e.g. install/build it on the host of that HA VM), that would be helpful. In the end, you are managing physical power of physical computers, and whatever automation a mere VM does to e.g. command the UPS to power off can be minutes away from the other VMs or the host itself being ready to kick the bucket (ordering it to go down last in case of shutdown can help though, but...)

Also, this is one of a couple of drivers (along with `nutdrv_atcl_usb`) that even in current revisions are not (yet?) fully converged into common NUT codebase and so do not share code of `drivers/usb-common.c` and some of those features and settings that other USB-capable drivers may have, such as the common `vendorid` and `productid` options I think I've mentioned earlier and that may be suggested by `nut-scanner` in that NUT release.

Perhaps ask that community or docs to see how the `ups.conf` is ultimately generated, and what checkboxes or YAMLs in the GUI correspond to that, and how to verify if these two options are in fact present, and how to inject `debug_min` there.

jimklimov, 2024-08-11:
Regarding custom NUT builds, please take a look at https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests and the docs it mentions for dependency lists for various platforms. You can test a driver from the build workspace, without changing the OS there. If you are crafty, you can also custom-build the dependencies (e.g. libusb) to use from the temporary or homedir based installation, and avoid putting those into the OS too as long as you are "just testing". Be sure to NOT have the device actively forwarded to a VM when trying to use it in the host OS context.

Another idea is to follow the community-contributed notes https://github.com/networkupstools/nut/wiki/Building-NUT-integration-for-Home-Assistant about building your own newer HA plugin with a specified NUT version. I believe originally it relies on packaged NUT, so you would have to craft that, or adapt the recipe to install dependencies and build NUT as part of creating the container.

pr0fetul, 2024-08-15:
Hi,
I am sorry, but I am kinda lost :) I don't think I can do the build by myself. 
I will try to extract the drivers from the CD it came with and when I do I will upload them here - maybe it can help the development and adding a new driver.

jimklimov, 2024-08-15:
I wonder if you "may" do such re-publication of binaries here, license-wise. Do they have a download link of such ISO or files it contains?

jimklimov, 2024-08-15:
As for your existing deployment, to step back in complexity:

* Check with HA docs/community about detailed settings for NUT drivers
* Perhaps investigate other issues labeled https://github.com/networkupstools/nut/labels/Home%20Assistant%20%28HA%29 here for more inspiration and key words that might tell you more than they do to me :)
* I expect that there is a way to edit them somehow, maybe as YAML blocks that HA NUT plugin then converts into NUT `ups.conf` sections
* I also expect that with NUT 2.8.0 or newer under the hood, you would be able to set and use the `debug_min` number value, e.g. to `7` for the highest practically implemented verbosity in NUT
* And then the plugin restart would log a wall of text from the driver, with details about what it saw and what it disliked, to copy-paste here :)

pr0fetul, 2024-08-15:
Ok, I managed to get the software online (https://powermanagerlite.software.informer.com/download/) I connected my laptop to the USB of the UPS after installing the PowerManager lite software and it detects the UPS. So it means that the driver from this package works. And it also has Linux Drivers in the package. 
As it is free to download - I think it can be worked with.... what do you think?

pr0fetul, 2024-08-21:
Hi @jimklimov ,
Not wanting to back down, I did the following:
1. Installed nut, nut-server, nut-client on a Raspbery - connected the UPS to it: ```lsusb``` brings:
```
vladc@raspberrypi:~ $ lsusb
**Bus 001 Device 004: ID 0925:1234 Lakeview Research UPS USB Mon V2.0**
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
2. Scanned using the nut-scanner:
```
vladc@raspberrypi:~ $ sudo nut-scanner -U
Scanning USB bus.
[nutdev1]
        driver = "richcomm_usb"
        port = "auto"
        vendorid = "0925"
        productid = "1234"
        product = "UPS USB Mon V2.0"
        vendor = "RICHCOMM"
        bus = "001"
```
So it detects ok the USB as Richcomm usb driver
3. Configured the main files:
```
sudo nano /etc/nut/ups.conf
  GNU nano 7.2                          /etc/nut/ups.conf
pollinterval = 1
maxretry = 3

[tedelectric]
    driver = "richcomm_usb"
    port = "auto"
    descr = "TedElectric 1800W UPS"
    vendorid = "0925"
    productid = "1234"
    product = "UPS USB Mon V2.0"
    vendor = "RICHCOMM"
```

```
sudo nano /etc/nut/upsmon.conf
RUN_AS_USER root

MONITOR tedelectric@localhost 1 user secret master
```

```
sudo nano /etc/nut/upsd.conf
  GNU nano 7.2                          /etc/nut/upsd.conf
LISTEN 0.0.0.0 3493
```

```
sudo nano /etc/nut/nut.conf
  GNU nano 7.2                          /etc/nut/nut.conf
MODE=netserver
```

```
sudo nano /etc/nut/upsd.users

[monuser]
  password = secret
  user master
```
4. Then restarted all services:
```
sudo service nut-server restart
sudo service nut-client restart
sudo systemctl restart nut-monitor
sudo upsdrvctl stop
Network UPS Tools - UPS driver controller 2.8.0
Can't open /run/nut/richcomm_usb-tedelectric.pid: No such file or directory
Can't open /run/nut/richcomm_usb-auto.pid either: No such file or directory
sudo upsdrvctl start
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Richcomm dry-contact to USB driver 0.10 (2.8.0)
Warning: This is an experimental driver.
Some features may not function correctly.

Fatal error: 'descr' is not a valid variable name for this driver.

Look in the man page or call this driver with -h for a list of
valid variable names and flags.
```

Any idea on how to debug further why is not working? Also, running: ``` lsusb -D /dev/bus/usb/001/004``` gives:
```
Device: ID 0925:1234 Lakeview Research UPS USB Mon V2.0
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x0925 Lakeview Research
  idProduct          0x1234
  bcdDevice            0.00
  iManufacturer           1 RICHCOMM
  iProduct                2 UPS USB Mon V2.0
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          4
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower               70mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              5
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      34
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
```

jimklimov, 2024-08-21:
One clue is the reported fatal error.

Another would (probably) be that this driver does not have options for `vendorid`/`productid` and would next complain about these.

Also with NUT 2.8.0 and newer, on OSes with systemd, the nut-driver-enumerator may create a service unit for each driver which would conflict with manual `upsdrvctl` runs.

pr0fetul, 2024-08-21:
Hi Jim,

So, what are my next steps?
Mulțumesc,
Vlad Chiriloiu

Sent from mobile phone

jimklimov, 2024-08-21:
First, get the driver program running: comment away `ups.conf` lines with `desc` and whatever else it would complain about.

Check `systemctl status nut-driver@tedelectric.service` - quite probably it would exist, in that case manipulate it via `systemctl` or NUT's `upsdrvsvcctl` wrapper (for services), not `upsdrvctl` designed for older systems.

pr0fetul, 2024-08-22:
Hi,
Thank you very much for the fast reply. 
So, I edited the ```ups.conf``` file to look like:
```
pollinterval = 1
maxretry = 3

[tedelectric]
    driver = "richcomm_usb"
    port = "auto"
    vendor = "RICHCOMM"
```
Then, I ran the following commands:
```
vladc@raspberrypi:~ $ sudo upsdrvctl start
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Richcomm dry-contact to USB driver 0.10 (2.8.0)
Warning: This is an experimental driver.
Some features may not function correctly.

USB device [0925:1234] matches, but driver callback failed: Entity not found
Driver failed to start (exit status=1)
Network UPS Tools - Richcomm dry-contact to USB driver 0.10 (2.8.0)
Warning: This is an experimental driver.
Some features may not function correctly.

USB device [0925:1234] matches, but driver callback failed: Entity not found
Driver failed to start (exit status=1)
Network UPS Tools - Richcomm dry-contact to USB driver 0.10 (2.8.0)
Warning: This is an experimental driver.
Some features may not function correctly.

USB device [0925:1234] matches, but driver callback failed: Entity not found
Driver failed to start (exit status=1)
```
If I run the following commands, I get:
```
vladc@raspberrypi:~ $ sudo service nut-server restart
vladc@raspberrypi:~ $ sudo service nut-client restart
Broadcast message from root@raspberrypi (somewhere) (Thu Aug 22 09:27:03 2024):  #->I understood that this is normal - so no worry about it#

Communications with UPS tedelectric@localhost lost

Broadcast message from root@raspberrypi (somewhere) (Thu Aug 22 09:27:08 2024):

UPS tedelectric@localhost is unavailable

vladc@raspberrypi:~ $ sudo systemctl restart nut-monitor
vladc@raspberrypi:~ $ sudo upsdrvsvcctl stop
=== The currently defined service units are:
Stopping nut-driver@tedelectric.service ...
vladc@raspberrypi:~ $ sudo upsdrvsvcctl start
No service units detected    #-> what does this mean?
```
Also, by running ```systemctl status nut-driver@tedelectric.service```, it outputs:
```
vladc@raspberrypi:~ $ sudo systemctl status nut-driver@tedelectric.service
○ nut-driver@tedelectric.service - Network UPS Tools - device driver for NUT device 'tedelectric'
     Loaded: loaded (/lib/systemd/system/nut-driver@.service; enabled; preset: enabled)
    Drop-In: /etc/systemd/system/nut-driver@tedelectric.service.d
             └─nut-driver-enumerator-generated-checksum.conf, nut-driver-enumerator-generated.conf
     Active: inactive (dead) since Thu 2024-08-22 09:42:54 EEST; 26s ago
    Process: 3725 ExecStart=/bin/sh -c NUTDEV="`/usr/libexec/nut-driver-enumerator.sh --get-device-for-service tedelectric`" && [ -n "$NUTDEV" ] || { echo "FATAL: Could not find a NUT devi>
        CPU: 271ms

Aug 22 09:42:51 raspberrypi systemd[1]: Starting nut-driver@tedelectric.service - Network UPS Tools - device driver for NUT device 'tedelectric'...
Aug 22 09:42:51 raspberrypi nut-driver@tedelectric[3753]: USB device [0925:1234] matches, but driver callback failed: Entity not found
Aug 22 09:42:51 raspberrypi nut-driver@tedelectric[3753]: Network UPS Tools - Richcomm dry-contact to USB driver 0.10 (2.8.0)
Aug 22 09:42:51 raspberrypi nut-driver@tedelectric[3753]: Warning: This is an experimental driver.
Aug 22 09:42:51 raspberrypi nut-driver@tedelectric[3753]: Some features may not function correctly.
[... 36 lines cut ...]
░░ Automatic restarting of the unit nut-driver@tedelectric.service has been scheduled, as the result for
░░ the configured Restart= setting for the unit.
Aug 22 09:48:43 raspberrypi systemd[1]: Stopped nut-driver@tedelectric.service - Network UPS Tools - device driver for NUT device 'tedelectric'.
░░ Subject: A stop job for unit nut-driver@tedelectric.service has finished
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A stop job for unit nut-driver@tedelectric.service has finished.
░░
░░ The job identifier is 4216 and the job result is done.
Aug 22 09:48:43 raspberrypi systemd[1]: Starting nut-driver@tedelectric.service - Network UPS Tools - device driver for NUT device 'tedelectric'...
░░ Subject: A start job for unit nut-driver@tedelectric.service has begun execution
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit nut-driver@tedelectric.service has begun execution.
░░
░░ The job identifier is 4216.
Aug 22 09:48:43 raspberrypi nut-driver@tedelectric[4219]: USB device [0925:1234] matches, but driver callback failed: Entity not found
Aug 22 09:48:43 raspberrypi nut-driver@tedelectric[4219]: Network UPS Tools - Richcomm dry-contact to USB driver 0.10 (2.8.0)
Aug 22 09:48:43 raspberrypi nut-driver@tedelectric[4219]: Warning: This is an experimental driver.
Aug 22 09:48:43 raspberrypi nut-driver@tedelectric[4219]: Some features may not function correctly.
Aug 22 09:48:43 raspberrypi nut-driver@tedelectric[4218]: Driver failed to start (exit status=1)
```
I hope this helps in pinpointing the issue - or at least I hope so :) - can you advise further?

jimklimov, 2024-08-22:
Thanks! So far this one is the most important part:
`````
USB device [0925:1234] matches, but driver callback failed: Entity not found
Driver failed to start (exit status=1)
`````

The driver did get past its configuration options, but a callback to check if the device looks like it is supported did not recognize it.

Logic-wise, it is called at https://github.com/networkupstools/nut/blob/c9132a5749ede3e3da76a65fd0ce6cb25b9da40f/drivers/richcomm_usb.c#L511 and goes to https://github.com/networkupstools/nut/blob/c9132a5749ede3e3da76a65fd0ce6cb25b9da40f/drivers/richcomm_usb.c#L236-L281 in the current codebase (https://github.com/networkupstools/nut/blob/v2.8.0/drivers/richcomm_usb.c#L500 and https://github.com/networkupstools/nut/blob/v2.8.0/drivers/richcomm_usb.c#L235-L276 in your older NUT release).

The logic here is simpler than in other USB-capable drivers:
* `callback()` decides if the device handle can be used ("configuration", "interface" and "endpoint" can all be used), returns -1 otherwise.
* if it failed, a loop tries (up to 3 times) to detach the kernel driver in case one is holding the device, and try the callback again.
* if all loop cycles failed, the driver exits with the message we saw

If you add `debug_min = 6` to the `ups.conf` section and restart the driver, we can see more details about what exactly failed here.

Also check if your system has `/usr/lib/udev/rules.d/62-nut-usbups.rules` (or maybe in some other location, e.g. under `/etc/udev*` - may depend on packaging, or under a different name - it might have change during 2.8.x timeframe), and if such file contains 0925 and 1234. Maybe for some reason the `udev`, `upower` or similar facility on your box is not told to hand-off this USB device from kernel to NUT.

It is also a possibility that this device does not fit this driver (billed as "dry-contact to USB driver") - maybe your UPS is newer and smarter than posting a few yes-no bits over USB, and needs some `usbhid-ups` or talks a Megatec Qx protocol and needs `nutdrv_qx` instead altogether. For the latter you can manually try different `protocol` and `subdriver` options to override built-in match-ups known when the driver version was built; `usbhid-ups` can also do such options but only since NUT v2.8.2 I think.

As a quick test with even older NUT versions, `usbhid-ups -x explore -x vendorid=0925 -x productid=1234 -s test` might expose if this device reports anything via USB HID interface.

> ````
> :; sudo upsdrvsvcctl start
> No service units detected
> ````
> what does this mean?

This is actually an odd one, I can't quickly say. The `nut-driver@tedelectric.service` systemd unit apparently exists according to other screenshots. Maybe this call happened in the split-second that you edited something, and the NDE service was re-defining the unit, so it was deleted for a few seconds?

jimklimov, 2024-08-22:
> I don't see anything regarding Richcomm in the rules file

It's here. Glued to CPS block for some reason, but present:

> ````
> #  Sweex 1000VA  - richcomm_usb
> ATTR{idVendor}=="0925", ATTR{idProduct}=="1234", MODE="664", GROUP="nut"
> ````

As for the driver startup, logs (thanks!) indicate that the loop of 3 retries does behave differently on first and later cycles. With responsible code lines (to help track this fault down):

* before loop: https://github.com/networkupstools/nut/blob/v2.8.0/drivers/richcomm_usb.c#L495
````
   0.026294     [D2] auto detached kernel driver from USB device
````

* cycle#1 (in callback): https://github.com/networkupstools/nut/blob/v2.8.0/drivers/richcomm_usb.c#L271
````
   0.027664     [D5] Can't reset USB endpoint
````

* cycle#1:
````
   0.027749     [D4] failed to detach kernel driver from USB device: Entity not found
````

* cycle#2 (in callback): https://github.com/networkupstools/nut/blob/v2.8.0/drivers/richcomm_usb.c#L247
````
   0.027843     [D5] Can't set USB configuration
````

* cycle#2: https://github.com/networkupstools/nut/blob/v2.8.0/drivers/richcomm_usb.c#L524 or https://github.com/networkupstools/nut/blob/v2.8.0/drivers/richcomm_usb.c#L533
````
   0.027881     [D4] failed to detach kernel driver from USB device: Entity not found
````

* cycle#3 same as cycle #2:
````
   0.027931     [D5] Can't set USB configuration
   0.027964     [D4] failed to detach kernel driver from USB device: Entity not found
````

So on the first cycle it did pass "USB configuration" and a few other operations in the `callback` method before failing; maybe one of those corrupted something? (e.g. "set USB alternate interface" was just before it, and if later detachments failed - this is where the libusb connection stayed). If that device does not have an alt interface (or not at that point), it might be why the other calls fail (and "Entity not found" verdict seemingly fits).

*Maybe* the "auto detach kernel driver from USB device" just before the loop is also at fault, at least it is not available on *all* systems out there.

I think at this point I can only recommend a custom build of NUT from master branch per https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests and fiddling with the driver source to comment away the suspected culprits (one then another) and see if your device would become recognized.

pr0fetul, 2024-08-23:
Hi @jimklimov 
Thank you very much for taking the time to go through this. Unfortunately "I think at this point I can only recommend a custom build of NUT from master branch per https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests and fiddling with the driver source to comment away the suspected culprits (one then another) and see if your device would become recognized." - I think this may be over my knowledge to test. I can setup a screen-share so you can maybe (if you want/can) debug directly the issue - and maybe improve the overall NUT driver system, but I am afraid that I have no idea on how to start-complete the driver creation from scratch...
If you have any other idea, please let me know.
Thank you very much.

jimklimov, 2024-08-23:
Cheers, the Wiki article refers to `config-prereqs` document which details installation of third-party packages with tools and dependencies for building NUT on a variety of platforms, and normally the `./ci_build.sh inplace` should take care of configuring a new build to be similar to whatever you had installed before (so should benefit from same configs and permissions).

Then you can run a built driver from the build workspace, e.g. stop the existing services and run `./drivers/richcomm_usb -a tedelectric -DDDDDD -d 1` (maybe prefix by `sudo` and add `-u root` to CLI options, if it complains about permissions), and it should try to talk to the device and run one data-collection loop and dump the output in `upsc` style, if successful.

By editing your copy of `drivers/richcomm_usb.c` (e.g. to comment away suspicious code, or add debug printouts with `upsdebugx()` lines similar to those nearby), and running `make` again, you can iterate with modifications to the driver and check those theories above about which USB operations could have lost sight of the device for us. Maybe as a big blunt hammer for the first shot, just `return 1` from the `callback` method and comment away the rest of it.

JamesGDiaz, 2025-03-13:
For future reference, and to help anyone who might be using these kind of devices:

My UPS is a HikVision DS-UPS3000 (3000VA/1800W), it gets detected by nut-scanner the same way as this device, and many others, probably
I've been trying to monitor it under a Debian server anmd the latest version of the NUT suite in Debian as of March 13th, 2025 is 2.8.0

I tested in Arch Linux, with nut version 2.8.2 and nut-scanner reports:
``` nut-scanner 
[nutdev-usb1]
	driver = "nutdrv_qx"	# alternately: richcomm_usb
	port = "auto"
	vendorid = "0925"
	productid = "1234"
	product = "UPS USB Mon V2.0"
	vendor = "RICHCOMM"
	# bus = "003"
	# device = "015"
	# busport = "002"
```

Added this to `nut.conf` and voila!! Now `upsc nutdev-usb1` outputs the following:
```
[root@lunar nut]# upsc nutdev-usb1
battery.charge: 100
battery.voltage: 27.0
battery.voltage.high: 26.00
battery.voltage.low: 20.80
battery.voltage.nominal: 24.0
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: UPS USB Mon V2.0
driver.parameter.productid: 1234
driver.parameter.synchronous: auto
driver.parameter.vendor: RICHCOMM
driver.parameter.vendorid: 0925
driver.state: quiet
driver.version: 2.8.2
driver.version.data: Megatec 0.07
driver.version.internal: 0.36
driver.version.usb: libusb-1.0.27 (API: 0x100010a)
input.current.nominal: 5.0
input.frequency: 60.0
input.frequency.nominal: 50
input.voltage: 120.0
input.voltage.fault: 0.0
input.voltage.nominal: 220
output.voltage: 120.0
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: V3.65
ups.load: 5
ups.productid: 1234
ups.status: OL
ups.temperature: 20.8
ups.type: offline / line interactive
ups.vendorid: 0925
```