networkupstools/nut #2598: Need Help with Windows NUT Server install

ScottReadTCF, 2024-08-20:
Hi Everyone,

So the issue I have is when I install NUT-Installer-2.6.5-6.exe as admin I get the error message:-

NUT UPS driver installer.
-------------------------
Searching for known UPS...
No known UPS device found.
Try installing libUSB manually.
Hit enter to continue

The UPS is a Riello Sentinel Dual 4kVA SDU 4000 and is connected via USB (this all works fine in the PowerShield software they suggest using). I've looked into this and tried to 'install' libUSB but this seems to be a  a folder with "files" in it but no exe file to 'install' from looking at the NUT site it suggests that SDU 4000 is compatible using the [riello_usb] driver https://networkupstools.org/docs/man/riello_usb.html)

I'm not sure exactly what I need to do to get the NUT install to use the riello_usb and get the software to see the UPS.

Thank you so much if you can help and i'm sorry if this seems an obvious question.

jimklimov, 2024-08-20:
Hello, the 2.6.5 baseline is very old (Aug 2012, so ~12 years now), although unfortunately it is what the MSI installer was made from and now the community knowledge about that arcane magic got a bit lost. I am not certain which exact commits went into that package file, but `riello_usb` was only introduced shortly after NUT v2.6.5 "vanilla" release. The driver code is present and a few commits old in Windows-v2.6.5-6 tag (Mar 2014) though, at which point that side project fizzled out for a decade.

The NUT for Windows effort was revived a couple of years back, so there are builds of current codebase being built by Appveyor CI, with downloadable archives (no installer yet) as detailed at https://github.com/networkupstools/nut/wiki/NUT-for-Windows

You may still need a few tricks about getting libusb based drivers seen, as commented there.

With your existing installation, please see if a `riello_ups.exe` exists - maybe same tricks can help get it to run and see the device, even if the device detection tool in that build does not know of it...

ScottReadTCF, 2024-08-20:
Thanks for your help!

do you happen to know if there will be an installer soonish (before I try to force this to work?), I can survive for the short term if its "coming soon".

jimklimov, 2024-08-20:
Unfortunately, probably not "soonish" -- needs some time from a determined developer to make it all come together on the platform, maybe with same or a different packaging technology. I had some interest about portable code base to get that it buildable and testable on the platform, and eventually merged, but struggled to find time for the finishing touches like the packager or certain known-missing equivalent code blocks.

jimklimov, 2024-08-20:
> Also riello_ups.exe doesn't exist in the install.

Sorry, I fat-fingered - `riello_usb.exe`

ScottReadTCF, 2024-08-21:
Hmm, so not sure how I should proceed...

jimklimov, 2024-08-21:
Re-check if that driver is at all present in the old package you have; then we might try to somehow get it running and at least check if it sees your UPS.

Otherwise forgo the package and try to use the archived installation prepared by CI farm, per wiki.

ScottReadTCF, 2024-08-22:
I can confirm there is no file that starts with 'rie' in the 'NUT' folder structure, where is the archived installation located please? (sorry I feel like i'm missing something here)...

Thanks

jimklimov, 2024-08-22:
Per https://github.com/networkupstools/nut/wiki/NUT-for-Windows 

> Current NUT for Windows archives from CI are available at https://ci.appveyor.com/project/nut-travis/nut/build/artifacts - note this is a rolling URL for a "latest" build that may include PR checks; for a somewhat-stable codebase (PR merges to "master" branch) one needs to open the "History" tab (NOT "Current build") and find the newest AppVeyor build link from a green-ticked commit at https://github.com/networkupstools/nut/commits/master history. In AppVeyor Web UI, a master-branch build would have a small gray comment with only "master" branch name and a commit ID, without another branch name and commit as is done for pull request builds (they also have a gray title to that effect).

I've updated it with a screenshot to help navigate now.

ScottReadTCF, 2024-08-22:
Hiya,

Yeah, I think I'm going to look at other options for this like hosting NUT
on a Lite Linux system.

Thanks for your help though!

jimklimov, 2024-08-22:
At the moment that may be more flexible and functional :) But if you do try Windows builds - would be great to hear how it went in real life :)

ScottReadTCF, 2024-08-22:
I'd love to hear once the install is on a Microsoft install package :-)

smith-48116, 2024-08-30:
I was able to install the NUT server on a Windows 7 machine successfully. I believe the instructions would be similar for Windows 10 with minor variations. I used the this build: NUT-for-Windows-x86_64-SNAPSHOT-2.8.2.2096-master.

In order to run the server I had to learn about libusb-win32 driver and all that comes with the installation using zadig-2.9.exe. In the end the driver from CyperPower (CP1500PFCLCD) was replaced with the libusb-win32 driver.

It was a painful learning curve to figure out how to hijack the existing UPS USB Driver using libusb-win32. Before installing NUT, I have never heard of libusb-win32 or the zadig-2.9.exe that is used replace the driver from CyperPower (CP1500PFCLCD) with the libusb-win32 driver.

[image]

The only remaining mystery that I cannot figure out is whether upsdrvctl is supposed to self-terminate after it has started the usbhid-ups.exe process or whether it is supposed to hang around (doing what?). It states that it will exit but the console window hangs around. I can minimize it to the taskbar but it would be nice to have it go away if it is finished starting usbhid-ups.exe in another process space that it appears to be doing.

This is what the console shows after executing upsdrvctl in a console.

 Network UPS Tools upsdrvctl - UPS driver controller 2.8.2.986.9-995-g943ed3c06 (development iteration after 2.8.2)
   0.000000     [D1] upsdrvctl commanding all drivers (1 found): start
   0.001000     [D1] Starting UPS: apc1500
Network UPS Tools driver 2.8.2.986.9-995-g943ed3c06 (development iteration after 2.8.2) - Generic HID driver 0.56
USB communication driver (libusb 0.1) 0.48
Using subdriver: CyberPower HID 0.81
  45.018575     [D1] upsdrvctl: successfully finished
  45.021575     [D1] Completed the job of upsdrvctl tool, cleaning up and exiting now
  45.023575     [D1] Completed the job of upsdrvctl tool, **clean-up finished, exiting now**

I am starting it via the Task Scheduler but you could likely just drag a shortcut to a batch file into the Startup folder to accomplish the same thing. Note that Task Scheduler under Windows 7 has issues with starting batch processes that I can cover in a write-up.

The Windows netclient is a breeze to set up on Windows 10 having set up Linux and MacOS NUT packages in the past. It took only a few minutes and it was up and running. Windows 7 is not so nice and several issues prevent a successful installation. I spent a good part of a day debugging it.

As for an installer, really the only thing that is needed is to create the C:\Program Files\NUT folder and copy the files to it. After that, it should update the system PATH (add - C:\Program Files\NUT\bin and C:\Program Files\NUT\sbin).

I may do an installation write-up for how to use the Windows NUT package for Windows 7 and Windows 10. If I decide to do so, where could I post it? Can I add to your Wiki page?

https://github.com/networkupstools/nut/wiki/NUT-for-Windows

smith-48116, 2024-08-30:
I should have added this comment. If you launch upsdrvctl from a .cmd file (Windows batch file) and close the console from which upsdrvctl is executing (and reportedly 'finished', you will get an error message and the usbhid-ups.exe has crashed. My guess is that the usbhid-ups.exe is spawned as a child process of upsdrvctl.exe (or of the .cmd) rather than opening a new process environment. I have also tested launching upsdrvctl.exe directly from Task Scheduler with the same result.

Before closing the console (or taskeng window):

[image]

After the close:

[image]

jimklimov, 2024-08-30:
Sounds great, thanks for the report. A Wiki page (maybe a new one linked from the one you mentioned) would be also great! :)

While at it, can you please check the `nut.exe` wrapper which is supposed to start `upsd`, `upsmon` and `upsdrvctl` for all driver(s) configured via `ups.conf`, or just `upsmon` (depending on `MODE` setting in `nut.conf`)? It should register as a Windows service, so having better privileges to run for an eventual UPS shutdown after user-space programs are killed (also headless - an area where I think Task Scheduler would be lacking).

ScottReadTCF, 2024-09-24:
Hi Jim,

So I have deployed NUT onto an Ubuntu machine I have 99% of the setup done I can Access the information under localhost but when I use the ip address to test upsc <UPSName>@<machineIPAddress> on the same machine I get:-

`Error: Connection failure: Connection refused`

I've tried upsc <UPSName>@<machineIPAddress> 1 <user> <password> that was setup under /etc/nut/upsd.users as per below:-

[user]

        password = <password>
        actions = SET FSD
        instcmds = ALL
        upsmon master

the rough guide i've followed is this but I feel i'm getting the network part wrong somewhere.....

Any help would be appreciated

jimklimov, 2024-09-24:
What is the `LISTEN` setting for `upsd.conf`? By default it is localhost-only...

ScottReadTCF, 2024-09-24:
The values are:-

Hiya,

The values are :-

LISTEN 127.0.0.1 3493
LISTEN ::1 3493
LISTEN 192.168.xxx.xxx 3493

the xxx obviously is the local IP for the machine in which the NUT server is installed.

Thanks

jimklimov, 2024-09-24:
Just in case, the NUT server is the Windows machine, right? Or a Linux etc. one? Which version, then?

IIRC 2.8.1 had a bug #2012 about multiple LISTEN directives, whose handling was inverted which could lead to some funny situations (though mostly about IPv4 vs IPv6 localhost with dual-stack OSes). If you are using that or perhaps older NUT version as server, try inverting the line order. Or leave one line to just `LISTEN * 3493`.

Surely, another point to check is plain old firewall rules on both server and client sides...

Also, `netstat -an` should show the currently LISTEN'ed ports on a server, to make sure the setup worked there. Logs of `upsd` can report if it failed to set up a listener; it might not-fail if at least one endpoint succeeded (trying to remember if there's a toggle for that behavior).

ScottReadTCF, 2024-09-25:
Hi Jim,

Sorry the NUT Server is setup as Ubuntu 24.04.1 LTS using NUT 2.8.1

Added LISTEN * 3493

Added Firewall exception for 3493 to the machine IP in out Unifi UDM Pro

Checked  netstat -an the result doesn't feature '3493' anywhere.

tried 'upsc @IPAddress' 

Still no banana...

ScottReadTCF, 2024-09-25:
But I have just cleared the examples out on the machine conf's and this might have it cracked just testing

ScottReadTCF, 2024-09-25:
YES!!! its working thank you so much!