networkupstools/nut #2603: missing input/output voltage variables on APC SMT2200RMI2UC (2U rackmount model)

olliecook, 2024-08-25:
I have an APC SMT2200RMI2UC (a 2U rack-mount model) with "Cloud-based remote monitoring connected to a Synology RS1221+ NAS by the included USB cable. `nuts` is configured through the Synology UI as follows:

```
# grep -v -e ^$ -e ^# /etc/ups/ups.conf 
pollinterval = 5
[ups]
	driver = usbhid-ups
	port = auto
```

It is 'working' in the sense that `upsc` is reporting some information (such as when it's supporting load via battery) but not others I am interested in such as the input voltage:

```
# upsc ups
Init SSL without certificate database
battery.charge: 99
battery.charge.low: 10
battery.charge.warning: 50
battery.runtime: 8973
battery.runtime.low: 150
battery.type: PbAc
battery.voltage: 54.3
battery.voltage.nominal: 48.0
device.mfr: American Power Conversion 
device.model: Smart-UPS_2200
device.serial: AS2345261237    
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 5
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: DSM7-2-1-NewModel-repack-64570-230831
driver.version.data: APC HID 0.96
driver.version.internal: 0.41
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.firmware: UPS 06.0 / ID=1027
ups.mfr: American Power Conversion 
ups.mfr.date: 2023/11/07
ups.model: Smart-UPS_2200
ups.productid: 0003
ups.serial: AS2345261237    
ups.status: OL CHRG
ups.timer.reboot: -1
ups.timer.shutdown: -1
ups.vendorid: 051d
```

I had expected to see variables such as these from https://github.com/networkupstools/nut-ddl/blob/master/APC/APC__Smart-UPS_2200__usbhid-ups__2.7.4__01.dev:

```
input.voltage:
output.voltage:
```

The voltages are showing on the APC SmartConnect web dashboard, so the figures are available (at least to APC):
 
[image]

The UPS firmware is showing as being the latest version on the SmartConnect website:

[image]

(although I am a bit sceptical because the [documentation link](https://download.schneider-electric.com/files?p_File_Name=Smart-UPS_SMT_SMC_Series-Firmware_Release_Notes_EN.pdf&p_Doc_Ref=SPD_CLII-AXYUKZ_EN&p_enDocType=Firmware) suggests the latest firmware version is `17.2` which is far from `06.0`)

Running `usbhid-ups` manually, I don't see anything relating to input/output voltage, only battery voltage:

```
# /usr/bin/usbhid-ups -a ups -D
[... 57 lines cut ...]
   0.177701	Path: UPS.PowerSummary.PresentStatus.VoltageNotRegulated, Type: Input, ReportID: 0x07, Offset: 11, Size: 1, Value: 0
   0.177709	Path: UPS.PowerSummary.PresentStatus.VoltageNotRegulated, Type: Feature, ReportID: 0x07, Offset: 11, Size: 1, Value: 0
   0.180527	Path: UPS.ff8600fd, Type: Input, ReportID: 0x89, Offset: 0, Size: 8, Value: 0
   0.183511	Path: UPS.ff8600fc, Type: Output, ReportID: 0x90, Offset: 0, Size: 8, Value: 0
   0.186512	Path: UPS.ff8600f1, Type: Feature, ReportID: 0x96, Offset: 0, Size: 8, Value: 0
   0.189513	Path: UPS.ff8600f7, Type: Feature, ReportID: 0x8d, Offset: 0, Size: 8, Value: 0
   0.192499	Path: UPS.ff8600f4, Type: Feature, ReportID: 0x92, Offset: 0, Size: 8, Value: 127
   0.195496	Path: UPS.ff8600f3, Type: Feature, ReportID: 0x93, Offset: 0, Size: 8, Value: 160
   0.198508	Path: UPS.ff8600f2, Type: Feature, ReportID: 0x94, Offset: 0, Size: 8, Value: 224
   0.201497	Path: UPS.ff8600f6, Type: Feature, ReportID: 0x8e, Offset: 0, Size: 8, Value: 0
   0.201522	Detected a UPS: American Power Conversion /Smart-UPS_2200 FW:UPS 06.0 / ID=1027
   0.207743	upsdrv_initinfo...
   0.207752	upsdrv_updateinfo...
   0.207755	Not using interrupt pipe...
   0.207760	Quick update...
...
```

I am running the latest version of the host OS ([DSM 7.2.1-69057 Update 5](https://www.synology.com/en-global/releaseNote/DSM?model=RS1221%2B#ver_69057-5)) for the Synology RS1221+ NAS. I don't know which version of `nuts` they have bundled. Is there a command to run to find out?

I'll see if I can upgrade the device firmware as a next step. If there's anything else anyone can recommend, that'd be great.

One question that springs to mind is whether the `usbhid-ups` driver is the most appropriate for this model of UPS. Should I be using `apcsmart` or `apcupsd-ups` instead?

Thank you.

jimklimov, 2024-08-25:
TBH, the version puzzled me too. It should be exposed in program banners and `driver.version`, but this is obe of those vendors who obfuscate stuff with their own product version.

Srill, side clues reveal they have (a variant of?) NUT 2.7.4 under the hood, ~8 years old now (but sadly still shipped by many stable distros as it was the latest published release for too long):

````
~/nut $ git blame v2.7.4 drivers/apc-hid.c | grep _VERS
f679a2820e (Charles Lepple 2016-01-24 17:33:48 -0500  34) #define APC_HID_VERSION "APC HID 0.96"

~/nut $ git blame v2.7.4 drivers/usbhid-ups.c | grep _VERS
c730b9ff78 drivers/usbhid-ups.c (Charles Lepple 2015-07-02 08:53:32 -0400   30) #define DRIVER_VERSION          "0.41"
````

jimklimov, 2024-08-25:
The diver should know about input/output voltage however, even at that version:

````
2b2d6758b6 (Arjen de Korte 2007-09-17 11:56:03 +0000 377)   { "input.voltage", 0, 0, "UPS.Input.Voltage", NULL, "%.1f", 0, NULL },
2b2d6758b6 (Arjen de Korte 2007-09-17 11:56:03 +0000 378)   { "input.voltage.nominal", 0, 0, "UPS.Input.ConfigVoltage", NULL, "%.0f", 0, NULL },
2b2d6758b6 (Arjen de Korte 2007-09-17 11:56:03 +0000 387)   { "output.voltage", 0, 0, "UPS.Output.Voltage", NULL, "%.1f", 0, NULL },
2b2d6758b6 (Arjen de Korte 2007-09-17 11:56:03 +0000 388)   { "output.voltage.nominal", 0, 0, "UPS.Output.ConfigVoltage", NULL, "%.1f", 0, NULL },
````

I suppose this new APC device follows the path they seem to have started circa 2010, by caring less about standard protocols like USB HID and promoting their own more. 

Maybe the device can also talk variants of microsol/solis, or apcsmart indeed. The `apcupsd-ups` is a relay from readings by `apcupsd` program which, incidentally, might also talk APC Modbus based protocol (added in NUT v2.8.2, maybe 2.8.1, and still needs a custom build of libmodbus for USB media; should be ok out of the box for serial or tcp).

bornecu, 2024-10-29:
I am having exactly the same issue with an APC SMC1000IC, which I connected via USB to a synolgy DS720+ running DSM 7.2.1-69057 Update 5. No clue how to add input.voltage output.voltage sensors... 
```
 :~$ upsc ups
Init SSL without certificate database
battery.charge: 97
battery.charge.low: 10
battery.charge.warning: 50
battery.runtime: 3025
battery.runtime.low: 150
battery.type: PbAc
battery.voltage: 27.0
battery.voltage.nominal: 24.0
device.mfr: American Power Conversion
device.model: Smart-UPS_1000
device.serial: AS2421152896
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 5
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: DSM7-2-1-NewModel-repack-64570-230831
driver.version.data: APC HID 0.96
driver.version.internal: 0.41
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.firmware: UPS 02.0 / ID=1061
ups.mfr: American Power Conversion
ups.mfr.date: 2024/05/24
ups.model: Smart-UPS_1000
ups.productid: 0003
ups.serial: XXXXXXX
ups.status: OL CHRG
ups.timer.reboot: -1
ups.timer.shutdown: -1
ups.vendorid: 051d

```

jimklimov, 2024-11-01:
Sadly, I've got no new answers compared to the above, although thank you both for adding clues and facts for someone who would eventually be able to debug this.

In the meanwhile, are any of you in position to attach this UPS to a system with a more recent build of NUT (ideally master branch), to see if current code base fares better? See e.g. https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests about building it and running a driver to probe the device straight from the build workspace, no changes required to the OS where it was built.

bornecu, 2024-11-03:
Hi Jim
Unfortunately I don't really get how to run that test with the latest code that you propose there. But I have a raspberry pi that I use for some other purposes and I was thinking to maybe try a NUT server on it. If there is a possibility to use the latest NUT build with the pi, I would definitely try it.

jimklimov, 2024-11-04:
Well, one idea was to take a plain Linux system, and follow that wiki to install tools and build the current NUT code base, and run driver programs to see what they report about your device.

Another idea, as I was reminded by HA in this context, is to build a new plugin with current NUT, as detailed at https://github.com/networkupstools/nut/wiki/Building-NUT-integration-for-Home-Assistant (I don't use HA yet, so can't help more about it).

Testing-wise, I am wary of extra moving parts: the plugin and dashboards integration can be an extra level of complexity that might lose something (not pass all configs through yaml=>conf conversion from HA to NUT, not reflect all data that NUT did report, etc.)

jimklimov, 2025-11-10:
Latest release tarball is 2.8.4. Its build of the driver should also say about the RTU types it supports (Serial, TCP, maybe USB if the library could support it when NUT `configure` and build happened).

jimklimov, 2025-11-10:
> Do I have to build NUT after I build this shared library?

Yes, and you have to either install it into a visible location, or pass `configure --with-...` options to the build/private-install location. In this latter case, static library archive may be easier to link against (no run-time resolution paths are needed).