networkupstools/nut #2605: ups.conf: allow_killpower flag has no effect

desertwitch, 2024-08-27:
I'm trying to get the `allow_killpower` flag of `ups.conf` to work with a user, but we can't get it active no matter what.
Despite all our tries, the `driver.flag.allow_killpower` attribute shows as `0` and the `driver.killpower` command has no effect.

The flag seems to get set here:
https://github.com/networkupstools/nut/blob/f31b1ebf76edd7bdb125ab5de91923c6d6da2a71/drivers/main.c#L923-L931

Is it possible this later line overwrites the previously set flag with what seems to be a default?
https://github.com/networkupstools/nut/blob/f31b1ebf76edd7bdb125ab5de91923c6d6da2a71/drivers/main.c#L2684-L2686

It seems to be done a bit differently for other parameters (checking against a previously set value), an example:
https://github.com/networkupstools/nut/blob/f31b1ebf76edd7bdb125ab5de91923c6d6da2a71/drivers/main.c#L2645-L2647

desertwitch, 2024-08-27:
I dove into this some more (as good as possible from mobile phone), and perhaps it was overlooked here that the `allow_killpower` configuration flag can already set a different value at startup... and maybe this default of `0` was hard-coded with only runtime changes via `upsrw` in mind, not configurational ones? Possibly we need an integer with the default, later to be overwritten by either the configuration flag or at runtime? Not sure...

jimklimov, 2024-08-31:
Good catch. For history linking, the flag was added in or just before #1664 follow-ups like #1923 (there were several PRs about extending driver soclet protocol), as much as I can see from a phone too now :)

PR #2606 (2024-08-31): drivers/main.c: do not overwrite set allow_killpower flag with defaults
only apply the defaults for `driver.flag.allow_killpower` when `allow_killpower` flag was not already set.
current state applies defaults regardless of this, overwriting a previously set `allow_killpower` flag with `0`.
fixes #2605 , tested and working with APC device using `usbhid-ups` on Slackware 15.0 w/ gcc 11.2.0.
closed 2024-08-31 by jimklimov