networkupstools/nut #2608: Megatec NetagentA SNMP UPS (Ts Shara 3200VA) invalid ups.status:

juslex, 2024-08-30:
Olá, tenho um UPS Ts Shara 3200 VA com comunicação SNMP usando o netagentA da Megatec. No entanto, seja na versão 2.8.0 ou na 2.8.2 do NUT é informado um valor equivocado do status do ups:

Log do upsc

Init SSL without certificate database
battery.charge: 91
battery.charge.low: 15
device.contact: Administrator
device.description: UPS Agent
device.location: My Office
device.mfr: Tripp Lite / Phoenixtec
device.model: Senoid  32
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: snmp-ups
driver.parameter.ondelay: 0
driver.parameter.override.battery.charge.low: 15
driver.parameter.pollfreq: 5
driver.parameter.pollinterval: 2
driver.parameter.port: 10.0.1.10
driver.parameter.snmp_version: v1
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.2.986-986-gf31b1ebf7
driver.version.data: xppc MIB 0.40
driver.version.internal: 1.31
input.voltage: 226
output.frequency: 60
output.voltage: 115
ups.load: 27
ups.mfr: Tripp Lite / Phoenixtec
ups.model: Senoid  32
ups.status: OFF
ups.temperature: 29

Segue dados do snmpwalk -v 1:

iso.3.6.1.2.1.1.1.0 = STRING: "UPS Agent"
iso.3.6.1.2.1.1.2.0 = OID: iso.3.6.1.4.1.935.1.1.1
iso.3.6.1.2.1.1.3.0 = Timeticks: (13658170) 1 day, 13:56:21.70
iso.3.6.1.2.1.1.4.0 = STRING: "Administrator
"
iso.3.6.1.2.1.1.5.0 = STRING: "UPS Agent
"
iso.3.6.1.2.1.1.6.0 = STRING: "My Office
"
iso.3.6.1.2.1.1.7.0 = INTEGER: 79
iso.3.6.1.2.1.1.8.0 = Timeticks: (1049) 0:00:10.49
iso.3.6.1.2.1.1.9.1.2.1 = OID: iso.3.6.1.6.3.10.3.1.1
iso.3.6.1.2.1.1.9.1.2.2 = OID: iso.3.6.1.6.3.11.3.1.1
iso.3.6.1.2.1.1.9.1.2.3 = OID: iso.3.6.1.6.3.15.2.1.1
iso.3.6.1.2.1.1.9.1.2.4 = OID: iso.3.6.1.6.3.1
iso.3.6.1.2.1.1.9.1.2.5 = OID: iso.3.6.1.6.3.16.2.2.1
iso.3.6.1.2.1.1.9.1.2.6 = OID: iso.3.6.1.2.1.49
iso.3.6.1.2.1.1.9.1.2.7 = OID: iso.3.6.1.2.1.4
iso.3.6.1.2.1.1.9.1.2.8 = OID: iso.3.6.1.2.1.50
iso.3.6.1.2.1.1.9.1.2.9 = OID: iso.3.6.1.6.3.13.3.1.3
iso.3.6.1.2.1.1.9.1.2.10 = OID: iso.3.6.1.2.1.92
iso.3.6.1.2.1.1.9.1.3.1 = STRING: "The SNMP Management Architecture MIB."
iso.3.6.1.2.1.1.9.1.3.2 = STRING: "The MIB for Message Processing and Dispatching."
iso.3.6.1.2.1.1.9.1.3.3 = STRING: "The management information definitions for the SNMP User-based Security Model."
iso.3.6.1.2.1.1.9.1.3.4 = STRING: "The MIB module for SNMPv2 entities"
iso.3.6.1.2.1.1.9.1.3.5 = STRING: "View-based Access Control Model for SNMP."
iso.3.6.1.2.1.1.9.1.3.6 = STRING: "The MIB module for managing TCP implementations"
iso.3.6.1.2.1.1.9.1.3.7 = STRING: "The MIB module for managing IP and ICMP implementations"
iso.3.6.1.2.1.1.9.1.3.8 = STRING: "The MIB module for managing UDP implementations"
iso.3.6.1.2.1.1.9.1.3.9 = STRING: "The MIB modules for managing SNMP Notification, plus filtering."
iso.3.6.1.2.1.1.9.1.3.10 = STRING: "The MIB module for logging SNMP Notifications."
iso.3.6.1.2.1.1.9.1.4.1 = Timeticks: (1046) 0:00:10.46
[... 967 lines cut ...]
iso.3.6.1.2.1.88.1.4.2.1.5.6.95.115.110.109.112.100.95.109.116.101.84.114.105.103.103.101.114.70.97.108.108.105.110.103 = INTEGER: 1
iso.3.6.1.2.1.88.1.4.2.1.5.6.95.115.110.109.112.100.95.109.116.101.84.114.105.103.103.101.114.70.105.114.101.100 = INTEGER: 1
iso.3.6.1.2.1.88.1.4.2.1.5.6.95.115.110.109.112.100.95.109.116.101.84.114.105.103.103.101.114.82.105.115.105.110.103 = INTEGER: 1
iso.3.6.1.2.1.88.1.4.3.1.1.6.95.115.110.109.112.100.95.108.105.110.107.68.111.119.110 = OID: iso.3.6.1.6.3.1.1.5.3
iso.3.6.1.2.1.88.1.4.3.1.1.6.95.115.110.109.112.100.95.108.105.110.107.85.112 = OID: iso.3.6.1.6.3.1.1.5.4
iso.3.6.1.2.1.88.1.4.3.1.1.6.95.115.110.109.112.100.95.109.116.101.84.114.105.103.103.101.114.70.97.105.108.117.114.101 = OID: iso.3.6.1.2.1.88.2.0.4
iso.3.6.1.2.1.88.1.4.3.1.1.6.95.115.110.109.112.100.95.109.116.101.84.114.105.103.103.101.114.70.97.108.108.105.110.103 = OID: iso.3.6.1.2.1.88.2.0.3
iso.3.6.1.2.1.88.1.4.3.1.1.6.95.115.110.109.112.100.95.109.116.101.84.114.105.103.103.101.114.70.105.114.101.100 = OID: iso.3.6.1.2.1.88.2.0.1
iso.3.6.1.2.1.88.1.4.3.1.1.6.95.115.110.109.112.100.95.109.116.101.84.114.105.103.103.101.114.82.105.115.105.110.103 = OID: iso.3.6.1.2.1.88.2.0.2
iso.3.6.1.2.1.88.1.4.3.1.2.6.95.115.110.109.112.100.95.108.105.110.107.68.111.119.110 = STRING: "_snmpd"
iso.3.6.1.2.1.88.1.4.3.1.2.6.95.115.110.109.112.100.95.108.105.110.107.85.112 = STRING: "_snmpd"
iso.3.6.1.2.1.88.1.4.3.1.2.6.95.115.110.109.112.100.95.109.116.101.84.114.105.103.103.101.114.70.97.105.108.117.114.101 = STRING: "_snmpd"
iso.3.6.1.2.1.88.1.4.3.1.2.6.95.115.110.109.112.100.95.109.116.101.84.114.105.103.103.101.114.70.97.108.108.105.110.103 = STRING: "_snmpd"
iso.3.6.1.2.1.88.1.4.3.1.2.6.95.115.110.109.112.100.95.109.116.101.84.114.105.103.103.101.114.70.105.114.101.100 = STRING: "_snmpd"
iso.3.6.1.2.1.88.1.4.3.1.2.6.95.115.110.109.112.100.95.109.116.101.84.114.105.103.103.101.114.82.105.115.105.110.103 = STRING: "_snmpd"
iso.3.6.1.2.1.88.1.4.3.1.3.6.95.115.110.109.112.100.95.108.105.110.107.68.111.119.110 = STRING: "_linkUpDown"
iso.3.6.1.2.1.88.1.4.3.1.3.6.95.115.110.109.112.100.95.108.105.110.107.85.112 = STRING: "_linkUpDown"
iso.3.6.1.2.1.88.1.4.3.1.3.6.95.115.110.109.112.100.95.109.116.101.84.114.105.103.103.101.114.70.97.105.108.117.114.101 = STRING: "_triggerFail"
iso.3.6.1.2.1.88.1.4.3.1.3.6.95.115.110.109.112.100.95.109.116.101.84.114.105.103.103.101.114.70.97.108.108.105.110.103 = STRING: "_triggerFire"
iso.3.6.1.2.1.88.1.4.3.1.3.6.95.115.110.109.112.100.95.109.116.101.84.114.105.103.103.101.114.70.105.114.101.100 = STRING: "_triggerFire"
iso.3.6.1.2.1.88.1.4.3.1.3.6.95.115.110.109.112.100.95.109.116.101.84.114.105.103.103.101.114.82.105.115.105.110.103 = STRING: "_triggerFire"
iso.3.6.1.2.1.92.1.1.1.0 = Gauge32: 1000
iso.3.6.1.2.1.92.1.1.2.0 = Gauge32: 1440
iso.3.6.1.2.1.92.1.2.1.0 = Counter32: 0
iso.3.6.1.2.1.92.1.2.2.0 = Counter32: 0

jimklimov, 2024-08-31:
Hello, I am currently traveling so would only look in detail in a couple of weeks. Maybe someone would beat me and answer first.

What status did NUT 2.8.0 report? I think `OFF` was introduced later...

Also, this `snmpwalk` seems to have only the standard IETF data branch; can you please also query the "sysoid" used in your driver?

juslex, 2024-09-02:
What status did NUT 2.8.0 report?

There is:

Init SSL without certificate database
battery.charge: 100
battery.charge.low: 20
device.contact: Administrator
device.description: UPS Agent
device.location: My Office
device.mfr: Tripp Lite / Phoenixtec
device.model: Senoid  32
device.type: ups
driver.name: snmp-ups
driver.parameter.offdelay: 120
driver.parameter.ondelay: 0
driver.parameter.pollfreq: 10
driver.parameter.pollinterval: 2
driver.parameter.port: 10.0.1.10
driver.parameter.snmp_version: v1
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.data: xppc MIB 0.4
driver.version.internal: 1.21
input.voltage: 229
output.frequency: 60
output.voltage: 116
ups.load: 37
ups.mfr: Tripp Lite / Phoenixtec
ups.model: Senoid  32
ups.status: OFF
ups.temperature: 29

One more thing: How do I do this: can you please also query the "sysoid" for vendor extensions, as used in your sub-driver?

jimklimov, 2024-09-02:
Per https://github.com/networkupstools/nut/blob/master/drivers/xppc-mib.c#L29 try `.1.3.6.1.4.1.935`

The file also lists a couple of meanings for `OFF` state: sleeping or standby reported by the UPS at `.1.3.6.1.4.1.935.1.1.1.4.1.1.0`

Also, there are rather few mappings here, maybe check if it is handled better by other vendor MIBs (if it exposes several) or plain IETF?..

jimklimov, 2024-09-02:
> SNMP communication using Megatec's netagentA

Out of curiosity - what is the meaning of this set of words? Did not pass my dictionary check :)

juslex, 2024-09-02:
[image]

NetAgentA Is the name of the SNMP management application provided by Megatec http://www.megatec.com.tw

> > SNMP communication using Megatec's netagentA
> 
> Out of curiosity - what is the meaning of this set of words? Did not pass my dictionary check :)

juslex, 2024-09-02:
@jimklimov It seems to be a precise problem with this one of mine:

https://github.com/networkupstools/nut/issues/171

juslex, 2024-09-02:
I run gen-snmp-subdriver.sh and generated this 2 files:

https://drive.google.com/drive/folders/10GHjpkhUSbqTrDjc2PMFJjDgVTr1LgZE?usp=share_link

jimklimov, 2024-09-02:
Thanks! I've double-checked that `.1.3.6.1.4.1.935` branch seems to be XPPC MIB after all (our info is correct on that part): https://mibs.observium.org/mib/XPPC-MIB/

So the lines you posted are a good start to add into existing subdriver file and pick the correct data point names per `docs/nut-names.txt`.

I might try in a couple of weeks, or you can beat me to it ;)

closed 2026-03-20 by juslex