networkupstools/nut #2609: APC_MODBUS _apc_modbus_read_registers Timeouts

LucAce, 2024-09-01:
On an APC SMT1500C device using the rtu_usb version of libmodbus and a USB cable, reads fail with a timeout.

Since support for USB based APC Modbus is not mainlined, I am building the required RTU_USB version of libmodbus and compiling the main branch of NUT with usb and modbus support per the https://github.com/networkupstools/nut/wiki/APC-UPS-with-Modbus-protocol wiki page.  The build appears to be successful but I am not able to at get the apc_modbus driver to communicate with the UPS.

I have attached build logs and included the apc_modbus output below.  I have tested the connection using APC's PowerChute program (on a different system) and PowerChute is able to collect all of the ModBus only fields (like load and temperatures, etc).  So I am fairly confident it is not a physical connection issue.  Also when using the standard Debian bookworm build of NUT I am able to reliably communicate using the usbhid-ups driver but as it only provides a few fields, the Modbus driver is prefered.

Any pointers would be appreciated but I am not sure where to go from here.

NUT configuration indicating the new RTU USB features are included in the NUT build
```
checking for libmodbus version via pkg-config... 3.1.10 found
checking for libmodbus cflags... -I/usr/include/modbus
checking for libmodbus ldflags... -lmodbus
checking for modbus.h... yes
checking for modbus_new_rtu... yes
checking for modbus_new_tcp... yes
checking for modbus_set_byte_timeout... yes
checking for modbus_set_response_timeout... yes
checking for modbus_new_rtu_usb... yes
checking types of arguments for modbus_set_byte_timeout... sec_usec_uint32_cast_timeval_fields
Found types to use for modbus_set_byte_timeout: sec_usec_uint32_cast_timeval_fields
```

Using apc_modbus to probe the unit this is the output:
```
$ sudo ./apc_modbus -DDDDDD -s ups -x port=auto
   0.000001     [D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools driver 2.8.2.989-989-g4bf05e115 (development iteration after 2.8.2) - NUT APC Modbus driver 0.10
   0.000408     [D5] send_to_all: SETINFO driver.version.usb "libusb-1.0.26 (API: 0x1000109)"
   0.000447     [D1] Using USB implementation: libusb-1.0.26 (API: 0x1000109)
   0.000504     [D3] main_arg: var='port' val='auto'
   0.000533     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000561     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000604     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000624     [D1] Network UPS Tools version 2.8.2.989-989-g4bf05e115 (development iteration after 2.8.2) built with gcc (Debian 12.2.0-14) 12.2.0 and configured with flags: --prefix=/usr --with-doc=no --with-drivers=apc_modbus,usbhid-ups --with-usb --with-modbus --sysconfdir=/etc/nut --includedir=/usr/include --mandir=/usr/share/man --libdir=/usr/lib/aarch64-linux-gnu --libexecdir=/usr/libexec --with-ssl --with-nss --with-cgi --with-dev --with-statepath=/run/nut --with-altpidpath=/run/nut --with-drvpath=/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut --with-htmlpath=/usr/share/nut/www --with-pidpath=/run/nut --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib/aarch64-linux-gnu/pkgconfig --with-user=nut --with-group=nut --with-udev-dir=/lib/udev --with-systemdsystemunitdir=/lib/systemd/system --with-systemdshutdowndir=/lib/systemd/system-shutdown --with-systemdtmpfilesdir=/usr/lib/tmpfiles.d
   0.000695     [D1] debug level is '6'
   0.000724     [D5] send_to_all: SETINFO driver.debug "6"
   0.000754     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.001755     [D1] Succeeded to become_user(nut): now UID=106 GID=112
   0.001830     [D1] Signalling UPS [ups]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.001900     Can't open /run/nut/apc_modbus-ups: No such file or directory
   0.001938     [D1] Request for other driver to exit returned code -1
   0.001967     [D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
   0.002003     [D5] send_to_all: SETINFO device.type "ups"
   0.002034     [D5] send_to_all: SETINFO driver.state "init.device"
Number of USB devices: 5
Considering device #0 (051d:0003)
   0.047244     [D2] _apc_modbus_usb_callback: Matched American Power Conversion  Smart-UPS_1500 FW:UPS 15.5 / ID=1015 (USB VID/PID 051d:0003)
   0.047332     [D2] _apc_modbus_usb_callback: Checking American Power Conversion  Smart-UPS_1500 FW:UPS 15.5 / ID=1015 (USB VID/PID 051d:0003) report descriptors
   0.048073     [D1] _apc_modbus_usb_callback: Found report ids RX=0x90 TX=0x89
   0.048160     [D5] send_to_all: SETINFO ups.vendorid "051d"
   0.048223     [D5] send_to_all: SETINFO ups.productid "0003"
Found Device 0 (Path 1-1.1.2):
  Vendor ID: 0x051d
  Product ID: 0x0003
Bytes flushed (0)
   0.058556     [D5] send_to_all: SETINFO driver.state "init.quiet"
   0.058602     [D5] send_to_all: SETINFO driver.version "2.8.2.989-989-g4bf05e115"
   0.058623     [D5] send_to_all: SETINFO driver.version.internal "0.10"
   0.058645     [D5] send_to_all: SETINFO driver.name "apc_modbus"
   0.058663     [D5] send_to_all: SETINFO driver.state "init.info"
[01][03][02][04][00][78]
Waiting for a confirmation...
ERROR Connection timed out: select
   0.567189     _apc_modbus_read_registers: Read of 516:636 failed: Connection timed out (auto)
Bytes flushed (0)
   1.577519     Can't read inventory information from the UPS
   1.577610     [D5] send_to_all: SETINFO driver.state "cleanup.upsdrv"
   1.578214     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
   1.578280     upsnotify: notify about state 4 with libsystemd: was requested, but not running as a service unit now, will not spam more about it
   
   1.578310     upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
```

*Host Details*
* Hardware: Raspberry Pi 3B+
* OS: Raspberry Pi OS Lite (64-bit)
* Kernel version: 6.6
* Debian version: 12 (bookworm)
* uname: Linux rpi 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

**ups.conf**
```
[ups]
    driver = "apc_modbus"
    port = "auto"
```

**upsd.conf**
```
LISTEN 0.0.0.0 3493
```

**nut.conf**
```
MODE=netserver
```

LucAce, 2024-09-01:
Example build scripts.

[build_libmodbus.sh.txt](https://github.com/user-attachments/files/16830476/build_libmodbus.sh.txt)
[build_nut.sh.txt](https://github.com/user-attachments/files/16830477/build_nut.sh.txt)

LucAce, 2024-09-01:
Build scripts used to build libmodbus and NUT (from GIT).

[build_libmodbus.txt](https://github.com/user-attachments/files/16830481/build_libmodbus.txt)
[build_nut.txt](https://github.com/user-attachments/files/16830482/build_nut.txt)

LucAce, 2024-09-01:
Build command for libmodbus and NUT

*build_libmodbus.sh*
```bash
#!/bin/bash

if [ "$EUID" -ne 0 ]; then
    echo -e "ERROR: Please run as root\n"
    exit
fi

# Install dependencies
apt-get install -y \
    autoconf libusb-1.0-0-dev libusb-dev git libtool gawk

# Build RTU USB Library
git clone -b rtu_usb https://github.com/networkupstools/libmodbus
cd libmodbus
./autogen.sh

export DEB_HOST_MULTIARCH=aarch64-linux-gnu

./configure \
    --prefix=/usr \
    --sysconfdir=/etc/nut \
    --includedir=/usr/include \
    --mandir=/usr/share/man \
    --libdir=/usr/lib/${DEB_HOST_MULTIARCH} \
    --libexecdir=/usr/libexec \
    --with-libusb \
    --enable-static \
    --enable-shared \
    --with-pkgconfig-dir=/usr/lib/${DEB_HOST_MULTIARCH}/pkgconfig

make -j4 && make install
```

*build_nut.sh*
```bash
#!/bin/bash

if [ "$EUID" -ne 0 ]; then
    echo -e "ERROR: Please run as root\n"
    exit
fi

# Install dependencies
apt-get install -y \
    libssl-dev libnss3-dev libgd-dev augeas-tools \
    libaugeas-dev augeas-lenses libcppunit-dev \
    ccache time perl curl make autoconf automake \
    libltdl-dev libtool cppcheck pkg-config gcc g++ \
    clang valgrind libsystemd-dev

# Build NUT
git clone https://github.com/networkupstools/nut
cd nut
./autogen.sh

export DEB_HOST_MULTIARCH="aarch64-linux-gnu"

./configure \
    --prefix=/usr \
    --with-doc=no \
    --with-drivers="apc_modbus,usbhid-ups" \
    --with-usb \
    --with-modbus \
    --sysconfdir=/etc/nut \
    --includedir=/usr/include \
    --mandir=/usr/share/man \
    --libdir=/usr/lib/${DEB_HOST_MULTIARCH} \
    --libexecdir=/usr/libexec \
    --with-ssl \
    --with-nss \
    --with-cgi \
    --with-dev \
    --with-statepath=/run/nut \
    --with-altpidpath=/run/nut \
    --with-drvpath=/lib/nut \
    --with-cgipath=/usr/lib/cgi-bin/nut \
    --with-htmlpath=/usr/share/nut/www \
    --with-pidpath=/run/nut \
    --datadir=/usr/share/nut \
    --with-pkgconfig-dir=/usr/lib/${DEB_HOST_MULTIARCH}/pkgconfig \
    --with-user=nut \
    --with-group=nut \
    --with-udev-dir=/lib/udev \
    --with-systemdsystemunitdir=/lib/systemd/system \
    --with-systemdshutdowndir=/lib/systemd/system-shutdown \
    --with-systemdtmpfilesdir=/usr/lib/tmpfiles.d

make -j4

echo ""
read -p "Do you wish to continue? [Y/N] " yesno
case $yesno in
    [Yy]* )
        echo "Installing..."
    ;;
    * )
        echo "Exiting..."
        exit
    ;;
esac

make install
```

jimklimov, 2024-09-01:
CC @EchterAgo : does this ring any bells, please?

LucAce, 2024-09-02:
In an effort to rule out the RPi being a factor, I recreated the build on an x86 system with Ubuntu 24.04.1 which results in:

```
$ ./apc_modbus -DDDDDD -a ups -x port=auto
   0.000001     [D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools driver 2.8.2.1004-1004-g38d8078ab (development iteration after 2.8.2) - NUT APC Modbus driver 0.10
   0.000145     [D5] send_to_all: SETINFO driver.version.usb "libusb-1.0.27 (API: 0x100010a)"
   0.000160     [D1] Using USB implementation: libusb-1.0.27 (API: 0x100010a)
   0.000311     [D5] do_upsconf_args: confupsname=(null), var=maxretry, val=3
   0.000316     [D5] do_upsconf_args: call do_global_args()
   0.000319     [D3] do_global_args: var='maxretry' val='3'
   0.000388     [D5] do_upsconf_args: confupsname=ups, var=driver, val=apc_modbus
   0.000392     [D5] do_upsconf_args: call main_arg()
   0.000396     [D3] main_arg: var='driver' val='apc_modbus'
   0.000399     [D5] do_upsconf_args: not a main_arg()
   0.000402     [D5] do_upsconf_args: this is a 'driver' setting, may we proceed?
   0.000406     [D6] testval_reloadable: var=driver, oldval=apc_modbus, newval=apc_modbus, reloadable=0, reload_flag=0
   0.000410     [D6] testval_reloadable: verdict for (re)loading var=driver value: -1
   0.000413     [D5] do_upsconf_args: 'driver' setting already applied with this value
   0.000417     [D5] do_upsconf_args: confupsname=ups, var=port, val=auto
   0.000421     [D5] do_upsconf_args: call main_arg()
   0.000424     [D3] main_arg: var='port' val='auto'
   0.000428     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000433     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000439     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000449     [D3] main_arg: var='port' val='auto'
   0.000453     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000456     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000460     [D1] Network UPS Tools version 2.8.2.1004-1004-g38d8078ab (development iteration after 2.8.2) built with gcc (Ubuntu 13.2.0-23ubuntu4) 13.2.0 and configured with flags: --prefix=/usr --with-doc=no --with-drivers=apc_modbus,usbhid-ups --with-usb --with-modbus --sysconfdir=/etc/nut --includedir=/usr/include --mandir=/usr/share/man --libdir=/usr/lib/x86_64-linux-gnu --libexecdir=/usr/libexec --with-ssl --with-nss --with-cgi --with-dev --with-statepath=/run/nut --with-altpidpath=/run/nut --with-drvpath=/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut --with-htmlpath=/usr/share/nut/www --with-pidpath=/run/nut --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib/x86_64-linux-gnu/pkgconfig --with-user=nut --with-group=nut --with-udev-dir=/lib/udev --with-systemdsystemunitdir=/lib/systemd/system --with-systemdshutdowndir=/lib/systemd/system-shutdown --with-systemdtmpfilesdir=/usr/lib/tmpfiles.d
   0.000484     [D1] debug level is '6'
   0.000489     [D5] send_to_all: SETINFO driver.debug "6"
   0.000495     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.001387     [D1] Succeeded to become_user(nut): now UID=110 GID=111
   0.001413     [D1] Signalling UPS [ups]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.001425     Can't open /run/nut/apc_modbus-ups: No such file or directory
   0.001430     [D1] Request for other driver to exit returned code -1
   0.001434     [D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
   0.001440     [D5] send_to_all: SETINFO device.type "ups"
   0.001444     [D5] send_to_all: SETINFO driver.state "init.device"
Number of USB devices: 3
Considering device #1 (051d:0003)
   0.025093     [D2] _apc_modbus_usb_callback: Matched American Power Conversion  Smart-UPS_1500 FW:UPS 15.5 / ID=1015 (USB VID/PID 051d:0003)
   0.025120     [D2] _apc_modbus_usb_callback: Checking American Power Conversion  Smart-UPS_1500 FW:UPS 15.5 / ID=1015 (USB VID/PID 051d:0003) report descriptors
   0.025264     [D1] _apc_modbus_usb_callback: Found report ids RX=0x90 TX=0x89
   0.025279     [D5] send_to_all: SETINFO ups.vendorid "051d"
   0.025287     [D5] send_to_all: SETINFO ups.productid "0003"
Found Device 1 (Path 1-8):
  Vendor ID: 0x051d
  Product ID: 0x0003
Bytes flushed (63)
   0.036922     [D5] send_to_all: SETINFO driver.state "init.quiet"
   0.036947     [D5] send_to_all: SETINFO driver.version "2.8.2.1004-1004-g38d8078ab"
   0.036966     [D5] send_to_all: SETINFO driver.version.internal "0.10"
   0.036982     [D5] send_to_all: SETINFO driver.name "apc_modbus"
   0.036995     [D5] send_to_all: SETINFO driver.state "init.info"
[01][03][02][04][00][78]
Waiting for a confirmation...
*** bit out of range 0 - FD_SETSIZE on fd_set ***: terminated
Aborted (core dumped)
```

```
$ ./upsdrvctl start
Network UPS Tools upsdrvctl - UPS driver controller 2.8.2.1004-1004-g38d8078ab (development iteration after 2.8.2)
Network UPS Tools driver 2.8.2.1004-1004-g38d8078ab (development iteration after 2.8.2) - NUT APC Modbus driver 0.10
*** bit out of range 0 - FD_SETSIZE on fd_set ***: terminated
Driver exited abnormally
```

jimklimov, 2024-09-02:
Thanks for the diligence about multi-platform check. RPi's USB chips were also suspects a few times.

But the segfault was reported in several issues recently - so seeing it misbehave differently per platform is at least "most intriguing!" :)

CC @EchterAgo again

I might at best look at code in a couple of weeks or more, and have no similar APC device to check against.

LucAce, 2024-09-03:
@jimklimov Thanks for the reply.  I will keep a look out for updates, let me know if there is anything you'd like me to try.

adlerweb, 2024-09-23:
To add another device: I'm seeing a similar behavior using a Smart-UPS 750 running FW 9.3 on NUT 2.8.2 with [libmodbus](https://github.com/networkupstools/libmodbus) c02f62b58aa4ce7538e087ef46266d2a048b0bd6.

```
/lib64/nut/apc_modbus  -DDDDDD -s ups -x port=auto
   0.000000     [D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools - NUT APC Modbus driver 0.10 (2.8.2)
   0.000105     [D5] send_to_all: SETINFO driver.version.usb "libusb-1.0.27 (API: 0x100010a)"
   0.000114     [D1] Using USB implementation: libusb-1.0.27 (API: 0x100010a)
   0.000128     [D3] main_arg: var='port' val='auto'
   0.000135     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000141     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000156     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000162     [D1] Network UPS Tools version 2.8.2 (release/snapshot of 2.8.2) built with x86_64-pc-linux-gnu-gcc (Gentoo 13.3.1_p20240614 p17) 13.3.1 20240614 and configured with flags: --prefix=/usr --build=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --mandir=/usr/share/man --infodir=/usr/share/info --datadir=/usr/share --sysconfdir=/etc --localstatedir=/var/lib --datarootdir=/usr/share --disable-dependency-tracking --disable-silent-rules --disable-static --docdir=/usr/share/doc/nut-2.8.2-r1 --htmldir=/usr/share/doc/nut-2.8.2-r1/html --with-sysroot=/ --libdir=/usr/lib64 --datadir=/usr/share/nut --datarootdir=/usr/share/nut --disable-static --disable-strip --disable-Werror --sysconfdir=/etc/nut --with-dev --with-drvpath=/lib64/nut --with-group=nut --with-htmlpath=/usr/share/nut/html --with-logfacility=LOG_DAEMON --with-statepath=/var/lib/nut --with-systemdsystemunitdir=/lib/systemd/system --with-systemdtmpfilesdir=/usr/lib/tmpfiles.d --with-udev-dir=/lib/udev --with-user=nut --without-powerman --without-python --without-python2 --with-altpidpath=/run/nut --with-pidpath=/run/nut --with-usb --with-modbus --with-modbus-includes=-I/usr/local/include/modbus --with-modbus-libs='-L/usr/local/lib -lmodbus' --without-cgi --without-gpio --without-linux_i2c --without-freeipmi --without-ipmi --without-nut_monitor --without-pynut --without-serial --without-snmp --with-ssl --without-libsystemd --without-wrap --with-usb --without-neon --without-avahi --with-doc=man --without-python3
   0.000197     [D1] debug level is '6'
   0.000203     [D5] send_to_all: SETINFO driver.debug "6"
   0.000210     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.000409     [D1] Succeeded to become_user(nut): now UID=84 GID=84
   0.000433     [D5] send_to_all: SETINFO device.type "ups"
   0.000440     [D5] send_to_all: SETINFO driver.state "init.device"
Number of USB devices: 25
Considering device #0 (9022:d483)
libusb_open for device #0 failed: Access denied (insufficient permissions)
Considering device #1 (1c4f:0002)
libusb_open for device #1 failed: Access denied (insufficient permissions)
Considering device #2 (10c4:ea60)
libusb_open for device #2 failed: Access denied (insufficient permissions)
Considering device #4 (10c4:ea60)
libusb_open for device #4 failed: Access denied (insufficient permissions)
Considering device #5 (1a40:0201)
libusb_open for device #5 failed: Access denied (insufficient permissions)
Considering device #6 (051d:0003)
   0.026713     [D2] _apc_modbus_usb_callback: Matched American Power Conversion  Smart-UPS 750 FW:UPS 09.3 / ID=18 (USB VID/PID 051d:0003)
   0.026745     [D2] _apc_modbus_usb_callback: Checking American Power Conversion  Smart-UPS 750 FW:UPS 09.3 / ID=18 (USB VID/PID 051d:0003) report descriptors
   0.026906     [D1] _apc_modbus_usb_callback: Found report ids RX=0x90 TX=0x89
   0.026926     [D5] send_to_all: SETINFO ups.vendorid "051d"
   0.026945     [D5] send_to_all: SETINFO ups.productid "0003"
Found Device 6 (Path 12-1):
  Vendor ID: 0x051d
  Product ID: 0x0003
Bytes flushed (63)
   0.047954     [D5] send_to_all: SETINFO driver.state "init.quiet"
   0.047967     [D5] send_to_all: SETINFO driver.version "2.8.2"
   0.047975     [D5] send_to_all: SETINFO driver.version.internal "0.10"
   0.047981     [D5] send_to_all: SETINFO driver.name "apc_modbus"
   0.047997     [D5] send_to_all: SETINFO driver.state "init.info"
[01][03][02][04][00][78]
Waiting for a confirmation...
*** bit out of range 0 - FD_SETSIZE on fd_set ***: terminated
```

jimklimov, 2024-09-24:
To summarize current findings in one place, we see several similar patterns ending up in driver crash (typically with `bit out of range 0 - FD_SETSIZE on fd_set` or `buffer overflow detected`), or a graceful exit (with `ERROR Connection timed out: select`; not sure if `_apc_modbus_read_registers ... timed out` cases are all in the same bucket):

* https://github.com/networkupstools/nut/issues/2609#issue-2499677698 - on Raspberry Pi 3B+ 64-bit build (Pi-OS lite based on Debian 12 bookworm)
````
...
   0.058663     [D5] send_to_all: SETINFO driver.state "init.info"
[01][03][02][04][00][78]
Waiting for a confirmation...
ERROR Connection timed out: select
   0.567189     _apc_modbus_read_registers: Read of 516:636 failed: Connection timed out (auto)
Bytes flushed (0)
   1.577519     Can't read inventory information from the UPS
   1.577610     [D5] send_to_all: SETINFO driver.state "cleanup.upsdrv"
   1.578214     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
````

* https://github.com/networkupstools/nut/issues/2609#issuecomment-2323657557 - same device as above, on an x86 system with Ubuntu 24.04.1:
````
...
   0.036995     [D5] send_to_all: SETINFO driver.state "init.info"
[01][03][02][04][00][78]
Waiting for a confirmation...
*** bit out of range 0 - FD_SETSIZE on fd_set ***: terminated
Aborted (core dumped)
````

* https://github.com/networkupstools/nut/pull/2063#issuecomment-1905299561
````
Network UPS Tools - UPS driver controller 2.8.1
Network UPS Tools - NUT APC Modbus driver 0.01 (2.8.1)
*** buffer overflow detected ***: terminated
Driver exited abnormally
````
** Notably, `Jan 23 05:00:27 nut kernel: [  250.593376] usb 2-1: usbfs: process 918 (apc_modbus) did not claim interface 0 before use`

* https://github.com/networkupstools/nut/issues/2289#issue-2104336348
````
...
   0.050384     [D5] send_to_all: SETINFO driver.state "init.info"
[01][03][02][04][00][58]
Waiting for a confirmation...
*** buffer overflow detected ***: terminated
Aborted
````

* https://github.com/networkupstools/nut/issues/2289#issuecomment-2310851725
````
...
   0.040156     [D5] send_to_all: SETINFO driver.state "init.info"
[01][03][02][04][00][78]
Waiting for a confirmation...
*** bit out of range 0 - FD_SETSIZE on fd_set ***: terminated
Aborted (core dumped)
````

* https://github.com/networkupstools/nut/issues/2289#issuecomment-2313515279
````
...
5.438365 [D5] send_to_all: SETINFO driver.state "init.info"
[01][03][02][04][00][78]
Waiting for a confirmation...
***** buffer overflow detected *: terminated
Aborted
````

Separately from this (maybe... seems similar to first sample above), in the later after-party discussion of original PR #2063 there were at least occasional (fixed by cable re-plug) faults with:
````
Network UPS Tools - UPS driver controller 2.8.1
Network UPS Tools - NUT APC Modbus driver 0.01 (2.8.1)
_apc_modbus_read_registers: Read of 516:604 failed: Connection timed out (auto)
[... 22 lines cut ...]
I suppose a run with `gdb` or similar would be helpful to identify the line at which it actually crashes - is it in `FD_SET` or in some later loop? Note that `FD_SET()` is probably a macro, and probably defined differently depending on architecture, e.g. here are some hits from the systems I could look at quickly:
````
:; grep -rw FD_SET /usr/include/
...
/usr/include/aarch64-linux-gnu/sys/select.h:#define     FD_SET(fd, fdsetp)      __FD_SET (fd, fdsetp)
/usr/include/x86_64-linux-gnu/sys/select.h:#define      FD_SET(fd, fdsetp)      __FD_SET (fd, fdsetp)
...

:; grep -rw -A5  __FD_SET /usr/include/
...
usr/include/aarch64-linux-gnu/bits/select.h:#define __FD_SET(d, s) \
/usr/include/aarch64-linux-gnu/bits/select.h-  ((void) (__FDS_BITS (s)[__FD_ELT(d)] |= __FD_MASK(d)))
...
/usr/include/x86_64-linux-gnu/bits/select.h:#define __FD_SET(d, s) \
/usr/include/x86_64-linux-gnu/bits/select.h-  ((void) (__FDS_BITS (s)[__FD_ELT(d)] |= __FD_MASK(d)))
...
````

There are other FD macros nearby, all defined in terms of access to `__FDS_BITS (s)[]` array - hence the likely overflow if that array is too small for the offset number that lands into it as an index.

https://stackoverflow.com/questions/22827510/how-to-avoid-bad-fd-set-buffer-overflow-crash suggests there are problems with 1000+ socket/FD numbers, and also reminds to check if there are FD leaks (loops with a FD not closed so another number gets opened and issued for some next operation).

Another question is what `ctx->s` is in `FD_SET(ctx->s, &rset);` expression, my random guess is it might be mis-populated in the new code for libmodbus `rtu_usb` branch?..

CC @EchterAgo

jimklimov, 2024-09-24:
I suppose `ctx->s` is initialized to `-1` in `_modbus_init_common()` and remains that way. At least, I do not see any calls to `modbus_set_socket()` or direct access to the field in `src/modbus-rtu-usb.c`.

That's about all I can say for now, lacking a device to test against, time (gotta run in fact), etc.

But I really hope the data-dig above can get someone started!

ZerglingGo, 2024-09-25:
@jimklimov I can provide you with system. If you want, Please send me a mail.

jimklimov, 2024-09-25:
@ZerglingGo : thanks I will, but first...

@ all : were all those attempts done with the proper NUT setup, drivers running unprivileged by default?

That "Resource temporarily unavailable" clue above bothers me since I saw it yesterday. Can you please try running the driver as `root` (with `-x root` option if running on command line, or `user=root` in its `ups.conf` section)?

The hypothesis I'm trying to check here is if permissions for `nut` or `ups` etc. account to use the devfs node for that USB device did not get set up correctly, so the driver does not succeed (fully?) opening the device, does not abort either, and gets into some sort of undefined state (e.g. polling 0 open/responding file descriptors or something).

In the debug dumps posted earlier I see `_apc_modbus_usb_callback` lines with apparently device details, but *maybe* it is something cached by the OS USB handler (name/model/version) and not reliant on access to the actual device from the driver program.

adlerweb, 2024-09-25:
I guess that should be `-u root` and not `-x root`? If so: Same output. https://gist.github.com/adlerweb/8803c7c4c088a62b4036f4179f0c246b

jimklimov, 2024-09-25:
Sorry, yes. Thought about `-x user=root` then edited and lost a bit :D

ZerglingGo, 2024-09-26:
https://gist.github.com/ZerglingGo/1cc0d27ae30e26d4000d674b0f7da6ec#file-strace_apc_modbus-log-2
Here it is running result with `-u root`

RadekHvizdos, 2024-12-14:
I was able to replicate the issue with the latest code in git. What can we do to push this forward?

Thanks

jimklimov, 2024-12-14:
> What can we do to push this forward?

Generally speaking, and ideally -- have someone with the device and the basic programming/debugging skills run the NUT driver through a debugger...
* I was earlier able to wrangle NetBeans and VSCode into wrapping GDB to step through drivers, etc.
* If there is a core dump produced by the failed program (creation of such files may be forbidden in modern OSes by default), you can try to load it into `gdb` and at least print out the call stack trace, to make sure which methods and lines are the problematic code path. Similarly if the program fails while running under `gdb`, may not be constrained by permissions to create a core file as such.

These clues can help a lot with the guesswork about what went wrong. My two primary suspect areas would be new code in the driver or in libmodbus, possibly their interactions with further libraries like libusb.

jimklimov, 2024-12-15:
I haven't finished my read through the changes, but it seems there are two variants of modbus+usb init (in lib changes, and one of those is used by the driver): one way has libmodbus use the USB devfs path, and another gets a libusb context like NUT drivers do (so some USB codepath here might end up with a socket or not - still figuring out between other weekend chores).

I think I'll provisionally try to post a variant of libmodbus code that would check if "s" (socket) field is invalid and skip trying to `select` on it. If you'd be able to build NUT against that and see if it behaves better - that would be great.

RadekHvizdos, 2024-12-15:
@jimklimov, that is great. Yes, please do let us know when your variant is ready and I can test in in my setup.

jimklimov, 2024-12-19:
For now I got stuck on this, wanted to reproduce the fault locally (without an APC UPS, just poking a random USB-Serial converter and faking successes in libmodbus), but it behaves sanely and does not segfault even though `ctx->s==-1` :\

jimklimov, 2024-12-20:
In any case, per https://stackoverflow.com/a/69786982/4715872 we should check for, and avoid, using -1 value here (gotta check how skipping this method in case of libusb backend plays with libmodbus native logic):

> An fd_set is a fixed size buffer. Executing FD_CLR() or FD_SET() with a value of fd that is negative or is equal to or larger than FD_SETSIZE will result in undefined behavior. Moreover, POSIX requires fd to be a valid file descriptor.

Just in case, do your devices also present as a serial port (e.g. `/dev/ttyUSB0`)? That might work differently as a serial Modbus RTU.

An alternative experiment could be to hack around apc_modbus.c and use the second connection method from the library (usb but for paths).

jimklimov, 2025-01-12:
Cheers all! I've got a moment to look at this again.

The "second connection method" won't help, they converge to same logic.

I've tried a couple of fixes in libmodbus fork:
* do not `select()` on `ctx->s` of it is invalid, but do all the rest of the receiving method
* try to get polling file descriptors from libusb context, and if there's exactly one returned, use it as `ctx->s`

As I have nowhere to actually test this change, can you please build that branch, and then NUT against the resulting library (instructions mentioned above)?

````
:; git clone https://github.com/jimklimov/libmodbus -b issue-nut-2609 libmodbus-issue-nut-2609
:; cd libmodbus-issue-nut-2609
:; ./autogen.sh
:; ./configure --enable-static --disable-shared ...
````

moritzkraemer, 2025-01-13:
I'm getting this error:
`./apc_modbus: symbol lookup error: ./apc_modbus: undefined symbol: modbus_new_rtu_usb`

moritzkraemer, 2025-01-13:
it seems to work now:

```
   0.000008     [D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools 2.8.2.1877-1877-ge7788b97d (development iteration after 2.8.2) - NUT APC Modbus driver with USB support 0.11
   0.000110     [D5] send_to_all: SETINFO driver.version.usb "libusb-1.0.27 (API: 0x0100010A)"
   0.000126     [D1] Using USB implementation: libusb-1.0.27 (API: 0x0100010A)
   0.000139     [D1] This build of the driver is USB-capable; also Serial and TCP Modbus RTU are supported
   0.000166     [D3] main_arg: var='port' val='auto'
   0.000203     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000225     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000242     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000254     [D1] Network UPS Tools version 2.8.2.1877-1877-ge7788b97d (development iteration after 2.8.2) built with gcc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0 and configured with flags: --prefix=/usr --with-doc=no --with-drivers=apc_modbus,usbhid-ups --with-usb --with-modbus --sysconfdir=/etc/nut --includedir=/usr/include --mandir=/usr/share/man --libdir=/usr/lib/aarch64-linux-gnu --libexecdir=/usr/libexec --with-ssl --with-nss --with-cgi --with-dev --with-statepath=/run/nut --with-altpidpath=/run/nut --with-drvpath=/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut --with-htmlpath=/usr/share/nut/www --with-pidpath=/run/nut --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib/aarch64-linux-gnu/pkgconfig --with-user=nut --with-group=nut --with-udev-dir=/lib/udev --with-systemdsystemunitdir=/lib/systemd/system --with-systemdshutdowndir=/lib/systemd/system-shutdown --with-systemdtmpfilesdir=/usr/lib/tmpfiles.d
   0.000350     [D1] debug level is '6'
   0.000366     [D5] send_to_all: SETINFO driver.debug "6"
   0.000379     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.002457     [D1] Succeeded to become_user(nut): now UID=110 GID=111
   0.002489     [D1] Signalling UPS [test]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.002518     Can't open /run/nut/apc_modbus-test: No such file or directory
   0.002532     [D1] Request for other driver to exit returned code -1
   0.002547     [D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
   0.002564     [D5] send_to_all: SETINFO device.type "ups"
   0.002585     [D5] send_to_all: SETINFO driver.state "init.device"
Number of USB devices: 4
Considering device #1 (8087:0aaa)
libusb_open for device #1 failed: Access denied (insufficient permissions)
Considering device #2 (051d:0003)
   0.044408     [D2] _apc_modbus_usb_callback: Matched American Power Conversion  Smart-UPS_1500 FW:UPS 18.0 / ID=1027 (USB VID/PID 051d:0003)
   0.044444     [D2] _apc_modbus_usb_callback: Checking American Power Conversion  Smart-UPS_1500 FW:UPS 18.0 / ID=1027 (USB VID/PID 051d:0003) report descriptors
   0.044671     [D1] _apc_modbus_usb_callback: Found report ids RX=0x90 TX=0x89
   0.044693     [D5] send_to_all: SETINFO ups.vendorid "051d"
   0.044708     [D5] send_to_all: SETINFO ups.productid "0003"
Found Device 2 (Path 1-4):
  Vendor ID: 0x051d
  Product ID: 0x0003
Got a list of 3 libusb file descriptors to pollBytes flushed (0)
   0.054991     [D5] send_to_all: SETINFO driver.state "init.quiet"
   0.055012     [D5] send_to_all: SETINFO driver.version "2.8.2.1877-1877-ge7788b97d"
   0.055031     [D5] send_to_all: SETINFO driver.version.internal "0.11"
   0.055049     [D5] send_to_all: SETINFO driver.name "apc_modbus"
   0.055066     [D5] send_to_all: SETINFO driver.state "init.info"
[01][03][02][04][00][78]
Waiting for a confirmation...
Using a backend without a file descriptor, will not select() on it.
ERROR Connection timed out: select
<01><03><F0><55><50><53><20><31><38><2E><30><00><00><00><00><00><00><00><00><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><53><6D><61><72><74><2D><55><50><53><20><31><35><30><30><20><20><20><20><20><20><20><20><20><20><20><20><20><20><20><20><20><20><53><4D><54><31><35><30><30><52><4D><49><32><55><43><20><20><20><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><41><53><32><33><31><33><33><35><31><35><37><36><20><20><20><20><41><50><43><52><42><43><31><35><39><20><20><20><20><20><20><20><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><05><DC><03><E8><00><03><21><29><00><20><00><00><00><00><21><7A><41><50><43><20><55><50><53><20><20><20><20><20><20><20><20><20><55><6E><73><77><69><74><63><68><65><64><20><47><72><6F><75><70><4F><75><74><6C><65><74><20><47><72><6F><75><70><20><31><20><20><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00>
   1.056350     [D5] send_to_all: SETINFO outlet.group.count "2"
   1.056506     [D5] send_to_all: SETINFO ups.firmware "UPS 18.0"
   1.056578     [D5] send_to_all: SETFLAGS ups.firmware STRING
   1.056802     [D5] send_to_all: SETAUX ups.firmware 16
   1.056840     [D5] send_to_all: SETINFO ups.model "Smart-UPS 1500"
   1.056868     [D5] send_to_all: SETFLAGS ups.model STRING
   1.056897     [D5] send_to_all: SETAUX ups.model 32
   1.056931     [D5] send_to_all: SETINFO ups.serial "ASxxx"
   1.056966     [D5] send_to_all: SETFLAGS ups.serial STRING
   1.056999     [D5] send_to_all: SETAUX ups.serial 16
   1.057038     [D5] send_to_all: SETINFO ups.power.nominal "1500"
[... cut ...]
ups.mfr: American Power Conversion
ups.mfr.date: 2023-03-30
ups.model: Smart-UPS 1500
ups.power: 478.71
ups.power.nominal: 1500
ups.productid: 0003
ups.realpower: 267.89
ups.realpower.nominal: 1000
ups.serial: ASxxx
ups.status: OL HE
ups.test.result: Passed, Source: LocalUI
ups.timer.reboot: -1
ups.timer.shutdown: -1
ups.timer.start: -1
ups.vendorid: 051d
  14.089881     [D5] send_to_all: SETINFO driver.state "cleanup.upsdrv"
  14.090455     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
```

jimklimov, 2025-01-13:
Looks great, thanks for testing!

jimklimov, 2025-01-14:
@moritzkraemer : hello again, may I ask you to also check if bringing in the newer upstream code per https://github.com/jimklimov/libmodbus/tree/bump-libmodbus-3.1.11 also still works for our RTU USB devices?

````
:; git clone https://github.com/jimklimov/libmodbus -b bump-libmodbus-3.1.11 libmodbus-rtu_usb-3.1.11
:; cd libmodbus-rtu_usb-3.1.11
:; ./autogen.sh
:; ./configure --with-libusb --prefix=/path/to/prefix --enable-static --disable-shared
:; make -j 4 && make install
````

moritzkraemer, 2025-01-14:
Looks good!

jimklimov, 2025-01-14:
@moritzkraemer : meaning, also passed and collected the info and did not crash? Thanks :)

jimklimov, 2025-01-14:
@moritzkraemer : I believe you've also had a use-case for Modbus TCP with APC in #2757? Does the updated library+driver work for that setup as well?

moritzkraemer, 2025-01-14:
> @moritzkraemer : meaning, also passed and collected the info and did not crash? Thanks :)
Yes, one information thats missing for my usecase is the last time the selftest passed but i guess that information isnt available by the ups?

> @moritzkraemer : I believe you've also had a use-case for Modbus TCP with APC in #2757? Does the updated library+driver work for that setup as well?
Yes

jimklimov, 2025-01-15:
Super, thanks!

No idea really about the missing data, maybe someone with protocol info could develop that.

Did you check if `apcupsd` can get that info bit?

moritzkraemer, 2025-01-16:
apcupsd also doesnt get that info so i guess its not available by the ups

karelkryda, 2025-01-16:
Hello @jimklimov,
thanks to your recent changes in `libmodbus` I tried to add USB Modbus support for my **APC SRT3000XLi** UPS.

---
**System:** RPi 4 Model B (4GB RAM)
**OS:** Debian 12 (not RaspiOS) - installed using [Debian raspi builds](https://raspi.debian.net/tested-images/)
**Arch:** arm64

---
I cloned the latest code from the `rtu_usb` branch for **libmodbus** and the latest `master` branch for **NUT** and compiled the code with the following commands:
```bash
# libmodbus
cd libmodbus
./autogen.sh
./configure --with-libusb --enable-static --disable-shared --prefix=/tmp
make install
```
and
```bash
# nut
cd nut
./autogen.sh
export DEB_HOST_MULTIARCH=aarch64-linux-gnu
./configure --prefix=/usr --sysconfdir=/etc/nut --includedir=/usr/include --mandir=/usr/share/man --libdir=\/usr/lib/${DEB_HOST_MULTIARCH} --libexecdir=/usr/libexec --with-ssl --with-nss --with-cgi --with-dev --enable-static --with-statepath=/run/nut --with-altpidpath=/run/nut --with-drvpath=/usr/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut --with-htmlpath=/usr/share/nut/www --with-pidpath=/run/nut --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib/${DEB_HOST_MULTIARCH}/pkgconfig --with-user=nut --with-group=nut --with-udev-dir=/usr/lib/udev --with-systemdsystemunitdir=/usr/lib/systemd/system --with-systemdshutdowndir=/usr/lib/systemd/system-shutdown --with-systemdtmpfilesdir=/usr/lib/tmpfiles.d --with-python=python3 --with-python3=/usr/bin/python3 --with-doc=man --with-modbus+usb --with-modbus-includes=-I/tmp/include/modbus --with-modbus-libs="-L/tmp/lib -lmodbus"
make
```
I then tried to run the `apc_modbus` driver itself from the drivers folder using:
```bash
./drivers/apc_modbus -DDDDDDD -s ups -x user=root -x port=auto
```
and it seemed to load the data - the console always jumped once in a while and dumped a few lines of values from UPS.
I assumed it was fine like this and as recommended in the manual, I installed the nut from the Debian distribution (**2.8.0**) and then installed the drivers from the `drivers` folder using:
```bash
make install
```
Everything looked promising, the message `Communications with UPS homelab@localhost established` was displayed and I was able to use `upsc homelab@localhost` to get a lot of information from UPS that was never available using `usbhid`.
However, after a while I received the message `Communications with UPS homelab@localhost lost` and the journal is full of `nut-monitor[712]: poll UPS [homelab@localhost] failed - Data stale` messages, also upsc reports `Init SSL without certificate database
Error: data stale`.
I tried disconnecting the UPS from the USB and plugging it into a different port on the RPi (3.0 -> 2.0). At that point everything came back to life and the data started returning - I got the following output in the journal:
```bash
led 16 17:51:46 caleb kernel: usb 1-1.4: new full-speed USB device number 4 using xhci_hcd
led 16 17:51:46 caleb kernel: usb 1-1.4: New USB device found, idVendor=051d, idProduct=0003, bcdDevice= 0.01
led 16 17:51:46 caleb kernel: usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
led 16 17:51:46 caleb kernel: usb 1-1.4: Product: Smart-UPS SRT     FW:UPS 15.6 / ID=1025
led 16 17:51:46 caleb kernel: usb 1-1.4: Manufacturer: American Power Conversion
led 16 17:51:46 caleb kernel: usb 1-1.4: SerialNumber: <serial_num>
led 16 17:51:46 caleb kernel: hid-generic 0003:051D:0003.0002: hiddev0,hidraw0: USB HID v1.11 Device [American Power Conversion  Smart-UPS SRT     FW:UPS 15.6 / ID=1025] on usb-0000:01:00.0-1.4/input0
led 16 17:51:48 caleb kernel: usb 1-1.4: usbfs: process 803 (apc_modbus) did not claim interface 0 before use
led 16 17:51:48 caleb nut-server[709]: UPS [homelab] data is no longer stale
led 16 17:51:48 caleb upsd[709]: UPS [homelab] data is no longer stale
led 16 17:51:50 caleb nut-monitor[712]: Communications with UPS homelab@localhost established
led 16 17:51:50 caleb upssched[826]: Timer daemon started
led 16 17:51:50 caleb upssched[826]: Cancel commbad, event: commok
led 16 17:51:50 caleb upssched-cmd[829]: Communication with the UPS has been restored
led 16 17:51:50 caleb upssched[826]: exec_cmd(/etc/nut/upssched-cmd commok) returned 127
led 16 17:51:50 caleb nut-monitor[821]: Network UPS Tools upsmon 2.8.0
led 16 17:51:50 caleb upssched[826]: Timer queue empty, exiting
```
but after about 5 minutes the situation repeated again and since then the journal is again full of `Poll UPS [homelab@localhost] failed - Data stale` messages, see log:
```bash
led 16 17:56:31 caleb nut-server[709]: Data for UPS [homelab] is stale - check driver
led 16 17:56:31 caleb upsd[709]: Data for UPS [homelab] is stale - check driver
led 16 17:56:35 caleb nut-monitor[712]: Poll UPS [homelab@localhost] failed - Data stale
led 16 17:56:35 caleb nut-monitor[712]: Communications with UPS homelab@localhost lost
led 16 17:56:35 caleb upssched[1188]: Timer daemon started
led 16 17:56:35 caleb upssched[1188]: New timer: commbad (30 seconds)
led 16 17:56:35 caleb nut-monitor[1183]: Network UPS Tools upsmon 2.8.0
led 16 17:56:40 caleb nut-monitor[712]: Poll UPS [homelab@localhost] failed - Data stale
led 16 17:56:45 caleb nut-monitor[712]: Poll UPS [homelab@localhost] failed - Data stale
led 16 17:56:50 caleb nut-monitor[712]: Poll UPS [homelab@localhost] failed - Data stale
[... 1 lines cut ...]
led 16 17:57:00 caleb nut-monitor[712]: Poll UPS [homelab@localhost] failed - Data stale
led 16 17:57:05 caleb upssched[1188]: Event: commbad
[... cut ...]
led 16 18:42:50 caleb apc_modbus[1706]: upsnotify: failed to notify about state NOTIFY_STATE_READY_WITH_PID: no notification tech defined, will not spam more about it
led 16 18:42:50 caleb apc_modbus[1706]: upsnotify: logged the systemd watchdog situation once, will not spam more about it
led 16 18:42:50 caleb systemd[1]: Started nut-driver@homelab.service - Network UPS Tools - device driver for homelab.
led 16 18:42:50 caleb apc_modbus[1706]: sock_connect: enabling asynchronous mode (auto)
```

I would appreciate any advice, thank you in advance

jimklimov, 2025-01-16:
@karelkryda : I suppose your packaging overrode something in the custom-built NUT, possibly the statepath - I'm not sure if the packages follow NUT default with `/var/state/ups` or not, but your build uses `/run/nut`. Ultimately it may be unable (question of permissions) to write the socket files for `upsd` to see, or places them in a different place from where `upsd` looks for them. That's one first guess. Similarly, drvpath of `(/usr)/lib/nut` seems different. so the systemd units may be looking in the wrong place.

If you got the custom build working, you just probably don't need the packaged one at all, and avoid confusion.

Out of necessity, it is somewhat common to make a custom build to overlay the packaged one (with same user accounts, config locations, etc.) so the system would run your new build that is mostly compatible to the old one in terms of management. But running a mix is more error-prone (probably possible, may be fragile in terms of careful coordination).

I suggest to remove the package, verify that user/group accounts remain defined (or re-create them), and `sudo make install-as-root` from the source area (if using current master).

karelkryda, 2025-01-16:
Interesting, yesterday I tried unsuccessfully to perform an in-place upgrade following the instructions [here](https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests#replacing-a-nut-deployment). I used version **2.8.1** which I managed to get and install from the Debian mirror, despite the fact that only version **2.8.0** is officially available for **bookworm**.
I have now tried the procedure described [here](https://github.com/networkupstools/nut/wiki/Building-NUT-on-Debian,-Raspbian-and-Ubuntu), which recommends just installing the system version and then installing only the drivers.
As for the parameters, I based them on the Debian repo [here](https://salsa.debian.org/debian/nut/-/blob/debian/debian/rules).

However, I should point out that once I posted my previous comment, the UPS has not disconnected from the NUT since then, and even after restarting the RPi everything booted as expected.
Except the fact that the log in the journal returned the following errors:
```bash
led 16 19:58:30 caleb systemd[1]: Reached target nut.target - Network UPS Tools - target for power device drivers, data server and monitoring client (if enabled) on this system.
led 16 19:58:30 caleb nut-server[704]: fopen /run/nut/upsd.pid: No such file or directory
led 16 19:58:30 caleb nut-server[704]: Could not find PID file '/run/nut/upsd.pid' to see if previous upsd instance is already running!
led 16 19:58:30 caleb nut-monitor[705]: fopen /run/nut/upsmon.pid: No such file or directory
led 16 19:58:30 caleb nut-monitor[705]: Could not find PID file to see if previous upsmon instance is already running!
led 16 19:58:30 caleb nut-server[704]: listening on 0.0.0.0 port 3493
```

Now I don't know whether to leave things as they are, as they seem to be stable for now, or to do the procedure you recommended.
What would you advise me?

_Note: during the day when I was doing various testing, I often got stuck with the driver service returning information that there is a PID file and therefore another modbus instance and an attempt to terminate it is in progress. However, the service was never able to recover and just kept restarting with this output. Even manually deleting the .pid file didn't help._

jimklimov, 2025-01-16:
Well, if it works - leave it be I suppose :)

Conflicts about older instance - were they when you used `upsdrvctl` as in some older instructions? In that case with NUT v2.8.x you could be conflicting with a service wrapper for the driver, see https://github.com/networkupstools/nut/wiki/nut%E2%80%90driver%E2%80%90enumerator-(NDE)

The lack of PID files is not an error if the older daemon (driver, upsd, upsmon...) instance is not running and should not be. It is also not an error if the daemon is only managed under systemd/SMF and nobody outside needs to know its PID. Newer 2.8.x service definitions (and daemon implementations) gradually ran the daemons with `-F` to not-fork needlessly under systemd, and later with `-FF` to remain foregrounded and still write the PID file for older management techniques to see. If you have some mix of 2.8.1 and current master, in terms of both daemons' abilities and systemd unit definitions, it is an unpredictable mixed bag of results really :)

karelkryda, 2025-01-16:
I understand it's not ideal. Honestly, I'm glad that after 2 days it started to work - albeit rather miraculously. I'm honestly afraid to interfere with it now 😄.  Maybe something will go wrong again and I'll have to go back to it.

Anyway, question a bit off topic, are there any plans to use the NUT-forked `libmodbus` repo as source instead of the official one, thus allowing `apc_modbus` driver distribution directly with NUT versions? _- and hopefully shipped directly by Linux distributions_

jimklimov, 2025-01-16:
It depends on what the future holds. Hopefully we've got some gears moving in upstream project (CLA signed before posting commits, etc.) so there's a chance the changes made for RTU USB will become part of upstream.

Alternately, or for older systems that don't have current/any libmodbus packaged, I did think about providing a generic way to pull and build dependencies. Wouldn't be the first project doing so; in fact in the days of tarballs before packaging everyone had `contrib` directories with "liked" copies of deps sources. We may well be the last major project introducing something of the sort :)

Got some head start tracking an open issue about it, documenting prereq builds for Windows, and scripted a bit of PoC builder there. The sister 42ity/FTY project had a whole ecosystem built by my scripts from dozens of repos and/or tarballs, similar to how OS userlands do it. So it is certainly a possibility. Maybe not inside NUT code base nowadays, but as a nearby repo for such integration from scratch where NUT is just one of components.

karelkryda, 2025-01-18:
Hi, @jimklimov,
unfortunately, I'm coming back to the problem. I started getting `Data stale` messages again this morning. Restarting the RPi and modifying configuration did not solve the problem and I always get `Data stale` again after a few minutes.
I've gone through all the NUT service logs including the journal and nothing anywhere to indicate a problem.
Here is the journal log:
```bash
# ***DHCP REQUEST***
led 18 14:31:48 caleb systemd[1]: Reloaded postfix.service - Postfix Mail Transport Agent.
led 18 14:31:48 caleb systemd[1]: postfix-resolvconf.service: Deactivated successfully.
led 18 14:37:08 caleb nut-server[752]: Data for UPS [homelab] is stale - check driver
led 18 14:37:08 caleb upsd[752]: Data for UPS [homelab] is stale - check driver
led 18 14:37:11 caleb nut-monitor[758]: Poll UPS [homelab@localhost] failed - Data stale
led 18 14:37:11 caleb nut-monitor[758]: Communications with UPS homelab@localhost lost
led 18 14:37:11 caleb upssched[2973]: Timer daemon started
led 18 14:37:11 caleb upssched[2973]: New timer: commbad (30 seconds)
led 18 14:37:11 caleb nut-monitor[2968]: Network UPS Tools upsmon 2.8.0
led 18 14:37:16 caleb nut-monitor[758]: Poll UPS [homelab@localhost] failed - Data stale
...
```

I tried, according to the research, to reduce the `pollinterval` from 1s to 5s and then 10s, but it seemed to me that it made the problem worse. The loss and establishment of the connection changed every few seconds.

_I also tried updating the RPi EEPROM to the latest one, but that didn't help either._

---
I decided to uninstall the NUT package from APT and install it according to your recommendations, but before that I called the modbus driver directly from the build folder and it seems to me that the intervals between each data poll are too high - according to the comments above the data arrives within seconds, while my delays are easily around 30 seconds.
So I decided to attach the driver log output and hope that together we can identify the problem.
<details>
<summary>Driver log</summary>

```
root@caleb:~/nut_build/nut# ./drivers/apc_modbus -DDDDDDD -s ups -x user=root -x port=auto
   0.000000     [D5] send_to_all: SETINFO driver.state "init.starting"
   0.000365     [D5] send_to_all: SETINFO driver.version.usb "libusb-1.0.26 (API: 0x01000109)"
   0.000395     [D1] Using USB implementation: libusb-1.0.26 (API: 0x01000109)
   0.000420     [D1] This build of the driver is USB-capable; also Serial and TCP Modbus RTU are supported
   0.000451     [D3] main_arg: var='user' val='root'
   0.000471     [D6] testval_reloadable: var=user, oldval=nut, newval=root, reloadable=0, reload_flag=0
   0.000494     [D1] testval_reloadable: setting 'user' exists and differs: new value 'root' vs. 'nut'
   0.000511     [D6] testval_reloadable: verdict for (re)loading var=user value: 1
   0.000528     [D1] Overriding previously specified user 'nut' with 'root' specified for driver section
   0.000544     [D3] main_arg: var='port' val='auto'
   0.000560     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000575     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000594     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000615     [D1] Network UPS Tools version 2.8.2.1898-1898-gaeeed24fc (development iteration after 2.8.2) built with gcc (Debian 12.2.0-14) 12.2.0 and configured with flags: --prefix=/usr --sysconfdir=/etc/nut --includedir=/usr/include --mandir=/usr/share/man --libdir=/usr/lib/aarch64-linux-gnu --libexecdir=/usr/libexec --with-ssl --with-nss --with-cgi --with-dev --enable-static --with-statepath=/run/nut --with-altpidpath=/run/nut --with-drvpath=/usr/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut --with-htmlpath=/usr/share/nut/www --with-pidpath=/run/nut --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib/aarch64-linux-gnu/pkgconfig --with-user=nut --with-group=nut --with-udev-dir=/usr/lib/udev --with-systemdsystemunitdir=/usr/lib/systemd/system --with-systemdshutdowndir=/usr/lib/systemd/system-shutdown --with-systemdtmpfilesdir=/usr/lib/tmpfiles.d --with-python=python3 --with-python3=/usr/bin/python3 --with-doc=man --with-modbus+usb --with-modbus-includes=-I/tmp/include/modbus --with-modbus-libs='-L/tmp/lib -lmodbus'
   0.000680     [D1] debug level is '7'
   0.000702     [D5] send_to_all: SETINFO driver.debug "7"
   0.000721     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.001373     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.001428     [D1] Signalling UPS [ups]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.001471     Can't open /run/nut/apc_modbus-ups: No such file or directory
   0.001489     [D1] Request for other driver to exit returned code -1
   0.001507     [D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
   0.001530     [D5] send_to_all: SETINFO device.type "ups"
   0.001551     [D5] send_to_all: SETINFO driver.state "init.device"
   0.026501     [D2] _apc_modbus_usb_callback: Matched American Power Conversion  Smart-UPS SRT     FW:UPS 15.6 / ID=1025 (USB VID/PID 051d:0003)
   0.026544     [D2] _apc_modbus_usb_callback: Checking American Power Conversion  Smart-UPS SRT     FW:UPS 15.6 / ID=1025 (USB VID/PID 051d:0003) report descriptors
   0.026750     [D1] _apc_modbus_usb_callback: Found report ids RX=0x90 TX=0x89
[... cut ...]
^C 170.612186   [D1] set_exit_flag: raising exit flag due to signal 2
 192.022346     [D5] send_to_all: SETINFO driver.state "quiet"
 192.022432     Signal 2: exiting
 192.022487     [D5] send_to_all: SETINFO driver.state "cleanup.upsdrv"
 192.022980     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
``` 
</details> 

---
_If it could affect this, I would also be glad if you could advise me what paths should be set correctly in the NUT parameters, given your comment above._

Thanks in advance

karelkryda, 2025-01-18:
I installed the NUT from repo using `make install-as-root`, commented out the `pollinterval` and `maxretry` options in `ups.conf` with no success and connection loss/reconnection every few seconds.

I'm attaching the output from debug level *7* for `apc_modbus` and `upsd`:
<details>
<summary>Log</summary>

```
led 18 17:43:15 caleb upsd[2251]: Data for UPS [homelab] is stale - check driver
led 18 17:43:23 caleb apc_modbus[1246]: [D5] send_to_all: SETINFO driver.state "quiet"
led 18 17:43:23 caleb apc_modbus[1246]: [D6] send_to_all: write 29 bytes to socket 5 succeeded (ret=29): SETINFO driver.state "quiet"
led 18 17:43:23 caleb upsd[2251]: UPS [homelab] data is no longer stale
led 18 17:43:23 caleb apc_modbus[1246]: [D6] sock_arg: Driver on /run/nut/apc_modbus-homelab is now handling PING with 1 args
led 18 17:43:23 caleb apc_modbus[1246]: [D2] send_to_one: sending PONG
led 18 17:43:23 caleb apc_modbus[1246]: [D5] send_to_one: PONG
led 18 17:43:23 caleb apc_modbus[1246]: [D6] send_to_one: write 5 bytes to socket 5 succeeded (ret=5): PONG
led 18 17:43:23 caleb apc_modbus[1246]: [D6] sock_arg: Driver on /run/nut/apc_modbus-homelab is now handling PING with 1 args
led 18 17:43:23 caleb apc_modbus[1246]: [D2] send_to_one: sending PONG
led 18 17:43:23 caleb apc_modbus[1246]: [D5] send_to_one: PONG
led 18 17:43:23 caleb apc_modbus[1246]: [D6] send_to_one: write 5 bytes to socket 5 succeeded (ret=5): PONG
led 18 17:43:23 caleb apc_modbus[1246]: [D6] sock_arg: Driver on /run/nut/apc_modbus-homelab is now handling PING with 1 args
led 18 17:43:23 caleb apc_modbus[1246]: [D2] send_to_one: sending PONG
led 18 17:43:23 caleb apc_modbus[1246]: [D5] send_to_one: PONG
led 18 17:43:23 caleb apc_modbus[1246]: [D6] send_to_one: write 5 bytes to socket 5 succeeded (ret=5): PONG
led 18 17:43:23 caleb apc_modbus[1246]: [D6] sock_arg: Driver on /run/nut/apc_modbus-homelab is now handling PING with 1 args
led 18 17:43:23 caleb apc_modbus[1246]: [D2] send_to_one: sending PONG
led 18 17:43:23 caleb apc_modbus[1246]: [D5] send_to_one: PONG
led 18 17:43:23 caleb apc_modbus[1246]: [D6] send_to_one: write 5 bytes to socket 5 succeeded (ret=5): PONG
led 18 17:43:23 caleb apc_modbus[1246]: [D5] send_to_all: SETINFO driver.state "updateinfo"
led 18 17:43:23 caleb apc_modbus[1246]: [D6] send_to_all: write 34 bytes to socket 5 succeeded (ret=34): SETINFO driver.state "updateinfo"
led 18 17:43:39 caleb upsd[2251]: Data for UPS [homelab] is stale - check driver
led 18 17:43:50 caleb apc_modbus[1246]: [D5] send_to_all: SETINFO battery.runtime "3439"
led 18 17:43:50 caleb apc_modbus[1246]: [D6] send_to_all: write 31 bytes to socket 5 succeeded (ret=31): SETINFO battery.runtime "3439"
led 18 17:43:50 caleb apc_modbus[1246]: [D5] send_to_all: SETINFO battery.temperature "22.00"
led 18 17:43:50 caleb upsd[2251]: UPS [homelab] data is no longer stale
led 18 17:43:50 caleb apc_modbus[1246]: [D6] send_to_all: write 36 bytes to socket 5 succeeded (ret=36): SETINFO battery.temperature "22.00"
led 18 17:43:50 caleb apc_modbus[1246]: [D5] send_to_all: SETINFO ups.load "12.77"
led 18 17:43:50 caleb apc_modbus[1246]: [D6] send_to_all: write 25 bytes to socket 5 succeeded (ret=25): SETINFO ups.load "12.77"
led 18 17:43:50 caleb apc_modbus[1246]: [D5] send_to_all: SETINFO ups.realpower "344.88"
led 18 17:43:50 caleb apc_modbus[1246]: [D6] send_to_all: write 31 bytes to socket 5 succeeded (ret=31): SETINFO ups.realpower "344.88"
led 18 17:43:50 caleb apc_modbus[1246]: [D5] send_to_all: SETINFO ups.power "403.48"
led 18 17:43:50 caleb apc_modbus[1246]: [D6] send_to_all: write 27 bytes to socket 5 succeeded (ret=27): SETINFO ups.power "403.48"
led 18 17:43:50 caleb apc_modbus[1246]: [D5] send_to_all: SETINFO output.current "1.66"
led 18 17:43:50 caleb apc_modbus[1246]: [D6] send_to_all: write 30 bytes to socket 5 succeeded (ret=30): SETINFO output.current "1.66"
led 18 17:43:50 caleb apc_modbus[1246]: [D5] send_to_all: SETINFO output.voltage "235.86"
led 18 17:43:50 caleb apc_modbus[1246]: [D6] send_to_all: write 32 bytes to socket 5 succeeded (ret=32): SETINFO output.voltage "235.86"
led 18 17:43:50 caleb apc_modbus[1246]: [D5] send_to_all: SETINFO experimental.output.energy "2996599"
led 18 17:43:50 caleb apc_modbus[1246]: [D6] send_to_all: write 45 bytes to socket 5 succeeded (ret=45): SETINFO experimental.output.energy "2996599"
led 18 17:43:50 caleb apc_modbus[1246]: [D5] send_to_all: SETINFO input.voltage "235.86"
led 18 17:43:50 caleb apc_modbus[1246]: [D6] send_to_all: write 31 bytes to socket 5 succeeded (ret=31): SETINFO input.voltage "235.86"
led 18 17:43:55 caleb apc_modbus[1246]: [D5] send_to_all: SETINFO driver.state "quiet"
led 18 17:43:55 caleb apc_modbus[1246]: [D6] send_to_all: write 29 bytes to socket 5 succeeded (ret=29): SETINFO driver.state "quiet"
led 18 17:43:55 caleb apc_modbus[1246]: [D6] sock_arg: Driver on /run/nut/apc_modbus-homelab is now handling PING with 1 args
led 18 17:43:55 caleb apc_modbus[1246]: [D2] send_to_one: sending PONG
led 18 17:43:55 caleb apc_modbus[1246]: [D5] send_to_one: PONG
led 18 17:43:55 caleb apc_modbus[1246]: [D6] send_to_one: write 5 bytes to socket 5 succeeded (ret=5): PONG
led 18 17:43:55 caleb apc_modbus[1246]: [D6] sock_arg: Driver on /run/nut/apc_modbus-homelab is now handling PING with 1 args
led 18 17:43:55 caleb apc_modbus[1246]: [D2] send_to_one: sending PONG
led 18 17:43:55 caleb apc_modbus[1246]: [D5] send_to_one: PONG
[... cut ...]
I am attaching my `ups.conf` file as well:
```bash
#pollinterval = 1
#maxretry = 3
debug_min = 7
LIBUSB_DEBUG = 7

[homelab]
#  driver = "usbhid-ups"
  driver = "apc_modbus"
  port = "auto"
  desc = "APC Smart-UPS SRT 3 000 VA"
  vendorid = "051D"
  productid = "0003"
  product = "Smart-UPS SRT     FW:UPS 15.6 / ID=1025"
  serial = "ASxxxx"
  vendor = "American Power Conversion"
  bus = "001"
  device = "003"
  busport = "004"
#  default.input.voltage.nominal = 230
  override.battery.charge.low = 20
```

jimklimov, 2025-01-18:
I am not sure why the latest logs lack a message similar to `Got a list of 3 libusb file descriptors to poll` - did you rebuild libmodbus and something changed?

Regarding the delay - I suppose that in your system `select()` did some default work based on the provided "fdset", maybe it defaults to some system-defined streams, not sure. At least, it is not polling a file descriptor for libusb (because it did not find exactly one, and current library code is geared towards checking one).

I also feel a bit stupid, as I overlooked that the method involved in `_modbus_receive_msg()` is not a standard `select()`, but a specific `ctx->backend->select()` so the one here `_modbus_rtu_usb_select()` does not care about `rset` and talks to the libusb context. The fixes in that PR to avoid `FD_SET(-1)` are valid (to avoid illegal memory access), but are not the full answer here as I hoped it could be.

Either way, it seems that `select()` in libmodbus sleeps for up to 0.5 seconds (default `_RESPONSE_TIMEOUT` or `_BYTE_TIMEOUT`) waiting for a message from the device, in a reading loop in case of USB RTU.

Inter-loop delays done in NUT common `drivers/main.c` file are dictated by `pollinterval` (default 5 sec) and may soak up the time it took to communicate with a device, in a driver-specific implementation.

Is there anything in `dmesg` around the time your connections may have got broken?

jimklimov, 2025-01-18:
@moritzkraemer : hello, I was working on libmodbus this weekend, to make it buildable and testable on various platforms of the NUT CI farm. Some changes could impact the non-test code base...
* https://github.com/networkupstools/libmodbus/pull/3

Now I've got a new branch to test, may I ask for your help again (USB. TCP; serial if possible):
````
:; git clone https://github.com/jimklimov/libmodbus -b fix-mingw-build libmodbus-fix-mingw-build
:; cd libmodbus-fix-mingw-build 
...
````

karelkryda, 2025-01-19:
Hi @jimklimov,
during the night I had NUT running with the `usbhid-ups` driver set. I have now stopped all nut* services and recompiled the driver with the latest code for libmodbus (rtu_usb) and NUT and attached the output from the `apc_modbus` driver.

<details>
<summary>Driver log</summary>

```
root@caleb:~/nut_build/nut# ./drivers/apc_modbus -DDDDDDD -s ups -x user=root -x port=auto
   0.000001     [D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools 2.8.2.1915-1915-g8f95466f5 (development iteration after 2.8.2) - NUT APC Modbus driver with USB support 0.11
   0.000241     [D5] send_to_all: SETINFO driver.version.usb "libusb-1.0.26 (API: 0x01000109)"
   0.000264     [D1] Using USB implementation: libusb-1.0.26 (API: 0x01000109)
   0.000289     [D1] This build of the driver is USB-capable; also Serial and TCP Modbus RTU are supported
   0.000321     [D3] main_arg: var='user' val='root'
   0.000340     [D6] testval_reloadable: var=user, oldval=nut, newval=root, reloadable=0, reload_flag=0
   0.000363     [D1] testval_reloadable: setting 'user' exists and differs: new value 'root' vs. 'nut'
   0.000380     [D6] testval_reloadable: verdict for (re)loading var=user value: 1
   0.000396     [D1] Overriding previously specified user 'nut' with 'root' specified for driver section
   0.000414     [D3] main_arg: var='port' val='auto'
   0.000430     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000445     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000464     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000477     [D1] Network UPS Tools version 2.8.2.1915-1915-g8f95466f5 (development iteration after 2.8.2) built with gcc (Debian 12.2.0-14) 12.2.0 and configured with flags: --prefix=/usr --sysconfdir=/etc/nut --includedir=/usr/include --mandir=/usr/share/man --libdir=/usr/lib/aarch64-linux-gnu --libexecdir=/usr/libexec --with-ssl --with-nss --with-cgi --with-dev --enable-static --with-statepath=/run/nut --with-altpidpath=/run/nut --with-drvpath=/usr/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut --with-htmlpath=/usr/share/nut/www --with-pidpath=/run/nut --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib/aarch64-linux-gnu/pkgconfig --with-user=nut --with-group=nut --with-udev-dir=/usr/lib/udev --with-systemdsystemunitdir=/usr/lib/systemd/system --with-systemdshutdowndir=/usr/lib/systemd/system-shutdown --with-systemdtmpfilesdir=/usr/lib/tmpfiles.d --with-python=python3 --with-python3=/usr/bin/python3 --with-doc=man --with-modbus+usb --with-modbus-includes=-I/tmp/include/modbus --with-modbus-libs='-L/tmp/lib -lmodbus'
   0.000511     [D1] debug level is '7'
   0.000529     [D5] send_to_all: SETINFO driver.debug "7"
   0.000547     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.001192     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.001251     [D1] Signalling UPS [ups]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.001294     Can't open /run/nut/apc_modbus-ups: No such file or directory
   0.001314     [D1] Request for other driver to exit returned code -1
   0.001331     [D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
   0.001355     [D5] send_to_all: SETINFO device.type "ups"
   0.001374     [D5] send_to_all: SETINFO driver.state "init.device"
Number of USB devices: 4
Considering device #1 (051d:0003)
   0.027979     [D2] _apc_modbus_usb_callback: Matched American Power Conversion  Smart-UPS SRT     FW:UPS 15.6 / ID=1025 (USB VID/PID 051d:0003)
   0.028038     [D2] _apc_modbus_usb_callback: Checking American Power Conversion  Smart-UPS SRT     FW:UPS 15.6 / ID=1025 (USB VID/PID 051d:0003) report descriptors
   0.028220     [D1] _apc_modbus_usb_callback: Found report ids RX=0x90 TX=0x89
   0.028247     [D5] send_to_all: SETINFO ups.vendorid "051d"
   0.028264     [D5] send_to_all: SETINFO ups.productid "0003"
Found Device 1 (Path 1-1.4):
  Vendor ID: 0x051d
  Product ID: 0x0003
Got a list of 3 libusb file descriptors to poll
Bytes flushed (0)
   0.038545     [D5] send_to_all: SETINFO driver.state "init.quiet"
   0.038586     [D5] send_to_all: SETINFO driver.version "2.8.2.1915-1915-g8f95466f5"
   0.038607     [D5] send_to_all: SETINFO driver.version.internal "0.11"
   0.038712     [D5] send_to_all: SETINFO driver.name "apc_modbus"
   0.038741     [D5] send_to_all: SETINFO driver.state "init.info"
[01][03][02][04][00][78]
Waiting for a confirmation...
Using a backend without a file descriptor, will not select() on it.
<01><03><F0><00><00><00><00><00><00><00><00><55><50><53><20><31><35><2E><36><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><53><6D><61><72><74><2D><55><50><53><20><53><52><54><20><20><20><20><33><30><30><30><20><20><20><20><20><20><20><20><20><20><20><53><52><54><33><30><30><30><58><4C><49><20><20><20><20><20><20><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><41><53><31><37><34><33><32><39><31><35><35><31><20><20><20><20><41><50><43><52><42><43><31><35><32><20><20><20><20><20><20><20><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><0B><B8><0A><8C><00><06><19><6C><00><20><00><01><00><00><19><6C><41><50><43><20><55><50><53><20><20><20><20><20><20><20><20><20><55><6E><73><77><69><74><63><68><65><64><20><47><72><6F><75><70><48><6C><61><76><6E><69><20><6E><61><70><61><6A><65><6E><69><20><5A><61><6C><6F><7A><6E><69><20 [... cut ...]
led 18 17:31:13 caleb nut-monitor[1048]: Communications with UPS homelab@localhost lost
led 18 17:31:18 caleb nut-monitor[1048]: Poll UPS [homelab@localhost] failed - Data stale
led 18 17:31:20 caleb nut-server[1005]: UPS [homelab] data is no longer stale
led 18 17:31:20 caleb upsd[1005]: UPS [homelab] data is no longer stale
led 18 17:31:23 caleb nut-monitor[1048]: Communications with UPS homelab@localhost established
led 18 17:31:42 caleb nut-server[1005]: Data for UPS [homelab] is stale - check driver
.....
```
</details>

moritzkraemer, 2025-01-21:
When i run the driver directly it gives me this Output so it looks like it is still getting Data from the UPS

<details>
<summary>Output</summary>

```
   0.000008     [D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools 2.8.2.1929-1929-g9066236ec (development iteration after 2.8.2) - NUT APC Modbus driver with USB support 0.11
   0.000117     [D5] send_to_all: SETINFO driver.version.usb "libusb-1.0.27 (API: 0x0100010A)"
   0.000133     [D1] Using USB implementation: libusb-1.0.27 (API: 0x0100010A)
   0.000168     [D1] This build of the driver is USB-capable; also Serial and TCP Modbus RTU are supported
   0.000195     [D3] main_arg: var='port' val='auto'
   0.000208     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000278     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000320     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000332     [D1] Network UPS Tools version 2.8.2.1929-1929-g9066236ec (development iteration after 2.8.2) built with gcc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0 and configured with flags: --with-drivers=apc_modbus --with-usb --with-modbus --with-modbus-includes=-I/usr/local/include/modbus --with-modbus-libs='-L/usr/local/lib -lmodbus'
   0.000360     [D1] debug level is '6'
   0.000375     [D5] send_to_all: SETINFO driver.debug "6"
   0.000390     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.001400     [D1] Succeeded to become_user(nobody): now UID=65534 GID=65534
   0.001436     [D1] Signalling UPS [test]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.001461     Can't open /var/state/ups/apc_modbus-test: No such file or directory
   0.001474     [D1] Request for other driver to exit returned code -1
   0.001487     [D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
   0.001501     [D5] send_to_all: SETINFO device.type "ups"
   0.001516     [D5] send_to_all: SETINFO driver.state "init.device"
Number of USB devices: 4
Considering device #1 (8087:0aaa)
libusb_open for device #1 failed: Access denied (insufficient permissions)
Considering device #2 (051d:0003)
   0.040831     [D2] _apc_modbus_usb_callback: Matched American Power Conversion  Smart-UPS_1500 FW:UPS 18.0 / ID=1027 (USB VID/PID 051d:0003)
   0.040868     [D2] _apc_modbus_usb_callback: Checking American Power Conversion  Smart-UPS_1500 FW:UPS 18.0 / ID=1027 (USB VID/PID 051d:0003) report descriptors
   0.041080     [D1] _apc_modbus_usb_callback: Found report ids RX=0x90 TX=0x89
   0.041102     [D5] send_to_all: SETINFO ups.vendorid "051d"
   0.041118     [D5] send_to_all: SETINFO ups.productid "0003"
Found Device 2 (Path 1-1):
  Vendor ID: 0x051d
  Product ID: 0x0003
Got a list of 3 libusb file descriptors to poll
Bytes flushed (0)
   0.055967     [D5] send_to_all: SETINFO driver.state "init.quiet"
   0.055988     [D5] send_to_all: SETINFO driver.version "2.8.2.1929-1929-g9066236ec"
   0.056000     [D5] send_to_all: SETINFO driver.version.internal "0.11"
   0.056037     [D5] send_to_all: SETINFO driver.name "apc_modbus"
   0.056049     [D5] send_to_all: SETINFO driver.state "init.info"
[01][03][02][04][00][78]
Waiting for a confirmation...
Using a backend without a file descriptor, will not natively select() on it.
<01><03>
<F0>
<55><50><53><20><31><38><2E><30><00><00><00><00><00><00><00><00><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><53><6D><61><72><74><2D><55><50><53><20><31><35><30><30><20><20><20><20><20><20><20><20><20><20><20><20><20><20><20><20><20><20><53><4D><54><31><35><30><30><52><4D><49><32><55><43><20><20><20><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><41><53><32><33><31><33><33><35><31><35><37><36><20><20><20><20><41><50><43><52><42><43><31><35><39><20><20><20><20><20><20><20><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><05><DC><03><E8><00><03><21><29><00><20><00><00><00><00><21><7A><41><50><43><20><55><50><53><20><20><20><20><20><20><20><20><20><55><6E><73><77><69><74><63><68><65><64><20><47><72><6F><75><70><4F><75><74><6C><65><74><20><47><72><6F><75><70><20><31><20><20><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00><00>

   0.158085     [D5] send_to_all: SETINFO outlet.group.count "2"
   0.158133     [D5] send_to_all: SETINFO ups.firmware "UPS 18.0"
   0.158171     [D5] send_to_all: SETFLAGS ups.firmware STRING
   0.158208     [D5] send_to_all: SETAUX ups.firmware 16
   0.158243     [D5] send_to_all: SETINFO ups.model "Smart-UPS 1500"
   0.158263     [D5] send_to_all: SETFLAGS ups.model STRING
   0.158282     [D5] send_to_all: SETAUX ups.model 32
   0.158303     [D5] send_to_all: SETINFO ups.serial "ASxxxx"
   0.158320     [D5] send_to_all: SETFLAGS ups.serial STRING
   0.158337     [D5] send_to_all: SETAUX ups.serial 16
   0.158364     [D5] send_to_all: SETINFO ups.power.nominal "1500"
   0.158389     [D5] send_to_all: SETINFO ups.realpower.nominal "1000"
   0.158544     [D5] send_to_all: SETINFO ups.mfr.date "2023-03-30"
   0.158586     [D5] send_to_all: SETINFO battery.date "2023-06-19"
   0.158616     [D5] send_to_all: SETFLAGS battery.date RW
   0.158655     [D5] send_to_all: SETINFO ups.id "APC UPS"
   0.158690     [D5] send_to_all: SETFLAGS ups.id RW STRING
   0.158725     [D5] send_to_all: SETAUX ups.id 16
[... 696 lines cut ...]
[... cut ...]
   0.072186     upsnotify: notify about state NOTIFY_STATE_STOPPING with libsystemd: was requested, but not running as a service unit now, will not spam more about it
   0.072210     [D1] On systems without service units, consider `export NUT_QUIET_INIT_UPSNOTIFY=true`
   0.072224     upsnotify: failed to notify about state NOTIFY_STATE_STOPPING: no notification tech defined, will not spam more about it
```
</details>

I'm not sure if the issue was introduced with the new version of the driver or something on my system changed that could cause this issue

jimklimov, 2025-01-21:
@moritzkraemer : thanks for testing! I think I saw messages about unexpected length before, in other reports.

"Reply from 32 when 1 was expected" or some such popped ot too, I think, maybe in unit tests. The thing is flaky :\

My guess is that connection flow (or even reconnection) does some flush on one side and not on the other, or something, so partial messages are seen and confuse the next lifetime. Maybe a re-request should be done to receive a reply whole and anew?..

@EchterAgo: are you online? Able to take a look please?

As for `upsdrvctl`, seems like abother copy of a driver was running. Could systemd intervene with its copy (by NDE)? Or did it fail to exit/disconnect when killed?..

moritzkraemer, 2025-01-21:
Im getting different issues everytime i replug the system:

First Time:
<details>
<summary>Output</summary>

```
   0.000009     [D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools 2.8.2.1929-1929-g9066236ec (development iteration after 2.8.2) - NUT APC Modbus driver with USB support 0.11
   0.000194     [D5] send_to_all: SETINFO driver.version.usb "libusb-1.0.27 (API: 0x0100010A)"
   0.000208     [D1] Using USB implementation: libusb-1.0.27 (API: 0x0100010A)
   0.000222     [D1] This build of the driver is USB-capable; also Serial and TCP Modbus RTU are supported
   0.000249     [D3] main_arg: var='port' val='auto'
   0.000271     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000305     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000324     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000338     [D1] Network UPS Tools version 2.8.2.1929-1929-g9066236ec (development iteration after 2.8.2) built with gcc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0 and configured with flags: --with-drivers=apc_modbus --with-usb --with-modbus --with-modbus-includes=-I/usr/local/include/modbus --with-modbus-libs='-L/usr/local/lib -lmodbus'
   0.000388     [D1] debug level is '6'
   0.000405     [D5] send_to_all: SETINFO driver.debug "6"
   0.000435     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.001595     [D1] Succeeded to become_user(nobody): now UID=65534 GID=65534
   0.001634     [D1] Signalling UPS [text]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.001675     Can't open /var/state/ups/apc_modbus-text: No such file or directory
   0.001690     [D1] Request for other driver to exit returned code -1
   0.001704     [D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
   0.001721     [D5] send_to_all: SETINFO device.type "ups"
   0.001737     [D5] send_to_all: SETINFO driver.state "init.device"
Number of USB devices: 4
Considering device #1 (8087:0aaa)
libusb_open for device #1 failed: Access denied (insufficient permissions)
Considering device #2 (051d:0003)
   0.044656     [D2] _apc_modbus_usb_callback: Matched American Power Conversion  Smart-UPS_1500 FW:UPS 18.0 / ID=1027 (USB VID/PID 051d:0003)
   0.044692     [D2] _apc_modbus_usb_callback: Checking American Power Conversion  Smart-UPS_1500 FW:UPS 18.0 / ID=1027 (USB VID/PID 051d:0003) report descriptors
   0.044903     [D1] _apc_modbus_usb_callback: Found report ids RX=0x90 TX=0x89
   0.044925     [D5] send_to_all: SETINFO ups.vendorid "051d"
   0.044940     [D5] send_to_all: SETINFO ups.productid "0003"
Found Device 2 (Path 1-1):
  Vendor ID: 0x051d
  Product ID: 0x0003
Got a list of 3 libusb file descriptors to poll
Bytes flushed (0)
   0.059873     [D5] send_to_all: SETINFO driver.state "init.quiet"
   0.059895     [D5] send_to_all: SETINFO driver.version "2.8.2.1929-1929-g9066236ec"
   0.059908     [D5] send_to_all: SETINFO driver.version.internal "0.11"
   0.059921     [D5] send_to_all: SETINFO driver.name "apc_modbus"
   0.059930     [D5] send_to_all: SETINFO driver.state "init.info"
[01][03][02][04][00][78]
Waiting for a confirmation...
Using a backend without a file descriptor, will not natively select() on it.
<20><20>
<20>

The responding slave 32 isn't the requested slave 1
   0.105737     _apc_modbus_read_registers: Read of 516:636 failed: Response not from requested slave (auto)
   0.105753     _apc_modbus_handle_error: Closing connection
   0.106160     Can't read inventory information from the UPS
   0.106199     [D5] send_to_all: SETINFO driver.state "cleanup.upsdrv"
   0.106227     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
   0.106255     upsnotify: notify about state NOTIFY_STATE_STOPPING with libsystemd: was requested, but not running as a service unit now, will not spam more about it
   0.106349     [D1] On systems without service units, consider `export NUT_QUIET_INIT_UPSNOTIFY=true`
   0.106380     upsnotify: failed to notify about state NOTIFY_STATE_STOPPING: no notification tech defined, will not spam more about it
```
</details>

Then i unplug and replug the ups and the following issue appears:

<details>
<summary>Output</summary>

```
[... 30 lines cut ...]
  Vendor ID: 0x051d
  Product ID: 0x0003
Got a list of 3 libusb file descriptors to poll
Bytes flushed (0)
   0.059297     [D5] send_to_all: SETINFO driver.state "init.quiet"
   0.059319     [D5] send_to_all: SETINFO driver.version "2.8.2.1929-1929-g9066236ec"
   0.059331     [D5] send_to_all: SETINFO driver.version.internal "0.11"
   0.059343     [D5] send_to_all: SETINFO driver.name "apc_modbus"
   0.059352     [D5] send_to_all: SETINFO driver.state "init.info"
[01][03][02][04][00][78]
Waiting for a confirmation...
Using a backend without a file descriptor, will not natively select() on it.
<01><03>
<F0>
ERROR Connection timed out: ctx->backend->select
   0.653937     _apc_modbus_read_registers: Read of 516:636 failed: Connection timed out (auto)
Bytes flushed (63)
   1.667660     Can't read inventory information from the UPS
   1.667696     [D5] send_to_all: SETINFO driver.state "cleanup.upsdrv"
   1.668182     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
   1.668220     upsnotify: notify about state NOTIFY_STATE_STOPPING with libsystemd: was requested, but not running as a service unit now, will not spam more about it
   1.668268     [D1] On systems without service units, consider `export NUT_QUIET_INIT_UPSNOTIFY=true`
   1.668298     upsnotify: failed to notify about state NOTIFY_STATE_STOPPING: no notification tech defined, will not spam more about it
```
</details>

karelkryda, 2025-01-21:
It looks like we ran into the same problem.

I'm just letting you know that I'll be happy to help test any changes, if required 🙂.

somejavadev, 2025-01-22:
@jimklimov @EchterAgo not sure if this might help you or someone else looking into this. I can consistently replicate this and I think it happens if 2 drivers are running at the same time.

I have 2 ups's connected and noticed that only one reports information on the web but it wasn't consistent which one it was, sometimes it was UPS2, sometimes it was UPS3

[image]

I then opened 2 terminals:

Terminal 1:

```bash
sudo ./apc_modbus -DDD -d 1  -s apc  -x port=auto -x porttype=usb -x serial=ASXXXX
```

This runs perfectly fine on it's own.

Terminal 2:
```bash
sudo ./apc_modbus -DDD -d 1  -s apc1  -x port=auto -x porttype=usb -x serial=ASYYYY
```

This runs perfectly fine on it's own.

But when I run Terminal 1 and Terminal 2's commands as close as possible at the same time, the one run second always times out with: `_apc_modbus_read_registers: Read of 516:636 failed: Input/output error (auto)`

Anton2079, 2025-01-24:
Also have the same issue, if you need me to test anything let me know.

jimklimov, 2025-01-24:
@somejavadev : did you start the driver program with higher debug verbosity? Do they surely connect to different devices (use the `serial` matcher successfully) and do not try to actually fetch bytes from the same device's stream, breaking it for one or both readers?

@Anton2079 : ultimately we have discussed a number of issues here, which "same" do you have?
* timeouts as such
* FD-SET problem and program crash - hopefully solved by current rtu_usb branch tip
* messages too short and/or wrong "slave" counterpart ID (maybe both due to some wrong flush)

Which commit of libmodbus is involved in your situation? Does it work fine for a while and then spiral into madness, possibly after reconnect attempts?

somejavadev, 2025-01-24:
@jimklimov, I have started it with a higher debug level and I can see it failed the regex match:

sudo ./apc_modbus -DDDDDDDDDD -d 1  -s apc1  -x port=auto -x porttype=usb -x serial=ASYYYY

```
Considering device #2 (051d:0003)
   0.050179	[D3] match_function_regex: matching a device...
   0.050451	[D2] match_function_regex: failed match of Serial: ASXXXX
   0.050515	[D2] _apc_modbus_usb_callback: Matched American Power Conversion  Smart-UPS_1500 FW:UPS 03.5 / ID=1015 (USB VID/PID 051d:0003)
   0.050592	[D2] _apc_modbus_usb_callback: Checking American Power Conversion  Smart-UPS_1500 FW:UPS 03.5 / ID=1015 (USB VID/PID 051d:0003) report descriptors
   0.051071	[D1] _apc_modbus_usb_callback: Found report ids RX=0x90 TX=0x89
   0.051138	[D5] send_to_all: SETINFO ups.vendorid "051d"
   0.051194	[D5] send_to_all: SETINFO ups.productid "0003"
```

but strangely further down in the logs it seems to be reading the UPS information from the UPS it failed the match on:

```
battery.runtime: 13927
battery.temperature: 27.62
battery.voltage: 25.84
device.mfr: American Power Conversion
device.model: Smart-UPS 1500
device.serial: ASXXXX
device.type: ups
driver.debug: 10
```
libmodbus commit:

```
git rev-parse rtu_usb
0f10ae4fe4d1b7a916c58228f7a59a689c2406d8
```

nut commit:

```
git rev-parse master
9066236ec9c4a7cb3432859b7f718a93d8d53d6c
```

jimklimov, 2025-01-24:
Thanks! I think the driver should have ignored the mismatched device and went on to try another, but in fact called the callback (on it?) anyway...

somejavadev, 2025-01-25:
Thanks @jimklimov, let me know if I must open a separate issue for it but I wonder if it's not related to the problem @LucAce  is having.

jimklimov, 2025-01-29:
For some past days I've tried to bring the NUT fork of libmodbus up to speed, both regarding updates from the upstream project and about my own fixes here and there to pass NUT CI. Oh, and also ensuring those NUT CI builds.

Overall, the library seems shaky and flaky in its own tests, sometimes passing or sometimes failing, in same or varying test cases. It feels some are related to timing (and then as one test case is non-fatally skipped over, the next one receives a response to the skipped one's older query and not its own newer one), but in any case as we also deal with timeouts as a problem here, this may be related. Increasing some timeouts in test cases that probed them specifically did increase the chance of passing libmodbus `make check`, gut not to 100%.

I've updated the https://github.com/networkupstools/libmodbus/commits/rtu_usb fork/branch with some of the fixes that basically let it pass builds on our different platforms, but do not modify any test behavior - that part remains in a separate branch for now. FWIW also posted some of those changes that apply to upstream code, will see how long it would take for someone to take a look...

I am not sure I can pull this particular cart much further, only theorizing about where the issues might be (especially as the library's own self-tests are not reliable either) and not diving in to become an expert on modbus as such :\

So if anyone is willing to roll up some sleeves and try their luck with a debugger, to step through the actually running code and see how it works, that would be very welcome. I'm afraid however that any timing-related behaviors would be smudged by seconds or minutes spent paused in the debugger, and/or by slower performance of instrumented binaries generally. Maybe this is a case where an horde of `printf` (or NUT `upsdebugx`) statements is a better choice for troubleshooting and debug analysis...

jimklimov, 2025-01-29:
Regarding the mis-match of a device by `serial` (and possibly an attempt to use again the device held by another driver instance), I think it is a flaw in this bit of code:
* https://github.com/networkupstools/nut/blob/e3ef649ea6b188848b16fa1ff9241d3ef6efe9b2/drivers/apc_modbus.c#L1709-L1721

So indeed if we did not match with one matcher, we match on to another (with lowest-priority being the match by built-in vendor/product IDs which succeeds); it should differentiate the failure causes (unknown or rejected) and abort on the hard fail.

jimklimov, 2025-01-29:
Posted a PR to help with multiple-UPS monitoring and maybe saner reconnections, please try a custom build from https://github.com/jimklimov/nut/tree/issue-2609

somejavadev, 2025-01-31:
Thanks @jimklimov, really appreciate all the effort. I have rebuild nut from your branch and also rebuild libmodbus pulling the latest changes from rtu_usb

I can see the driver now selects the correct UPS but seems to timeout when trying to read information from it, I have increased the timeout value but it doesn't seem to make a difference, it just takes longer before timing out.

```
sudo ./apc_modbus -DDDDDDD -d 1  -s apc1  -x port=auto -x porttype=usb -x serial=YYY -x response_timeout_ms=30000
   0.000001	[D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools 2.7.4.11303-11303-ge3ef649ea (development iteration after 2.7.4) - NUT APC Modbus driver with USB support 0.11
   0.000220	[D5] send_to_all: SETINFO driver.version.usb "libusb-1.0.27 (API: 0x0100010A)"
   0.000243	[D1] Using USB implementation: libusb-1.0.27 (API: 0x0100010A)
   0.000261	[D1] This build of the driver is USB-capable; also Serial and TCP Modbus RTU are supported
   0.000294	[D3] main_arg: var='port' val='auto'
   0.000316	[D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000334	[D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000362	[D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000383	[D3] main_arg: var='porttype' val='usb'
   0.000407	[D5] send_to_all: SETINFO driver.parameter.porttype "usb"
   0.000426	[D3] main_arg: var='serial' val='YYY'
   0.000454	[D5] send_to_all: SETINFO driver.parameter.serial "YYY"
   0.000474	[D3] main_arg: var='response_timeout_ms' val='30000'
   0.000516	[D5] send_to_all: SETINFO driver.parameter.response_timeout_ms "30000"
   0.000535	[D1] Network UPS Tools version 2.7.4.11303-11303-ge3ef649ea (development iteration after 2.7.4) built with gcc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0 and configured with flags: --build=aarch64-linux-gnu --prefix=/usr --includedir=/usr/local/ups/include --mandir=/usr/local/ups/share/man --infodir=/usr/local/ups/share/info --sysconfdir=/etc --localstatedir=/var --disable-option-checking --disable-silent-rules --libdir=/usr/local/ups/lib/aarch64-linux-gnu --runstatedir=/run --disable-maintainer-mode --disable-dependency-tracking --prefix= --sysconfdir=/etc/nut --includedir=/usr/include --mandir=/usr/share/man --libdir=/usr/local/ups/lib/aarch64-linux-gnu --libexecdir=/usr/libexec --with-ssl --with-nss --with-cgi --with-dev --enable-static --with-statepath=/run/nut --with-altpidpath=/run/nut --with-drvpath=/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut --with-htmlpath=/usr/share/nut/www --with-pidpath=/run/nut --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib/aarch64-linux-gnu/pkgconfig --with-user=nut --with-group=nut --with-udev-dir=/lib/udev --with-systemdsystemunitdir=/lib/systemd/system --with-systemdshutdowndir=/lib/systemd/system-shutdown --with-systemdtmpfilesdir=/usr/lib/tmpfiles.d --with-python=python3 --with-python3=/usr/bin/python3 --with-doc=man --sbindir=/usr/sbin --bindir=/usr/bin --enable-Wcolor --enable-warnings --enable-Werror --enable-keep_nut_report_feature --with-all=auto --with-cgi=auto --with-serial=auto --with-dev=auto --with-doc=skip --with-nut_monitor=auto --with-pynut=auto --disable-force-nut-version-header --enable-check-NIT --enable-maintainer-mode --with-debuginfo=auto --with-drivers=apc_modbus --with-usb --with-modbus --with-modbus-includes=-I/usr/local/ups/include/modbus --with-modbus-libs='-L/usr/local/ups/lib -lmodbus' --enable-inplace-runtime='2.8.2.1929-1929-g9066236ec (development iteration after 2.8.2) built with gcc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0'
   0.000766	[D1] debug level is '7'
   0.000797	[D5] send_to_all: SETINFO driver.debug "7"
   0.000819	[D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.004097	[D1] Succeeded to become_user(nut): now UID=109 GID=111
   0.004159	[D1] Signalling UPS [apc1]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.004206	Can't open /run/nut/apc_modbus-apc1: No such file or directory
   0.004233	[D1] Request for other driver to exit returned code -1
   0.004253	[D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
   0.004278	[D5] send_to_all: SETINFO device.type "ups"
   0.004300	[D5] send_to_all: SETINFO driver.state "init.device"
Number of USB devices: 6
Considering device #2 (051d:0003)
   0.047123	[D2] - VendorID: 051d
   0.047200	[D2] - ProductID: 0003
   0.047250	[D2] - Manufacturer: American Power Conversion 
   0.047299	[D2] - Product: Smart-UPS_1500 FW:UPS 03.5 / ID=1015
   0.047347	[D2] - Serial Number: XXX    
   0.047394	[D2] - Bus: 001
   0.047440	[D2] - Bus Port: 004
   0.047487	[D2] - Device: 004
   0.047534	[D2] - Device release number: 0001
   0.047640	[D2] Trying to match device
   0.047691	[D5] _apc_modbus_usb_callback: current_matcher=0xaaaadb26c880
   0.047799	[D3] match_function_regex: matching a device...
   0.047987	[D2] match_function_regex: failed match of Serial: XXX    
   0.048045	[D5] _apc_modbus_usb_callback: Tried matcher 0xaaaadb26c880 returned 0
   0.048124	[D2] _apc_modbus_usb_callback: Device does not match - skipping
Considering device #3 (051d:0003)
   0.072247	[D2] - VendorID: 051d
   0.072296	[D2] - ProductID: 0003
   0.072328	[D2] - Manufacturer: American Power Conversion 
   0.072359	[D2] - Product: Smart-UPS_1500 FW:UPS 03.5 / ID=1015
   0.072390	[D2] - Serial Number: YYY    
   0.072422	[D2] - Bus: 001
   0.072452	[D2] - Bus Port: 003
[... cut ...]
   0.084451	[D5] send_to_all: SETINFO driver.version.internal "0.11"
   0.084499	[D5] send_to_all: SETINFO driver.name "apc_modbus"
   0.084544	[D5] send_to_all: SETINFO driver.state "init.info"
[01][03][02][04][00][78]
Waiting for a confirmation...
Using a backend without a file descriptor, will not natively select() on it.
```

I am going to do a fresh install of the OS and try again since I see there are references of version 2.8.2 and 2.7.4 when executing the command. Will update here if it makes a difference.

jimklimov, 2025-02-04:
The "screenshot" ends at initial data collection. Does the driver hang/timeout at this point?

Is there any specific message about the timeout, and what is the exit code?

Can you please check - with one UPS connected and with no matching by attributes (e.g. no vendor, product, serial parameters), so it would use only the default ID-based matching, does it also hang?

Overall, I wonder if I messed up the recent commits in that PR, changing the work with the collection of "matchers", so it would do some infinite loop/recursion, or lose sight of the stored matcher list, etc. when (re-)connecting to the device?

> 2.7.4

Development Version string construction (as amount of commits since a point in time) relies on known tags. If your repo did not `git fetch --tags`, this becomes problematic in terms of counting from an unknown newer release. I am not sure why 2.7.4 does pop up in many custom rebuilds, possibly it was a different tag type (annotated vs. plain) and is fetched differently during `git clone`?..

jimklimov, 2025-02-26:
@somejavadev : sounds interesting. So a bumped timeout value (maybe a better built-in default) can fix this?

MrPvTDagger, 2025-02-27:
Not sure if it helps but when the driver times out this is the output I get with the first line being the last successful pull for data before it disconnects.

```nut/drivers/apc_modbus -s nutdev-usb1 -x port=auto -DDDDDDD```

```  14.511575     [D5] send_to_all: SETINFO driver.state "quiet"
  14.511593     [D6] send_to_all: write 29 bytes to socket 10 succeeded (ret=29): SETINFO driver.state "quiet"

  16.241704     sock_connect: enabling asynchronous mode (auto)
  16.241718     [D3] sock_connect: new connection on fd 11
  16.241725     [D6] sock_arg: Driver on /run/nut/apc_modbus-nutdev-usb1 is now handling NOBROADCAST with 1 args
  16.241730     [D1] sock_arg: socket 11 requested NOBROADCAST mode
  16.241732     [D6] sock_arg: Driver on /run/nut/apc_modbus-nutdev-usb1 is now handling PING with 1 args
  16.241736     [D2] send_to_one: sending PONG
  16.241739     [D5] send_to_one: PONG
  16.241744     [D6] send_to_one: write 5 bytes to socket 11 succeeded (ret=5): PONG

  16.241833     [D6] sock_arg: Driver on /run/nut/apc_modbus-nutdev-usb1 is now handling INSTCMD with 4 args
  16.241837     [D3] sock_arg: TRACKING = 2DDC5036-F665-4D29-BE4C-0374D091D263
  16.241839     [D2] entering main_instcmd(driver.exit, (null)) for [nutdev-usb1] on socket 11
  16.241845     [D1] set_exit_flag: raising exit flag due to signal -2
  16.241846     [D2] send_to_one: sending TRACKING 2DDC5036-F665-4D29-BE4C-0374D091D263 0
  16.241847     [D5] send_to_one: TRACKING 2DDC5036-F665-4D29-BE4C-0374D091D263 0
  16.241853     [D6] send_to_one: write 48 bytes to socket 11 succeeded (ret=48): TRACKING 2DDC5036-F665-4D29-BE4C-0374D091D263 0

  16.241857     Signal -2: exiting
  16.241863     [D5] send_to_all: SETINFO driver.state "cleanup.upsdrv"
  16.241870     [D6] send_to_all: write 38 bytes to socket 10 succeeded (ret=38): SETINFO driver.state "cleanup.upsdrv"

  16.241941     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
  16.241953     [D6] send_to_all: write 36 bytes to socket 10 succeeded (ret=36): SETINFO driver.state "cleanup.exit"

  16.241991     [D3] sock_disconnect: disconnecting socket 11
  16.241997     [D5] sock_disconnect: finishing parsing context
  16.242000     [D5] sock_disconnect: relinking the chain of connections
  16.242003     [D5] sock_disconnect: freeing the conn object
  16.242004     [D3] sock_disconnect: disconnecting socket 10
  16.242011     [D5] sock_disconnect: finishing parsing context
  16.242014     [D5] sock_disconnect: relinking the chain of connections
  16.242015     [D5] sock_disconnect: freeing the conn object
```

Reconnecting the USB cable allows me to connect again

[21 more replies]