networkupstools/nut #2612: Service failed to start

Sssnipercat1, 2024-09-02:
[image]

Let me preface by saying that I am running Debian on a raspberry pi 3 and I followed the below guide to setup NUT. https://forums.unraid.net/topic/93341-tutorial-networked-nut-for-cyberpower-ups/

I have tried setting it to only listen on 127.0.0.1 but it still fails to start. I have no clue how to access the logs, I have tried looking but no dice. I am trying to figure out how to diagnose and fix the issue.

My ups.conf:
[image]

jimklimov, 2024-09-02:
What NUT version do you have? There's a wiki article about getting debug logs from different NUT versions. The `upsd -F` implies a recent-ish NUT v2.8.x, so that's more promising than 2.7.4 alas still too common too :)

Did you start the actual driver? Likely NDE would have spawned a `nut-driver@cyberpower.service` instance based on your config.

Sssnipercat1, 2024-09-02:
APT lists the version of NUT as: `nut/stable,now 2.8.0-7 all [installed]`

I did run `upsdrvctl start` with this output: 
```
root@rpi3-20231109:~# upsdrvctl start
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
Duplicate driver instance detected (PID file /run/nut/usbhid-ups-cyberpower.pid exists)! Terminating other driver!
Using subdriver: CyberPower HID 0.6
using 'battery.charge' to set battery low state
using 'battery.runtime' to set battery low state
```
Although the guide that I followed before the one linked in the issue had used parts of the automatically generated config, I couldn’t get that to work either.

jimklimov, 2024-09-02:
Well, the part about "Duplicate driver instance" suggests NDE did make a service for the driver. Check if `systemctl status nut-driver@cyberpower` says something about its health (e.g. did it recognize the device or keeps looping to start?)

For more debugging details you can write e.g. `DEBUG_MIN 6` into `upsd.conf` (for `nut-server` part) to see more about why it fails to start.

Use `journalctl -flu nut-server` to watch its logs as the unit is restarted from another terminal.

jimklimov, 2024-09-02:
Also, did it fail to start before or after you defined the device section in `ups.conf`? Are config file ownership and permissions okay?

jimklimov, 2024-09-02:
> P.P.S I have a remote management card in my UPS but I don’t have the drivers to support SNMP UPSs and wondering how to get them in my installation so I don’t have to connect to the UPS using a usb cable

Might be in a separate package like `nut-snmp`? IIRC `apt-cache search snmp-ups` should locate the package.

Sssnipercat1, 2024-09-02:
The status of `nut-driver@cybepower` shows that the service is active and had successful started, full output here:
[image]

And I added `DEBUG_MIN 6` to `upsd.conf` and the logs didn’t change at all. This is from a reboot: 
[image]

As far as I know all files should be owned by root, I only have the root user and I haven’t done anything to file permissions/ownership.

Sssnipercat1, 2024-09-02:
Also directly calling `upsd -h` outputs nothing, it just returns straight back to the console as if the process exited, without outputting anything, very shortly after starting.

Sssnipercat1, 2024-09-02:
> Also, did it fail to start before or after you defined the device section in `ups.conf`? Are config file ownership and permissions okay?

I am not sure every time I have tried to start the nut-server service it has failed.

Sssnipercat1, 2024-09-02:
As for file permissions on the configs, I ran `tree /etc/nut/ -pugaf` to result in this:
```
root@rpi3-20231109:~# tree /etc/nut/ -pugaf
[drwxr-xr-x root     nut     ]  /etc/nut
|-- [-rw-r--r-- root     root    ]  /etc/nut/.upssched.conf.swp
|-- [-rw-r----- root     nut     ]  /etc/nut/nut.conf
|-- [-rw-r----- root     nut     ]  /etc/nut/ups.conf
|-- [-rw-r----- root     nut     ]  /etc/nut/upsd.conf
|-- [-rw-r----- root     nut     ]  /etc/nut/upsd.users
|-- [-rw-r----- root     nut     ]  /etc/nut/upsmon.conf
`-- [-rw-r----- root     nut     ]  /etc/nut/upssched.conf

1 directory, 7 files
```

Sssnipercat1, 2024-09-02:
That was an accident

Sssnipercat1, 2024-09-02:
I directly called the the `upsd` executable to receive this: 
[image]

Sssnipercat1, 2024-09-02:
I just got my windows nut client connected to my raspberry pi, I directly invoked `/lib/nut/upsd` and the nut-server daemon started immediately.

Sssnipercat1, 2024-09-02:
Here is a screenshot from `htop`: 
[image]

Sssnipercat1, 2024-09-02:
I manually edited the system service files for the `nut-server` service and the `nut-client` service so they directly point at the executable files instead of the files in `/sbin/` so that they startup correctly.

jimklimov, 2024-09-02:
Hm, what's in `/sbin/upsd`? Some wrapper script? Is it from package? Seems like distro problem then :\

Sssnipercat1, 2024-09-02:
Yeah it looks like a wrapper script.
```
#!/bin/sh

# Include NUT nut.conf
[ -r /etc/nut/nut.conf ] && . /etc/nut/nut.conf

case "$MODE" in
  standalone|netserver)
    exec /lib/nut/upsd "$@"
  ;;
  none|netclient)
    echo "upsd disabled, please adjust the configuration to your needs"
    echo "Then set MODE to a suitable value in /etc/nut/nut.conf to enable it"
    # exit success to avoid breaking the install process!
    exit 0
  ;;
  *)
    exit 1
  ;;
esac
```

jimklimov, 2024-09-03:
Does not look sinister. Maybe some setting in `nut.conf` precludes setver start? Maybe directly `MODE`, maybe some other envvar?

Sssnipercat1, 2024-09-03:
That is what it looks like to me, that it is a wrapper script that warns the user if the mode=setting is not appropriately. It just seems like it wasn’t working correctly causing the upsd to not get called correctly.

jimklimov, 2024-09-03:
FWIW, can you post your `nut.conf` (and did you modify anything there)?

For kicks, you can also `set -xv` at the start of script and use it again in the systemd unit, to trace what the shell does...

Sssnipercat1, 2024-09-03:
My nut.conf is just mode = netserver

jimklimov, 2024-09-03:
uppercase, no spaces?

````
MODE=netserver
````

like this?

Sssnipercat1, 2024-09-03:
Whoops I was doing that off of memory, it is set as `MODE=nutserver`

Sssnipercat1, 2024-09-03:
The wrapper scripts are working after the typo was fixed

jimklimov, 2024-09-03:
So, mystery solved - issue can be closed?

Sssnipercat1, 2024-09-03:
Yeah.

closed 2024-09-03 by jimklimov