networkupstools/nut #2614: USB string langid should be cached somewhere (in `USBDevice_t` or a new `USBHandle_t`?)

jimklimov, 2024-09-03:
Currently we query for this value in every string request. Given repetitive query loops, caching can help.

Bonus question: Can different strings/queries have different langid's? Would such cache be one int to rule them all, or a map/dict?

_Originally posted in https://github.com/networkupstools/nut/issues/2604_

tormodvolden, 2024-09-03:
No, different strings cannot have different langid's. This is per device, and I believe it can be safely read once per enumeration, with no reason to read it more often. I suppose e.g. an OS shouldn't store the langid in a registry or so like some do for storing driver information etc), however, it is not expected that a device change its langid's, which declare the supported languages for the device's string descriptors.

Note that the request for langid ("string" descriptor zero) returns an array of supported langid's, but typically there is only one value, and most software only look at the first anyway, which should be the preferred langid. But say it returns English and Chinese, then the software could, e.g. depending on locale or user preference, choose to retrieve the English strings or the Chinese ones (or both).

So to turn your question around a bit, if the device reports multiple langid's, you /can/ choose to read one string descriptor e.g in Chinese and another one e.g. in English, if that would ever make sense for an application.

jimklimov, 2024-09-04:
Thanks a lot for the details, much appreciated!