networkupstools/nut #2617: [Support] Epyc ION UPS (lsusb listed as cyber power CP1500PFCLCD) connection refused

darkmattercoder, 2024-09-05:
Hello,

I am absolutely new with UPSes. I grabbed a relatively cheap UPS with good reviews from amazon to protect my home server. The unit is named Epyc Ion: https://www.justepyc.com/en/ups/14-57-epyc-ion-ups.html#/power-800va

since I do not necessarily want to install the suspicious management software that they offer on my server (Power Master Plus): https://www.powermonitor.software/#PowerMasterPlusSoftware (no imprint, no company name, looks really strange to me), I wanted to try a well known software like nut. 

The brand name EPYC is not mentioned anywhere on the official documentation website. But I found one dev mailing list conversation between a dev and a person who is apparently a brand representative: https://alioth-lists.debian.net/pipermail/nut-upsdev/2021-August/007583.html

I am not sure whether that is solely about putting the device on the website or adding its communication to the drivers. I also cannot see the attached files that the person mentioned in their e-mail.

When I plug the unit into an ubuntu 22.04 pc, it is listed as  CP1500PFCLCD via `lsusb`, which is a device that is in the compatibility matrix via `usbhid-psu`.

So I tried to configure it via

```
[epyc]
    driver = usbhid-ups
    port = auto
    desc = "Epyc Ion"
```

and started the daemon:

```
sudo upsdrvctl -D start                     
Network UPS Tools - UPS driver controller 2.7.4
   0.000000	Starting UPS: epyc
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
```

I then tried the following command to see whether the communication works:

```
sudo upsc epyc       
Error: Connection failure: Connection refused
```

A deeper look into the driver output shows:

```
0.024842	Checking device (0764/0601) (001/034)
   3.093006	- VendorID: 0764
   3.093048	- ProductID: 0601
   3.093060	- Manufacturer: unknown
   3.093072	- Product: unknown
   3.093083	- Serial Number: unknown
   3.093095	- Bus: 001
   3.093110	- Device release number: 0200
   3.093128	Trying to match device
   3.093152	Device matches
   3.285504	Unable to get HID descriptor (error sending control message: Value too large for defined data type)
   3.285553	HID descriptor length 739
   8.469330	Unable to get Report descriptor: Connection timed out
```

I guess that the identification as CyberPower device is wrong and I need a specific subdriver. Maybe those textfiles from the mailing list would help, however how to get them and what to do with them....? 

One other person also wanted to use an epyc device: https://github.com/networkupstools/nut/issues/2362 but that is somehow homeassistant related and out of scope for me for now.

I am a bit lost, maybe someone has hints how to proceed. 

Thanks in advance

jimklimov, 2024-09-05:
Well, HA is typically also Linux (with containers for plugins like NUT, from what I gather) - so suggestion in that thread to try `nutdrv_qx` or `blazer_usb` also applies.

In your case the driver failed to start, but generally per error message, `upsc` client failed (it seems) primarily because the `nut-server` service (or `upsd` program) is not running.

Also, see if you can update the distro to one that ships current NUT (2.8.x) or if you can compile your own, see our Github wiki footer links (traveling now, hard to post).

darkmattercoder, 2024-09-07:
I updated to NUT 2.8.0 via an available ppa for ubuntu 22.04. This made the UPS accessible.

closed 2024-09-07 by darkmattercoder