networkupstools/nut #2619: Eaton UPS with SNMP not changing to LB ups.status on low battery

Chris3773, 2024-09-08:
When the low battery warning is triggered the ups.status doesn't update to LB and stays on OB, DISCHRG. The warning is triggered and is visible through the Eaton Network-M3 web UI, but the status is never updated for NUT. I have tested with low the warning trigger at 5% with 0 runtime and 5% with 300 runtime with the same results. Since the ups.status isn't set to LB status the shutdown is never triggered.

OS: pfSense 24.03-RELEASE
Nut Version: 2.8.2
UPS: Eaton 5PX 3000 RT2U G2
Network Card: Eaton Network-M3
Driver: SNMPv3

jimklimov, 2024-09-08:
Just in case, you don't have `ignorelb` in `ups.conf` section for the device, right?

Chris3773, 2024-09-08:
> Just in case, you don't have `ignorelb` in `ups.conf` section for the device, right?

Correct, I do not have `ignorelb` in `ups.conf`.

xkszltl, 2025-02-10:
Same question for this model. Also it doesn't report the battery.low.level configured in web, only battery.runtime.low is available but also rounded to 60s.

xkszltl, 2025-02-10:
Also:
- battery.charge and battery.runtime are correctly reported.
- ups.status is always "ALARM OB DISCHRG" on low battery.

jimklimov, 2025-02-10:
@xkszltl : just in case, are you using a packaged NUT or a custom build of recent master branch, e.g. per https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests ? Note there were some changes regarding Eaton/MGE devices since NUT v2.8.2 release, although I don't quickly see any about SNMP...

ugo-zaber, 2025-11-12:
I'm also experiencing the same issue. I updated NUT to 2.8.4 and ups.status goes from OL to OB, DISCHRG but never LB.
Because of that, NOTIFYFLAG LOWBATT doesn't kick in either.

Are there any information I can provide to help?