networkupstools/nut #2621: Restricted Eaton Software License

xTeixeira, 2024-09-10:
Commit a69179a6484126f04d90db93ae1f0ec6e8f0ca95 added a license.txt file which places restrictions on copying and modifying, so it seems to be incompatible with the GPL.

This can also create problems for redistributing source code tarballs e.g. in rpm and deb source packages.

Maybe the Eaton scripts should be relicensed or moved to a separate repo?

jimklimov, 2024-09-10:
Good point, Eaton open-source people did approve of the publication, not in the least because the code is scripted and inherently open-sourced when delivered to customers of the freely downloaded companion software. And arranges field configuration of GPL NUT. FWIW, the initial upload was deliberately "as is" to be a starting point for any trackable changes.

Perhaps the license file can be updated or removed, or indeed these helpers can be served in a separate repo.

WDYT, @arnaudquette-eaton ?

arnaudquette-eaton, 2024-09-11:
thanks for raising the point @xTeixeira 
as pointed by @jimklimov, these files were given to NUT as an attempt to help a bit more.
The relicensing (to GPL2+) is implied, and the pointed license.txt file is just the one that was presented to users upon installation.
Jim, please do everything needed to realign with NUT licensing, spasibo

jimklimov, 2024-09-11:
Ok, I will remove the file from current Git to avoid confusion, and make a note in README and/or some other doc.

PR #2623 (2024-09-11): docs: finalize re-licensing of `scripts/installer` …
…from Eaton to NUT terms (GPLv2+)

Follows up from: #2288

Closes: #2621

CC @arnaudquette-eaton - looks right? :)
closed 2024-09-11 by jimklimov