networkupstools/nut #2622: A wrong `usb_config_index` (1) caused a segfault

jimklimov, 2024-09-11:
> I rebased on master and force-pushed an identical change to libusb0.c.
> This was compile and runtime tested, but not confirmed to fix the issue since I can't repro. I intentionally set a wrong `usb_hid_rep_index` value in ups.conf and confirmed that things still broke as expected, though. A wrong `usb_config_index` (1) caused a segfault, didn't look too much into it though:

```
   0.268093	[D2] Trying to match device
   0.268094	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.268095	[D3] match_function_regex: matching a device...
   0.268110	[D2] Device matches
   0.268118	[D3] nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.271084	[D2] Retrieved HID descriptor (expected 9, got 9)
   0.271089	[D3] HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 ae 01
   0.271090	[D3] HID descriptor length (method 1) 430
Segmentation fault
```

_Originally posted by @tofurky in https://github.com/networkupstools/nut/issues/2611#issuecomment-2332995263_

jimklimov, 2024-09-11:
@tofurky : did this only happen with the libusb-0.1 build for you, or also libusb-1.0 (how much pressing does this issue seem)?

tofurky, 2024-09-11:
It doesn't happen with libusb-1.0, just 0.1. Some more info (I didn't want to turn my PR into a bug report):

valgrind:
```
   0.941232	[D2] Device does not match - skipping
   0.941310	[D2] Checking device (1D6B/0002) (003/001)
   0.941357	[D1] nut_usb_get_string_descriptor: string descriptor 0 request failed, retrying...
   0.991450	[D1] nut_usb_get_string_descriptor: string descriptor 0 request failed, retrying...
   1.041550	[D1] nut_libusb_open: get Manufacturer string failed
   1.041659	[D1] nut_usb_get_string_descriptor: string descriptor 0 request failed, retrying...
   1.091752	[D1] nut_usb_get_string_descriptor: string descriptor 0 request failed, retrying...
   1.141852	[D1] nut_libusb_open: get Product string failed
   1.141879	[D1] nut_usb_get_string_descriptor: string descriptor 0 request failed, retrying...
   1.191981	[D1] nut_usb_get_string_descriptor: string descriptor 0 request failed, retrying...
   1.242079	[D1] nut_libusb_open: get Serial Number string failed
   1.242100	[D2] - VendorID: 1d6b
   1.242113	[D2] - ProductID: 0002
   1.242126	[D2] - Manufacturer: unknown
   1.242138	[D2] - Product: unknown
   1.242149	[D2] - Serial Number: unknown
   1.242159	[D2] - Bus: 003
   1.242171	[D2] - Device: 001
   1.242182	[D2] - Device release number: 0606
   1.242192	[D2] Trying to match device
   1.242208	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   1.242229	[D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   1.242240	[D2] Device does not match - skipping
   1.242323	[D2] Checking device (0764/0501) (003/002)
   1.259297	[D2] - VendorID: 0764
   1.259319	[D2] - ProductID: 0501
   1.259331	[D2] - Manufacturer: CP1500PFCLCD
   1.259344	[D2] - Product: CRCA102#BJ1
   1.259354	[D2] - Serial Number: CPS
   1.259365	[D2] - Bus: 003
   1.259375	[D2] - Device: 002
   1.259386	[D2] - Device release number: 0001
   1.259396	[D2] Trying to match device
   1.259410	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   1.259832	[D3] match_function_regex: matching a device...
   1.266585	[D2] Device matches
   1.266991	[D3] nut_libusb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   1.269366	[D2] Retrieved HID descriptor (expected 9, got 9)
   1.269901	[D3] HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 ae 01
   1.270083	[D3] HID descriptor length (method 1) 430
==684660== Invalid read of size 8
==684660==    at 0x136170: ??? (in /usr/lib/nut/usbhid-ups)
==684660==    by 0x12868D: ??? (in /usr/lib/nut/usbhid-ups)
==684660==    by 0x499E1C9: (below main) (libc_start_call_main.h:58)
==684660==  Address 0x4bd1b10 is 16 bytes before a block of size 34 free'd
==684660==    at 0x484988F: free (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==684660==    by 0x4881FCD: usb_os_find_devices (in /usr/lib/x86_64-linux-gnu/libusb-0.1.so.4.4.4)
==684660==    by 0x488277C: usb_find_devices (in /usr/lib/x86_64-linux-gnu/libusb-0.1.so.4.4.4)
==684660==    by 0x13546E: ??? (in /usr/lib/nut/usbhid-ups)
==684660==    by 0x12868D: ??? (in /usr/lib/nut/usbhid-ups)
==684660==    by 0x499E1C9: (below main) (libc_start_call_main.h:58)
==684660==  Block was alloc'd at
==684660==    at 0x4846828: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==684660==    by 0x4882030: usb_os_find_devices (in /usr/lib/x86_64-linux-gnu/libusb-0.1.so.4.4.4)
==684660==    by 0x488277C: usb_find_devices (in /usr/lib/x86_64-linux-gnu/libusb-0.1.so.4.4.4)
==684660==    by 0x13546E: ??? (in /usr/lib/nut/usbhid-ups)
==684660==    by 0x12868D: ??? (in /usr/lib/nut/usbhid-ups)
==684660==    by 0x499E1C9: (below main) (libc_start_call_main.h:58)
==684660== 
==684660== Invalid read of size 8
==684660==    at 0x136174: ??? (in /usr/lib/nut/usbhid-ups)
==684660==    by 0x12868D: ??? (in /usr/lib/nut/usbhid-ups)
==684660==    by 0x499E1C9: (below main) (libc_start_call_main.h:58)
==684660==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==684660== 
==684660== 
[... 81 lines cut ...]
        do_forceshutdown = <optimized out>
        update_count = 0
        cmd = <optimized out>
        oldpid = <optimized out>
        optstring = "+a:s:kDFBd:hx:Lqr:u:g:Vi:c:P:"
```

Seems line 600 of libusb0.c is this:
```
                        iface = &dev->config[usb_subdriver.usb_config_index].interface[usb_subdriver.hid_rep_index].altsetting[0];
                        for (i=0; i<iface->extralen; i+=iface->extra[i]) {
                                upsdebugx(4, "i=%d, extra[i]=%02x, extra[i+1]=%02x", i,
                                        iface->extra[i], iface->extra[i+1]);

                                if (i+9 <= iface->extralen
                                &&  iface->extra[i] >= 9
                                &&  iface->extra[i+1] == 0x21
                                ) {
                                        p = (usb_ctrl_char *)&iface->extra[i];
                                        upsdebug_hex(3, "HID descriptor, method 2", p, 9);
                                        rdlen2 = ((uint8_t)p[7]) | (((uint8_t)p[8]) << 8);
                                        break;
                                }
                        }
```

tofurky, 2024-09-11:
Updated valgrind output above; didn't have dbgsym installed in the original comment.

tofurky, 2024-09-11:
And also, to clarify - it only occurred because I intentionally specified an invalid `usb_config_index` to ensure that the changes in the PR weren't stepping on the override. So, not pressing at all - was just a bit surprised to see a segfault rather than it being gracefully handled.

jimklimov, 2024-09-11:
So, comparing these lines for two implementations:

* https://github.com/networkupstools/nut/blob/6da2fd9c3a5769c96e7d695a1ae389b01e15540f/drivers/libusb0.c#L599
* https://github.com/networkupstools/nut/blob/6da2fd9c3a5769c96e7d695a1ae389b01e15540f/drivers/libusb1.c#L666

I can only guess the issue is about looking into array elements which are not there?..

In `libusb1.c` variant, it is fetched here:
* https://github.com/networkupstools/nut/blob/6da2fd9c3a5769c96e7d695a1ae389b01e15540f/drivers/libusb1.c#L498-L500

I do not see any practical equivalent in `libusb0.c` :\

jimklimov, 2024-09-11:
Seems to be this `struct usb_device` : https://www.kernel.org/doc/html/v6.1/driver-api/usb/usb.html

```
...
  struct usb_host_config *config;
  struct usb_host_config *actconfig;
...
```

* config
    all of the device’s configs

* actconfig
    the active configuration

Very helpful :\ no counter. Maybe gotta loop until a null entry?