networkupstools/nut #2632: [HCL] PowerWalker VFI 1000 AT supported by usbhid-ups

evgeniy-harchenko, 2024-09-16:
[PowerWalker VFI 1000 AT](https://powerwalker.com/product/10122180/)

`upsc`:

```
battery.charge: 94
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: 1 
battery.runtime: 58740
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 27.4
battery.voltage.nominal: 24
device.mfr: 1 
device.model: 1000
device.serial: [redacted]
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.offdelay: 20
driver.parameter.ondelay: 30
driver.parameter.pollfreq: 5
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0601
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0764
driver.state: quiet
driver.version: 2.8.1
driver.version.data: CyberPower HID 0.8
driver.version.internal: 0.52
driver.version.usb: libusb-1.0.27 (API: 0x100010a)
input.frequency: 50.0
input.transfer.high: 300
input.transfer.low: 160
input.voltage: 210.5
input.voltage.nominal: 230
output.frequency: 49.9
output.voltage: 24.8
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 0
ups.mfr: 1 
ups.model: 1000
ups.productid: 0601
ups.realpower.nominal: 900
ups.serial: [redacted]
ups.status: OL CHRG
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```

`upsrw`:

```
[battery.charge.low]
Remaining battery level when UPS switches to LB (percent)
Type: STRING
Maximum length: 10
Value: 10

[battery.runtime.low]
Remaining battery runtime when UPS switches to LB (seconds)
Type: STRING
Maximum length: 10
Value: 300

[driver.debug]
Current debug verbosity level of the driver program
Type: NUMBER
Value: 0

[driver.flag.allow_killpower]
Safety flip-switch to allow the driver daemon to send UPS shutdown command (accessible via driver.killpower)
Type: NUMBER
Value: 0

[input.transfer.high]
High voltage transfer point (V)
Type: STRING
Maximum length: 10
Value: 300

[input.transfer.low]
Low voltage transfer point (V)
Type: STRING
Maximum length: 10
Value: 160

[ups.delay.shutdown]
Interval to wait after shutdown with delay command (seconds)
Type: STRING
Maximum length: 10
Value: 20

[ups.delay.start]
Interval to wait before (re)starting the load (seconds)
Type: STRING
Maximum length: 10
Value: 30
```

`upscmd`:

```
beeper.disable - Disable the UPS beeper
beeper.enable - Enable the UPS beeper
beeper.mute - Temporarily mute the UPS beeper
beeper.off - Obsolete (use beeper.disable or beeper.mute)
beeper.on - Obsolete (use beeper.enable)
driver.killpower - Tell the driver daemon to initiate UPS shutdown; should be unlocked with driver.flag.allow_killpower option or variable setting
driver.reload - Reload running driver configuration from the file system (only works for changes in some options)
driver.reload-or-error - Reload running driver configuration from the file system (only works for changes in some options); return an error if something changed and could not be applied live (so the caller can restart it with new options)
driver.reload-or-exit - Reload running driver configuration from the file system (only works for changes in some options); exit the running driver if something changed and could not be applied live (so service management framework can restart it with new options)
load.off - Turn off the load immediately
load.off.delay - Turn off the load with a delay (seconds)
load.on - Turn on the load immediately
load.on.delay - Turn on the load with a delay (seconds)
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start.deep - Start a deep battery test
test.battery.start.quick - Start a quick battery test
test.battery.stop - Stop the battery test
```

My `ups.conf`:

```
[nutdev1]
        driver = "usbhid-ups"
        port = "auto"
        desc = "PowerWalker VFI 1000 AT"        
        vendorid = "0764"
        productid = "0601"
        pollfreq = 5
        offdelay = 20
        ondelay = 30
        default.ups.timer.shutdown = 20        
        default.ups.timer.start = 30
```
Shutdown of clients, server and UPS works. But it seems it ignores `ondelay` and `offdelay`. I tried different values for them but the result is always the same. After `upsmon -c fsd` in `/bin/upssched-cmd` invoked server starts to shutdown. After ~15 seconds timer appears on the UPS's display. After even ~10 seconds UPS shuts down outlets (timer is still on dislay). After all of that UPS continues to work with outlets turned off. And only after 2 more minutes UPS shuts down.

evgeniy-harchenko, 2024-09-17:
By the way. `output.voltage` is wrong in 2.8.1. I don't know what it shows. There was no problem in 2.7.4. Unfortunately, I can't test latest git version now, Ubuntu repository has not provide it yet.

jimklimov, 2024-09-17:
Can you try building current NUT to test the driver from workspace? No need to replace the existing installation, although you would need to add prerequisite and tool packages...

evgeniy-harchenko, 2024-09-18:
Just checked on clear Ubuntu 22.04 with last git version `2.8.2.1068-1068-g967cb5403`. Unfortunately, same problems (delay and `output.voltage`).

```
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: 1 
battery.runtime: 45480
battery.runtime.low: 300
battery.status: 100%
battery.type: PbAcid
battery.voltage: 27.5
battery.voltage.nominal: 24
device.mfr: 1 
device.model: 1000
device.serial: [redacted]
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: 1000
driver.parameter.productid: 0601
driver.parameter.serial: [redacted]
driver.parameter.synchronous: auto
driver.parameter.vendor: 1
driver.parameter.vendorid: 0764
driver.state: quiet
driver.version: 2.8.2.1068-1068-g967cb5403
driver.version.data: CyberPower HID 0.81
driver.version.internal: 0.57
driver.version.usb: libusb-1.0.25 (API: 0x1000109)
input.frequency: 50.0
input.transfer.high: 300
input.transfer.low: 160
input.voltage: 231.0
input.voltage.nominal: 230
output.frequency: 50.0
output.voltage: 24.8
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 0
ups.mfr: 1 
ups.model: 1000
ups.productid: 0601
ups.realpower.nominal: 900
ups.serial: [redacted]
ups.status: OL
ups.timer.shutdown: 20
ups.timer.start: 30
ups.vendorid: 0764
```

evgeniy-harchenko, 2024-09-18:
Not helped :disappointed:. By the way, I don't know what this value shows, but sometimes it changes its value.

evgeniy-harchenko, 2024-11-09:
OK. I found what `output.voltage` is showing now. It changes when I change this value:
[image]
[image]
I set `0bP` and `output.voltage` is ~`9.2` now.

evgeniy-harchenko, 2025-04-30:
Hmm. After multiple reinstallations of different versions (built from source), I completely removed NUT and reinstalled it using the latest Git version. The problem has now disappeared, and I didn't change any configuration files. So, closing.

closed 2025-04-30 by evgeniy-harchenko