networkupstools/nut #2633: Eaton ups recommendation

erlendoyen, 2024-09-18:
Hi

i am looking for a new UPS to get something where UPS shutdown after load shutdown command is supported and working

i think i would like to get a Eaton ups, since that brand seems to provide information to the NUT project about the hw, and is supported by the usbhid-ups

price: 200-250$

desertwitch, 2024-09-19:
I've just read from another user their Eaton Ellipse ECO 650 works well in that regard (kill power after shutdown).
I can also report personally that my 5P 850 works extremely well with NUT (also supporting the kill power commands).
You can also explore the "Issues" here for "HCL" reports, which usually state which commands were tested and worked... 😎

desertwitch, 2024-09-19:
> ok, thank you.
> 
> 
> 
> where did you read about Eaton Ellipse ECO 650?

On the Unraid forums. 😃