networkupstools/nut #2642: Mac OS X Sonoma / Sequoia on Mac Mini M2 - NUT 2.8.2 upsdrvctl / usbhid-ups fail to fork the driver as a daemon process.

pjkerly, 2024-09-26:
Mac OS X Sonoma / Sequoia (I have not tried other OS versions) on Mac Mini M2 - NUT 2.8.2 - I can start both upsdrvctl / usbhid-ups as long as it runs in the foreground.  As long as I use -D -F -FF with runs in foreground all works as expected.  If I try to run in background mode default or with -B, the driver will find the UPS device and run correctly all the way up until it forks the process.  Right now, I have workaround but it does not work as expected.

sudo ../sbin/upsdrvctl -FF start CP1500PFCLCD 
 
Network UPS Tools - UPS driver controller 2.8.2
Network UPS Tools - Generic HID driver 0.53 (2.8.2)
USB communication driver (libusb 1.0) 0.47
Using subdriver: CyberPower HID 0.80
Running as foreground process, but saving a PID file anyway
upsnotify: failed to notify about state 2: no notification tech defined, will not spam more about it
sock_connect: enabling asynchronous mode (auto)

Then I can stop it successfully using sudo ../sbin/upsdrvctl stop which means that it is generating the PID file success and can use it to stop the driver.

I can also start the usbhid-ups driver manually as well as long as it runs in the foreground.  You can see that the driver does find the UPS device and processes correctly just to the point when it would fork the process.

sudo ../bin/usbhid-ups -D -B -u root -a CP1500PFCLCD                                                         
Password:
Network UPS Tools - Generic HID driver 0.53 (2.8.2)
USB communication driver (libusb 1.0) 0.47
   0.000000	[D1] upsdrv_makevartable...
   0.000185	[D1] Using USB implementation: libusb-1.0.27 (API: 0x100010a)
   0.000193	[D1] Built-in default or configured user for drivers '_nut' was ignored due to 'root' specified on command line
   0.000534	[D1] Network UPS Tools version 2.8.2 (release/snapshot of 2.8.2) built with Apple clang version 16.0.0 (clang-1600.0.26.3); Target: arm64-apple-darwin24.0.0; Thread model: posix and configured with flags: --prefix=/opt/local --with-serial=no --with-usb=auto --with-snmp=no --with-hal=auto --with-cgi=no --with-neonxml=no --with-ssl=yes --with-ipv6=no --with-statepath=/opt/local/var/db/ups --with-pidpath=/opt/local/var/run --with-hotplug-dir=/opt/local/etc/nut/hotplug --with-udev-dir=/opt/local/etc/nut/udev --with-user=_nut --with-group=_nut
   0.000539	[D1] debug level is '1'
   0.001381	[D1] Succeeded to become_user(root): now UID=0 GID=0
   0.001442	[D1] Saving PID 2939 into /opt/local/var/db/ups/usbhid-ups-CP1500PFCLCD.pid
   0.001485	[D1] upsdrv_initups (non-SHUT)...
   0.029139	Using subdriver: CyberPower HID 0.80
   0.029144	[D1] 53 HID objects found
   0.032078	[D1] Path: UPS.PowerSummary.iProduct, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
   0.035025	[D1] Path: UPS.PowerSummary.iSerialNumber, Type: Feature, ReportID: 0x02, Offset: 0, Size: 8, Value: 2
   0.038078	[D1] Path: UPS.PowerSummary.ff0100d0, Type: Feature, ReportID: 0x1b, Offset: 0, Size: 8, Value: 5
   0.041063	[D1] Path: UPS.PowerSummary.iDeviceChemistry, Type: Feature, ReportID: 0x03, Offset: 0, Size: 8, Value: 4
   0.044084	[D1] Path: UPS.PowerSummary.iOEMInformation, Type: Feature, ReportID: 0x04, Offset: 0, Size: 8, Value: 3
   0.047075	[D1] Path: UPS.PowerSummary.Rechargeable, Type: Feature, ReportID: 0x05, Offset: 0, Size: 8, Value: 1
   0.050041	[D1] Path: UPS.PowerSummary.CapacityMode, Type: Feature, ReportID: 0x06, Offset: 0, Size: 8, Value: 2
   0.053027	[D1] Path: UPS.PowerSummary.DesignCapacity, Type: Feature, ReportID: 0x07, Offset: 0, Size: 8, Value: 100
   0.053037	[D1] Path: UPS.PowerSummary.CapacityGranularity1, Type: Feature, ReportID: 0x07, Offset: 8, Size: 8, Value: 5
   0.053045	[D1] Path: UPS.PowerSummary.CapacityGranularity2, Type: Feature, ReportID: 0x07, Offset: 16, Size: 8, Value: 10
   0.053053	[D1] Path: UPS.PowerSummary.WarningCapacityLimit, Type: Feature, ReportID: 0x07, Offset: 24, Size: 8, Value: 20
   0.053060	[D1] Path: UPS.PowerSummary.RemainingCapacityLimit, Type: Feature, ReportID: 0x07, Offset: 32, Size: 8, Value: 10
   0.053064	[D1] Path: UPS.PowerSummary.FullChargeCapacity, Type: Feature, ReportID: 0x07, Offset: 40, Size: 8, Value: 100
   0.056030	[D1] Path: UPS.PowerSummary.RemainingCapacity, Type: Input, ReportID: 0x08, Offset: 0, Size: 8, Value: 100
   0.056040	[D1] Path: UPS.PowerSummary.RemainingCapacity, Type: Feature, ReportID: 0x08, Offset: 0, Size: 8, Value: 100
   0.056049	[D1] Path: UPS.PowerSummary.RunTimeToEmpty, Type: Input, ReportID: 0x08, Offset: 8, Size: 16, Value: 8970
   0.056056	[D1] Path: UPS.PowerSummary.RunTimeToEmpty, Type: Feature, ReportID: 0x08, Offset: 8, Size: 16, Value: 8970
   0.056064	[D1] Path: UPS.PowerSummary.RemainingTimeLimit, Type: Input, ReportID: 0x08, Offset: 24, Size: 16, Value: 300
   0.056066	[D1] Path: UPS.PowerSummary.RemainingTimeLimit, Type: Feature, ReportID: 0x08, Offset: 24, Size: 16, Value: 300
   0.059028	[D1] Path: UPS.PowerSummary.ConfigVoltage, Type: Feature, ReportID: 0x09, Offset: 0, Size: 8, Value: 24
   0.062021	[D1] Path: UPS.PowerSummary.Voltage, Type: Feature, ReportID: 0x0a, Offset: 0, Size: 8, Value: 24
   0.065036	[D1] Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Input, ReportID: 0x0b, Offset: 0, Size: 1, Value: 1
   0.065047	[D1] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Input, ReportID: 0x0b, Offset: 1, Size: 1, Value: 0
   0.065053	[D1] Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Input, ReportID: 0x0b, Offset: 2, Size: 1, Value: 0
   0.065059	[D1] Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Input, ReportID: 0x0b, Offset: 3, Size: 1, Value: 0
   0.065066	[D1] Path: UPS.PowerSummary.PresentStatus.FullyCharged, Type: Input, ReportID: 0x0b, Offset: 4, Size: 1, Value: 1
   0.065068	[D1] Path: UPS.PowerSummary.PresentStatus.RemainingTimeLimitExpired, Type: Input, ReportID: 0x0b, Offset: 5, Size: 1, Value: 0
   0.065070	[D1] Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x0b, Offset: 0, Size: 1, Value: 1
   0.065072	[D1] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x0b, Offset: 1, Size: 1, Value: 0
   0.065074	[D1] Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x0b, Offset: 2, Size: 1, Value: 0
   0.065076	[D1] Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x0b, Offset: 3, Size: 1, Value: 0
   0.065078	[D1] Path: UPS.PowerSummary.PresentStatus.FullyCharged, Type: Feature, ReportID: 0x0b, Offset: 4, Size: 1, Value: 1
   0.065080	[D1] Path: UPS.PowerSummary.PresentStatus.RemainingTimeLimitExpired, Type: Feature, ReportID: 0x0b, Offset: 5, Size: 1, Value: 0
   0.068052	[D1] Path: UPS.PowerSummary.AudibleAlarmControl, Type: Feature, ReportID: 0x0c, Offset: 0, Size: 8, Value: 2
   0.068059	[D1] Path: UPS.PowerSummary.AudibleAlarmControl, Type: Input, ReportID: 0x0c, Offset: 0, Size: 8, Value: 2
   0.070990	[D1] Path: UPS.PowerSummary.iManufacturer, Type: Feature, ReportID: 0x0d, Offset: 0, Size: 8, Value: 3
   0.074033	[D1] Path: UPS.Input.ConfigVoltage, Type: Feature, ReportID: 0x0e, Offset: 0, Size: 8, Value: 120
   0.077031	[D1] Path: UPS.Input.Voltage, Type: Feature, ReportID: 0x0f, Offset: 0, Size: 16, Value: 119
   0.080036	[D1] Path: UPS.Input.LowVoltageTransfer, Type: Feature, ReportID: 0x10, Offset: 0, Size: 16, Value: 88
   0.080042	[D1] Path: UPS.Input.LowVoltageTransfer, Type: Input, ReportID: 0x10, Offset: 0, Size: 16, Value: 88
   0.080046	[D1] Path: UPS.Input.HighVoltageTransfer, Type: Feature, ReportID: 0x10, Offset: 16, Size: 16, Value: 139
   0.080050	[D1] Path: UPS.Input.HighVoltageTransfer, Type: Input, ReportID: 0x10, Offset: 16, Size: 16, Value: 139
   0.083005	[D1] Path: UPS.Output.Voltage, Type: Feature, ReportID: 0x12, Offset: 0, Size: 16, Value: 119
   0.086030	[D1] Path: UPS.Output.PercentLoad, Type: Feature, ReportID: 0x13, Offset: 0, Size: 8, Value: 3
   0.089029	[D1] Path: UPS.Output.Test, Type: Feature, ReportID: 0x14, Offset: 0, Size: 8, Value: 6
   0.089033	[D1] Path: UPS.Output.Test, Type: Input, ReportID: 0x14, Offset: 0, Size: 8, Value: 6
   0.092024	[D1] Path: UPS.Output.DelayBeforeShutdown, Type: Feature, ReportID: 0x15, Offset: 0, Size: 16, Value: -60
   0.095017	[D1] Path: UPS.Output.DelayBeforeStartup, Type: Feature, ReportID: 0x16, Offset: 0, Size: 16, Value: -60
   0.098029	[D1] Path: UPS.Output.Boost, Type: Feature, ReportID: 0x17, Offset: 0, Size: 1, Value: 0
[... cut ...]
   0.119052	[D1] Got 6 HID objects...
   0.119064	[D1] Quick update...
   0.119565	[D1] Group and/or user account for this driver was customized ('root:_nut') compared to built-in defaults. Fixing socket '/opt/local/var/db/ups/usbhid-ups-CP1500PFCLCD' ownership/access.
   0.119574	[D1] WARNING: opening socket file for stat/chown failed (102), which is rather typical for Unix socket handling: Operation not supported on socket
   0.119604	[D1] Group access for this driver successfully fixed (using file name based methods)

jimklimov, 2024-09-26:
Can you please try that later command with more verbosity e.g. `-DDDDDD`?

Trying to reproduce it on a VM from the CI farm, so with a `dummy-ups` driver in a setup made by `make check-NIT-sandbox` - and it forks well...

````
Abuild-iMac-Pro:nut abuild$ sudo -E /Users/abuild/nut/drivers/dummy-ups -a dummy -DDDDDD -B -u root
   0.000000     [D5:70761] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools 2.8.2.1081-1081-g78e0525d3 (development iteration after 2.8.2) - Device simulation and repeater driver 0.19
   0.000102     [D5:70761] do_upsconf_args: confupsname=(null), var=maxretry, val=3
   0.000108     [D5:70761] do_upsconf_args: call do_global_args()
   0.000115     [D3:70761] do_global_args: var='maxretry' val='3'
   0.000125     [D5:70761] do_upsconf_args: confupsname=(null), var=driverpath, val=/Users/abuild/nut/drivers
   0.000129     [D5:70761] do_upsconf_args: call do_global_args()
   0.000134     [D3:70761] do_global_args: var='driverpath' val='/Users/abuild/nut/drivers'
   0.000141     [D5:70761] do_upsconf_args: confupsname=dummy, var=driver, val=dummy-ups
   0.000146     [D5:70761] do_upsconf_args: call main_arg()
   0.000150     [D3:70761] main_arg: var='driver' val='dummy-ups'
   0.000154     [D5:70761] do_upsconf_args: not a main_arg()
   0.000158     [D5:70761] do_upsconf_args: this is a 'driver' setting, may we proceed?
   0.000164     [D6:70761] testval_reloadable: var=driver, oldval=dummy-ups, newval=dummy-ups, reloadable=0, reload_flag=0
   0.000172     [D6:70761] testval_reloadable: verdict for (re)loading var=driver value: -1
   0.000177     [D5:70761] do_upsconf_args: 'driver' setting already applied with this value
   0.000198     [D5:70761] do_upsconf_args: confupsname=dummy, var=desc, val=Crash Dummy
   0.000203     [D5:70761] do_upsconf_args: call main_arg()
   0.000207     [D3:70761] main_arg: var='desc' val='Crash Dummy'
   0.000218     [D5:70761] do_upsconf_args: confupsname=dummy, var=port, val=dummy.seq
   0.000223     [D5:70761] do_upsconf_args: call main_arg()
   0.000227     [D3:70761] main_arg: var='port' val='dummy.seq'
   0.000232     [D6:70761] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=dummy.seq, reloadable=0, reload_flag=0
   0.000237     [D6:70761] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000247     [D5:70761] send_to_all: SETINFO driver.parameter.port "dummy.seq"
   0.000265     [D5:70761] do_upsconf_args: confupsname=UPS1, var=driver, val=dummy-ups
   0.000273     [D5:70761] do_upsconf_args: confupsname=UPS1, var=desc, val=Example event sequence
   0.000280     [D5:70761] do_upsconf_args: confupsname=UPS1, var=port, val=evolution500.seq
   0.000295     [D5:70761] do_upsconf_args: confupsname=UPS2, var=driver, val=dummy-ups
   0.000303     [D5:70761] do_upsconf_args: confupsname=UPS2, var=desc, val=Example ePDU data dump
   0.000314     [D5:70761] do_upsconf_args: confupsname=UPS2, var=port, val=epdu-managed.dev
   0.000321     [D5:70761] do_upsconf_args: confupsname=UPS2, var=mode, val=dummy-once
   0.000339     [D1:70761] Built-in default or configured user for drivers 'nobody' was ignored due to 'root' specified on command line
   0.000351     [D1] Network UPS Tools version 2.8.2.1081-1081-g78e0525d3 (development iteration after 2.8.2) built with Apple clang version 14.0.0 (clang-1400.0.29.202); Target: x86_64-apple-darwin21.6.0; Thread model: posix and configured with flags: --with-snmp-includes='-isystem /usr/local/opt/net-snmp/include -I/usr/local/opt/net-snmp/include -isystem /usr/local/include -I/usr/local/include' --with-snmp-libs='-L/usr/local/opt/net-snmp/lib -lnetsnmp' --with-libltdl-includes='-isystem /usr/local/include -I/usr/local/include' --with-libltdl-libs='-L/usr/local/lib -lltdl' PKG_CONFIG_PATH=/Users/abuild/nut/tmp/lib/pkgconfig:/usr/local/lib/pkgconfig:/usr/local/opt/net-snmp/lib/pkgconfig CFLAGS='-I/Users/abuild/nut/tmp/include  -Wno-poison-system-directories -Wno-deprecated-declarations -isystem /usr/local/include -I/usr/local/include' CPPFLAGS='-I/Users/abuild/nut/tmp/include ' CXXFLAGS='-I/Users/abuild/nut/tmp/include  -Wno-poison-system-directories -isystem /usr/local/include -I/usr/local/include' LDFLAGS='-L/Users/abuild/nut/tmp/lib  -L/usr/local/lib' --enable-keep_nut_report_feature --prefix=/Users/abuild/nut/tmp --sysconfdir=/Users/abuild/nut/tmp/etc/nut --with-udev-dir=/Users/abuild/nut/tmp/etc/udev --with-devd-dir=/Users/abuild/nut/tmp/etc/devd --with-hotplug-dir=/Users/abuild/nut/tmp/etc/hotplug --enable-check-NIT --with-nut_monitor=force --with-pynut=auto --with-nut-scanner=auto --with-nutconf=auto --with-doc=man CC=/usr/local/opt/ccache/libexec/clang CXX=/usr/local/opt/ccache/libexec/clang++ CPP='clang -E' --enable-Wcolor
   0.000394     [D1:70761] debug level is '6'
   0.000406     [D5:70761] send_to_all: SETINFO driver.debug "6"
   0.000416     [D5:70761] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.001908     [D1:70761] Succeeded to become_user(root): now UID=0 GID=0
   0.001937     [D1:70761] Signalling UPS [dummy]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.001948     Can't open /Users/abuild/nut/tests/NIT/tmp/run/dummy-ups-dummy: No such file or directory
   0.001951     [D1:70761] Request for other driver to exit returned code -1
   0.001956     [D1:70761] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
[... cut ...]
   1.016244     [D1:70775] upsdrv_updateinfo...
Abuild-iMac-Pro:nut abuild$    2.018299 [D1:70775] entering parse_data_file()
   2.018320     [D1:70775] leaving (paused)...
   2.018328     [D5:70775] send_to_all: SETINFO driver.state "quiet"
   3.018255     [D5:70775] send_to_all: SETINFO driver.state "updateinfo"
   3.018276     [D1:70775] upsdrv_updateinfo...
````

So the part where your log ends, in mine is followed by the `Startup successful` line and saving the other PID file after it forks to (IIRC) detach from console.

jimklimov, 2024-09-26:
https://stackoverflow.com/questions/31045575/how-to-trace-system-calls-of-a-program-in-mac-os-x has some suggestions about tracing programs (ktrace, dtruss, lowering system protection to be able to trace). Maybe this would expose what the forking in your system blocks on.

The CI farm worker is on Monterey, I think, so maybe something in the platform has changed?..

pjkerly, 2024-09-26:
BTW, I also disable System Integrity Protection ... have the same issue enabled or disabled.

pjkerly, 2024-09-26:
I tried your dummy-ups and it still fails on the fork.  I'll have to look into dtrace.

sudo -E ../bin/dummy-ups -a dummyups -DDDDDD -B -u root
   0.000000	[D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools - Device simulation and repeater driver 0.19 (2.8.2)
   0.000378	[D5] do_upsconf_args: confupsname=(null), var=user, val=root
   0.000383	[D5] do_upsconf_args: call do_global_args()
   0.000385	[D3] do_global_args: var='user' val='root'
   0.000387	[D6] testval_reloadable: var=user, oldval=_nut, newval=root, reloadable=0, reload_flag=0
   0.000389	[D1] testval_reloadable: setting 'user' exists and differs: new value 'root' vs. '_nut'
   0.000390	[D6] testval_reloadable: verdict for (re)loading var=user value: 1
   0.000392	[D1] Overriding previously specified user '_nut' with 'root' specified in global section
   0.000396	[D5] do_upsconf_args: confupsname=CP1500PFCLCD, var=driver, val=usbhid-ups
   0.000398	[D5] do_upsconf_args: confupsname=CP1500PFCLCD, var=port, val=auto
   0.000402	[D5] do_upsconf_args: confupsname=CP1500PFCLCD, var=desc, val=Cyber Power CP1500PFCLCD
   0.000405	[D5] do_upsconf_args: confupsname=dummyups, var=driver, val=dummy-ups
   0.000406	[D5] do_upsconf_args: call main_arg()
   0.000408	[D3] main_arg: var='driver' val='dummy-ups'
   0.000409	[D5] do_upsconf_args: not a main_arg()
   0.000411	[D5] do_upsconf_args: this is a 'driver' setting, may we proceed?
   0.000412	[D6] testval_reloadable: var=driver, oldval=dummy-ups, newval=dummy-ups, reloadable=0, reload_flag=0
   0.000414	[D6] testval_reloadable: verdict for (re)loading var=driver value: -1
   0.000416	[D5] do_upsconf_args: 'driver' setting already applied with this value
   0.000418	[D5] do_upsconf_args: confupsname=dummyups, var=port, val=/dev/null
   0.000419	[D5] do_upsconf_args: call main_arg()
   0.000421	[D3] main_arg: var='port' val='/dev/null'
   0.000422	[D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=/dev/null, reloadable=0, reload_flag=0
   0.000424	[D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000427	[D5] send_to_all: SETINFO driver.parameter.port "/dev/null"
   0.000429	[D5] do_upsconf_args: confupsname=dummyups, var=desc, val=dummy
   0.000430	[D5] do_upsconf_args: call main_arg()
   0.000432	[D3] main_arg: var='desc' val='dummy'
   0.000687	[D1] Built-in default or configured user for drivers 'root' was ignored due to 'root' specified on command line
   0.000689	[D1] Network UPS Tools version 2.8.2 (release/snapshot of 2.8.2) built with Apple clang version 16.0.0 (clang-1600.0.26.3); Target: arm64-apple-darwin24.0.0; Thread model: posix and configured with flags: --prefix=/opt/local --with-serial=no --with-usb=auto --with-snmp=no --with-hal=auto --with-cgi=no --with-neonxml=no --with-ssl=yes --with-ipv6=no --with-statepath=/opt/local/var/db/ups --with-pidpath=/opt/local/var/run --with-hotplug-dir=/opt/local/etc/nut/hotplug --with-udev-dir=/opt/local/etc/nut/udev --with-user=_nut --with-group=_nut
   0.000693	[D1] debug level is '6'
   0.000695	[D5] send_to_all: SETINFO driver.debug "6"
   0.000697	[D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.001941	[D1] Succeeded to become_user(root): now UID=0 GID=0
   0.002171	[D1] Saving PID 3519 into /opt/local/var/db/ups/dummy-ups-dummyups.pid
   0.002237	[D5] send_to_all: SETINFO device.type "ups"
   0.002241	[D5] send_to_all: SETINFO driver.state "init.device"
   0.002244	[D1] Dummy (simulation) mode default (looping infinitely)
   0.002246	[D5] send_to_all: SETINFO driver.parameter.mode "dummy"
   0.002251	[D2] Located /dev/null for device simulation data: /dev/null
   0.002253	[D5] send_to_all: SETINFO driver.state "init.quiet"
   0.002258	[D5] send_to_all: SETINFO driver.version "2.8.2"
   0.002260	[D5] send_to_all: SETINFO driver.version.internal "0.19"
   0.002262	[D5] send_to_all: SETINFO driver.name "dummy-ups"
   0.002264	[D5] send_to_all: SETINFO driver.state "init.info"
   0.002266	[D5] send_to_all: SETINFO ups.mfr "Dummy Manufacturer"
   0.002267	[D5] send_to_all: SETFLAGS ups.mfr RW STRING
   0.002269	[D5] send_to_all: SETAUX ups.mfr 32
   0.002271	[D5] send_to_all: SETINFO ups.model "Dummy UPS"
   0.002272	[D5] send_to_all: SETFLAGS ups.model RW STRING
   0.002274	[D5] send_to_all: SETAUX ups.model 32
   0.002276	[D5] send_to_all: SETINFO ups.status "OL"
   0.002277	[D5] send_to_all: SETFLAGS ups.status RW STRING
   0.002279	[D5] send_to_all: SETAUX ups.status 32
   0.002280	[D1] entering parse_data_file()
   0.002311	[D5] send_to_all: DATAOK
   0.002314	[D5] send_to_all: ADDCMD load.off
   0.002316	[D5] send_to_all: SETINFO driver.state "init.updateinfo"
   0.002317	[D1] upsdrv_updateinfo...
   1.007412	[D1] entering parse_data_file()
   1.007740	[D5] send_to_all: SETINFO driver.state "init.quiet"
   1.009418	[D2] dstate_init: sock /opt/local/var/db/ups/dummy-ups-dummyups open on fd 3
   1.010395	[D1] Group and/or user account for this driver was customized ('root:_nut') compared to built-in defaults. Fixing socket '/opt/local/var/db/ups/dummy-ups-dummyups' ownership/access.
   1.010425	[D1] WARNING: opening socket file for stat/chown failed (102), which is rather typical for Unix socket handling: Operation not supported on socket
   1.010524	[D1] Group access for this driver successfully fixed (using file name based methods)
   1.010539	[D5] send_to_all: SETINFO driver.parameter.pollinterval "2"
   1.010545	[D5] send_to_all: SETINFO driver.parameter.synchronous "auto"
   1.010550	[D5] send_to_all: SETINFO device.mfr "Dummy Manufacturer"
   1.010554	[D5] send_to_all: SETINFO device.model "Dummy UPS"

jimklimov, 2024-09-27:
For dummy-ups setup, I used NUT sources and prepared the prerequisites for MacOS/Homebrew combo per https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests and referenced docs.

Then after building NUT, I just `make check-NIT-sandbox` and it prepares `upsd` and some dummies, and prints where to source an env file with needed variables to run NUT programs from the build area using the mock setup.

Afterwards I killed one of the spawned dummy-ups'es and started one with that config name manually.

pjkerly, 2024-09-27:
Another quick update ...

Using dtrace for fork entry / exit results ... From the results below, it looks like the fork return is actually successful and returns PID of the forked usbhid-ups which matches the PID found in the usbhid-ups-CP1500PFCLCD.PID file.

So I assume the child process is created and then must crash after forking. 

syscall::fork:entry ' matched 10 probes
CPU     ID                    FUNCTION:NAME
  2    173                      fork:return 480 zsh fork returned 3764
  2    173                      fork:return 

  4    173                      fork:return 3764 sudo fork returned 3765
  4    173                      fork:return 

  5    173                      fork:return 3765 usbhid-ups fork returned 3766
  5    173                      fork:return

jimklimov, 2024-09-27:
Thinking of it, your "2.8.2" version of NUT is likely from packaging? Are you in position to build from git per instructions liked above? On one hand, there were some fixes and changes generally since the last release, on another - maybe the tooling/settings/libs used for package build differ from whatever the documented Homebrew-based approach provides.

pjkerly, 2024-09-28:
Further investigation ... 

So it appears that the usbhib-ups driver is crashing on a null pointer (Faulting instruction pointer: 0).  Based on the error info, I assume it is using dynamic function pointer(s) into Core Foundation.  Seeing this in the write sys call.  I assume this is being write to an error or system log.  But I looked in logs and did see any errors before I started debugging.  

But my take is that some functionality in CoreFoundation can be used by a forked process.  Sound likes it needs to an exec() call in order to use that functionality.  I will look to see if there are any limitations using fork and Apple libraries.

The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().

  1    176                      write:entry PID 16949 TID 1526720 syscall write called
Data: SETINFO driver.state "updateinfo"

  1    176                      write:entry PID 16949 TID 1526720 syscall write called
Data: SETINFO driver.state "reconnect.trying"

  1    176                      write:entry PID 16949 TID 1526720 syscall write called
Data: 

  1    176                      write:entry PID 16949 TID 1526720 syscall write called
Data: The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().

  1    176                      write:entry PID 16949 TID 1526720 syscall write called
Data: Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.

  1    176                      write:entry PID 16949 TID 1526720 syscall write called
Data: The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().

  1    176                      write:entry PID 16949 TID 1526720 syscall write called
Data: Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.

  1    176                      write:entry PID 16949 TID 1526720 syscall write called
Data: The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().

  1    176                      write:entry PID 16949 TID 1526720 syscall write called
Data: Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.

  1    176                      write:entry PID 16949 TID 1526720 syscall write called
Data: The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().

  1    176                      write:entry PID 16949 TID 1526720 syscall write called
Data: Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.

  1    176                      write:entry PID 16949 TID 1526720 syscall write called
Data: The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().

  1    176                      write:entry PID 16949 TID 1526720 syscall write called
Data: Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.

  1    176                      write:entry PID 16949 TID 1526720 syscall write called
Data: The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().

  1    176                      write:entry PID 16949 TID 1526720 syscall write called
Data: Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.

  1    176                      write:entry PID 16949 TID 1526720 syscall write called
Data: The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().

  1    176                      write:entry PID 16949 TID 1526720 syscall write called
Data: Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.

  1 395514                          .:fault Fault detected in process: usbhid-ups (PID: 16949, TID: 1526720)
Fault type: 1
Faulting address: fffffe24d9fc5658
Faulting instruction pointer: 0
Timestamp: 143417880310375
Generated signal: 1

              0x1957232dc
              0x104e8e2f8
              0x104e88630
              0x104e08fc8
              0x104e15578
              0x19514c274

  1 395786                           .:exit Process exited: PID=16949, TID=1526720, Executable=usbhid-ups, Exit Status=1

[usbhid-ups.pdf](https://github.com/user-attachments/files/17175367/usbhid-ups.pdf)

pjkerly, 2024-09-28:
Let me know if someone can make these changes.  If not, I'll need to spend some time getting a build/debug environment setup to build NUT for Mac.  I'm not really familiar with the requirements to build/test any changes.

Thanks,
Phil  

From ChatGPT so need to take that into account ... but it sounds reasonable especially since I was getting those messages in my dtrace output.

The error message "The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec()" is typically seen in macOS (or any system using CoreFoundation) when a process uses certain CoreFoundation functions after a fork() but before an exec().

Why This Happens:
Fork-Exec Model: In Unix-like operating systems, when a process forks, the child process is an exact copy of the parent, sharing most of its memory and resources. However, after the fork, the child should typically replace its memory space by calling exec(), which loads a new program into memory.
CoreFoundation's Limitations: CoreFoundation (and related libraries, like Cocoa or AppKit) maintain internal state and rely on various resources that aren’t safely copied or reset in the child process after a fork. Forking a process that uses these libraries and not immediately calling exec() to replace the process image can cause issues, as the child process might access shared or inconsistent states, leading to crashes or undefined behavior.
To prevent these unsafe operations, macOS and CoreFoundation actively block such functionality after a fork() and warn you that you must call exec() immediately after forking, which replaces the child process' memory space with a new executable image and safely resets the state.

Key Points:
fork() without exec(): If you fork and do not follow it immediately with exec(), you risk running into issues with CoreFoundation, which is not fork-safe.
Internal Resources: CoreFoundation maintains internal resources (file descriptors, memory allocators, threads, etc.) that are not properly re-initialized in a child process created via fork().
Thread Safety: CoreFoundation is not thread-safe across fork() boundaries because threads are not copied over to the child process, but their related states might remain in a broken state.
How to Fix This:
You must ensure that after calling fork(), the child process immediately calls exec() to replace its memory space with a new process image. Here's how you can fix the issue:

Call exec() After fork():
When you fork a process, follow it immediately by calling exec() in the child process. The exec() family of functions replaces the current process image with a new one, ensuring a fresh environment and avoiding conflicts with CoreFoundation.
Avoid Using CoreFoundation Functions After fork():
If you cannot use exec() immediately, make sure to avoid any CoreFoundation (or Cocoa/AppKit) function calls in the child process.
Example of Proper Fork-Exec Pattern:

pid_t pid = fork();
if (pid == -1) {
    // Handle fork error
} else if (pid == 0) {
    // This is the child process
    // Replace current process image with a new one
    execvp("/path/to/executable", argv);  // Replace with actual exec call
    // If execvp() fails:
    perror("exec");
    exit(1);
} else {
    // This is the parent process, continue as normal
}

jimklimov, 2024-09-28:
Not really versed with Mac ecosystem, sorry.
CC @clepple comes to mind, if there would be any more ideas or people to poll?

pjkerly, 2024-09-28:
OK ... do you have any references/links on how to setup a build/debug environment for MacPorts and how someone goes about submitting / recommending any changes.  I don't have a full suite of Mac OS versions so I would be limited to testing on a small number of systems / OS.

jimklimov, 2024-09-28:
If ChatGPT is right :) then we might have to add some `#ifdef NUT_PLATFORM_APPLE_OSX` blocks in `drivers/main.c` (and probably all other NUT daemon programs) to detect the path to program binary (its `argv[0]` if nobody fiddled with it, possibly relative to initial value of current working directory), and when we `fork()` to detach from console - re-`exec()` the same program. 

I have no firm idea what it would do to the C variables already collected and expected to be populated post-detachment, but probably nothing good, given that it would cause execution of a new process from scratch, essentially. Some research may be needed to check what exactly is not fork-safe, perhaps it may suffice to close/open the syslog socket, or something.

Given that forking to detach from console and go into background (or other sort of multi-threading purposes) is common practice for decades, probably there's a gazillion other programs that have solved this one way or another, including open-source ones to take inspiration from. I'd probably start at known multi-process stuff like Apache httpd, perhaps nginx or sendmail - staples of the ecosystem.

Sadly, so far I have zero knowledge of MacPorts beside the name; setting up a cloud VM for CI and going with HomeBrew as the first option I saw is as much as I know of the platform.

That said, if you do unravel this - additions to `docs/config-prereqs.txt`, and possibly tweaks into `ci_build.sh` (needed `HOMEBREW_PREFIX` to work around some header and pkg-config issues), `configure.ac` etc. would be most welcome.

If those two ecosystems can co-exist on the same VM - I might add scenarios to NUT CI farm as well then...

jimklimov, 2024-09-28:
Also, if the two ecosystems can co-exist (or if you can spawn a Sonoma VM for experiments), and if you can check if current NUT source built with Homebrew works or not, it can help rule out OS problems (something in common libs shared by both ecosystems) vs. ecosystem problems (e.g. HomeBrew might have libs with an implementation of `fork()` that behaves as expected by other systems, and that can inspire solutions that would help on MacPorts?)

clepple, 2024-09-30:
Why not just run the driver in the foreground under launchd? I thought libusb had some better protections against using CF in the wrong process, but I also haven't been doing much OS X + USB in the past few years.

pjkerly, 2024-09-30:
Exactly what I'm doing now ... but without documentation that I could find it was a process to get to this point to figure why upsdrvctl and usbhid-ups failed.  BTW the forking in uspd and upsmon don't have the same issue.  In this case, I'm using -FF option to force generating the PID so that my LaunchDaemon can shutdown the usbhid-ups driver.

Kraiggers, 2024-10-19:
I ran into this, using the brew package.  (As well some additional issues.) I've been tearing my hair out trying to figure out why I can't get the driver and upsd to run in background. It's almost a relief that there is a bug of some kind. 

Like the OP, I am able to run upsdrvctl, (only in the foreground):

> jks4e@Athena ~ % sudo upsdrvctl -u root -FF start
> Network UPS Tools - UPS driver controller 2.8.2
> Network UPS Tools - Generic HID driver 0.53 (2.8.2)
> USB communication driver (libusb 1.0) 0.47
> Using subdriver: CyberPower HID 0.80
> Running as foreground process, but saving a PID file anyway
> upsnotify: failed to notify about state 2: no notification tech defined, will not spam more about it
> sock_connect: enabling asynchronous mode (auto)

And, I am also able to run upsd:

> jks4e@Athena ~ % sudo upsd -u root
> Network UPS Tools upsd 2.8.2
> fopen /opt/homebrew/var/state/ups/upsd.pid: No such file or directory
> Could not find PID file '/opt/homebrew/var/state/ups/upsd.pid' to see if previous upsd instance is already running!
> /opt/homebrew/etc/nut/upsd.conf is world readable
> listening on 127.0.0.1 port 3493
> /opt/homebrew/var/state/ups is world readable
> Connected to UPS [CyberPower]: usbhid-ups-CyberPower
> Found 1 UPS defined in ups.conf
> /opt/homebrew/etc/nut/upsd.users is world readable

However, when I try to stop upsdrvctl, I do get a sigterm, but I also get this:

> jks4e@Athena ~ % sudo upsdrvctl stop
> Network UPS Tools - UPS driver controller 2.8.2
> fopen /opt/homebrew/var/state/ups/usbhid-ups-CyberPower.pid: No such file or directory

There *should* be a .pid file, shouldn't there?

And I don't know if it's related, but I also am baffled by the permissions. I seem to only be able to start upsdrvctl (or usbhid-ups directly) and upsd successfully if I use sudo *and* -u root.

What am I doing wrong (but almost right)?

Kraiggers, 2024-10-19:
And, by the by, how did you get them to run as a launch agent?  I've been struggling with that too... my .plist seems like it should work, but it's not. Maybe a permissions thing again?

jimklimov, 2024-10-19:
Can't really speak to MacOS as a platform to run things; in the cloud I only have a (Monterey?) VM to build them in CI. It may well be that by Sonoma (v12 vs. v14 major) they changed a lot of things that I might test on that VM and would "work for me" there :\

That said, so now regarding the general NUT code and behavior:
* Note that for `upsdrvctl` the `-D` option(s) apply to logging from the tool itself; an additional `-d` passes the selected level to the driver(s) also. For more details, see https://github.com/networkupstools/nut/wiki/Changing-NUT-daemon-debug-verbosity
* the PID files should be there (unless running with a single `-F` foreground option), at least FS permissions permitting (check who owns `/opt/homebrew/var/state/ups` and if that matches the run-time user which you or the package builder `configure`'d; in debug mode that would be seen around `become_user` log lines).
** It does say `Running as foreground process, but saving a PID file anyway`... Again, debug might show more.
* Permissions should not be a problem when you troubleshoot to run as `root` though.
** You may want to constrain the configuration files to only be readable by e.g. `root:nut` and only writable by owner `root`.

I am not sure if and how Apple limits access to USB devices. On most of the Unix-like platforms, there is eventually a device file system and each device has a file-like "node" there, with POSIX permissions attached (maybe ACL somewhere, haven't seen that). 

Most frequently the problem about starting not as `root` (or dropping privileges to a run-time `nut` or similar user account) is about not having I/O permission to a USB/serial/... port at that low level. OS facilities like `devd`, `udev` and some others usually allow to manage such permissions as part of plug-and-play; persistent devfs or init-scripts and cron jobs calling `chown+chmod` take care of it on others (at least where some path name to the devfs/sysfs node is predictable).

With USB there may be a separate layer of complexity, that some other program or kernel handler grabbed the device node (especially if it poses as a HID - Human Interface Device) exclusively, and so won't let the NUT driver have a piece. This is also part that `udev` and the likes usually solve.

jimklimov, 2024-10-19:
> Let me know if someone can make these changes. If not, I'll need to spend some time getting a build/debug environment setup to build NUT for Mac. I'm not really familiar with the requirements to build/test any changes.

Unfortunately, my practice with the platform is limited to setting up the CI build agent.

That said, I think the experience should be all documented at https://networkupstools.org/docs/user-manual.chunked/_build_prerequisites_to_make_nut_from_scratch_on_various_operating_systems.html#_macos_with_homebrew for the platform setup (with Homebrew as the one I tried - a chapter with Fink etc. would be welcome), and then running `./ci_build.sh` for the common building and testing.

* Looking at it, there may have been a step about installing XCode if it was not part of the OS. In the CI cloud, it was there.
* https://github.com/networkupstools/nut/blob/master/.circleci/config.yml is the complete automation used in the cloud builds.
* You may have to export `HOMEBREW_PREFIX` for `ci_build.sh` to know to apply some hacks for this build, and avoid distrust to system headers, e.g. as
```
HOMEBREW_PREFIX="`brew config | grep HOMEBREW_PREFIX: | awk '{print $2}'`"
```

jimklimov, 2024-10-19:
With the tests prepared, you can `make check-NIT` to run the NUT Integration Tests with `upsd` talking to some `dummy-ups` instances and different scenarios about having and not having configs, as well as testing the Python/C++/etc. client implementations (if enabled).

This would at least rule out where exactly the forking problem is - NUT daemons themselves, or some third party code (e.g. libusb vs. CF as suggested above).

With `make check-NIT-sandbox` you can prepare an environment to run other built NUT programs against these `upsd + dummy-ups` or to edit the generated config files in `tests/NIT/tmp/etc/` and e.g. test a built `drivers/usbhid-ups` program. The sandbox will suggest to source a generated env file with the random `$NUT_PORT` and configuration paths and stuff, into your current shell. I think it would prepare the `LD_LIBRARY_PATH` using the built artifacts, too - not fully sure though.

Note that depending on tooling, your actual libs and programs may be in `.libs` subdirs, e.g. `drivers/.libs/usbhid-ups` and the ones near the sources being GNU libtool wrappers to set up paths and debugging. For tracing or GDB stepping, you need the actual binaries (and manually reproduce the path setup that libtool does).

Kraiggers, 2024-10-19:
So, the command

> sudo upsdrvctl -u root -FF start

*does* start usbhid-ups in the foreground successfully, and it *does* spawn 

> /opt/homebrew/var/state/ups/usbhid-ups-CyberPower.pid

with the correct process ID, owned by system.  And, the foreground process receives a signal 15: exiting.  But I still get 

> jks4e@Athena StartupItems % sudo upsdrvctl stop
> Password:
> Network UPS Tools - UPS driver controller 2.8.2
> fopen /opt/homebrew/var/state/ups/usbhid-ups-CyberPower.pid: No such file or directory

Maybe that's an erroneous error message?  How else could the process be terminated successfully?

pjkerly, 2024-10-19:
I get the same behavior with MacPorts ... I also wondered about this message.  I concluded that it is expected behavior.  I've noticed that I don't get the fopen message until after the upsdrvctl with start has already exited before the upsdrvctl with stop gets the message.  I concluded that the original upsdrvctl that was started once it receives the SIGTERM, it checks for and removes the PID file.  The upsdrvctl that is executed with the stop command is following the same path on exit to clean up the PID file but it has already been removed and not finding it is expected if all goes well.

I didn't pursue it further since it works without any issues.

BTW, I'm using MacPorts daemondo to start and stop both upsdrvctl and upsd in the background on Mac OS X boot and shutdown which works out really well for clean startup and shutdown.

I've uploaded what I'm using to start and stop at OS boot / shutdown.  

[org.macports.upsdrvctl.plist.txt](https://github.com/user-attachments/files/17445362/org.macports.upsdrvctl.plist.txt)
[org.macports.upsd.plist.txt](https://github.com/user-attachments/files/17445363/org.macports.upsd.plist.txt)

[upsdrvctl.wrapper.txt](https://github.com/user-attachments/files/17445357/upsdrvctl.wrapper.txt)
[upsd.wrapper.txt](https://github.com/user-attachments/files/17445358/upsd.wrapper.txt)

jimklimov, 2024-10-19:
> How else could the process be terminated successfully?

If you ran those tools and drivers with debug, it would be a bit less of guesswork. But in recent releases, NUT grew some ways to not depend on PID files (e.g. using the same socket/pipe as the one for driver-to-`upsd` communications, to send an `exit` command).

Conversely, recent versions also try to check the file name associated with a PID, to not signal/kill unsuspecting bystanders that happen to have the same PID (e.g. if a PID-file is written into location which survives a reboot).

jimklimov, 2024-10-19:
Hm, I wonder which program and when complains about `/opt/homebrew/var/state/ups/usbhid-ups-CyberPower.pid` - `upsdrvctl` or the driver? Maybe one deletes it before the other one checks for it?

jimklimov, 2024-10-19:
> I've uploaded what I'm using to start and stop at OS boot / shutdown.

@pjkerly : would you care to make a PR in your name, e.g. adding these files and a README to a new `scripts/macos/macports` location, as a reference for future adventurers?

pjkerly, 2024-10-19:
What's a PR?  I'm not really familiar with MacPorts other than as a user.  I would have attempted to actually fix both upsdrvctl and upsd (at least for the Mac OS fork bug) but I'm not familiar with how to actually setup a build environment, process for getting changes accepted, nor can I test on many different platforms.

pjkerly, 2024-10-19:
OK ... So a PR ... Pull Request.  Let me look into it.  Would you be the reviewer / maintainer to merge in?

jimklimov, 2024-10-19:
Yes. The  "how to set up" is in this thread less than 10 posts above, at least as much as I know about MacOS as seen on workers for CI :)

Kraiggers, 2024-10-19:
OK.  I managed to get launch agents to run successfully, but I had to add a user-level sudo permission for upsd & upsdrvctl in order to do it.  There is something funky that I'm definitely not seeing/understanding/both about the permissions.

This is almost assuredly a dumb question, but how do I force the upsdrvctl agent to run *before* the upsd agent?

Finally, after many days of farting around with this, it appears that the CyberPower oem software service can't run *alongside* the NUT service.  What I wanted, was for the oem software to alert me with its (local) email notification service (is there another way to get this kind of notification?), and then be able to use an app I found (Mac, iOS) that connects to NUT so I can get real-time status information.  Because, of course, the Cyber Power cloud service that does this is both 1) incredibly ugly and 2) behind a paywall.

Ugh.

Kraiggers, 2024-10-19:
FYI

https://apps.apple.com/us/app/ups-power-monitor/id1500563567

https://apps.apple.com/us/app/ups-power-monitor/id1500180529?mt=12

I don't see any specific information regarding notifications...

pjkerly, 2024-10-19:
Order shouldn't matter ... They should sync up in either order.  I used the oem software but haven't since switching.  Have you look at upsmon - can use a command option to send e-mail or (to text)?  Or have you looked at the UPS Power Monitor and Power Guard (from App Store for Mac).  I personally have a Home Assistant setup with an integration for ups ...

Kraiggers, 2024-10-19:
Oh I didn't know upsmon could do that.  

Christ, another NUT service to try to figure out how to run...

pjkerly, 2024-10-19:
several options for command line mail notifications ...

1. Using mail Command
The mail command is a simple tool you can use to send an email. Here's an example:

bash
Copy code
echo "This is the body of the email" | mail -s "Subject of the Email" recipient@example.com
To attach a file, use the -A option:

bash
Copy code
echo "This is the body of the email" | mail -s "Subject of the Email" -A /path/to/file.txt recipient@example.com
You may need to install the mail command if it’s not available:

bash
Copy code
brew install mailutils
2. Using sendmail Command
Another option is sendmail, which provides more flexibility but requires configuration. Here's an example command:

bash
Copy code
echo -e "Subject: Test Email\n\nThis is a test email" | sendmail recipient@example.com
This assumes sendmail is correctly configured on your system.

3. Using s-nail or mailx
You can use s-nail (or mailx), which is similar to mail but has more options and is often available by default on macOS. Here's an example:

bash
Copy code
echo "This is the body" | s-nail -s "Subject of the email" recipient@example.com
For sending through SMTP (if needed, for external services like Gmail), you’ll need to configure s-nail or mailx by editing the ~/.mailrc file:

bash
Copy code
set smtp="smtp://smtp.gmail.com:587"
set smtp-auth=login
set smtp-auth-user="your-email@gmail.com"
set smtp-auth-password="your-password"
set ssl-verify=ignore
set from="your-email@gmail.com"
After configuration, you can send emails using s-nail as shown earlier.

4. Using Python
If you have Python installed, you can use a simple Python script to send an email via the command line:

bash
Copy code
python3 -c "
import smtplib
server = smtplib.SMTP('smtp.gmail.com', 587)
server.starttls()
server.login('your-email@gmail.com', 'your-password')
server.sendmail('from-email@gmail.com', 'to-email@example.com', 'Subject: Test\nThis is the body of the email.')
server.quit()
"
This method gives you flexibility to use Gmail or other SMTP services.

pjkerly, 2024-10-19:
sysmon was easier since it doesn't have the same fork() bug that upsd and upsdrvctl has.  But I still used the daemondo, plist, and wrapper approach just to keep all of them consistent in implementation.

jimklimov, 2024-10-19:
I'll be off for today but it seems you've started a fruitful discussion here, so just adding a few bits :)

> it appears that the CyberPower oem software service can't connect to the UPS concurrently with the NUT service.

On most systems only one program can fully attach to a USB device (or serial port), so yes - either one or another. Part of the issues discussed above with `udev`/`devd`/... part where something in the OS would grab a HID device by default, because why not.

As for `upsmon` - not sure I get the surprise in the discovery. It is the NUT client daemon which drives the safe shutdown of your system in case of outage, which is why you run NUT, right? So I suppose it should be available in the packaging or made to run otherwise. And yes, it can send notifications (syslog, wall, script e.g. for mail)...

pjkerly, 2024-10-19:
I spent many days debugging / root causing the background / fork() issue so I understand.  I use sysmon because I actually have two systems plugged in my UPS.  One is a server which I want to run longer.  The other isn't critical so that if the UPS is running on battery for ~10 minutes, I have it shutdown more quickly to conserve battery.  The server on the other hand runs longer until the battery is down to the last 15% and then it shutdowns a VM cleanly which is running my Home Assistant and shuts down the server cleanly.  I can check the status of the UPS through my Home Assistant remotely.  Notification is actually secondary because if I lose power, it's unlikely I will be able to send any notifications.  I also have my internet router and cable modem on the UPS but the ISP may also be impacted.  But if the internet is still available it  notifies me if running on battery for more than 10 minutes and just before shutdown.  When power is restored I have the UPS send power to the devices only if the power has been restored continuously for more than 10 minutes.

Kraiggers, 2024-10-19:
Nice.  I don't need any of that.  

This started off as just, hmmm I wonder if I can NOT use the Cyber Power dumpster fire OEM software, that also has a subscription (of course it does).

I wan't really intending to get eyeballs deep in a terminal / launch agent project...  But here we are.

Kraiggers, 2024-10-19:
All right.  Well, I've got upsmon going as well, set up to text me on NOTIFY events as per upsmon.conf.  I won't lie it was pretty satisfying.  It's not perfectly clean, as I'm basically sending messages to myself through iMessage, but I think I can live with that.  Better than emails.  Yuck.

AND I even got the UPS Power Monitor software working on my phone (bounced through my DDNS).  So that's pretty neat.  

I'm not super pumped about running the three processes via sudo, but I don't see what harm it could cause...?

Kraiggers, 2024-10-19:
[image]
[image]

jimklimov, 2024-10-19:
Run-time estimates depend on calibration results that should be done with same sort of load as you're running, and repeated (becaude batteries do age). If the device supports such a test, you may be able to `upscmd` the `battery.test.start` (beware that if the UPS does overestimate itself, load like computers may be suddenly powered off). If not, with some drivers there's a `runtimecal` setting with estimates you measure and record there (see man pages).

pjkerly, 2024-10-20:
It's not unusual for a UPS to drop estimated runtime soon after it switches to battery particularly for an older battery as it ages.  You have to start with what are your requirements / needs.  My requirements are simple... 1) A must - Prevent system corruptions from catastrophic and instantaneous power failures.  2) A must / highly desirable -  bridge power interruptions / flickers that last a few seconds to tens of minutes.  If the power interruption last more than ~10 minutes, it's likely to be longer than my UPS will support any way.  3). Nice to have -  Maintain internet connectivity for as long as possible for use with laptop / iPad / iPhones with no specific time-frame.     

If your projected run time drops from 47 minutes to 20 minutes, then I wouldn't call the battery toast.  After the initial drop, do the rate of runtime seem constant with wall clock time?  A shutdown sequence should be on the order of 60 secs or so depending on what you are running.  And 20 minutes is definitely enough time to bridge power flickers/fluctuations during storms as well as support clean shutdown.  But your requirements on how long you want to run at your desired load will dictate the size and performance of the UPS you needs.

I'm not familiar with the battery you listed nor whether it is compatible with your UPS.  I have a a CyberPower CP1500PFCLCD UPS from Costco  which meets my needs.  I can find replacement batteries on Amazon for it but haven't done so.  My system is also probably ~4 yrs old.  I also see a good drop in initial runtime right after switching to battery.

clepple, 2024-10-20:
> Do you have any experience or opinions about putting a LifePO4 batter in place of a Pb Acid?

from the product page: "LiFePO4 charger recommended" which I'm guessing doesn't include most consumer Pb-acid UPSes.

6 years is beyond the usual 3-5 year lifetime for an UPS battery. Maybe 20 minutes is still sufficient for your purposes, but it's like driving on old tires: they're not going to get any better over time, and you've hit the wear indicator.

fmnamado, 2025-04-08:
> <img alt="Screenshot 2024-10-19 at 14 30 26" width="356" src="https://private-user-images.githubusercontent.com/16720677/378113951-e2f81a4a-ed85-4eb6-84ce-c862d208597f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDQxMjQ2MjcsIm5iZiI6MTc0NDEyNDMyNywicGF0aCI6Ii8xNjcyMDY3Ny8zNzgxMTM5NTEtZTJmODFhNGEtZWQ4NS00ZWI2LTg0Y2UtYzg2MmQyMDg1OTdmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDA4VDE0NTg0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM4NjkzMWZkYjBlYWI3ZGMzNzc1ZWI0ZGM3YTViYzZhNTI4MDAyYWZiNDI4NjA4NDc5MTA2OWQ5NzFmNTY1MjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.YYXgUKeQeyMhH78EOiiiNuL6R7TwiC0K9I4niXAcdqg"> <img alt="Screenshot 2024-10-19 at 14 30 58" width="535" src="https://private-user-images.githubusercontent.com/16720677/378113953-100d4c0c-8f63-44b2-8d3c-d47cb0ab926f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDQxMjQ2MjcsIm5iZiI6MTc0NDEyNDMyNywicGF0aCI6Ii8xNjcyMDY3Ny8zNzgxMTM5NTMtMTAwZDRjMGMtOGY2My00NGIyLThkM2MtZDQ3Y2IwYWI5MjZmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDA4VDE0NTg0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc2YzE0OWM0YTQyNzE3NDBmMGE0MDU5ZDdkMjNiN2RmZTU0MTc2MjcyODQ2OTBmYjVhNThhYmFjNTBkNWI5MDkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.P4_zCFQdeTSo4HXd6pLfs_DKnRmR2WcUnoVKCfQC_98">

Hello,
Did you manager to shutdown your mac using this app in conjunction with powerguard?
Because I tried to use nut ups, installed by brew follwing techno tim.
My UPS, Cyberpower CP550EPFCLCD is connected to my Synology NAS DS720+
In may mac, configured as a nut client, I get the messege "executing automatic power-fail shutdown" but it never shuts down...

With that app, did you manage to shutdown the mac or even with the app you don't get it to shutdown?

Thank you.

pjkerly, 2025-04-11:
I did not use Power Guard.  I use upsmon from the NUT package.  I needed to setup/config ups and upsd anyway.  Then I used Mac's native LaunchDaemon launchctl that then uses the macports daemondo and a wrapper to start and stop upsmon on Mac OS X boot up and shutdown.  In upsmon.conf, I use SHUTDOWNCMD "/sbin/shutdown -h +5m" which schedules the shutdown gracefully.

fmnamado, 2025-04-11:
Can you please share your conf files for Nut and lanchctl?
With the least information redacted possible?

Because i get to the message in terminal “mac is going to shutdown” and it never does.

I tried everything and several times, including installing VMs to start over, etc.

You help would be very appreciated, thank
You.