networkupstools/nut #2644: nut_libusb_get_interrupt: Connection timed out on debug

masterwishx, 2024-09-27:
Using NUT plugin for Unraid
When debug is on have many interrupt transfers :  

have a lot of `nut_libusb_get_interrupt: Connection timed out` in debug log and 
maybe also affect on https://github.com/networkupstools/nut/issues/2643
or  its just closing connection every time for cycle of reading data?

From https://github.com/libusb/libusb/issues/1548#issuecomment-2378734313 by @tormodvolden (have issues with interrupt transfers **)**

Mostly looks like this :  
```
send_to_all: SETINFO driver.state "quiet"
Sep 27 09:37:30 DaRKNaS usbhid-ups[24218]: [D6:24218] send_to_all: write 29 bytes to socket 16 succeeded (ret=29): SETINFO driver.state "quiet"
Sep 27 09:37:31 DaRKNaS usbhid-ups[24218]: [D5:24218] send_to_all: SETINFO driver.state "updateinfo"
Sep 27 09:37:31 DaRKNaS usbhid-ups[24218]: [D6:24218] send_to_all: write 34 bytes to socket 16 succeeded (ret=34): SETINFO driver.state "updateinfo"
Sep 27 09:37:31 DaRKNaS usbhid-ups[24218]: [D1:24218] upsdrv_updateinfo...
Sep 27 09:37:32 DaRKNaS usbhid-ups[24218]: [D2:24218] nut_libusb_get_interrupt: Connection timed out
Sep 27 09:37:32 DaRKNaS usbhid-ups[24218]: [D1:24218] Got 0 HID objects...
Sep 27 09:37:32 DaRKNaS usbhid-ups[24218]: [D1:24218] Quick update...
Sep 27 09:37:32 DaRKNaS usbhid-ups[24218]: [D4:24218] 
```

[image]

masterwishx, 2024-09-30:
@tormodvolden @jimklimov do you think it's related to how nut proccess libusb or I can test it with xusb if it related to libusb itself?

I have also nut client on windows that read values from nut, also telegraf that read values and write to influxdbv2 for grafana dashboard.
Not shure if it can affect on isshue...

tormodvolden, 2024-09-30:
Is this causing a problem or is it just that you see the debug message in the log? You'll have to find out what the nut code tries to do, then what to expect from the device and libusb calls.

If you have suspicions about a libusb issue, you must provide LIBUSB_DEBUG=4 logs to start.

xusb can do some HID interrupt transfers, but I don't know if it fits with your device.

masterwishx, 2024-10-01:
> Is this causing a problem or is it just that you see the debug message in the log? You'll have to find out what the nut code tries to do, then what to expect from the device and libusb calls.

Yes the problem upsrw commands not working for me , hovewer it write OK or Success https://github.com/networkupstools/nut/issues/2643

masterwishx, 2024-10-01:
> Is this causing a problem or is it just that you see the debug message in the log? You'll have to find out what the nut code tries to do, then what to expect from the device and libusb calls.
>
> If [...]

[image]

tormodvolden, 2024-10-01:
Personally I am allergic to text being pasted as images :) There are so many great tools to efficiently search and process textual information.

masterwishx, 2024-10-01:
Sorry, got it, will post the log here..

masterwishx, 2024-10-01:
> Personally I am allergic to text being pasted as images :) There are so many great tools to efficiently search and process textual information.

[nut debug log 1.10.24.txt](https://github.com/user-attachments/files/17202676/nut.debug.log.1.10.24.txt)

masterwishx, 2024-10-01:
Also tryed to add `exported LIBUSB_DEBUG = 4`  to ups.conf with `debug_min = 6` but seems no effect with  `LIBUSB_DEBUG = 4`

tormodvolden, 2024-10-01:
Please see #2616.

masterwishx, 2024-10-01:
> Please see #2616.

Thanks, by #2616
I tryed to add LIBUSB_DEBUG = 4, but seems not worked for, I mean only debug_min=6 working by log I posted

tormodvolden, 2024-10-01:
Maybe bring that up in #2616, if you can't figure out how to enable this, the documentation is definitely not good enough.

tormodvolden, 2024-10-01:
Did you already read docs/man/nut.conf.txt ?

masterwishx, 2024-10-01:
> Did you already read docs/man/nut.conf.txt ?

No just was reading #2616

masterwishx, 2024-10-01:
> exported LIBUSB_DEBUG = 4

seems `exported LIBUSB_DEBUG = 4` not documented yet in docs/man/nut.conf.txt
im using nut plugin for Unraid ,
So i just select UPS Driver Debug Level: 6 and its adding debug_min = 6 in ups.cong file

tormodvolden, 2024-10-01:
The LIBUSB_DEBUG variable is mentioned in docs/man/nut_usb_addvars.txt but docs/man/nut.conf.txt explains more in general how to set variables correctly.

tormodvolden, 2024-10-01:
What does docs/man/nut.conf.txt say about spaces? :)

masterwishx, 2024-10-01:
> What does docs/man/nut.conf.txt say about spaces? :)

:) OK Got it, i will try it 
but in nut plugin for Unraid that im using all config with spaces around equal , sorry for picture:

[image]

[image]

mb-grinn, 2024-11-06:
Hi,

I'm experiencing the similar issue: after a certain period, the UPS (APC Back-UPS BX750MI) stops responding without any indication of stale data.

```
Nov 04 09:52:55 taylor usbhid-ups[16704]: [D2] nut_libusb_get_interrupt: Connection timed out
Nov 04 09:52:55 taylor usbhid-ups[16704]: [D1] Got 0 HID objects...
Nov 04 09:52:55 taylor usbhid-ups[16704]: [D1] Quick update...
```

Under normal operation, the HIDGetEvents function returns a positive non-zero value (e.g., 1). However, when the data appears stale, HIDGetEvents returns 0.

```
/* On success, returns an item count > 0. When no notifications are available,
 * returns an error or 'no event' code.
 */
int HIDGetEvents(hid_dev_handle_t udev, HIDData_t **event, int eventsize);
```

The documentation specifies that, on success, the return value should be greater than 0. Perhaps we should interpret a return value of 0 as an error and attempt to reconnect in that case?

I've tested this approach with the attached patch, and it has worked without any data staleness for over two days.

[0001-apc-fix.patch.txt](https://github.com/user-attachments/files/17646570/0001-apc-fix.patch.txt)

michal4132, 2024-11-06:
Hi,

I'm experiencing the similar issue: after a certain period, the UPS (APC Back-UPS BX750MI) stops responding without any indication of stale data.

```
Nov 04 09:52:55 taylor usbhid-ups[16704]: [D2] nut_libusb_get_interrupt: Connection timed out
Nov 04 09:52:55 taylor usbhid-ups[16704]: [D1] Got 0 HID objects...
Nov 04 09:52:55 taylor usbhid-ups[16704]: [D1] Quick update...
```

Under normal operation, the HIDGetEvents function returns a positive non-zero value (e.g., 1). However, when the data appears stale, HIDGetEvents returns 0.

```
/* On success, returns an item count > 0. When no notifications are available,
 * returns an error or 'no event' code.
 */
int HIDGetEvents(hid_dev_handle_t udev, HIDData_t **event, int eventsize);
```

The documentation specifies that, on success, the return value should be greater than 0. Perhaps we should interpret a return value of 0 as an error and attempt to reconnect in that case?

I've tested this approach with the attached patch, and it has worked without any data staleness for over two days.

[0001-apc-fix.patch.txt](https://github.com/user-attachments/files/17646606/0001-apc-fix.patch.txt)

jimklimov, 2024-11-06:
Looks reasonable; would you care to make it a pull request in your name?

Also, with an APC BX###MI device, https://github.com/networkupstools/nut/pull/2565 and related issues and PRs may be relevant.

In your case, before the fix, did it begin reporting "0 HID objects" and then continued to do so indefinitely, or recovered after some time (~10-30 sec)?

michal4132, 2024-11-06:
Sure, I'll create a PR later.

I’ve looked through other issues, including #2565, and can confirm that the LB+RB bug is still present even with this fix.

Before applying the patch, the UPS would remain stuck at "0 HID objects" indefinitely (at least for several hours) and wouldn’t recover without a driver restart.

masterwishx, 2024-11-06:
> I'm experiencing similar issue: after a certain period, the UPS (APC Back-UPS BX750MI) stops responding without any indication of stale data.

In my case ups not stop responding but seems some upsrw commands have no effect, also seems I'm not mentioned that I have ups Eaton 9E2000i model.

masterwishx, 2024-11-16:
with LIBUSB_DEBUG = 4 i can see next on timeouts

@tormodvolden 
can you please check if you see some issues here : 

```
Nov 16 14:42:07 server rc.nut:    7.463963#011[D1:11606] upsdrv_updateinfo...
Nov 16 14:42:07 server rc.nut: [ 6.649112] [00002d56] libusb: debug [libusb_submit_transfer] transfer 0x480c40
Nov 16 14:42:07 server rc.nut: [ 6.649113] [00002d56] libusb: debug [add_to_flying_list] arm timer for timeout in 750ms (first in line)
Nov 16 14:42:07 server rc.nut: [ 6.649118] [00002d56] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 512
Nov 16 14:42:07 server rc.nut: [ 6.649128] [00002d56] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
Nov 16 14:42:07 server rc.nut: [ 6.649130] [00002d56] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
Nov 16 14:42:08 server rc.nut: [ 7.399124] [00002d56] libusb: debug [usbi_wait_for_events] poll() returned 1
Nov 16 14:42:08 server rc.nut: [ 7.399135] [00002d56] libusb: debug [libusb_cancel_transfer] transfer 0x480c40
Nov 16 14:42:08 server rc.nut: [ 7.399481] [00002d56] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
Nov 16 14:42:08 server rc.nut: [ 7.399489] [00002d56] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
Nov 16 14:42:08 server rc.nut: [ 7.399491] [00002d56] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
Nov 16 14:42:08 server rc.nut: [ 7.399494] [00002d56] libusb: debug [usbi_wait_for_events] poll() returned 1
Nov 16 14:42:08 server rc.nut: [ 7.399498] [00002d56] libusb: debug [reap_for_handle] urb type=1 status=-2 transferred=0
Nov 16 14:42:08 server rc.nut: [ 7.399500] [00002d56] libusb: debug [handle_bulk_completion] handling completion status -2 of bulk urb 1/1
Nov 16 14:42:08 server rc.nut: [ 7.399502] [00002d56] libusb: debug [handle_bulk_completion] abnormal reap: urb status -2
Nov 16 14:42:08 server rc.nut: [ 7.399503] [00002d56] libusb: debug [handle_bulk_completion] abnormal reap: last URB handled, reporting
Nov 16 14:42:08 server rc.nut: [ 7.399505] [00002d56] libusb: debug [usbi_handle_transfer_cancellation] detected timeout cancellation
Nov 16 14:42:08 server rc.nut: [ 7.399506] [00002d56] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
Nov 16 14:42:08 server rc.nut: [ 7.399508] [00002d56] libusb: debug [usbi_handle_transfer_completion] transfer 0x480c40 has callback 0x14aae8475630
Nov 16 14:42:08 server rc.nut: [ 7.399510] [00002d56] libusb: debug [sync_transfer_cb] actual_length=0
Nov 16 14:42:08 server rc.nut: [ 7.399512] [00002d56] libusb: debug [libusb_free_transfer] transfer 0x480c40
Nov 16 14:42:08 server rc.nut:    8.214372#011[D2:11606] nut_libusb_get_interrupt: Connection timed out
Nov 16 14:42:08 server rc.nut:    8.214378#011[D1:11606] Got 0 HID objects...
Nov 16 14:42:08 server rc.nut:    8.214381#011[D1:11606] Quick update...

```

tormodvolden, 2024-11-16:
I see a bulk transfer that times out, with the timeout set to 750ms. Have you tried a longer timeout?
What is the request? Is it something that the device should always handle?

tormodvolden, 2024-11-16:
Note that this is bulk transfer. Is it a bulk endpoint? I was confused by the talk about "interrupt transfers" in the other ticket. What is nut_libusb_get_interrupt() supposed to do?

masterwishx, 2024-11-16:
> Note that this is bulk transfer. Is it a bulk endpoint? I was confused by the talk about "interrupt transfers" in the other ticket. What is nut_libusb_get_interrupt() supposed to do?

It's for Eaton 9E 2000i ups that you fixed buggy firmware with wrong language ID.
Maybe @jimklimov can help here?

masterwishx, 2024-11-16:
```
Nov 16 11:47:33 server rc.nut:    0.961876#011[D1:31179] Checking device 10 of 11 (0463/FFFF)
Nov 16 11:47:33 server rc.nut:    0.961876#011[D1:31179] nut_usb_get_string: Broken language identifier, assuming en_US
Nov 16 11:47:33 server rc.nut:    1.016918#011[D1:31179] nut_usb_get_string: Broken language identifier, assuming en_US
Nov 16 11:47:33 server rc.nut:    1.082520#011[D1:31179] nut_usb_get_string: Broken language identifier, assuming en_US
Nov 16 11:47:33 server rc.nut:    1.131127#011[D2:31179] - VendorID: 0463
Nov 16 11:47:33 server rc.nut:    1.131148#011[D2:31179] - ProductID: ffff
Nov 16 11:47:33 server rc.nut:    1.131159#011[D2:31179] - Manufacturer: EATON
Nov 16 11:47:33 server rc.nut:    1.131161#011[D2:31179] - Product: Eaton 9E
Nov 16 11:47:33 server rc.nut:    1.131163#011[D2:31179] - Serial Number: reducted
Nov 16 11:47:33 server rc.nut:    1.131165#011[D2:31179] - Bus: 003
Nov 16 11:47:33 server rc.nut:    1.131166#011[D2:31179] - Bus Port: 003
Nov 16 11:47:33 server rc.nut:    1.131168#011[D2:31179] - Device: 002
Nov 16 11:47:33 server rc.nut:    1.131170#011[D2:31179] - Device release number: 0202
Nov 16 11:47:33 server rc.nut:    1.131173#011[D2:31179] Trying to match device
Nov 16 11:47:33 server rc.nut:    1.131176#011[D2:31179] match_function_subdriver (non-SHUT mode): matching a device...
Nov 16 11:47:33 server rc.nut:    1.131179#011[D3:31179] match_function_regex: matching a device...
Nov 16 11:47:33 server rc.nut:    1.131182#011[D2:31179] Device matches
Nov 16 11:47:33 server rc.nut:    1.131184#011[D2:31179] Reading configuration descriptor 1 of 1
Nov 16 11:47:33 server rc.nut:    1.131193#011[D3:31179] libusb_kernel_driver_active() returned 0: Success
Nov 16 11:47:33 server rc.nut:    1.131209#011[D2:31179] Claimed interface 0 successfully
Nov 16 11:47:33 server rc.nut:    1.131212#011[D3:31179] nut_libusb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
Nov 16 11:47:33 server rc.nut:    1.162016#011[D2:31179] Retrieved HID descriptor (expected 9, got 9)
Nov 16 11:47:33 server rc.nut:    1.162038#011[D3:31179] HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 05 08
Nov 16 11:47:33 server rc.nut:    1.162040#011[D3:31179] HID descriptor length (method 1) 2053
Nov 16 11:47:33 server rc.nut:    1.162042#011[D4:31179] i=0, extra[i]=09, extra[i+1]=21
Nov 16 11:47:33 server rc.nut:    1.162045#011[D3:31179] HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 f6 03
Nov 16 11:47:33 server rc.nut:    1.162049#011[D3:31179] HID descriptor length (method 2) 1014
Nov 16 11:47:33 server rc.nut:    1.162052#011[D1:31179] Eaton device v2.02. Using full report descriptor
Nov 16 11:47:33 server rc.nut:    1.162054#011[D2:31179] Warning: two different HID descriptors retrieved (Reportlen = 2053 vs. 1014)
Nov 16 11:47:33 server rc.nut:    1.162056#011[D2:31179] HID descriptor length 2053
Nov 16 11:47:37 server rc.nut:    4.427419#011[D2:31179] Report Descriptor size = 2053

```

masterwishx, 2024-11-16:
found only :  

https://github.com/networkupstools/nut/blob/0f6ae03925b139f64e8f45576dd9d5623cd1fd34/drivers/usbhid-ups.c#L1120C1-L1124C1

@jimklimov can it be related to `pollonly` ?

jimklimov, 2024-11-16:
Not really sure, it was before my time and USB code is still not something I am able to navigate well. But just tracing the method/variable names gets us quite far along:

* https://github.com/networkupstools/nut/blob/0f6ae03925b139f64e8f45576dd9d5623cd1fd34/drivers/libusb0.c#L920-L950 called `usb_interrupt_read()`
* https://github.com/networkupstools/nut/blob/0f6ae03925b139f64e8f45576dd9d5623cd1fd34/drivers/libusb1.c#L1032-L1097 calls `libusb_interrupt_transfer()`
* https://github.com/networkupstools/nut/blob/0f6ae03925b139f64e8f45576dd9d5623cd1fd34/drivers/usbhid-ups.c#L1271-L1274 Here the `pollonly` setting is translated `use_interrupt_pipe` boolean in C code, which goes to call or not `HIDGetEvents()` in `upsdrv_updateinfo()` at https://github.com/networkupstools/nut/blob/0f6ae03925b139f64e8f45576dd9d5623cd1fd34/drivers/usbhid-ups.c#L1144-L1187
* My understanding that the alternative to this is to just rely on calling `hid_ups_walk()` regularly, lower in the method.

masterwishx, 2024-11-16:
> Not really sure, it was before my time and USB code is still not something I am able to navigate well. But just tracing the method/variable names gets us quite far along:
 
Thanks for hint , take me some time to get on the code :(

> Note that this is bulk transfer. Is it a bulk endpoint? I was confused by the talk about "interrupt transfers" in the other ticket. What is nut_libusb_get_interrupt() supposed to do?

Maybe you can help here also to questions from @tormodvolden above ?

jimklimov, 2024-11-16:
Can't really elaborate on that authoritatively, but my understanding is that we can get device data from regular polling, which is probably "expensive" (walking all data points known to be served by the device, more so for a "Full update"), so done infrequently (`pollfreq` ~ 5-30 sec). The data loop in `main.c`

jimklimov, 2024-11-16:
Can't really elaborate on that authoritatively, but my understanding is that we can get device data from regular polling, which is probably "expensive" (walking all data points known to be served by the device, more so for a "Full update"), so done infrequently (`pollfreq` in some drivers ~ 5-30 sec and `poll_interval` in all via the data loop in `main.c` may both have an impact on timing).

So where interrupts are a possibility (and do not freeze the controller firmware, hence the toggles) they allow us to see individual events more frequently as they happen.

But that's "handwavium" - I can only hope the inherited code actually does that :)

masterwishx, 2024-11-16:
> But that's "handwavium" - I can only hope the inherited code actually does that :)

Thanks for explaining, just wanted to mention again that I dont have a big problem with these `connection timeout and 0 hid objects`, only suspect that some commands and r/w values not working sometime becose of it :(

tormodvolden, 2024-11-17:
About my question, I was doubly confused. Internally, libusb wraps interrupt transfers in the bulk transfer code path, so it is perfectly normal to see "bulk transfer" in the log for interrupt transfers.

A question is how often does the device "post" a new value on its interrupt endpoint? The polling interval is defined by the device, in its descriptors. If you are polling faster than that, I can imagine it is normal to have some timeouts, but I am not sure what exactly you will see.

Please see https://libusb.sourceforge.io/api-1.0/group__libusb__syncio.html#ga0f171a72904a552fc43e6e6564d108a3 about the `transferred` parameter and https://libusb.sourceforge.io/api-1.0/group__libusb__asyncio.html#asynctimeout which also explains that timeouts do not always mean failure, it can be a partial data transfer. So you should check for any transferred data in this case.

masterwishx, 2024-11-17:
> So you should check for any transferred data in this case.

Thanks i will check, I can see in debug log after upsdrv updateinfo and nut_libusb_get_interrupt: connection timed out, Got 0 HID object
and libusb_get_report all paths from NUT.

Also when send by r/w value or command I got OK or Success but need to check again  if it really worked and how we got OK/success in this case.

masterwishx, 2024-11-30:
just for the info by #2685 user log for Eaton 9SX have same : `nut_libusb_get_interrupt: connection timed out, Got 0 HID object` 
but not have issues with commands ...

masterwishx, 2024-12-08:
> just for the info by #2685 user log for Eaton 9SX have same : `nut_libusb_get_interrupt: connection timed out, Got 0 HID object` but not have issues with commands ...

Also in log I see `PONG` that I don't have, maybe related to OS.
@jimklimov not sure how to check if data received how @tormodvolden was said to check...

jimklimov, 2024-12-08:
I suppose "PONG" sighting can be related to having `upsd` running and the debug level involved: the data server occasionally `PING`'s the drivers, if there was otherwise no data from them coming in recently, and the driver is expected to reply `PONG` or is considered frozen => `Data stale` situation.

masterwishx, 2024-12-08:
> I suppose "PONG" sighting can be related to having `upsd` running and the debug level involved: the data server occasionally `PING`'s the drivers, if there was otherwise no data from them coming in [...]

That is very strange because i dont have any `PING` or `PONG` in my debug logs :(