networkupstools/nut #2645: DEXP CEE-E1500VA

big-vl, 2024-09-29:
I decided to try to raise nut for the subject, but alas it doesn’t work, and the application was written for Windows in 2018.

Could you maybe advance me a little?

```
server@debian:/etc/nut$ nut-scanner
Neon library not found. XML search disabled.
IPMI library not found. IPMI search disabled.
Scanning USB bus.
No start IP, skipping SNMP
No start IP, skipping NUT bus (old connect method)
[nutdev1]
	driver = "nutdrv_atcl_usb"
	port = "auto"
	vendorid = "0001"
	productid = "0000"
	bus = "003"
server@debian:/etc/nut$ sudo nano ups.conf
server@debian:/etc/nut$ sudo /lib/nut/nutdrv_atcl_usb -a ups -u root
Network UPS Tools - 'ATCL FOR UPS' USB driver 1.1 (2.7.4)
Warning: This is an experimental driver.
Some features may not function correctly.

Communications with UPS lost: Can't open USB device, retrying ...
Communications with UPS lost: Can't open USB device, retrying ...
Communications with UPS lost: Can't open USB device, retrying ...
Unable to find ATCL FOR UPS

Things to try:
 - Connect UPS device to USB bus
 - Run this driver as another user (upsdrvctl -u or 'user=...' in ups.conf).
   See upsdrvctl(8) and ups.conf(5).

Fatal error: unusable configuration
```

jimklimov, 2024-09-29:
Unfortunately, this device has a useless identification so it is a guesswork at what driver might work. 

In NUT HCL there's an entry for a DEXP though:

`data/driver.list.in:"DEXP"      "ups"   "2"     "MIX 850VA"     "USB"   "blazer_usb langid_fix=0x0409 runtimecal=240,100,720,50 default.battery.voltage.high=2.27 default.battery.voltage.low=1.72"     # https://github.com/networkupstools/nut/issues/721`

So you can try blazer_usb or nutdrv_qx drivers.

Also, NUT v2.7.4 is old, so see if you can get a newer package or build from source (see wiki).

big-vl, 2024-09-29:
```
server@debian:/usr/local/ups/bin$ ls /dev | grep hid
hidraw0
hidraw1
hidraw2
uhid

```

hidraw2  is the ups itself, after launching blazer_usb the device disappears.
```

sudo ./blazer_usb -DDD -a dexp
Network UPS Tools 2.8.2.1082-1082-g1560afe33 (development iteration after 2.8.2) - Megatec/Q1 protocol USB driver 0.20
   0.000001	[D1] Using USB implementation: libusb-0.1 (or compat)
   0.000138	[D3] main_arg: var='driver' val='blazer_usb'
   0.000181	[D3] main_arg: var='port' val='auto'
   0.000223	[D3] main_arg: var='desc' val='DEXP IEC LCD ONLINE 3000VA'
   0.000268	[D3] main_arg: var='vendorid' val='0001'
   0.000317	[D3] main_arg: var='productid' val='0000'
   0.000376	[D1] Network UPS Tools version 2.8.2.1082-1082-g1560afe33 (development iteration after 2.8.2) built with gcc (Debian 10.2.1-6) 10.2.1 20210110 and configured with flags: --with-user=ups --with-group=nut --with-usb --with-cgi
   0.000418	[D1] debug level is '3'
   0.002368	[D1] Succeeded to become_user(ups): now UID=1001 GID=1001
   0.002451	[D1] Signalling UPS [dexp]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.002535	Can't open /var/state/ups/blazer_usb-dexp: No such file or directory
   0.002574	[D1] Request for other driver to exit returned code -1
   0.002612	[D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
   0.264276	[D3] usb_busses=0x56004fb82ae0
   0.264315	[D2] Checking device (1D6B/0003) (004/001)
   0.290637	[D1] nut_usb_get_string_descriptor: string descriptor 0 request failed, retrying...
   0.340758	[D1] nut_usb_get_string_descriptor: string descriptor 0 request failed, retrying...
   0.390877	[D1] nut_libusb_open: get Manufacturer string failed
   0.390907	[D1] nut_usb_get_string_descriptor: string descriptor 0 request failed, retrying...
   0.441037	[D1] nut_usb_get_string_descriptor: string descriptor 0 request failed, retrying...
   0.492546	[D1] nut_libusb_open: get Product string failed
   0.492744	[D1] nut_usb_get_string_descriptor: string descriptor 0 request failed, retrying...
   0.542943	[D1] nut_usb_get_string_descriptor: string descriptor 0 request failed, retrying...
   0.593086	[D1] nut_libusb_open: get Serial Number string failed
   0.593302	[D2] - VendorID: 1d6b
   0.594768	[D2] - ProductID: 0003
   0.594817	[D2] - Manufacturer: unknown
   0.594855	[D2] - Product: unknown
   0.594954	[D2] - Serial Number: unknown
   0.594993	[D2] - Bus: 004
   0.595081	[D2] - Device: 001
   0.595126	[D2] - Device release number: 0510
   0.595218	[D2] Trying to match device
   0.595258	[D3] match_function_regex: matching a device...
   0.595371	[D2] match_function_regex: failed match of VendorID: 1d6b
   0.595468	[D2] Device does not match - skipping
   0.595682	[D2] Checking device (0001/0000) (003/007)
   0.597278	[D1] nut_usb_get_string: Broken language identifier, assuming en_US
   0.599991	[D2] - VendorID: 0001
   0.600225	[D2] - ProductID: 0000
   0.600264	[D2] - Manufacturer: unknown
   0.600342	[D2] - Product: MEC0003
   0.600381	[D2] - Serial Number: unknown
   0.600416	[D2] - Bus: 003
   0.600451	[D2] - Device: 007
   0.600487	[D2] - Device release number: 0100
   0.600523	[D2] Trying to match device
   0.600567	[D3] match_function_regex: matching a device...
   0.600658	[D2] Device matches
   0.600724	[D2] failed to claim USB device: could not claim interface 0: Device or resource busy
   0.601637	[D2] detached kernel driver from USB device...
   0.601728	[D3] nut_libusb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.601788	Please note that this driver is deprecated and will not receive
new development. If it works for managing your devices - fine,
but if you are running it to try setting up a new device, please
consider the newer nutdrv_qx instead, which should handle all 'Qx'
[... 433 lines cut ...]
   7.849028	[D1] upsdrv_updateinfo...
   7.849065	[D1] Quick update...
   7.849116	[D3] send: Q1
   7.849156	[D3] asking for: 0D
   7.852012	[D3] read: (243.0 000.0 210.0 016 49.9 26.9 29.0 00101000
   9.849278	[D1] upsdrv_updateinfo...
   9.849336	[D1] Quick update...
   9.849353	[D3] send: Q1
   9.850815	[D3] read: (243.0 000.0 210.0 016 49.9 26.9 29.0 00101000
  11.851337	[D1] upsdrv_updateinfo...
  11.851400	[D1] Quick update...
  11.851463	[D3] send: Q1
  11.851502	[D3] asking for: 0C
  11.861267	[D3] read: (243.0 000.0 210.0 016 50.0 27.0 29.0 00101000
  13.853265	[D1] upsdrv_updateinfo...
  13.853306	[D1] Quick update...
  13.853327	[D3] send: Q1
  13.853340	[D3] asking for: 0D
  13.856175	[D3] read: (243.0 000.0 210.0 016 50.0 27.0 29.0 00101000
^C  15.774749	[D1] set_exit_flag: raising exit flag due to signal 2
  15.774815	Signal 2: exiting
  15.774864	[D1] upsdrv_cleanup...
```

After startup, it disappears again from /dev, but it can be called again and shows the value, how to make sure whether the value is correct and whether it is reading the values ​​​​from the laptop battery.

jimklimov, 2024-09-30:
Not sure about the disappearing device, maybe their controller chip is confused by queries and restarts.

Does `lsusb` show any details like naming? This behavior is not too common:

```
4.001419	[D2] Checking device (0001/0000) (003/007) 
4.002754	[D1] nut_usb_get_string: Broken language identifier, assuming en_US 
4.004870	[D2] - VendorID: 0001 
4.004923	[D2] - ProductID: 0000 
4.004932	[D2] - Manufacturer: unknown 
4.004940	[D2] - Product: MEC0003 
4.004947	[D2] - Serial Number: unknown
...
```

> how to make sure whether the value is correct and whether it is reading the values ​​​​from the laptop battery.

Not sure what the question is... was "laptop battery" a typo here?

The logged `read: (<numbers>` lines reveal a variant of Megatec protocol, so at least that bit is a hit, with data coming from the UPS.

big-vl, 2024-10-16:
> Not sure what the question is... was "laptop battery" a typo here?

Yes, this is not a true statement, I thought that the value was being read from the laptop battery, but now I think I’m sure that there is a connection, I’ll look into the setup further over the weekend.