networkupstools/nut #2646: usbhid-ups: Output voltage reading showing wrong voltage

Aerol, 2024-09-29:
I've got a Tripp-Lite SmartLCD to provide power to my rack and desk, got the CGI scripts served up on a raspberry pi and noticed in upsstats.cgi that output voltage was 1162.0v instead of 116.2v.

I ran `upsc smartlcd@localhost` and the output.voltage says the same thing there
```
Init SSL without certificate database
battery.charge: 100
battery.runtime: 2112
battery.type: PbAc
battery.voltage: 26.9
battery.voltage.nominal: 24.0
device.mfr: Tripp Lite
device.model: TRIPP LITE UPS
device.serial: FW-2263 E
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.data: TrippLite HID 0.84
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.frequency: 59.7
input.voltage: 116.2
input.voltage.nominal: 120
output.frequency.nominal: 60
output.voltage: 1162.0
output.voltage.nominal: 120
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.mfr: Tripp Lite
ups.model: TRIPP LITE UPS
ups.power.nominal: 1500
ups.productid: 2007
ups.serial: FW-2263 E
ups.status: OL CHRG
ups.timer.reboot: 65535
ups.timer.shutdown: 65535
ups.vendorid: 09ae
ups.watchdog.status: 0
```

jimklimov, 2024-09-29:
Looking at drivers/tripplite-hid.c source, it is full of tweaks dependent on model ID. Curiously, here the `input.voltage` seems OK so it's not that both of them need to be scaled down.

But probably your best bet is to build current NUT configured for in-place upgrade (see wiki), with sources changed to separate `"io"_voltage` multiplier into two variables, and introduce a new method similar to existing tweaks bound to modelid `0x2007`. And post a PR with tested results :D

jimklimov, 2025-10-15:
@haarp : so give a shot to https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests to run the current code base

Also this issue is about Tripplite, and CPS issues (like yours, including earlier discussions and solutions to yours) take up a significant portion of the issue tracker.