networkupstools/nut #2648: APC UPS Modbus support. Can't get it going.

Plutoaurus, 2024-09-30:
Have tried a few times to get NUT installed with usb modbus support and haven't be successful. 

I last tried following this:
https://github.com/networkupstools/nut/wiki/APC-UPS-with-Modbus-protocol

I always end up frustrated when it endlessly fails with missing dependencies not in any guide and going back to the beginning to the  first line and wondering why 14 years later it's still so difficult. This is a bit of a rant and cry for help sorry. 

Stephane in the libmodbus original repo is active and still updating, what's holding this back and can we do anything to help push the commit through?

The NUT repo seems to have 2 releases with modbus support but not for usb. Is this planned at all?

jimklimov, 2024-09-30:
Yeah, I understand the general frustration all too well, but so far I have not seen systems set up to build NUT (per its `docs/config-prereqs.txt`) that would have qualms building the fork of libmodbus too. If you do determine some missing package in that list as needed for libmodbus (or NUT), PRs or wiki updates are welcome.

With their CLA however, I am not sure anyone other than the original author may propose PRs for upstream changes (e.g. myself on behalf of @EchterAgo), and it seems the OA's interest and availability fizzled put waiting for Stephane to become active again :\ Maybe asking him how to go about it in this case (and actually getting a reply) would be a good start :D

That said, "14 years" is an overkill in regard to both NUT and libmodbus. The `rtu_usb` branch there and driver here are only a couple of years old...

On a side note, there are reported issues with the driver (maybe the added lib code actually) that cause driver init to crash for some users, but not everyone it seems, so *maybe* it is a bit too early to upstream the change yet...

jimklimov, 2024-09-30:
> The NUT repo seems to have 2 releases with modbus support but not for usb. Is this planned at all?

Also, on the NUT side, support is present; it is up to libmodbus to become able to use libusb (done in the fork, bug above notwithstanding).

Plutoaurus, 2024-10-01:
@jimklimov 
Thanks for the patient reply, I hoped this would be taken as a what can be done post rather than just a rant. 

I will give it another go and will try to post if any dependencies have changed etc.

desertwitch, 2024-10-12:
For what it's worth, my plugin has +/- 30 users actively (= `upsmon` alive > 60min) using `apc_modbus` on Unraid OS.
Most seem to be in combination with the Smart-UPS series, but the driver does seem to work well for a portion of users.
I haven't run into any complicated dependency situations packaging up `libmodbus` with `rtu_usb` either on my side of things.

Deiskos, 2024-10-14:
I'm having problems building it too. I followed the [wiki article](https://github.com/networkupstools/nut/wiki/APC-UPS-with-Modbus-protocol) exactly as written, both libmodbus and nut configure and build successfully, but when I run the driver with `sudo ./drivers/apc_modbus -d 1 -s upstest -x porttype=usb -x port=auto` just to see if it works it tells me `driver was not compiled with USB support`.

jimklimov, 2024-10-14:
Did the build logs indicate that USB support was enabled? Do you have `libusb-dev` (or similarly named package with headers etc.)  installed? Docs with dependencies are linked from https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests referenced in this article.

Deiskos, 2024-10-15:
I have `libusbx` and `libusbx-devel` installed. That's what the system nut package is built with:`dnf repoquery --requires nut` shows `libusb-1.0.so.0()(64bit)`)

The `--with-usb` flag takes care of checking for that. If I uninstall `libusbx-devel` then configure fails with `configure: error: USB drivers requested, but libusb not found.` as expected.

<details>
<summary>Build logs: (click me)</summary>

  ```
$ make apc_modbus
    CC       apc_modbus.o
    CC       libusb1.o
    CC       hidparser.o
    CC       usb-common.o
    CC       main.lo
    CC       dstate.lo
    CCLD     libdummy.la
    CC       upsdrvquery.lo
    CCLD     libdummy_upsdrvquery.la
  make[1]: Entering directory '/home/user/nut/common'
    CC       libcommon_la-state.lo
    CC       libcommon_la-str.lo
    CC       libcommon_la-upsconf.lo
  make[2]: Entering directory '/home/user/nut/include'
  SEMVER=2.8.2; TRUNK='master'; BASE='539d9709a5560cd1ea113f1d826e25f9c00ac674'; DESC='v2.8.2-1095-g539d9709a' => TAG='v2.8.2' + SUFFIX='-1095-g539d9709a' => VER5='2.8.2.1095.0' => VER50='2.8.2.1095' => DESC50='2.8.2.1095-1095-g539d9709a'
  NUT_VERSION: "2.8.2.1095-1095-g539d9709a"  NUT_VERSION_IS_RELEASE:false  NUT_VERSION_SEMVER: "2.8.2"
  test -f "nut_version.h" || cp "nut_version.h.tmp" "nut_version.h"
  cmp -s "nut_version.h.tmp" "nut_version.h" || cp "nut_version.h.tmp" "nut_version.h"
  rm -f "nut_version.h.tmp"
  make[2]: Leaving directory '/home/user/nut/include'
    CC       libcommon_la-common.lo
    CC       parseconf.lo
    CCLD     libparseconf.la
    CCLD     libcommon.la
  make[1]: Leaving directory '/home/user/nut/common'
  make[1]: Entering directory '/home/user/nut/common'
  make[1]: 'libparseconf.la' is up to date.
  make[1]: Leaving directory '/home/user/nut/common'
    CCLD     apc_modbus
```
</details>