networkupstools/nut #2651: Techly 1000VA, no battery and very few info

urbaman, 2024-10-05:
Hi,

I'm trying to manage this thing from a raspi 4B via USB.
It's working pretty well, but has very few info to show (no battery percentual for example).

Is there a better way to monitor it?

Thank you.

```
# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0001:0000 Fry's Electronics MEC0003
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
# lsusb -D /dev/bus/usb/001/004
Device: ID 0001:0000 Fry's Electronics MEC0003
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0001 Fry's Electronics
  idProduct          0x0000 
  bcdDevice            1.00
  iManufacturer           0 
  iProduct                1 MEC0003
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     624
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
```

```
# sudo nut-scanner -U
Scanning USB bus.
[nutdev1]
	driver = "nutdrv_qx"
	port = "auto"
	vendorid = "0001"
	productid = "0000"
	product = "MEC0003"
	bus = "001"
```

ups.conf:

```
maxretry = 3

[techly1000]
        driver = "nutdrv_qx"
        subdriver="hunnox"
        desc="Techly 1000"
        port = "auto"
        vendorid = "0001"
        productid = "0000"
        product = "MEC0003"
        protocol="hunnox"
        langid_fix=0x0409
        novendor
        norating
        noscanlangid
        bus = "001"
```

jimklimov, 2024-10-07:
Hard to say, it is the first time I hear of such brand so can't say how good they are at implementing the protocol or what they report. Lacking a valid USB ID for the vendor and model is not too encouraging in this regard.

That said, `nutdrv_qx` does support many values for `protocol` and `subdriver` settings -- see `nutdrv_qx -h` for values supported in your build, and try different combinations.

You did not clarify which NUT version you have, but 2.8.x should support a "dump mode" to collect data, report it and exit - so you can stop your driver service and script a loop to run `nutdrv_qx -a techly1000 -x subdriver=$S -x protocol=$P -d 1` and see if any of those combos reports more info.

Good luck! :)

urbaman, 2024-10-11:
Hi @jimklimov 

Here's what I can find for NUT version:

```
sudo upsdrvctl -h
Network UPS Tools - UPS driver controller 2.8.0
```

I'll try to make the proposed tests, if you have some more thorough guide for it, I'll appreciate, thank you.

jimklimov, 2024-10-11:
Yes, today I'm a bit more online so can post the Wiki link :) 

In-place replacement compatible build : https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests -- just build for now, without installing and replacing the packaged code.

Stop the running driver (if any).

Check driver help for supported keywords, e.g.:
````
:; ./drivers/nutdrv_qx -h
...
Acceptable values for 'subdriver' via -x or ups.conf in this driver: cypress, phoenixtec, phoenix, ippon, krauler, fabula, hunnox, fuji, sgs, snr, ablerex, armac

Acceptable values for 'protocol' via -x or ups.conf in this driver: voltronic, voltronic-qs, voltronic-qs-hex, mustek, megatec/old, bestups, mecer, megatec, zinto, masterguard, hunnox, ablerex, q1
...
````

Lock-pick the dialect:
````
:; for S in cypress phoenixtec phoenix ippon krauler fabula hunnox fuji sgs snr ablerex armac ; do
    for P in voltronic voltronic-qs voltronic-qs-hex mustek megatec/old bestups mecer megatec zinto masterguard hunnox ablerex q1 ; do
      ./drivers/nutdrv_qx -a techly1000 -x subdriver=$S -x protocol=$P -d 1 2>&1 | tee "techly1000-$S-`echo $P | tr / _`.txt"
   done
  done
````

...and hope the controller is smart enough to not brick if it sees commands and signals it would not understand.

jimklimov, 2024-10-11:
Normally the driver does a similar loop on its own (if no args are given and USB IDs do not suggest a specific combo to use), but it may stop if it gets a reply - and so not benefit from another dialect that might be supported better.

urbaman, 2024-10-11:
Ok, so probably I can say I found myself in this same situation and device type: [https://github.com/networkupstools/nut/issues/1539](https://github.com/networkupstools/nut/issues/1539)

So probably I will not find the battery % I search for.

I'll try to find time to fiddle around, but it seems also their own software do not show battery %.

I think I can close this one, I'll be back if I find something else.