networkupstools/nut #2652: battery.charge.low is set to 100 for some reason. APC Back-UPS BGM1500B (Back-UPS Pro Gaming)

Gon-97, 2024-10-05:
Hello I am currently using NUT on Unraid and I have this problem where my server just shuts off randomly all the time and I think it has to do with this:  the status is always at "On Line - Discharging Battery" and the battery.charge.low is always at 100%. Am I doing something wrong? I also have WinNUT installed on my gaming PC and that PC never shuts off when the Unraid PC shuts off. Not sure what's going on. Also here are my settings:

[image]

**Output from upsc:**

battery.charge: 100
battery.charge.low: 100
battery.mfr.date: 1980/01/01
battery.runtime: 839
battery.runtime.low: 120
battery.type: PbAc
battery.voltage: 27.5
battery.voltage.nominal: 24.0
device.mfr: American Power Conversion
device.model: Back-UPS BGM1500B
device.serial: 0B2118P00686
device.type: ups
driver.debug: 6
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.2.1
driver.version.data: APC HID 0.100
driver.version.internal: 0.57
driver.version.usb: libusb-1.0.26 (API: 0x1000108)
input.sensitivity: medium
input.transfer.high: 147
input.transfer.low: 88
input.transfer.reason: input voltage out of range
input.voltage: 122.0
input.voltage.nominal: 120
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.firmware: 31316S12-31320S10
ups.load: 39
ups.mfr: American Power Conversion
ups.mfr.date: 2021/04/22
ups.model: Back-UPS BGM1500B
ups.productid: 0002
ups.realpower.nominal: 900
ups.serial: 0B2118P00686
ups.status: OL DISCHRG
ups.test.result: Done and passed
ups.timer.reboot: 0
ups.timer.shutdown: -1
ups.vendorid: 051d

jimklimov, 2024-10-07:
Not sure why `battery.charge.low` is reported as 100%, but seems this is the setting reported by the device. Maybe you can modify it with `upsrw`, but I am not sure if the UPS would persist it (or forget when it power-cycles).

You probably can set it as `override.battery.charge.low = 70` (for example) in `ups.conf` section for the device.

I suppose, if the reading of actual `battery.charge` ever falls to 99% or less (so under the current 100% minimum), it becomes a trigger for Unraid shutdown. 

Not sure further why the WinNUT client does not request a shutdown of the PC, it is a third-party project (actually several went by that name). In NUT `upsmon` terms - I am not sure if WinNUT `MONITOR`'s your Unraid NUT server as a `primary` (ex-"master") for the UPS, and that it powers the required count of power sources of the PC, and so if the critical state of the NUT server is seen at all and is treated as a reason for PC "FSD" (Forced Shut Down). That is something for you to investigate, I guess.

Likewise, I'd expect the NUT server integration on Unraid to tell the UPS to shut down and/or power-cycle after a timeout, in case the wall power is back. This would disrupt the gaming PC in case of your problem (so it's good it does not happen), but generally this is how the NUT primary shutdowns should happen to maximize uptime in a hands-off manner (so the server is not indefinitely "off") - so it's not right that this acts as it does now.

During the recent NUT releases there were also several tweak settings to deal with "OL+DISCHRG" state combo reported by some UPSes, notably modern APCs. Effectively it may mean (vendor/model dependent) that the wall power is supplied, but the load is fed from battery - whether due to dual power conversion or undergoing calibration, for example.

CCing @desertwitch for the Unraid part :)

desertwitch, 2024-10-07:
I've seen this problem before, essentially `battery.charge.low` being `100` it even dropping by 1% will trigger an emergency shutdown due to what it considers as a low battery situation. The UPS dropping 1% can happen e.g. during short power flickers or also (most likely what is happening here) by its periodic self-testing (which slightly discharges the battery).

This can be fixed e.g. by putting `lowbatt = 30` into `ups.conf` to set the UPS low battery threshold to 30% battery charge, which is more reasonable. With the Unraid plugin, make sure to put this **after line 9 using the GUI configuration editor** so it is not overwritten by the GUI settings (unless you are in manual configuration mode, which doesn't seem like it from your screenshot):
```
lowbatt = 30
```

If this telling the UPS to use another value does not work, another possibility is ignoring the low battery flag set by the UPS and overriding the value reported by the UPS with a custom value that is then evaluated by NUT for low battery conditions instead. This can be done by putting into `ups.conf` (**on Unraid again after line 9 using the GUI configuration editor**):
```
ignorelb
override.battery.charge.low = 30
```

The gaming PC not shutting off I also suspect that it may not be registered as secondary/slave/client.
You can check this by executing the following command on your Unraid server's terminal:
```
upsc -c ups
```
This should list, apart from 127.0.0.1, the IP of your gaming PC - if not, WinNUT is not properly configured.
You need to make sure to put into WinNUT the "NUT Slave Username" and "NUT Slave Password" from the GUI.

The UPS itself not killing the outlet power/shutting down/power cycling is because it's disabled in the settings.
Setting "Disable UPS Power after Shutdown" to "Yes" would activate the behaviour as described by @jimklimov.

In the current situation I'd keep this setting off as not to disrupt power while you're still having weird shutdowns.
As soon as the situation stabilizes you can enable it, if you want the UPS to disable all power after shutting down Unraid.

desertwitch, 2024-10-08:
P.S. Don't forget to set the "UPS Driver Debug Level" back to "Default" if it's not needed anymore.
Level 6 is extremely verbose and will fill up your SYSLOG and that will cause a number of problems on its own.

Gon-97, 2024-10-08:
Thank you both for the great replies. I have since fixed the battery level low issue and so far nothing has happened. Also I did check and it seems NUT does see my gaming PC. I had WinNUT set to shut of my gaming PC if the battery level falls below 20% and whenever I checked WinNUT after my unraid server shut down I would always see the battery level said it was at 20%. So I think maybe it wasn't shutting down because it was at 20% and not below it. I have since set it to shut down at 30%. Not sure if that is going to change anything or not. 

Also one other weird thing to note: when my unraid server "shuts down" it doesn't actually turn off the computer itself because I can still see the LED on the front of the case lit up and I have to manually shut it down myself in order to reboot it. All dockers and the main dashboard are unreachable and I can see in the logs that NUT is initiating the shut down so not sure what that is all about.

Gon-97, 2024-10-09:
So I decided to do a little test. I unplugged my UPS from the wall and my gaming PC shut down immediately and my unraid server went offline as well, while the actual pc stayed on like i mentioned in the previous comment. Not sure what's going on. The display on the UPS said the battery was full as well. Very confusing. @desertwitch

desertwitch, 2024-10-09:
> So I decided to do a little test. I unplugged my UPS from the wall and my gaming PC shut down immediately and my unraid server went offline as well, while the actual pc stayed on like i mentioned in the previous comment. Not sure what's going on. The display on the UPS said the battery was full as well. Very confusing. @desertwitch
> 
> Also now this is what NUT is reporting once back online. Says low battery at 92%. ![image](https://private-user-images.githubusercontent.com/171991652/374772342-b1972fec-ed8c-450b-b966-fff50ec28d60.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg0NTIyMjEsIm5iZiI6MTcyODQ1MTkyMSwicGF0aCI6Ii8xNzE5OTE2NTIvMzc0NzcyMzQyLWIxOTcyZmVjLWVkOGMtNDUwYi1iOTY2LWZmZjUwZWMyOGQ2MC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAwOVQwNTMyMDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zMmUxOGJmNWVhN2M3YmFjMmU0YTlkZDAzMDJlYzc0OTJhNGQ1NGVmMzM1MzhhNzFkYzJmN2FlMzlmNmNiOGI3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.oF-07Y1RgkCMDnsUvxaZzycX83a_jOi6_TBbIfJLX0c)

Please post the "NUT Debug Package" that can be found in the "NUT Settings" GUI page.

desertwitch, 2024-10-09:
> So I decided to do a little test. I unplugged my UPS from the wall and my gaming PC shut down immediately and my unraid server went offline as well, while the actual pc stayed on like i mentioned in [...]

Please post the "NUT Debug Package" that can be found in the "NUT Settings" GUI page.
This way I can take a look at your plugin configuration and try figure out what is going wrong here.

It seems the UPS still forces a low battery condition despite the charge being well above critical limits...
As for the Unraid server not shutting down, Unraid OS shutdowns can take a while - how long did you wait?
Since NUT is not set to kill the outlet power, that stalling probably is within the OS somewhere and not really NUT...

Gon-97, 2024-10-09:
> Please post the "NUT Debug Package" that can be found in the "NUT Settings" GUI page. This way I can take a look at your plugin configuration and try figure out what is going wrong here.
>
> It seems [...]

[nut-debug-20241009051204.zip](https://github.com/user-attachments/files/17305601/nut-debug-20241009051204.zip)

Well sometimes unraid shuts down while I am at work so it could be sitting for hours and it still doesn't shut down fully, but here is the requested file.

desertwitch, 2024-10-11:
How old is your UPS battery (guess a bit newer than the reported _battery.mfr.date: 1980/01/01_)? Is it possible it's not capable of sustaining the load (anymore) and drains fast enough to hit even a configured 30% threshold too fast? You can try doing a test with another non-critical electronic device of similar wattage, to see how fast the UPS loses its power without NUT involved at all. 

Otherwise you can try going the suggested second route with `ignorelb` and `override.battery.charge.low = 30`... but, in any case, I'd also enable the syslog server on Unraid so we can see what's going on pre-shutdown (Settings => Syslog Server).

Gon-97, 2024-10-11:
Well I ordered my UPS on June 27th 2024. Not sure what the actual manufacture date of the battery is though. But when I unplugged the UPS from the wall to test before and then both of my PC's shut down. I plugged it back in a few minutes later and it was at 92% and then slowly charged back up to 100. I'm going to try the ignorelb method and see what happens. Also I did have a syslog server running and had an app on my computer to view it but it had a trial period and it ran out. Do you have any recommendations on a free program to use for the syslog server? (windows)

desertwitch, 2024-10-12:
> Well I ordered my UPS on June 27th 2024. Not sure what the actual manufacture date of the battery is though. But when I unplugged the UPS from the wall to test before and then both of my PC's shut [...]

You don't really need an external server, you can just activate "Local syslog server" in Settings => Syslog Server.
Then any logs will also be preserved after a system shutdown or reboot, and you can see what was going on before.

`ignorelb` & `override.battery.charge.low = 30` will suppress any low battery flags set by the UPS, so that's your best bet now.

Gon-97, 2024-10-26:
I set the ignorelb and override.battery.charge.low flags and then I got the new preview build update for NUT and everything seems to be working fine now! Not sure if it was setting those flags that fixed it or the update. I get notifications on unraid that it will shutdown if the battery level drops to 20% and when I plug it back in it works as it should.  Thank you for all the help!

closed 2024-10-26 by Gon-97