networkupstools/nut #2653: APC Back-UPS BX1200MI connection problems nut_libusb_get_report: Input/Output Error

HyP3r-, 2024-10-06:
I keep having connection problems with my UPS. What could be the cause? What can I do here?

```
Okt 06 09:43:19 afu-hv01 systemd[1]: Started nut-server.service - Network UPS Tools - power devices information server.
Okt 06 09:43:19 afu-hv01 systemd[1]: Started nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller.
Okt 06 09:43:19 afu-hv01 systemd[1]: Reached target nut.target - Network UPS Tools - target for power device drivers, data server and monitoring client (if enabled) on this system.
Okt 06 09:43:19 afu-hv01 nut-monitor[922]: fopen /run/nut/upsmon.pid: No such file or directory
Okt 06 09:43:19 afu-hv01 nut-monitor[922]: Could not find PID file to see if previous upsmon instance is already running!
Okt 06 09:43:19 afu-hv01 nut-server[919]: fopen /run/nut/upsd.pid: No such file or directory
Okt 06 09:43:19 afu-hv01 nut-server[919]: Could not find PID file '/run/nut/upsd.pid' to see if previous upsd instance is already running!
Okt 06 09:43:19 afu-hv01 nut-monitor[922]: Using power down flag file /etc/killpower
Okt 06 09:43:19 afu-hv01 nut-monitor[922]: UPS: ups1@localhost (primary) (power value 1)
Okt 06 09:43:19 afu-hv01 upsd[919]: listening on 127.0.0.1 port 3493
Okt 06 09:43:19 afu-hv01 nut-server[919]: listening on 127.0.0.1 port 3493
Okt 06 09:43:19 afu-hv01 nut-server[919]: listening on ::1 port 3493
Okt 06 09:43:19 afu-hv01 upsd[919]: listening on ::1 port 3493
Okt 06 09:43:19 afu-hv01 upsd[919]: Connected to UPS [ups1]: usbhid-ups-ups1
Okt 06 09:43:19 afu-hv01 nut-server[919]: Connected to UPS [ups1]: usbhid-ups-ups1
Okt 06 09:43:19 afu-hv01 usbhid-ups[842]: sock_connect: enabling asynchronous mode (auto)
Okt 06 09:43:19 afu-hv01 upsd[919]: Running as foreground process, not saving a PID file
Okt 06 09:43:19 afu-hv01 nut-server[919]: Running as foreground process, not saving a PID file
Okt 06 09:43:19 afu-hv01 nut-monitor[929]: Init SSL without certificate database
Okt 06 09:43:19 afu-hv01 nut-server[919]: User upsmaster@::1 logged into UPS [ups1]
Okt 06 09:43:19 afu-hv01 nut-monitor[929]: UPS ups1@localhost battery is low
Okt 06 09:43:19 afu-hv01 upsd[919]: User upsmaster@::1 logged into UPS [ups1]
Okt 06 09:43:19 afu-hv01 nut-monitor[951]: Network UPS Tools upsmon 2.8.0
Okt 06 09:56:48 afu-hv01 nut-server[919]: Data for UPS [ups1] is stale - check driver
Okt 06 09:56:48 afu-hv01 upsd[919]: Data for UPS [ups1] is stale - check driver
Okt 06 09:56:49 afu-hv01 nut-monitor[929]: Poll UPS [ups1@localhost] failed - Data stale
Okt 06 09:56:49 afu-hv01 nut-monitor[929]: Communications with UPS ups1@localhost lost
Okt 06 09:56:49 afu-hv01 nut-monitor[7259]: Network UPS Tools upsmon 2.8.0
Okt 06 09:56:54 afu-hv01 nut-monitor[929]: Poll UPS [ups1@localhost] failed - Data stale
Okt 06 09:56:59 afu-hv01 nut-monitor[929]: Poll UPS [ups1@localhost] failed - Data stale
Okt 06 09:57:01 afu-hv01 nut-server[919]: UPS [ups1] data is no longer stale
Okt 06 09:57:01 afu-hv01 upsd[919]: UPS [ups1] data is no longer stale
Okt 06 09:57:04 afu-hv01 nut-monitor[929]: Communications with UPS ups1@localhost established
Okt 06 09:57:04 afu-hv01 nut-monitor[7377]: Network UPS Tools upsmon 2.8.0
Okt 06 11:26:49 afu-hv01 nut-monitor[929]: UPS ups1@localhost battery is low
Okt 06 11:26:49 afu-hv01 nut-monitor[43921]: Network UPS Tools upsmon 2.8.0
Okt 06 11:27:21 afu-hv01 usbhid-ups[842]: nut_libusb_get_report: Input/Output Error
Okt 06 11:37:35 afu-hv01 usbhid-ups[842]: nut_libusb_get_report: Input/Output Error
Okt 06 11:37:37 afu-hv01 usbhid-ups[842]: nut_libusb_get_report: Input/Output Error
Okt 06 12:11:50 afu-hv01 nut-monitor[929]: UPS ups1@localhost battery is low
Okt 06 12:11:50 afu-hv01 nut-monitor[929]: UPS ups1@localhost battery needs to be replaced
Okt 06 12:11:50 afu-hv01 nut-monitor[62321]: Network UPS Tools upsmon 2.8.0
Okt 06 12:11:50 afu-hv01 nut-monitor[62323]: Network UPS Tools upsmon 2.8.0
Okt 06 12:13:29 afu-hv01 usbhid-ups[842]: nut_libusb_get_report: Input/Output Error
Okt 06 13:14:01 afu-hv01 usbhid-ups[842]: nut_libusb_get_report: Input/Output Error
Okt 06 13:44:35 afu-hv01 usbhid-ups[842]: nut_libusb_get_report: Input/Output Error
Okt 06 14:10:13 afu-hv01 usbhid-ups[842]: nut_libusb_get_report: Input/Output Error
Okt 06 14:10:18 afu-hv01 usbhid-ups[842]: nut_libusb_get_report: Input/Output Error
Okt 06 14:36:29 afu-hv01 usbhid-ups[842]: nut_libusb_get_report: Input/Output Error
```

Output of `upsc ups1`:
```
Init SSL without certificate database
battery.charge: 97
battery.charge.low: 10
battery.mfr.date: 2001/01/01
battery.runtime: 2388
battery.runtime.low: 120
battery.type: PbAc
battery.voltage: 13.5
battery.voltage.nominal: 12.0
device.mfr: American Power Conversion
device.model: Back-UPS BX1200MI
device.serial: 9B2421A02741
device.type: ups
driver.name: usbhid-ups
driver.parameter.bus: 001
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: Back-UPS BX1200MI
driver.parameter.productid: 0002
driver.parameter.serial: 9B2421A02741
driver.parameter.synchronous: auto
driver.parameter.vendor: American Power Conversion
driver.parameter.vendorid: 051D
driver.version: 2.8.0
driver.version.data: APC HID 0.98
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.sensitivity: medium
input.transfer.high: 295
input.transfer.low: 145
input.voltage: 232.0
input.voltage.nominal: 230
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.load: 5
ups.mfr: American Power Conversion
ups.mfr.date: 2024/05/24
ups.model: Back-UPS BX1200MI
ups.productid: 0002
ups.realpower.nominal: 650
ups.serial: 9B2421A02741
ups.status: OL CHRG
ups.test.result: Done and passed
ups.timer.reboot: 0
ups.timer.shutdown: -1
ups.vendorid: 051d
```

jimklimov, 2024-10-07:
Seems like an issue solved by PR #2565 (in current master branch, due to be part of NUT 2.8.3 release).

HyP3r-, 2024-10-23:
I have compiled and started the new version as described in the instructions. It seems to have been running for several days and weeks now without any errors. That's why I'm closing the ticket.

closed 2024-10-23 by HyP3r-