networkupstools/nut #2656: Unable to Configure NUT using Cyberpower CP1500PFCLCDa UPS

ccruiser, 2024-10-12:
This is my first attempt to configure NUT (I used https://pimylifeup.com/raspberry-pi-nut-server/ for reference) and although my UPS was not listed in the compatibility items, I thought it might work. Here's my diagnostic information:
$ lsusb | grep Cyber
Bus 001 Device 003: ID 0764:0601 Cyber Power System, Inc. PR1500LCDRT2U UPS

$ sudo /lib/nut/usbhid-ups -a netups -u root -DDDDD
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000     debug level is '5'
   0.001235     upsdrv_initups...
   0.073600     Checking device (1D6B/0003) (002/001)
   0.116872     - VendorID: 1d6b
   0.116947     - ProductID: 0003
   0.117009     - Manufacturer: Linux 6.1.21-v8+ xhci-hcd
   0.117068     - Product: xHCI Host Controller
   0.117126     - Serial Number: 0000:01:00.0
   0.117184     - Bus: 002
   0.117243     - Device release number: 0601
   0.117300     Trying to match device
   0.117370     Device does not match - skipping
   0.117865     Checking device (0764/0601) (001/003)
   0.123212     - VendorID: 0764
   0.123281     - ProductID: 0601
   0.123345     - Manufacturer: CPS
   0.123404     - Product: CP1500PFCLCDa
   0.123461     - Serial Number: CXXKV2002995
   0.123515     - Bus: 001
   0.123572     - Device release number: 0200
   0.123634     Trying to match device
   0.123774     Device does not match - skipping
   0.123868     Checking device (2109/3431) (001/002)
   0.124621     - VendorID: 2109
   0.124690     - ProductID: 3431
   0.124751     - Manufacturer: unknown
   0.124810     - Product: USB2.0 Hub
   0.124869     - Serial Number: unknown
   0.124928     - Bus: 001
   0.124986     - Device release number: 0421
   0.125044     Trying to match device
   0.125108     Device does not match - skipping
   0.125198     Checking device (1D6B/0002) (001/001)
   0.125658     - VendorID: 1d6b
   0.125726     - ProductID: 0002
   0.125789     - Manufacturer: Linux 6.1.21-v8+ xhci-hcd
   0.125848     - Product: xHCI Host Controller
   0.125907     - Serial Number: 0000:01:00.0
   0.125965     - Bus: 001
   0.126023     - Device release number: 0601
   0.126080     Trying to match device
   0.126141     Device does not match - skipping
   0.126227     No appropriate HID device found
   0.126295     No matching HID UPS found

jimklimov, 2024-10-12:
Seems like it should work, at least with modern NUT (note that 2.7.4 is 8 years old).

What are your `ups.conf` contents? Try adding these lines to `[netups]` device section:
```
    vendorid = 0764
    productid = 0601
```

Not sure quickly if that NUT version already had sufficient mapping tables and just lacked knowledge of needed IDs.

jimklimov, 2024-10-12:
Alternately, https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests should not be difficult.

ccruiser, 2024-10-12:
> Seems like it should work, at least with modern NUT (note that 2.7.4 is 8 years old).
>
> What are your `ups.conf` contents? Try adding these lines to `[netups]` device section:
>
> ```
>     vendorid = [...]

Thanks for the reply; here's my configuration taken from ups.conf 
```
##############
# cyber power local closet UPS
[netups]
driver = usbhid-ups
desc = "Cyberpower UPS PR1500 LCD RT2 Network Rack"
port = auto
vendorid = 764
productid = 601
```

jimklimov, 2024-10-13:
I think those comparisons would be via regular expressions, but just in case - add the leading zeroes...

ccruiser, 2024-10-13:
It appears even with the updated version and drives (version 2.8.2) that I am seeing the same failure.  See output below. 

```
$ sudo ./drivers/usbhid-ups -a netups -u root -DDDDD
   0.000000     [D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools 2.8.2.1095-1095-g539d9709a (development iteration after 2.8.2) - Generic HID driver 0.57
USB communication driver (libusb 0.1) 0.48
   0.000215     [D1] upsdrv_makevartable...
   0.000286     [D5] send_to_all: SETINFO driver.version.usb "libusb-0.1 (or compat)"
   0.000317     [D1] Using USB implementation: libusb-0.1 (or compat)
   0.000572     [D5] do_upsconf_args: confupsname=netups, var=driver, val=usbhid-ups
   0.000602     [D5] do_upsconf_args: call main_arg()
   0.000627     [D3] main_arg: var='driver' val='usbhid-ups'
   0.000670     [D5] do_upsconf_args: not a main_arg()
   0.000695     [D5] do_upsconf_args: this is a 'driver' setting, may we proceed?
   0.000723     [D5] do_upsconf_args: 'driver' setting already applied with this value
   0.000753     [D5] do_upsconf_args: confupsname=netups, var=desc, val=Cyberpower UPS PR1500 LCD RT2 Network Rack
   0.000779     [D5] do_upsconf_args: call main_arg()
   0.000803     [D3] main_arg: var='desc' val='Cyberpower UPS PR1500 LCD RT2 Network Rack'
   0.000830     [D5] do_upsconf_args: confupsname=netups, var=port, val=auto
   0.000854     [D5] do_upsconf_args: call main_arg()
   0.000878     [D3] main_arg: var='port' val='auto'
   0.000906     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000941     [D5] do_upsconf_args: confupsname=netups, var=vendorid, val=764
   0.000965     [D5] do_upsconf_args: call main_arg()
   0.000989     [D3] main_arg: var='vendorid' val='764'
   0.001013     [D5] do_upsconf_args: not a main_arg()
   0.001037     [D5] do_upsconf_args: process as value
   0.001069     [D5] send_to_all: SETINFO driver.parameter.vendorid "764"
   0.001096     [D5] do_upsconf_args: confupsname=netups, var=productid, val=601
   0.001120     [D5] do_upsconf_args: call main_arg()
   0.001146     [D3] main_arg: var='productid' val='601'
   0.001170     [D5] do_upsconf_args: not a main_arg()
   0.001193     [D5] do_upsconf_args: process as value
   0.001224     [D5] send_to_all: SETINFO driver.parameter.productid "601"
   0.001276     [D1] Built-in default or configured user for drivers 'nut' was ignored due to 'root' specified on command line
   0.001303     [D1] Network UPS Tools version 2.8.2.1095-1095-g539d9709a (development iteration after 2.8.2) built with gcc (Debian 10.2.1-6) 10.2.1 20210110 and configured with flags: --prefix=/usr --sbindir=/usr/sbin --bindir=/usr/bin --sysconfdir=/etc/nut --with-group=nut --with-user=nut --with-usb=yes --enable-inplace-runtime='2.8.2.1095-1095-g539d9709a (development iteration after 2.8.2) built with gcc (Debian 10.2.1-6) 10.2.1 20210110 and'
   0.001336     [D1] debug level is '5'
   0.001363     [D5] send_to_all: SETINFO driver.debug "5"
   0.001390     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.002456     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.002536     [D1] Signalling UPS [netups]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.002580     Can't open /var/state/ups/usbhid-ups-netups: No such file or directory
   0.002607     [D1] Request for other driver to exit returned code -1
   0.002632     [D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
   0.002664     [D5] send_to_all: SETINFO device.type "ups"
   0.002692     [D5] send_to_all: SETINFO driver.state "init.device"
   0.002716     [D1] upsdrv_initups (non-SHUT)...
   0.002741     [D2] Initializing an USB-connected UPS with library libusb-0.1 (or compat) (NUT subdriver name='USB communication driver (libusb 0.1)' ver='0.48')
   0.080361     [D3] usb_busses=0x559a56f430
   0.080437     [D2] Checking device (1D6B/0003) (002/001)
   0.123708     [D2] - VendorID: 1d6b
   0.123768     [D2] - ProductID: 0003
   0.123815     [D2] - Manufacturer: Linux 6.1.21-v8+ xhci-hcd
   0.123862     [D2] - Product: xHCI Host Controller
   0.123909     [D2] - Serial Number: 0000:01:00.0
   0.123956     [D2] - Bus: 002
   0.124004     [D2] - Device: 001
   0.124052     [D2] - Device release number: 0601
   0.124098     [D2] Trying to match device
   0.124152     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.124216     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.124266     [D2] Device does not match - skipping
   0.124755     [D2] Checking device (0764/0601) (001/003)
   0.129787     [D2] - VendorID: 0764
   0.129843     [D2] - ProductID: 0601
   0.129890     [D2] - Manufacturer: CPS
   0.129937     [D2] - Product: CP1500PFCLCDa
   0.129982     [D2] - Serial Number: CXXKV2002995
   0.130029     [D2] - Bus: 001
[... 13 lines cut ...]
   0.131517     [D2] - Bus: 001
   0.131563     [D2] - Device: 002
   0.131610     [D2] - Device release number: 0421
   0.131656     [D2] Trying to match device
   0.131706     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.131762     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.131811     [D2] Device does not match - skipping
   0.131896     [D2] Checking device (1D6B/0002) (001/001)
   0.132350     [D2] - VendorID: 1d6b
   0.132405     [D2] - ProductID: 0002
[... cut ...]
   0.132781     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.132835     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.132884     [D2] Device does not match - skipping
   0.132967     [D2] libusb0: No appropriate HID device found
   0.133017     No matching HID UPS found
   0.133082     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
   0.133136     upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
```

ccruiser, 2024-10-13:
It appears that after a reboot, the code updates were applied. Thanks for the help!

```
$ upsc netups
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 4375
battery.runtime.low: 300
battery.status: 100%
battery.type: PbAcid
battery.voltage: 27.2
battery.voltage.nominal: 24
device.mfr: CPS
device.model: CP1500PFCLCDa
device.serial: CXXKV2002995
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0601
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0764
driver.state: quiet
driver.version: 2.8.2.1095-1095-g539d9709a
driver.version.data: CyberPower HID 0.81
driver.version.internal: 0.57
driver.version.usb: libusb-0.1 (or compat)
input.sensitivity: normal
input.transfer.high: 139
input.transfer.low: 100
input.voltage: 121.0
input.voltage.nominal: 120
output.voltage: 121.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: CR01802B7
ups.load: 9
ups.mfr: CPS
ups.model: CP1500PFCLCDa
ups.power: 141
ups.power.nominal: 1500
ups.productid: 0601
ups.realpower: 99
ups.realpower.nominal: 1000
ups.serial: CXXKV2002995
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```

closed 2024-10-13 by ccruiser