networkupstools/nut #2659: Not supported UPS in NUT

AppleCarmine, 2024-10-16:
Hello everyone, I have a slightly old UPS from Control Systems 2, model STD60S. There’s not much information about this UPS on the web. It has a serial port, and with the Powercom software (https://www.upspowercom.com/Monitoring.jsp#Software), I managed to get it working and monitored on Windows. I installed the drivers for the USB SERIAL adapter, the Powercom software, and it works perfectly.

However, I can’t get it to work on NUT. Unfortunately, I’ve tried several serial drivers, including powercom, usbhid-ups, blazer_ser, and nutdrv_qx, but nothing seems to work. It just can’t detect it, and it doesn’t function.

The strange thing is that if the Powercom software works with my UPS, I don’t understand why the NUT drivers for Powercom UPSs aren’t working.

Could you please help me? I’ve been struggling with this for days. I’ve tried it on a Proxmox node, on a Proxmox VM using passthrough, and on a Raspberry Pi, but nothing seems to work.

[image]
[image]

desertwitch, 2024-10-16:
Have you tried what running `nut-scanner` returns?
https://networkupstools.org/docs/man/nut-scanner.html

jimklimov, 2024-10-16:
Well, for one - it would be beneficial to start directly the driver programs, not via `upsdrvctl`, so you can pass debug options to them. It may also help to custom-build recent NUT sources for the investigation - more drivers were added, and code may be more talkative for the troubleshooting part. See https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests about that.

Some ideas OTOH:
* Higher debug verbosity would expose how the driver tries to match the device and why it decides that it can not talk to it.
  * With `usbhid-ups` it might get to dumping the data exchange, if there were HID reports.
  * With `nutdrv_qx` (in both serial and USB modes) it can show which Megatec Q<x> protocol family commands it sent, and perhaps which replies it got. If there are replies starting with a parenthesis and a sequence of floating-point numbers - that is likely some dialect of Qx.
* USB-capable drivers like `usbhid-ups` and `nutdrv_qx` (in not-serial mode with `port=auto`) may require you to pass `vendorid` and `productid` options if the device uses some IDs not known to the driver. With NUT v2.8.2 and newer you can also specify a `subdriver` to try for `usbhid-ups`; the `nutdrv_qx` and `blazer*` drivers had that ability for longer (and also a `protocol` option).
* Also check that no other program grabbed the device (especially if it identifies as USB HID, so `udev` or similar might lay hands on it).
* I don't have direct experience, but in many issues saw USB-level sniffing used with Wireshark or similar, to check what the vendor protocol looks like with their tools. Allegedly there are variants that work on Windows too.

> The strange thing is that if the Powercom software works with my UPS, I don’t understand why the NUT drivers for Powercom UPSs aren’t working.

Finally, on this point: many vendors use a lot of protocols over time that they are on the market. Sometimes this even changes for the same device as firmwares are updated. They may have started with one protocol (that someone developed a NUT driver for back in the day), then adopted or licensed another. There are vendors whose similarly named models talk USB HID in one version and Qx in another. APC scaled down on common USB HID and pushes for modbus since 2010's, but due to having purchased many companies is also represented by protocols with "smart", "sol", etc. in the name. TLDR - it is a very mixed bag.

In case of USB-connected devices done diligently, vendors would have an identifier for themselves, and the further product identifier helps know which protocol it talks (I saw firmwares even change the announced productid when different protocols become supported, kudos). Sadly, many vendors do take the cheap-o route of using bogus identifiers like a mix of `0000`, `0001` and `ffff` usually, which does not help discern them.

AppleCarmine, 2024-10-16:
> Have you tried what running `nut-scanner` returns? https://networkupstools.org/docs/man/nut-scanner.html

Yes, but without success.
[image]

AppleCarmine, 2024-10-17:
Someone can help me :(?

jimklimov, 2024-10-18:
Thanks. So from those replies, the UPS does not seem to talk any Megatec dialect...

Potentially, it could be talking at a different data rate, so it just can't interpret the signal it sees from the driver. It seems that some serial drivers and devices hard-coded e.g. 2400bps, with a few having baudrate/parity/databit/stopbit settings at all.

Do you have a chance to sniff how the UPS talks to its Windows program, e.g. using https://learn.microsoft.com/en-us/sysinternals/downloads/portmon as suggested at https://stackoverflow.com/questions/461794/whats-a-good-free-serial-port-monitor-for-reverse-engineering or some other tool from that discussion? Hopefully that would show some useful clues.

Note that the serial port name can change as you re-plug the USB-Serial adapter, check the Device Manager for current name (probably the starting point would be `COM5` or similar).

AppleCarmine, 2024-10-19:
> Thanks. So from those replies, the UPS does not seem to talk any Megatec dialect...
>
> Potentially, it could be talking at a different data rate, so it just can't interpret the signal it sees from the [...]

I can’t get this program to work for sniffing requests; I’m trying to find an alternative.

AppleCarmine, 2024-10-22:
> Grazie. Quindi da quelle risposte, l'UPS non sembra parlare alcun dialetto Megatec...
> 
> Potenzialmente, potrebbe parlare a una velocità dati diversa, quindi non riesce a interpretare il segnale che vede dal driver. Sembra che alcuni driver e dispositivi seriali abbiano codificato in modo rigido, ad esempio 2400 bps, con alcuni che hanno impostazioni baudrate/parità/bit dati/stopbit.
> 
> Hai la possibilità di annusare come l'UPS comunica con il suo programma Windows, ad esempio usando https://learn.microsoft.com/en-us/sysinternals/downloads/portmon come suggerito su https://stackoverflow.com/questions/461794/whats-a-good-free-serial-port-monitor-for-reverse-engineering o qualche altro strumento da quella discussione? Spero che ciò possa mostrare qualche indizio utile.
> 
> Tieni presente che il nome della porta seriale può cambiare se ricolleghi l'adattatore USB-Seriale; controlla il nome attuale in Gestione dispositivi (probabilmente il punto di partenza sarebbe `COM5`o simile).

https://pastebin.com/dJhb5YzH

I managed to sniff the requests with Portmon! Wow, it drove me crazy to get it working, I hope it will be useful! I had to use 32-bit Windows XP, and after several attempts, I finally did it. Let’s hope it was really worth it. If you need anything else, let me know :D

jimklimov, 2024-10-22:
To my eye the protocol did not associate with any pattern I see often (like Megatec's characteristic `(xxx.y zzz.w ...` lines); however I see the `K)` part repeating often. ChatGPT suggested that this may be APC Smart protocol's status polling/report.

> The "K)" messages in the log you referenced suggest that the protocol being used is likely related to the Uninterruptible Power Supply (UPS) communication, typically following the Smart Protocol or the APC (American Power Conversion) protocol for UPS devices. These protocols often involve queries and responses for status and configuration information.
> The repetitive reads might be part of a polling mechanism where the software continuously checks the status of the UPS. If you can share more specific details or examples from the log, I could help identify the protocol more accurately!

At least, it is worth trying some APC drivers (although I am not sure if anyone but APC-branded devices used it).

jimklimov, 2024-10-22:
It seems that ChatGPT only blurbs about "APC Smart UPS Protocol" as an example of UPS protocol it heard about, not a specific indication. (Still worth a try, why not)

FWIW, I filtered the second posted sniff log into actual R/W bytes if it helps later investigations, but so far am out of quick ideas, sorry :\

* `curl https://pastebin.com/raw/ZM0dedri > /tmp/z1`
* `grep -E 'IRP_MJ_(READ|WRITE) UP STATUS_SUCCESS' /tmp/z1 | awk '{print $4" "$7" "$8}' > /tmp/z3`
** https://pastebin.com/uxH5gASe - with HEX and ASCII values
* `grep -E 'IRP_MJ_(READ|WRITE) UP STATUS_SUCCESS' /tmp/z1 | awk '{print $4" "$7}' > /tmp/z2`
** https://pastebin.com/69D3rzfh - only HEX

If the Windows software reports some numbers (charge, temperature etc.) see if they resemble anything from these messages? e.g. `0x63` is 99 decimal, might be the near-100% battery charge?

jimklimov, 2024-10-22:
Hopefully, it would not hurt...

As a paying customer, you can also try asking the vendor for their protocol. In some countries they can be obliged to share it.

AppleCarmine, 2024-10-22:
> > What do you mean by ‘Hopefully, it would not hurt…’?
> 
> It depends on how "they" coded their firmware, with regard to unexpected input (or EMI noise on the wire misinterpreted as signal).
> 
> "Hopefully" they coded to ignore unexpectancies or at worst reboot the controller - not brick it etc.
> 
> Never underestimate the possibility of cut corners in economy of low margins :\

Anyway, the software page says it supports NUT (https://www.upspowercom.com/Support-NUT.jsp), I tried with these configurations, but it doesn’t work. Maybe it needs some subdriver or something like that? It could be a clue.

jimklimov, 2024-10-22:
Cool, did not know about that back-link :)

However, they list a number of models under their own brand - not sure if yours maps to any of those. Also if your communications go over a serial port, the `usbhid-ups` driver would not be it, I'm afraid...

I've had several different Powercoms early in my sysadmin career, decent bang for buck (and got me to meet NUT for the first time); at that time the serial-port oriented `powercom` driver did talk to them.

Which gives me another idea: if your UPS is old, it may be that over NUT's evolution a suitable driver was deprecated and removed (there were one or two large re-designs of the software so whatever was not ported to new architecture could not work and was eventually dropped for lack of interest). If you can build NUT, try giving a shot to some very old versions, just to check if they had a driver that understands your device and may be worth wedging into the modern design?..

Looking at `history.txt`, `UPGRADING.adoc` and `NEWS.adoc`, I think some of the critical points were around NUT 0.45.0, 1.4.0, 2.0.5, 2.4.3 and 2.6.5 as the last versions delivering some things summarily removed in later ones.

AppleCarmine, 2024-10-22:
> Fantastico, non sapevo di quel back-link :)
> 
> Tuttavia, elencano un certo numero di modelli sotto il loro marchio, non sono sicuro se il tuo sia mappato su uno di quelli. Inoltre, se le tue comunicazioni avvengono tramite una porta seriale, il `usbhid-ups`driver non sarebbe quello, temo...
> 
> All'inizio della mia carriera di amministratore di sistema ho avuto diversi Powercom, un buon rapporto qualità-prezzo (e mi hanno fatto incontrare NUT per la prima volta); a quel tempo, il `powercom`driver orientato alla porta seriale comunicava con loro.
> 
> Il che mi dà un'altra idea: se il tuo UPS è vecchio, potrebbe essere che durante l'evoluzione di NUT un driver adatto sia stato deprecato e rimosso (ci sono state una o due grandi riprogettazioni del software, quindi tutto ciò che non è stato portato sulla nuova architettura non poteva funzionare ed è stato infine abbandonato per mancanza di interesse). Se riesci a costruire NUT, prova a dare un'occhiata ad alcune versioni molto vecchie, solo per verificare se avevano un driver che capisce il tuo dispositivo e potrebbe valere la pena di essere inserito nel design moderno?..
> 
> Esaminando `history.txt`, `UPGRADING.adoc`e `NEWS.adoc`, penso che alcuni dei punti critici riguardassero NUT 0.45.0, 1.4.0, 2.0.5, 2.4.3 e 2.6.5 come ultime versioni, che includevano alcune cose sommariamente rimosse in quelle successive.

Unfortunately, I’m not able to rebuild NUT from scratch, which is a shame. I’ll abandon the project and buy another supported UPS or make do with the proprietary software. Thanks anyway.

jimklimov, 2024-10-22:
FWIW, for modern NUT, the toolkits for different platforms are listed at https://github.com/networkupstools/nut/blob/master/docs/config-prereqs.txt or prettily rendered at https://networkupstools.org/docs/user-manual.chunked/_build_prerequisites_to_make_nut_from_scratch_on_various_operating_systems.html

I suppose they should be able to build many older versions too, especially if you ignore the compiler warnings.

Out of curiosity, did you try the Linux version of their UPSMON PRO? https://www.upspowercom.com/Monitoring.jsp#Software

AppleCarmine, 2024-10-22:
> Per quel che vale, per i moderni NUT, i toolkit per le diverse piattaforme sono elencati su https://github.com/networkupstools/nut/blob/master/docs/config-prereqs.txt o resi in modo gradevole su https://networkupstools.org/docs/user-manual.chunked/_build_prerequisites_to_make_nut_from_scratch_on_various_operating_systems.html
> 
> Immagino che dovrebbero essere in grado di compilare anche molte versioni più vecchie, soprattutto se si ignorano gli avvisi del compilatore.
> 
> Per curiosità, hai provato la versione Linux del loro UPSMON PRO? https://www.upspowercom.com/Monitoring.jsp#Software

I tried installing it several times without success, it kept giving me error after error. Now I’m trying to install it on a Raspberry Pi 3 with an older version of Ubuntu.

jimklimov, 2024-10-22:
Trying to make sense/educated guess of your sniffed data:

````
WRITE 01 . 

READ 2d - : varied 2b..31 (41..47dec)

READ 63 c : 99dec (battery %)

READ 6b k : same as below and varied 6b..6d (around 110dec - could be voltage minus 100, common trick for 270V scale representation)

READ 6b k 

READ 32 2 : 50dec (Hz?)

READ ff ÿ : 255dec

READ 32 2 : 50Hz (in/out?)

READ 00 . 

READ 00 . 

READ 14 . : 20dec

READ 00 . 

READ 4b K : 75dec

READ 29 ) : 41dec

READ 00 . 

READ 00 . 

READ 00 .
````

probably some of 0's (or bits in some unidentified numbers) are state, alarms etc.?...

AppleCarmine, 2024-10-22:
> Trying to make sense/educated guess of your sniffed data:
>
> ```
> WRITE 01 .
>
> READ 2d - : varied 2b..31 (41..47dec)
>
> READ 63 c : 99dec (battery %)
>
> READ 6b k : same as below and varied 6b..6d (around [...]

If by alarms you mean beeps, no, at the exact moment I did the dump, I didn’t receive any alarm signals. Anyway, the proprietary Linux software is terrible, it just won’t start :(

AppleCarmine, 2024-10-22:
I confirm, the UPSMON Pro software for Linux is terrible. It’s made with Java but doesn’t work; it’s missing libraries, it won’t start—it’s just a nightmare. Really disgusting. What a shame.