networkupstools/nut #2661: Added  ups.beeper.status  support for Masterpower MF-UPS650VA

yonesmit, 2024-10-17:
Hi,
Recently I start using a Masterpower MF-UPS650VA with nut. The device is detected correctly, it uses usbhid-ups driver and MGE HID 1.49 subdriver.
My problem was I couldn't control the beeper to silence it (ups.beeper.status). After some debugging I found that the feature was found in a different path than the ones that driver support (driver already support the feature in 2 different paths and it uses the first one it  founds).
Driver support ups.beeper.status in this 2 paths:
```
UPS.BatterySystem.Battery.AudibleAlarmControl
UPS.PowerSummary.AudibleAlarmControl
```
And my ups uses this path:
`UPS.AudibleAlarmControl`

So I added the 3 lines needed to support the beeper status feature in the path used by my ups and compiled the driver.

```
	{ "ups.beeper.status", 0 ,0, "UPS.AudibleAlarmControl", NULL, "%s", HU_FLAG_SEMI_STATIC, beeper_info },
...
	{ "beeper.disable", 0, 0, "UPS.AudibleAlarmControl", NULL, "1", HU_TYPE_CMD, NULL },
	{ "beeper.enable", 0, 0, "UPS.AudibleAlarmControl", NULL, "2", HU_TYPE_CMD, NULL },
```

Now the feature is correctly supported.

```
battery.charge: 100
battery.charge.low: 10
battery.runtime: 7200
battery.type: PbAC
device.mfr: KSTAR
device.model: UPS HID
device.serial: HID-C01
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: UPS HID
driver.parameter.productid: 0001
driver.parameter.serial: HID-C01
driver.parameter.synchronous: auto
driver.parameter.vendor: KSTAR
driver.parameter.vendorid: 0463
driver.state: quiet
driver.version: 2.8.2.1081-1081-g78e0525d3
driver.version.data: MGE HID 1.49
driver.version.internal: 0.57
driver.version.usb: libusb-1.0.24 (API: 0x1000108)
input.frequency: 49.9
input.voltage: 230.5
outlet.1.status: on
output.voltage: 230.4
ups.beeper.status: disabled
ups.firmware:
ups.load: 8
ups.mfr: KSTAR
ups.model: UPS HID
ups.productid: 0001
ups.serial: HID-C01
ups.status: OL
ups.vendorid: 0463
```
I'm attaching the file with changes for your reference.

[mge-hid.zip](https://github.com/user-attachments/files/17406635/mge-hid.zip)

Can the changes be added to project so the this ups is supported directly in future releases?
Thanks in advance

Best Regards

yonesmit

jimklimov, 2024-10-17:
Sounds great, thanks for sharing!

Would you care to post the change as a pull request, so it is attributed in your name in Git history?

yonesmit, 2024-10-17:
> Sounds great, thanks for sharing!
> 
> Would you care to post the change as a pull request, so it is attributed in your name in Git history?
I would but I don't know how to create a git pull request. I'm not sure if I need to create a branch, make changes in it and then submit comparing it with master branch or there is any other easier way. I'm not used to git yet (I've been out of world for last years XD ) Can you add the changes? if not I will try to do as soon as I learn hoy to do....
TIA
best regards,

yonesmit, 2024-10-17:
Finally I managed (?) to create the pull request. Tell me if there is something wrong. Sorry this is my fist time.
best regards

callegar, 2024-10-17:
@yonesmit Nice to see better support for this UPS! Thanks! A quick question: does a manual exist for this thing? I have one and it came with no manual apart from a single page with no useful info.

yonesmit, 2024-10-17:
A single manual page, And another page to tell you the download link of kpower. That's all I got.
But it was really cheap and works quite well, the only problem I got is I needed to silence it.

jimklimov, 2024-10-17:
Thanks, looks good for a first shot! :)

To answer your earlier question, yes it is recommended to make a new branch in your repository for each feature you work on, so you don't have problems e.g. updating to changes from remote `master` branch as it changes over time while your large pull request awaits a review.

ferguilo, 2024-11-27:
> Hi, Recently I start using a Masterpower MF-UPS650VA with nut. The device is detected correctly, it uses usbhid-ups driver and MGE HID 1.49 subdriver. My problem was I couldn't control the beeper to silence it (ups.beeper.status). After some debugging I found that the feature was found in a different path than the ones that driver support (driver already support the feature in 2 different paths and it uses the first one it founds). Driver support ups.beeper.status in this 2 paths:
> 
> ```
> UPS.BatterySystem.Battery.AudibleAlarmControl
> UPS.PowerSummary.AudibleAlarmControl
> ```
> 
> And my ups uses this path:
> 
> `UPS.AudibleAlarmControl`
> 
> So I added the 3 lines needed to support the beeper status feature in the path used by my ups and compiled the driver.
> 
> ```
> 	{ "ups.beeper.status", 0 ,0, "UPS.AudibleAlarmControl", NULL, "%s", HU_FLAG_SEMI_STATIC, beeper_info },
> ...
> 	{ "beeper.disable", 0, 0, "UPS.AudibleAlarmControl", NULL, "1", HU_TYPE_CMD, NULL },
> 	{ "beeper.enable", 0, 0, "UPS.AudibleAlarmControl", NULL, "2", HU_TYPE_CMD, NULL },
> ```
> 
> Now the feature is correctly supported.
> 
> ```
> battery.charge: 100
> battery.charge.low: 10
> battery.runtime: 7200
> battery.type: PbAC
> device.mfr: KSTAR
> device.model: UPS HID
> device.serial: HID-C01
> device.type: ups
> driver.debug: 0
> driver.flag.allow_killpower: 0
> driver.name: usbhid-ups
> driver.parameter.pollfreq: 30
> driver.parameter.pollinterval: 2
> driver.parameter.port: auto
> driver.parameter.product: UPS HID
> driver.parameter.productid: 0001
> driver.parameter.serial: HID-C01
> driver.parameter.synchronous: auto
> driver.parameter.vendor: KSTAR
> driver.parameter.vendorid: 0463
> driver.state: quiet
> driver.version: 2.8.2.1081-1081-g78e0525d3
> driver.version.data: MGE HID 1.49
> driver.version.internal: 0.57
> driver.version.usb: libusb-1.0.24 (API: 0x1000108)
> input.frequency: 49.9
> input.voltage: 230.5
> outlet.1.status: on
> output.voltage: 230.4
> ups.beeper.status: disabled
> ups.firmware:
> ups.load: 8
> ups.mfr: KSTAR
> ups.model: UPS HID
> ups.productid: 0001
> ups.serial: HID-C01
> ups.status: OL
> ups.vendorid: 0463
> ```
> 
> I'm attaching the file with changes for your reference.
> 
> [mge-hid.zip](https://github.com/user-attachments/files/17406635/mge-hid.zip)
> 
> Can the changes be added to project so this ups is supported directly in future releases? Thanks in advance
> 
> Best Regards
> 
> yonesmit

Hi, I’m trying to configure the MasterPower MF-UPS650VA with NUT, but I’m encountering issues. The system responds with the following error:

root@server-ao45:/run/nut# upsdrvctl start
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
libusb1: Could not open any HID devices: insufficient permissions on everything
No matching HID UPS found
Driver failed to start (exit status=1)

jimklimov, 2024-11-27:
@ferguilo : please check if https://github.com/networkupstools/nut/wiki/%22Insufficient-permissions%22-when-starting-USB-drivers and https://github.com/networkupstools/nut/wiki/nut%E2%80%90driver%E2%80%90enumerator-(NDE) pages help? Specifically, you are running with NUT v2.8.x and using `upsdrvctl` which looks like this particular red flag :)

ferguilo, 2024-11-28:
That is the response to `usbhid-ups -DDDDDD -d 1 -a ups`

root@server-ao45:/etc/udev/rules.d# usbhid-ups -DDDDDD -d 1 -a ups
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D3] main_arg: var='driver' val='usbhid-ups'
   0.000011     [D3] main_arg: var='port' val='auto'
   0.000018     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000023     [D3] main_arg: var='vendorid' val='09d6'
   0.000029     [D5] send_to_all: SETINFO driver.parameter.vendorid "09d6"
   0.000034     [D3] main_arg: var='productid' val='0001'
   0.000038     [D5] send_to_all: SETINFO driver.parameter.productid "0001"
   0.000043     [D3] main_arg: var='pollonly' val='yes'
   0.000047     [D5] send_to_all: SETINFO driver.parameter.pollonly "yes"
   0.000052     [D3] main_arg: var='desc' val='MasterPower UPS'
   0.000060     [D1] debug level is '6'
   0.000720     [D5] send_to_all: SETINFO device.type "ups"
   0.000728     [D2] Initializing an USB-connected UPS with library libusb-1.0.26 (API: 0x1000109) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.43')
   0.000733     [D1] upsdrv_initups (non-SHUT)...
   0.003688     [D2] Checking device 1 of 7 (1D6B/0003)
   0.003703     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.003708     [D2] Checking device 2 of 7 (8087/0029)
   0.003716     [D1] Failed to open device (8087/0029), skipping: Access denied (insufficient permissions)
   0.003720     [D2] Checking device 3 of 7 (1D6B/0002)
   0.003727     [D1] Failed to open device (1D6B/0002), skipping: Access denied (insufficient permissions)
   0.003732     [D2] Checking device 4 of 7 (1D6B/0003)
   0.003739     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.003743     [D2] Checking device 5 of 7 (1D5C/7102)
   0.003750     [D1] Failed to open device (1D5C/7102), skipping: Access denied (insufficient permissions)
   0.003755     [D2] Checking device 6 of 7 (09D6/0001)
   0.020731     [D2] - VendorID: 09d6
   0.020740     [D2] - ProductID: 0001
   0.020745     [D2] - Manufacturer: KSTAR
   0.020752     [D2] - Product: UPS
   0.020757     [D2] - Serial Number: KHID0
   0.020763     [D2] - Bus: 001
   0.020768     [D2] - Device: unknown
   0.020773     [D2] - Device release number: 0110
   0.020778     [D2] Trying to match device
   0.020783     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.020792     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.020797     [D2] Device does not match - skipping
   0.020808     [D2] Checking device 7 of 7 (1D6B/0002)
   0.020821     [D1] Failed to open device (1D6B/0002), skipping: Access denied (insufficient permissions)
   0.020826     [D2] libusb1: No appropriate HID device found
   0.020832     libusb1: Could not open any HID devices: insufficient permissions on everything
   0.020837     No matching HID UPS found

ferguilo, 2024-11-29:
> Interesting, this is also (allegedly) `Manufacturer: KSTAR` as in the original post (or so `driver.parameter.vendor` configuration said and matched), however the `vendorid` is different. While that one was in MGE/Eaton namespace as 0x0463, yours is 0x09d6 which is not known/mentioned anywhere in even current NUT code base.
> 
> Please try a more recent NUT release, 2.8.1 or newer (perhaps a custom build from git master, see https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests for details).
> 
> * 2.8.1 delivered a `subdriver=...` option support for `usbhid-ups` so you can tell it to try MGE HID for the unknown device
> * master branch is where the fix discussed in this ticket has landed; it is not part of any formal release yet - so a custom build is needed

Thank you very much! It is now working with version 2.8.2 and subdriver MGE HID 1.49. However, creating a custom build was quite challenging for me…

jimklimov, 2024-11-29:
Thanks for confirmation! Can you please post an `upsc` output for the NUT DDL, and so I can add the IDs to be hard-coded known ones in the next NUT releases?

closed 2024-12-18 by jimklimov

pimiento4781, 2024-12-31:
> Thank you very much! It is now working with version 2.8.2 and subdriver MGE HID 1.49. However, creating a custom build was quite challenging for me…

Hello,
I am facing the exact same issue, and I am quite a beginner with this stuff. How did you manage to create a custom build? Any help would be much appreciated.

Thank you very much.

jimklimov, 2025-01-01:
Literally, follow the link you've quoted ;)

pimiento4781, 2025-01-01:
> Literally, follow the link you've quoted ;)

I am very new to all this, and it's so hard for me. Yesterday I tried, and I ended up having to install from scratch OMV 😅. That's why I asked him :D 

Anyways, I think I am gonna buy a different UPS from the hardware compatibility list, so I don't have to go through that.

PR #2723 (2024-12-18): usbhid-ups: added support for 0x09d6:0x0001 (KSTAR) UPSes using MGE HID
Closes: #2661

Note: previously same model names also manufactured by KSTAR were using MGE USB VendorID (0x0463).

Also note that per #230 some devices with this ID could use Megatec Qx protocol family instead.