networkupstools/nut #2665: Please report it to NUT developers, with an 'upsc' output for your device. - APC Smart-UPS X 3000 - SNMP

bergernetch, 2024-10-28:
The tool said I should report this.

```root@nut-test03:~# dpkg -l | grep UPS
ii  libnutscan2:amd64            2.8.0-7                        amd64        network UPS tools - scanner library
ii  libupsclient6:amd64          2.8.0-7                        amd64        network UPS tools - client library
ii  nut                          2.8.0-7                        all          network UPS tools - metapackage
ii  nut-client                   2.8.0-7                        amd64        network UPS tools - clients
ii  nut-server                   2.8.0-7                        amd64        network UPS tools - core system
ii  nut-snmp                     2.8.0-7                        amd64        network UPS tools - SNMP driver
```

ups.conf 

```[apc-ups]
driver = snmp-ups
port = 10.67.100.30
community = public
snmp_version = v1
desc = "APC AP9630 UPS"
```

```
root@nut-test03:~# systemctl status nut-driver@apc-ups
● nut-driver@apc-ups.service - Network UPS Tools - device driver for NUT device 'apc-ups'
     Loaded: loaded (/lib/systemd/system/nut-driver@.service; enabled; preset: enabled)
    Drop-In: /etc/systemd/system/nut-driver@apc-ups.service.d
             └─nut-driver-enumerator-generated-checksum.conf, nut-driver-enumerator-generated.conf
     Active: active (running) since Mon 2024-10-28 12:40:50 UTC; 2s ago
    Process: 204 ExecStart=/bin/sh -c NUTDEV="`/usr/libexec/nut-driver-enumerator.sh --get-device-for-service apc-ups`" && [ -n "$NUTDEV" ] || { echo "FATAL: Could not find a NUT device section for >
   Main PID: 386 (snmp-ups)
      Tasks: 1 (limit: 269849)
     Memory: 1.9M
        CPU: 44ms
     CGroup: /system.slice/system-nut\x2ddriver.slice/nut-driver@apc-ups.service
             └─386 /lib/nut/snmp-ups -a apc-ups

Oct 28 12:40:49 nut-test03 nut-driver@apc-ups[243]: Please report it to NUT developers, with an 'upsc' output for your device.
Oct 28 12:40:49 nut-test03 nut-driver@apc-ups[243]: Going back to the classic MIB detection method.
Oct 28 12:40:49 nut-test03 nut-driver@apc-ups[243]: Detected Smart-UPS X 3000 on host 10.67.100.30 (mib: apcc 1.6)
Oct 28 12:40:49 nut-test03 nut-driver@apc-ups[243]: [apc-ups] Warning: excessive poll failures, limiting error reporting (OID = .1.3.6.1.4.1.318.1.1.1.9.2.3.1.5.1.1.3)
Oct 28 12:40:49 nut-test03 snmp-ups[243]: [apc-ups] Warning: excessive poll failures, limiting error reporting (OID = .1.3.6.1.4.1.318.1.1.1.9.2.3.1.5.1.1.3)
Oct 28 12:40:50 nut-test03 nut-driver@apc-ups[243]: [apc-ups] Warning: excessive poll failures, limiting error reporting (OID = .1.3.6.1.4.1.318.1.1.1.9.3.3.1.6.1.1.1)
Oct 28 12:40:50 nut-test03 snmp-ups[243]: [apc-ups] Warning: excessive poll failures, limiting error reporting (OID = .1.3.6.1.4.1.318.1.1.1.9.3.3.1.6.1.1.1)
Oct 28 12:40:50 nut-test03 snmp-ups[386]: Startup successful
Oct 28 12:40:50 nut-test03 nut-driver@apc-ups[242]: Network UPS Tools - UPS driver controller 2.8.0
Oct 28 12:40:50 nut-test03 systemd[1]: Started nut-driver@apc-ups.service - Network UPS Tools - device driver for NUT device 'apc-ups'.
```

```
root@nut-test03:~# upsc apc-ups
Error: Connection failure: Connection refused
```

Furthermore, why does it not work? snmpwalk does work. I would be glad if someone could point me to the right direction. 

```
root@nut-test03:~# snmpwalk -v1 -c"public" 10.67.100.30 | head
SNMPv2-MIB::sysDescr.0 = STRING: APC Web/SNMP Management Card (MB:v4.1.0 PF:v6.5.0 PN:apc_hw05_aos_650.bin AF1:v6.5.0 AN1:apc_hw05_sumx_650.bin MN:AP9630 HR:08 SN: ZA1828021293 MD:07/10/2018) (Embedded PowerNet SNMP Agent SW v2.2 compatible)
SNMPv2-MIB::sysObjectID.0 = OID: SNMPv2-SMI::enterprises.318.1.3.27
DISMAN-EVENT-MIB::sysUpTimeInstance = Timeticks: (343130) 0:57:11.30
SNMPv2-MIB::sysContact.0 = STRING: manager
SNMPv2-MIB::sysName.0 = STRING: apc01
SNMPv2-MIB::sysLocation.0 = STRING: Rack1
SNMPv2-MIB::sysServices.0 = INTEGER: 72
SNMPv2-MIB::sysORLastChange.0 = Timeticks: (0) 0:00:00.00
SNMPv2-MIB::sysORID.1 = OID: SNMPv2-MIB::snmpMIB
SNMPv2-MIB::sysORID.2 = OID: SNMP-FRAMEWORK-MIB::snmpFrameworkMIBCompliance
```

bergernetch, 2024-10-28:
I just realized that nut-server is not running. Where do I find the logs indicating the issue?
There is nothing in the syslog.

```
root@nut-test03:~# systemctl status nut-server
× nut-server.service - Network UPS Tools - power devices information server
     Loaded: loaded (/lib/systemd/system/nut-server.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Mon 2024-10-28 12:49:58 UTC; 3s ago
   Duration: 503us
    Process: 648 ExecStart=/sbin/upsd -F (code=exited, status=1/FAILURE)
   Main PID: 648 (code=exited, status=1/FAILURE)
        CPU: 2ms

Oct 28 12:49:58 nut-test03 systemd[1]: Started nut-server.service - Network UPS Tools - power devices information server.
Oct 28 12:49:58 nut-test03 systemd[1]: nut-server.service: Main process exited, code=exited, status=1/FAILURE
Oct 28 12:49:58 nut-test03 systemd[1]: nut-server.service: Failed with result 'exit-code'.
```

bergernetch, 2024-10-28:
I figured that I had a wrong setting in upsd.conf:
I now replaced `LISTEN 10.67.100.0/24` with `LISTEN 10.67.100.249`.
Now upsd starts and upsc works:

```
# upsc apc-ups
Init SSL without certificate database
battery.charge: 99.40
battery.date: 01/15/2022
battery.packs: 1
battery.runtime: 8892
battery.runtime.low: 120
battery.voltage: 136
[...]
```

jimklimov, 2024-11-06:
FWIW, checked this with current code - it behaved slightly better, claiming `getaddrinfo: Name or service not known` at least, as a pointer where to dig:
````
Nov 06 16:32:59 pve systemd[1]: Starting Network UPS Tools - power devices information server...
Nov 06 16:32:59 pve nut-server[2228592]: Network UPS Tools upsd 2.8.2.1085.57-1142-gf76af4d71 (development iteration after 2.8.2)
Nov 06 16:32:59 pve nut-server[2228592]: getaddrinfo: Name or service not known
Nov 06 16:32:59 pve systemd[1]: nut-server.service: Main process exited, code=exited, status=1/FAILURE
Nov 06 16:32:59 pve systemd[1]: nut-server.service: Failed with result 'exit-code'.
Nov 06 16:32:59 pve systemd[1]: Failed to start Network UPS Tools - power devices information server.
Nov 06 16:32:59 pve systemd[1]: nut-server.service: Scheduled restart job, restart counter is at 1.
````

At a certain debug verbosity (via `DEBUG_MIN 6` in `upsd.conf`) it did reveal the tried IP address, but I see how reporting it by default along with a failed bind could be useful.

````
Nov 06 16:35:14 pve systemd[1]: Starting Network UPS Tools - power devices information server...
Nov 06 16:35:14 pve nut-server[2235769]: Network UPS Tools upsd 2.8.2.1085.57-1142-gf76af4d71 (development iteration after 2.8.2)
Nov 06 16:35:14 pve nut-server[2235769]:    0.000001        [D1:2235769] debug level is '6'
Nov 06 16:35:14 pve nut-server[2235769]:    0.000014        [D3:2235769] setuptcp: try to bind to 10.94.56.99/24 port 3493
Nov 06 16:35:14 pve nut-server[2235769]:    0.000190        getaddrinfo: Name or service not known
Nov 06 16:35:14 pve nut-server[2235769]:    0.000201        [D1:2235769] upsd_cleanup: starting the end-game
Nov 06 16:35:14 pve nut-server[2235769]:    0.000208        [D1:2235769] upsd_cleanup: finished
````

PR #2855 (2025-03-21): A few quick fixes before v2.8.3 release
Closes: #1899
Closes: #1900
Closes: #2665

Also revises several documentation related recipes, and retouches a few docs themselves.
commit 96f642d7 (2025-03-21): server/upsd.c, NEWS.adoc: failure due to getaddrinfo() now reports the disliked host name [#2665]

Closes: #2665
closed 2025-03-21 by jimklimov