networkupstools/nut #2666: NUT 2.8.1-3 " Can't claim USB device [051d:0002]@0/0: Entity not found" using usbhid-ups

ErikDB87, 2024-10-30:
Hi all,

Yesterday, I bought a UPS for the first time in my life, and was eager to dive into NUT. But not all is working as expected... I saw a similar thread started on 18 October, but it didn't help me. (I also spent a handful of hours searching the web for solutions, and of course read the manual and FAQ - "queequeg".)

I tried shutting my UPS (APC "Back-UPS BX750MI  FW:295202G -302202G") down with `sudo upsdrvctl shutdown`, but no response. Digging around, I found a few things that raised my suspicion, but I can't figure it out...

I followed this fine gentleman's guide (but tweaked it a bit - I don't know why he uses `master` and `slave` for example?): https://www.youtube.com/watch?v=vyBP7wpN72c.
I must add that I created a few users and user groups during installation and configuration, following the documentation. But in the end, I lost oversight, and everything didn't work. So it's entirely possible there's a permissions issue somewhere. But I don't have any idea where...

Because of the trial-and-error approach, in the `.conf` files (shown below), a lot of stuff is commented out. I assume I can delete it, but I also assume the `#` should be adequate? Anyway, I keep it in the output below for clarity.

I'll just paste various outputs here, I hope that's a reasonable approach?

Thanks in advance for anyone's help!

Kind regards,

Erik

```
erik@MinipcLG2:/$ sudo upsdrvctl shutdown apcupskelder
Network UPS Tools - UPS driver controller 2.8.1
Network UPS Tools - Generic HID driver 0.52 (2.8.1)
USB communication driver (libusb 1.0) 0.46
select with socket: Invalid argument
Can't claim USB device [051d:0002]@0/0: Entity not found
upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
Driver failed to start (exit status=1)
```

```
erik@MinipcLG2:/$ sudo nut-scanner -U
Scanning USB bus.
[nutdev1]
        driver = "usbhid-ups"
        port = "auto"
        vendorid = "051D"
        productid = "0002"
        product = "Back-UPS BX750MI  FW:295202G -302202G"
        serial = "9B2416A22014"
        vendor = "American Power Conversion"
        bus = "001"
        device = "004"
        busport = "002"
        ###NOTMATCHED-YET###bcdDevice = "0106"
```

```
erik@MinipcLG2:/$ usbhid-ups -a apcupskelder
usbhid-ups: opdracht niet gevonden
```
This seems strange? But after some googling, I found the below alternative - although I would expect it to work without `/lib/nut/`):
````
erik@MinipcLG2:/$ sudo /lib/nut/usbhid-ups -a apcupskelder
Network UPS Tools - Generic HID driver 0.52 (2.8.1)
USB communication driver (libusb 1.0) 0.46
Duplicate driver instance detected (PID file /run/nut/usbhid-ups-apcupskelder.pid exists)! Terminating other driver!

Bericht aan allen van root@MinipcLG2 (somewhere) (Wed Oct 30 10:27:32 2024):

Communications with UPS apcupskelder@localhost lost

Bericht aan allen van root@MinipcLG2 (somewhere) (Wed Oct 30 10:27:32 2024):

Communications with UPS apcupskelder@localhost lost

HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 255 in ReportID: 0x0c
HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 255 in ReportID: 0x22
HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 255 in ReportID: 0x40
Using subdriver: APC HID 0.100
[... 209 lines cut ...]
NOTIFYMSG FSD       "UPS %s: forced shutdown in progress"
NOTIFYMSG COMMOK    "Communications with UPS %s established"
NOTIFYMSG COMMBAD   "Communications with UPS %s lost"
NOTIFYMSG SHUTDOWN  "Auto logout and shutdown proceeding"
NOTIFYMSG REPLBATT  "UPS %s battery needs to be replaced"
NOTIFYMSG NOCOMM    "UPS %s is unavailable"
NOTIFYMSG NOPARENT  "upsmon parent process died - shutdown impossible"

NOTIFYFLAG ONLINE   SYSLOG+WALL+EXEC
NOTIFYFLAG ONBATT   SYSLOG+WALL+EXEC
NOTIFYFLAG LOWBATT  SYSLOG+WALL
NOTIFYFLAG FSD      SYSLOG+WALL+EXEC
NOTIFYFLAG COMMOK   SYSLOG+WALL+EXEC
NOTIFYFLAG COMMBAD  SYSLOG+WALL+EXEC
NOTIFYFLAG SHUTDOWN SYSLOG+WALL+EXEC
NOTIFYFLAG REPLBATT SYSLOG+WALL
NOTIFYFLAG NOCOMM   SYSLOG+WALL+EXEC
NOTIFYFLAG NOPARENT SYSLOG+WALL

RBWARNTIME 43200

NOCOMMWARNTIME 600

FINALDELAY 5
```

ErikDB87, 2024-11-02:
I thought removing and installing the package again might solve the problem. I also removed `/etc/nut`.

But somehow, the directory `/etc/nut` **is** created again, but the config files in it **are not**...?

So I did this:
```
sudo apt remove nut nut-client nut-server
sudo reboot
sudo apt autoremove
sudo reboot
sudo rm -r /etc/nut
sudo apt install nut nut-client nut-server
```

And now:
```
erik@MinipcLG2:/etc/nut$ ls -pal
totaal 16
drwxr-xr-x   2 root root  4096 nov  2 16:29 ./
drwxr-xr-x 182 root root 12288 nov  2 16:29 ../
```

Somehow, I made it worse... But I don't understand how installing the packages again fails to install these files...?

jimklimov, 2024-11-02:
I believe some answers around this general issue are in the mailing list.

As for packaging, NUT team does not impact it directly, so it is technically a distro matter. That said, probably different packages defined for NUT there which deliver files which might need configuration files, all deliver the `/etc/nut` directory. There may be some package which also delivers `*.conf.sample` files which are just that - well-commented samples (to copy into `*.conf` and adapt to your setup). Packaging can not divine your device details, IP addresses, etc. to recreate your customized config files.

ErikDB87, 2024-11-02:
Of course I didn't expect my amended config files to appear. :) But the standard ones - which **were** installed when I first installed NUT, didn't reappear... I manually added `hosts.conf` using `sudo nano hosts.conf`.

Anyway, I had hoped this reinstall would have reset all the permissions. But I was wrong...

```
erik@MinipcLG2:~$ sudo /lib/nut/usbhid-ups -DDD -a apcupskelder
Network UPS Tools - Generic HID driver 0.52 (2.8.1)
USB communication driver (libusb 1.0) 0.46
   0.000000     [D3] do_global_args: var='pollinterval' val='1'
   0.000013     [D1] testval_reloadable: setting 'pollinterval' exists and differs: new value '1' vs. '2'
   0.000023     [D3] do_global_args: var='maxretry' val='3'
   0.000034     [D3] main_arg: var='driver' val='usbhid-ups'
   0.000065     [D3] main_arg: var='port' val='auto'
   0.000082     [D3] main_arg: var='desc' val='Back-UPS BX750MI  FW:295202G -302202G'
   0.000089     [D3] main_arg: var='vendorid' val='051d'
   0.000095     [D3] main_arg: var='productid' val='0002'
   0.000106     [D3] main_arg: var='serial' val='9B2416A22014'
   0.000126     [D1] Network UPS Tools version 2.8.1 (release/snapshot of 2.8.1) built with gcc (Ubuntu 13.2.0-23ubuntu2) 13.2.0 and configured with flags: --build=x86_64-linux-gnu --prefix=/usr --includedir=${prefix}/include --mandir=${prefix}/share/man --infodir=${prefix}/share/info --sysconfdir=/etc --localstatedir=/var --disable-option-checking --disable-silent-rules --libdir=${prefix}/lib/x86_64-linux-gnu --runstatedir=/run --disable-maintainer-mode --disable-dependency-tracking --prefix= --sysconfdir=/etc/nut --includedir=/usr/include --mandir=/usr/share/man --libdir=${prefix}/lib/x86_64-linux-gnu --libexecdir=/usr/libexec --with-ssl --with-nss --with-cgi --with-dev --enable-static --with-statepath=/run/nut --with-altpidpath=/run/nut --with-drvpath=/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut --with-htmlpath=/usr/share/nut/www --with-pidpath=/run/nut --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib/x86_64-linux-gnu/pkgconfig --with-user=nut --with-group=nut --with-udev-dir=/lib/udev --with-systemdsystemunitdir=/lib/systemd/system --with-systemdshutdowndir=/lib/systemd/system-shutdown --with-systemdtmpfilesdir=/usr/lib/tmpfiles.d --with-python=python3 --with-python3=/usr/bin/python3 --with-doc=man,html-single,html-chunked,pdf
   0.000176     [D1] debug level is '3'
   0.001153     [D1] Succeeded to become_user(nut): now UID=121 GID=130
   0.001182     [D1] upsdrv_initups (non-SHUT)...
   0.001189     [D2] Initializing an USB-connected UPS with library libusb-1.0.27 (API: 0x100010a) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.46')
   0.004785     [D2] Checking device 1 of 4 (1D6B/0003)
   0.004813     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.004821     [D2] Checking device 2 of 4 (0403/6001)
   0.004833     [D1] Failed to open device (0403/6001), skipping: Access denied (insufficient permissions)
   0.004840     [D2] Checking device 3 of 4 (051D/0002)
   0.006646     [D2] - VendorID: 051d
   0.006656     [D2] - ProductID: 0002
   0.006663     [D2] - Manufacturer: American Power Conversion
   0.006669     [D2] - Product: Back-UPS BX750MI  FW:295202G -302202G
   0.006675     [D2] - Serial Number: 9B2416A22014
   0.006681     [D2] - Bus: 001
   0.006687     [D2] - Bus Port: 002
   0.006693     [D2] - Device: 002
   0.006699     [D2] - Device release number: 0106
   0.006705     [D2] Trying to match device
   0.006775     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.006788     [D3] match_function_regex: matching a device...
   0.006868     [D2] Device matches
   0.006874     [D2] Reading first configuration descriptor
   0.006890     [D3] libusb_kernel_driver_active() returned 0: Success
   0.006903     [D2] failed to claim USB device: Resource busy
   0.006913     [D2] Kernel driver already detached
   0.006923     [D2] failed to claim USB device: Resource busy
   0.006932     [D2] Kernel driver already detached
   0.006941     [D2] failed to claim USB device: Resource busy
   0.006950     [D2] Kernel driver already detached
   0.006967     [D2] failed to claim USB device: Resource busy
   0.006977     [D2] Kernel driver already detached
   0.006986     Can't claim USB device [051d:0002]@0/0: Entity not found
   0.006997     upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
```

There are **more** permission problems now...

I apologize for making matters worse :(

jimklimov, 2024-11-02:
Probably no foul. As you edited `ups.conf`, NDE could have picked up the device definition and created the `nut-driver@apcupskelder.service` instance again, so the manually-started driver conflicts with it again.

Re-check if the unit does exist and does not complain (like on the mailing list) - if so, go on to `upsd` (nut-server) and `upsc` client (to check it works) and finally `upsmon` for notifications and safe shutdown.

ErikDB87, 2024-11-02:
I'm not sure I fully understand what you mean... I'm not really a Linux master :)

But here are some outputs:
```
erik@MinipcLG2:~$ systemctl status nut-driver@apcupskelder.service
● nut-driver@apcupskelder.service - Network UPS Tools - device driver for NUT device 'apcupskelder'
     Loaded: loaded (/usr/lib/systemd/system/nut-driver@.service; enabled; preset: enabled)
    Drop-In: /etc/systemd/system/nut-driver@apcupskelder.service.d
             └─nut-driver-enumerator-generated-checksum.conf, nut-driver-enumerator-generated.conf
     Active: active (running) since Sat 2024-11-02 18:25:04 CET; 56min ago
    Process: 5250 ExecStartPre=/usr/bin/systemd-tmpfiles --create /usr/lib/tmpfiles.d/nut-common-tmpfiles.conf (code=exited, status=0/SUCCESS)
    Process: 5251 ExecStart=/bin/sh -c NUTDEV="`/usr/libexec/nut-driver-enumerator.sh --get-device-for-service apcupskelder`" && [ -n "$NUTDEV" ] || { echo>
   Main PID: 5300 (usbhid-ups)
      Tasks: 1 (limit: 9284)
     Memory: 856.0K (peak: 2.3M)
        CPU: 52min 49.078s
     CGroup: /system.slice/system-nut\x2ddriver.slice/nut-driver@apcupskelder.service
             └─5300 /lib/nut/usbhid-ups -a apcupskelder

nov 02 18:25:04 MinipcLG2 nut-driver@apcupskelder[5279]: USB communication driver (libusb 1.0) 0.46
nov 02 18:25:04 MinipcLG2 usbhid-ups[5300]: Startup successful
nov 02 18:25:04 MinipcLG2 usbhid-ups[5300]: upsnotify: failed to notify about state 2: no notification tech defined, will not spam more about it
nov 02 18:25:04 MinipcLG2 nut-driver@apcupskelder[5251]: Network UPS Tools - UPS driver controller 2.8.1
nov 02 18:25:04 MinipcLG2 systemd[1]: Started nut-driver@apcupskelder.service - Network UPS Tools - device driver for NUT device 'apcupskelder'.
nov 02 18:25:05 MinipcLG2 usbhid-ups[5300]: sock_connect: enabling asynchronous mode (auto)
nov 02 18:25:36 MinipcLG2 usbhid-ups[5300]: sock_connect: enabling asynchronous mode (auto)
nov 02 18:25:36 MinipcLG2 usbhid-ups[5300]: sock_connect: enabling asynchronous mode (auto)
nov 02 18:25:36 MinipcLG2 usbhid-ups[5300]: Requesting UPS [apcupskelder] to power off, as/if handled by its driver by default (may exit), due to socket pr>
nov 02 18:25:36 MinipcLG2 usbhid-ups[5300]: WARNING: send_to_one: write 48 bytes to socket 12 failed (ret=-1), disconnecting: Broken pipe
```

```
erik@MinipcLG2:~$ systemctl status nut-server.service
● nut-server.service - Network UPS Tools - power devices information server
     Loaded: loaded (/usr/lib/systemd/system/nut-server.service; enabled; preset: enabled)
     Active: active (running) since Sat 2024-11-02 18:11:22 CET; 1h 11min ago
    Process: 1328 ExecStartPre=/usr/bin/systemd-tmpfiles --create /usr/lib/tmpfiles.d/nut-common-tmpfiles.conf (code=exited, status=0/SUCCESS)
   Main PID: 1338 (upsd)
      Tasks: 1 (limit: 9284)
     Memory: 944.0K (peak: 1.6M)
        CPU: 569ms
     CGroup: /system.slice/nut-server.service
             └─1338 /lib/nut/upsd -F

nov 02 18:11:22 MinipcLG2 upsd[1338]: Running as foreground process, not saving a PID file
nov 02 18:11:22 MinipcLG2 nut-server[1338]: Running as foreground process, not saving a PID file
nov 02 18:11:22 MinipcLG2 nut-server[1338]: upsnotify: failed to notify about state 2: no notification tech defined, will not spam more about it
nov 02 18:11:22 MinipcLG2 upsd[1338]: upsnotify: failed to notify about state 2: no notification tech defined, will not spam more about it
nov 02 18:24:44 MinipcLG2 nut-server[1338]: Can't connect to UPS [apcupskelder] (usbhid-ups-apcupskelder): No such file or directory
nov 02 18:24:44 MinipcLG2 upsd[1338]: Can't connect to UPS [apcupskelder] (usbhid-ups-apcupskelder): No such file or directory
nov 02 18:24:49 MinipcLG2 nut-server[1338]: Connected to UPS [apcupskelder]: usbhid-ups-apcupskelder
nov 02 18:24:49 MinipcLG2 upsd[1338]: Connected to UPS [apcupskelder]: usbhid-ups-apcupskelder
nov 02 18:25:05 MinipcLG2 nut-server[1338]: Connected to UPS [apcupskelder]: usbhid-ups-apcupskelder
nov 02 18:25:05 MinipcLG2 upsd[1338]: Connected to UPS [apcupskelder]: usbhid-ups-apcupskelder
```

```
erik@MinipcLG2:~$ systemctl status upsd
● nut-server.service - Network UPS Tools - power devices information server
     Loaded: loaded (/usr/lib/systemd/system/nut-server.service; enabled; preset: enabled)
     Active: active (running) since Sat 2024-11-02 18:11:22 CET; 1h 12min ago
    Process: 1328 ExecStartPre=/usr/bin/systemd-tmpfiles --create /usr/lib/tmpfiles.d/nut-common-tmpfiles.conf (code=exited, status=0/SUCCESS)
   Main PID: 1338 (upsd)
      Tasks: 1 (limit: 9284)
     Memory: 944.0K (peak: 1.6M)
        CPU: 572ms
     CGroup: /system.slice/nut-server.service
             └─1338 /lib/nut/upsd -F

nov 02 18:11:22 MinipcLG2 upsd[1338]: Running as foreground process, not saving a PID file
[... 20 lines cut ...]
     Active: active (running) since Sat 2024-11-02 18:11:22 CET; 1h 12min ago
    Process: 1346 ExecStartPre=/usr/bin/systemd-tmpfiles --create /usr/lib/tmpfiles.d/nut-common-tmpfiles.conf (code=exited, status=0/SUCCESS)
   Main PID: 1353 (upsmon)
      Tasks: 2 (limit: 9284)
     Memory: 2.8M (peak: 4.0M)
        CPU: 328ms
     CGroup: /system.slice/nut-monitor.service
             ├─1353 /lib/nut/upsmon -F
             └─1359 /lib/nut/upsmon -F

nov 02 18:24:59 MinipcLG2 nut-monitor[1359]: Poll UPS [apcupskelder@localhost] failed - Driver not connected
nov 02 18:24:59 MinipcLG2 nut-monitor[1359]: Communications with UPS apcupskelder@localhost lost
[... cut ...]
nov 02 18:25:03 MinipcLG2 nut-monitor[1359]: Poll UPS [apcupskelder@localhost] failed - Driver not connected
nov 02 18:25:05 MinipcLG2 nut-monitor[1359]: Poll UPS [apcupskelder@localhost] failed - Driver not connected
nov 02 18:25:07 MinipcLG2 nut-monitor[1359]: Communications with UPS apcupskelder@localhost established
nov 02 18:25:07 MinipcLG2 nut-monitor[5306]: sh: 1: /usr/sbin/upssched.sh: not found
nov 02 18:25:07 MinipcLG2 nut-monitor[5302]: Network UPS Tools upsmon 2.8.1
```

There seem to be some problems...

The latter seems problematic?

ErikDB87, 2024-11-05:
I'm not sure I fully understand what you mean... I'm not really a Linux
master :)

But here are some outputs:
```
***@***.***:/etc/nut$ systemctl status ***@***.***
● ***@***.*** - Network UPS Tools - device driver for
NUT device 'apcupskelder'
     Loaded: loaded ***@***.***; enabled;
preset: enabled)
    Drop-In: ***@***.***
             └─nut-driver-enumerator-generated-checksum.conf,
nut-driver-enumerator-generated.conf
     Active: active (running) since Sat 2024-11-02 20:27:55 CET; 48s ago
    Process: 11894 ExecStartPre=/usr/bin/systemd-tmpfiles --create
/usr/lib/tmpfiles.d/nut-common-tmpfiles.conf (code=exited, status=0/SUCCESS)
    Process: 11896 ExecStart=/bin/sh -c
NUTDEV="`/usr/libexec/nut-driver-enumerator.sh --get-device-for-service
apcupskelder`" && [ -n "$NUTDEV" ] || { echo "FATAL: Could not find a NUT
device section for service unit apcupskelder" >&2 ; exit 1 ; } ; exec
/sbin/upsdrvctl  start "$NUTDEV" (code=exited, status=0/SUCCESS)
   Main PID: 11934 (usbhid-ups)
      Tasks: 1 (limit: 9284)
     Memory: 608.0K (peak: 2.6M)
        CPU: 99ms
     CGroup:
***@***.***
             └─11934 /lib/nut/usbhid-ups -a apcupskelder

nov 02 20:27:55 MinipcLG2 ***@***.***[11923]: HIDParse: LogMax
is less than LogMin. Vendor HID report descriptor may be incorrect;
interpreting LogMax -1 as 255 in ReportID: 0x22
nov 02 20:27:55 MinipcLG2 ***@***.***[11923]: HIDParse: LogMax
is less than LogMin. Vendor HID report descriptor may be incorrect;
interpreting LogMax -1 as 255 in ReportID: 0x40
nov 02 20:27:55 MinipcLG2 ***@***.***[11923]: Using subdriver:
APC HID 0.100
nov 02 20:27:55 MinipcLG2 ***@***.***[11923]: Network UPS Tools
- Generic HID driver 0.52 (2.8.1)
nov 02 20:27:55 MinipcLG2 ***@***.***[11923]: USB communication
driver (libusb 1.0) 0.46
nov 02 20:27:55 MinipcLG2 usbhid-ups[11934]: Startup successful
nov 02 20:27:55 MinipcLG2 usbhid-ups[11934]: upsnotify: failed to notify
about state 2: no notification tech defined, will not spam more about it
nov 02 20:27:55 MinipcLG2 ***@***.***[11896]: Network UPS Tools
- UPS driver controller 2.8.1
nov 02 20:27:55 MinipcLG2 systemd[1]: Started
***@***.*** - Network UPS Tools - device driver for NUT
device 'apcupskelder'.
nov 02 20:27:56 MinipcLG2 usbhid-ups[11934]: sock_connect: enabling
asynchronous mode (auto)
```

```
***@***.***:/etc/nut$ systemctl status nut-server.service
● nut-server.service - Network UPS Tools - power devices information server
     Loaded: loaded (/usr/lib/systemd/system/nut-server.service; enabled;
preset: enabled)
     Active: active (running) since Sat 2024-11-02 20:27:35 CET; 3min 24s
ago
    Process: 11815 ExecStartPre=/usr/bin/systemd-tmpfiles --create
/usr/lib/tmpfiles.d/nut-common-tmpfiles.conf (code=exited, status=0/SUCCESS)
   Main PID: 11818 (upsd)
      Tasks: 1 (limit: 9284)
     Memory: 464.0K (peak: 1.5M)
        CPU: 65ms
     CGroup: /system.slice/nut-server.service
             └─11818 /lib/nut/upsd -F

nov 02 20:27:35 MinipcLG2 nut-server[11818]: Running as foreground process,
[... 81 lines cut ...]

nov 02 20:27:35 MinipcLG2 nut-monitor[11839]: Using power down flag file
/etc/nut/killpower
nov 02 20:27:35 MinipcLG2 nut-monitor[11842]: Init SSL without certificate
database
nov 02 20:27:35 MinipcLG2 nut-monitor[11842]: upsnotify: failed to notify
about state 2: no notification tech defined, will not spam more about it
nov 02 20:27:35 MinipcLG2 nut-monitor[11842]: Login on UPS
***@***.*** failed - got [ERR ACCESS-DENIED]
nov 02 20:27:52 MinipcLG2 nut-monitor[11842]: Poll UPS
***@***.*** failed - Driver not connected
nov 02 20:27:52 MinipcLG2 nut-monitor[11842]: Communications with UPS
***@***.*** lost
nov 02 20:27:52 MinipcLG2 nut-monitor[11925]: Network UPS Tools upsmon 2.8.1
nov 02 20:27:54 MinipcLG2 nut-monitor[11842]: Poll UPS
***@***.*** failed - Driver not connected
nov 02 20:27:56 MinipcLG2 nut-monitor[11842]: Communications with UPS
***@***.*** established
nov 02 20:27:56 MinipcLG2 nut-monitor[11936]: Network UPS Tools upsmon 2.8.1
```

There seem to be some problems with `usbhid-ups-apcupskelder`...

(Again, I don't want to spam, but maybe I reach people here, who aren't
following GitHub closely... Apologies if this seems inappropriate.)

jimklimov, 2024-11-05:
Looks great actually: `nut-driver@apcupskelder.service` initialized around `20:27:55`, and at `20:27:56` accepted a socket connection. That's the same moment logged by `nut-server.service` (aliased as `upsd` to match the program name):
````
nov 02 20:27:56 MinipcLG2 upsd[11818]: Connected to UPS [apcupskelder]: usbhid-ups-apcupskelder
````

`upsc` is not a service, it is a command-line client. You should just run
```
:; upsc apcupskelder
```
(if on the same machine as `upsd`) and it should report a number of `key: value` pairs for some driver settings and all device readings it found.

WARNING: The `nut-monitor.service` (aliased as `upsmon`) did have a hiccup:
```
Login on UPS ***@***.*** failed - got [ERR ACCESS-DENIED]
```

This usually means either mismatch between `/etc/nut/upsd.users` and the credentials in `MONITOR` line of `upsmon.conf`, or bad permissions so one of the daemons can not read its config (or too open permissions so it can refuse to trust that file, whose passwords anyone can know).

I am a bit puzzled about `upsnotify: failed to notify about state 2: no notification tech defined, will not spam more about it` lines (daemon should have noticed it run under systemd, but that is not too consequential).

ErikDB87, 2024-11-06:
> WARNING: The nut-monitor.service (aliased as upsmon) did have a hiccup:\n\nLogin on UPS ***@***.*** failed - got [ERR ACCESS-DENIED]\nThis usually means either mismatch between /etc/nut/upsd.users and the credentials in MONITOR line of upsmon.conf, or bad permissions so one of the daemons can not read its config (or too open permissions so it can refuse to trust that file, whose passwords anyone can know).

I had hoped a fresh install would have reset all the permissions... Do you have any idea which commands might fix the permissions issues?

jimklimov, 2024-11-06:
Looking at your earlier post, the `upsd.users` you've posted seems very wrong. Please check the sample at https://github.com/networkupstools/nut/blob/master/conf/upsd.users.sample about syntax and expected contents. Also maybe change the passwords in the final configuration, just in case ;)

Also note, with an APC BX###MI device, #2565 and related issues and PRs may be relevant. That fix is not part of a released version yet though, so if you'd end up needing it before a release is cut and gets through distro packaging queues - a custom build would be required, e.g. per https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

jimklimov, 2024-11-06:
As for file permissions, can you post `ls -la /etc/nut` output?

ErikDB87, 2024-11-06:
> Looking at your earlier post, the `upsd.users` you've posted seems very wrong. Please check the sample at https://github.com/networkupstools/nut/blob/master/conf/upsd.users.sample about syntax and expected contents. Also maybe change the passwords in the final configuration, just in case ;)

I see... Shame on me for following the guide, and not reading the readme. I did that eventually for a few other conf files, but not this one...

I assume this would be better?
```
[admin]
  password  = indUStr*a
  actions = FSD
  instcmds = ALL
```
?

(I'll have access to the Linux server later today; I'll run `ls -la /etc/nut` then as well.)

jimklimov, 2024-11-06:
Well, coming from legacy long ago, before my time, that `upsd.users` lists two separate user concepts. I don't remember if I mixed both into one user entry:
* User that can subscribe to the data server for monitoring and stay with a persistent session, with an `upsmon <ROLE>` line - that's the one you need to define and use in a `MONITOR` line of `upsmon.conf`
* User who can set variables or send commands, "all" or a selection of some - optional, for NUT clients like `upsrw`, `upscmd` or CGI `upsset`, etc.

ErikDB87, 2024-11-06:
> Well, coming from legacy long ago, before my time, that `upsd.users` lists two separate user concepts. I don't remember if I mixed both into one user entry:
>
> * User that can subscribe to the data [...]

So it should be:
```
[admin]
  password  = indUStr*a
  actions = FSD
  instcmds = ALL
  upsmon primary
```
?

jimklimov, 2024-11-06:
Maybe, if it does work to mix it like that. If something still complains, separate this into two users for two use-cases, e.g.:

````
[admin]
  password  = indUStr*a
  actions = FSD
  instcmds = ALL

[upsmon]
  password  = indUStr*a
  upsmon primary
````

(and then use `MONITOR apcupskelder@localhost 1 upsmon indUStr*a primary` in `upsmon.conf`).

ErikDB87, 2024-11-06:
Okay, so now `/etc/nut/upsd.users` reads as follows:
```
[admin]
  password  = indUStr*a
  actions = FSD
  instcmds = ALL

[upsmon]
  password  = indUStr*a
  upsmon primary
```

`/etc/nut/upsmon.conf` now reads:
```
#RUN_AS_USER nutgebruiker
RUN_AS_USER root
#MONITOR apcupskelder@localhost 1 bekijker indUStr*a primary
#MONITOR apcupskelder@localhost 1 admin indUStr*a master
#MONITOR apcupskelder@localhost 1 admin indUStr*a primary
MONITOR apcupskelder@localhost 1 upsmon indUStr*a primary

MINSUPPLIES 1
SHUTDOWNCMD "/sbin/shutdown -h +0"
NOTIFYCMD /usr/sbin/upssched
POLLFREQ 2
POLLFREQALERT 1
HOSTSYNC 15
DEADTIME 15
POWERDOWNFLAG /etc/nut/killpower

NOTIFYMSG ONLINE    "UPS %s on line power"
NOTIFYMSG ONBATT    "UPS %s on battery"
NOTIFYMSG LOWBATT   "UPS %s battery is low"
NOTIFYMSG FSD       "UPS %s: forced shutdown in progress"
NOTIFYMSG COMMOK    "Communications with UPS %s established"
NOTIFYMSG COMMBAD   "Communications with UPS %s lost"
NOTIFYMSG SHUTDOWN  "Auto logout and shutdown proceeding"
NOTIFYMSG REPLBATT  "UPS %s battery needs to be replaced"
NOTIFYMSG NOCOMM    "UPS %s is unavailable"
NOTIFYMSG NOPARENT  "upsmon parent process died - shutdown impossible"

NOTIFYFLAG ONLINE   SYSLOG+WALL+EXEC
NOTIFYFLAG ONBATT   SYSLOG+WALL+EXEC
NOTIFYFLAG LOWBATT  SYSLOG+WALL
NOTIFYFLAG FSD      SYSLOG+WALL+EXEC
NOTIFYFLAG COMMOK   SYSLOG+WALL+EXEC
NOTIFYFLAG COMMBAD  SYSLOG+WALL+EXEC
NOTIFYFLAG SHUTDOWN SYSLOG+WALL+EXEC
NOTIFYFLAG REPLBATT SYSLOG+WALL
NOTIFYFLAG NOCOMM   SYSLOG+WALL+EXEC
NOTIFYFLAG NOPARENT SYSLOG+WALL

RBWARNTIME 43200

NOCOMMWARNTIME 600

FINALDELAY 5
```

Unfortunately, still:
```
erik@MinipcLG2:~$ sudo upsdrvctl shutdown
Network UPS Tools - UPS driver controller 2.8.1
Network UPS Tools - Generic HID driver 0.52 (2.8.1)
USB communication driver (libusb 1.0) 0.46
select with socket: Invalid argument
Can't claim USB device [051d:0002]@0/0: Entity not found
upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
Driver failed to start (exit status=1)
```

As requested:
```
erik@MinipcLG2:~$ ls -la /etc/nut
totaal 112
drwxr-xr-x   2 root nut   4096 nov  6 14:48 .
drwxr-xr-x 182 root root 12288 nov  6 03:30 ..
-rw-r--r--   1 root root    48 nov  2 18:01 hosts.conf
-rw-r-----   1 root nut     15 nov  2 16:42 nut.conf
-rw-r-----   1 root root  4156 nov  2 16:41 nut.example.conf
-rw-r-----   1 root root  4201 nov  2 16:41 upsched.example.conf
-rw-r-----   1 root nut    385 nov  2 18:01 ups.conf
-rw-r-----   1 root nut     20 nov  2 18:02 upsd.conf
-rw-r-----   1 root root  8247 nov  2 16:41 upsd.example.conf
-rw-r-----   1 root nut    116 nov  6 14:35 upsd.users
-rw-r-----   1 root root  9845 nov  2 16:41 ups.example.conf
-rw-r-----   1 root nut   1381 nov  6 14:35 upsmon.conf
-rw-r-----   1 root root 24344 nov  2 16:41 upsmon.example.conf
-rw-r--r--   1 root root  1120 nov  2 18:03 upssched-cmd.sh
-rw-r-----   1 root nut    745 nov  2 18:03 upssched.conf
```

jimklimov, 2024-11-06:
Looks cool about getting the driver and `upsmon` to play together, finally.

I think `sudo upsdrvctl shutdown` suffers from a similar conflict with a running driver. Would it show more clues if you try passing debug verbosity to this copy of the driver - `sudo upsdrvctl -DDDDDD -d shutdown` ?

I think with #1917 in NUT v2.8.1 the driver program should have tried to communicate with the previous instance over its local socket (same as communications with `upsd`), so the originally running program would remain connected to the UPS and pass it the shutdown command(s). Maybe `select with socket: Invalid argument` is about inability to find or use that local socket, so it tries from scratch (and for some reason does not try to kill off the earlier copy of the driver program - e.g. if it did not leave PID files)?

jimklimov, 2024-11-06:
Alternately, you can try sending commands to the UPS (using the `admin` account created earlier and `upscmd` client).

ErikDB87, 2024-11-06:
> Would it show more clues if you try passing debug verbosity to this copy of the driver - `sudo upsdrvctl -DDDDDD -d shutdown` ?

```
erik@MinipcLG2:~$ sudo upsdrvctl -DDDDDD -d shutdown
Network UPS Tools - UPS driver controller 2.8.1
   0.000000     [D1] upsdrvctl commanding all drivers (1 found): (null)
   0.000009     [D1] Shutdown UPS: apcupskelder
   0.000015     [D2] exec:  /lib/nut/usbhid-ups -a apcupskelder -k
   0.000106     [D2] Starting driver with debug but without explicit backgrounding: will not wait for it to fork and detach, continuing...
   0.000127     [D1] upsdrvctl: successfully finished
   0.000131     [D1] Completed the job of upsdrvctl tool, cleaning up and exiting now
   0.000136     [D1] Completed the job of upsdrvctl tool, clean-up finished, exiting now
```
The UPS didn't power cylce.

> You may be after `upsmon -c fsd` instead, so it would initiate the shutdown, and if the OS integration works well, that would call `sudo upsdrvctl shutdown` in the endgame (when NUT daemons are all down, along with other services), with no conflict for a running driver.
```
erik@MinipcLG2:~$ sudo upsmon -DDDDD -c fsd
Network UPS Tools upsmon 2.8.1
   0.000000     [D1] Signaled old daemon OK
```
The Linux box shut down, but the UPS didn't power cycle...

> Alternately, you can try sending commands to the UPS (using the `admin` account created earlier and `upscmd` client) directly.
You mean like so:
```
erik@MinipcLG2:/etc/nut$ upscmd -u admin -p indUStr*a apcupskelder shutdown.reboot
OK
```
?

Unfortunately, again no power cycle.

(I've got a table lamp lit, so a power loss should be visible.)

ErikDB87, 2024-11-06:
> Can you see if your UPS picks up the others?

I assume you mean the other options mentioned in that Java code?
```
erik@MinipcLG2:/etc/nut$ sudo upscmd -u admin -p indUStr*a apcupskelder shutdown.return
Unexpected response from upsd: ERR CMD-NOT-SUPPORTED
erik@MinipcLG2:/etc/nut$ sudo upscmd -u admin -p indUStr*a apcupskelder shutdown.reboot
OK
erik@MinipcLG2:/etc/nut$ sudo upscmd -u admin -p indUStr*a apcupskelder load.off.delay
OK
```
No power cycle...

jimklimov, 2024-11-06:
And that's plain old C code ;)

ErikDB87, 2024-11-06:
> "OK" there means the command was accepted by the driver. Please try `upscmd -w ...` to wait for the result, maybe it would report more details.

```
erik@MinipcLG2:/etc/nut$ sudo upscmd -u admin -p indUStr*a -w apcupskelder shutdown.reboot
SUCCESS
erik@MinipcLG2:/etc/nut$ sudo upscmd -u admin -p indUStr*a -w apcupskelder shutdown.return
Unexpected response from upsd: ERR CMD-NOT-SUPPORTED
erik@MinipcLG2:/etc/nut$ sudo upscmd -u admin -p indUStr*a -w apcupskelder load.off.delay
SUCCESS
erik@MinipcLG2:/etc/nut$ sudo upscmd -u admin -p indUStr*a -w apcupskelder load.off
SUCCESS
```

> It may also be that the device model/firmware does not actually support that command, or we poke a wrong USB endpoint for that (e.g. worked for other related devices but not this one).

What would that mean for my setup? End of the line? Or are there more avenues to be walked? :)

jimklimov, 2024-11-06:
> What would that mean for my setup? End of the line? Or are there more avenues to be walked? :)

Haven't got many ideas at the moment...

Is the lamp a heavy power-drawer (LED or incandescent)? Maybe the UPS does not notice it has any load, so decides to do nothing?

@desertwitch : You seem to have some recent experience with APC, any more ideas about why shutdown commands might not work?

ErikDB87, 2024-11-06:
> Is the lamp a heavy power-drawer?
No, less than 6 W... But if all devices are shutdown before the UPS reboots, it wouldn't have any load at that moment either, right?

ErikDB87, 2024-11-06:
> Also note, with an APC BX###MI device, #2565 and related issues and PRs may be relevant. That fix is not part of a released version yet though, so if you'd end up needing it before a release is cut [...]

Could this have anything to do with it? That's too high level for me to understand. :)

jimklimov, 2024-11-06:
> But if all devices are shutdown before the UPS reboots, it wouldn't have any load at that moment either, right?

That bit is up to UPS maker. More likely, this is one of devices where APC minimized support for USB HID in favor of Modbus. A chance remains that they moved the needed USB command to a different endpoint, though. You can find several ways of spelling same commands on different models at https://github.com/networkupstools/nut/blob/master/drivers/apc-hid.c#L424-L461 so another one would not be too surprising...

> Could this have anything to do with it?

With the lack of shutdown - probably no.

ErikDB87, 2024-11-06:
> More likely, this is one of devices where APC minimized support for USB HID in favor of Modbus.

Are you suggesting using [the `apc-modbus` driver](https://networkupstools.org/docs/man/apc_modbus.html) is a possible solution? Although... I read that I then should do something with the front panel of the UPS. But mine doesn't have a front panel... 

> A chance remains that they moved the needed USB command to a different endpoint, though. You can find several ways of spelling same commands on different models at https://github.com/networkupstools/nut/blob/master/drivers/apc-hid.c#L424-L461 so another one would not be too surprising...

How did these spellings got found? Not by just trying stuff, I hope? 😅 Maybe the "APC PowerShute" software can be read or something like that? I haven't tried that program yet, though...

jimklimov, 2024-11-06:
To a large extent, by increasing driver verbosity and/or  using an `-x explore` subdriver option, so `usbhid-ups` program dumps all "USB HID reports" it sees, interspersed with how it understood (some of) them. You can see a currently open pull request about Eaton ECO mode for a glimpse of the approach.

ErikDB87, 2024-11-07:
> To a large extent, by increasing driver verbosity and/or using an -x explore subdriver option, so usbhid-ups program dumps all \"USB HID reports\" it sees, interspersed with how it understood (some of) them.

I'm willing to dive into this, but I'd need a bit more guidance, I'm afraid. :)

> You can see a currently open pull request about Eaton ECO mode for a glimpse of the approach.

I read the discussion, but I only see the content being discussed, not the way to discover 'driver commands' (or how it's called). However, I didn't click on any links, so maybe the answer lays there. I'll try to do that tomorrow (or 'later today', technically :)).

jimklimov, 2024-11-07:
Well, at least trying `apc_modbus` might be worthwhile. One caveat is that it requires a custom build of both NUT and libmodbus (to add USB RTU support), see https://github.com/networkupstools/nut/wiki/APC-UPS-with-Modbus-protocol and https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests for links to pre-requisites and in-place build configuration.

Note you can run a driver program straight from the build workspace (via `sudo` probably, for actual device access) without changing the files installed by packaging until you are sure you like the new build better.

desertwitch, 2024-11-07:
I'm not sure if the BX series, being rather entry-level, supports shutdown commands at all.
My Back-UPS Pro takes almost 2 minutes to shutdown after the command - how long did you wait?

ErikDB87, 2024-11-07:
> how long did you wait?

A few seconds... :) I'll try waiting a few minutes!

> I'm not sure if the BX series, being rather entry-level, supports shutdown commands at all.

That would make the endeavor pointless, of course... Maybe I can test it with APC's PowerChute software?

ErikDB87, 2024-11-07:
> Maybe I can test it with APC's PowerChute software?

Apparently, that doesn't do more than shutdown the device. It doesn't power cycle.

So if the BX series don't support shutdown commands, there's no way to avoid a power race?

ErikDB87, 2024-11-07:
> Maybe I can test it with APC's PowerChute software?

@desertwitch, I tested this, and the UPS actually shut down (as also documented [here](https://community.se.com/t5/APC-UPS-for-Home-and-Office-Forum/Can-BX1400UI-UPS-be-powered-up-or-down-by-a-software-command/td-p/295302)). So there must be some command that the UPS understands to shut down.

jimklimov, 2024-11-10:
Looks about right.

You might need `libusb-dev` with headers (maybe named to suit the libusb 0.1 vs. 1.0 library), but if your builds succeeded - you have something of the sort anyway.

It can also help to use that `./configure --enable-inplace-runtime` argument, so it would set the paths and user/group accounts similar to the installed NUT, so the same `udev` and filesystem permissions that were installed by the package, and config/socket/... file locations, would apply to the custom build.

Ultimately, in the NUT workspace you would have the `./drivers/apc_modbus` program which you can test against your device (with packaged and systemd-wrapped NUT drivers stopped), e.g. as `root` (or via `sudo`):
````
:; sudo systemctl stop nut-driver@apcupskelder

### Run the driver to see how it probes the device etc.
### If this fully works, `-d1` should end with a device data dump
### similar to an `upsc` report after one collection loop.
:; sudo ./drivers/apc_modbus -s tempups  -d1 -DDDDDD -x port=auto

:; sudo systemctl start nut-driver@apcupskelder
````

ErikDB87, 2024-11-10:
Thanks!

> Ultimately, in the NUT workspace you would have the ./drivers/apc_modbus program which you can test against your device (with packaged and systemd-wrapped NUT drivers stopped), e.g. as root (or via sudo):

That is in the case I don't remove the package installs? Or do you advise against removing them altogether?

jimklimov, 2024-11-10:
That is a bit orthogonal.

* Just building custom NUT is not related to having or removing packaged NUT.
* Configuring the custom build as a possible "in-place" replacement of whatever is installed helps test the new programs, using configuration and user accounts that are already configured by the package or older build.
* You can test the newly built programs right from the build workspace, only need to stop the packaged equivalents (`systemctl stop` their services) while testing the custom build.
* Only when you are satisfied with the results (e.g. if `apc_modbus` works at all, and better than `usbhid-ups`, for you), you can proceed to `sudo make install` of the custom build where it would overwrite some of the packaged files and probably make some others irrelevant (e.g. Debian-packaged `/lib/nut` location can become largely useless if the custom drivers land elsewhere).
** That wiki page also details how to enable/restart some systemd units, do that :)
* Un-installation of packages may lead to loss of configs and/or deletion of `nut` user/group accounts, which is something you may not want. Having packages formally remaining in place also makes future updates easier, when (assuming) the packaged builds include what you custom-patched and more.

ErikDB87, 2024-11-10:
> Looks about right.
>
> You might need `libusb-dev` with headers (maybe named to suit the libusb 0.1 vs. 1.0 library), but if your builds succeeded - you have something of the sort anyway.
>
> It can also [...]

I ended up installing some more of the prerequisite stuff:
```
apt get-install \
    libusb-dev libusb-1.0-0-dev
apt-get install \
    libgd-dev
apt-get install \
    asciidoc source-highlight python3-pygments dblatex
```

But even then, the one to last step failed:
```
erik@MinipcLG2:~/nut$ sudo ./configure --enable-inplace-runtime --prefix=/usr/local/ --with-drivers=apc_modbus --with-usb --with-modbus --with-modbus-includes=-I/usr/local/include/modbus --with-modbus-libs="-L/usr/local/lib -lmodbus"
checking for CONFIG_FLAGS... --prefix=/usr/local/ --with-drivers=apc_modbus --with-usb --with-modbus --with-modbus-includes=-I/usr/local/include/modbus --with-modbus-libs='-L/usr/local/lib -lmodbus' --enable-inplace-runtime
Network UPS Tools version 2.8.2.1151 (2.8.2.1151-1151-g6941fb88d) development iteration
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... x86_64-pc-linux-gnu
checking for host system name... mint
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for autoconf macro to enable system extensions... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
checking for unistd.h... yes
checking for wchar.h... yes
checking for minix/config.h... no
checking for stdbool.h... yes
checking for sys/time.h... yes
checking for time.h... yes
checking for fcntl.h... yes
checking for sys/socket.h... yes
checking for netdb.h... yes
checking for ifaddrs.h... yes
checking for netinet/in.h... yes
checking for net/if.h... yes
checking whether it is safe to define __EXTENSIONS__... yes
checking whether _XOPEN_SOURCE should be defined... no
checking how to run the C preprocessor... gcc -E
checking for g++... g++
checking whether the compiler supports GNU C++... yes
[... 533 lines cut ...]
checking if asciidoc and a2x versions can build html-single (minimum required asciidoc-8.1.0 a2x-1.0.0)... <unknown>:1: SyntaxWarning: invalid escape sequence '\S'
<unknown>:1: SyntaxWarning: invalid escape sequence '\S'
yes
checking desire and ability to build html-chunked documentation... yes
checking if asciidoc and a2x versions can build html-chunked (minimum required asciidoc-8.1.0 a2x-1.0.0)... <unknown>:1: SyntaxWarning: invalid escape sequence '\S'
<unknown>:1: SyntaxWarning: invalid escape sequence '\S'
a2x: ERROR: "xsltproc"  --nonet --stringparam callout.graphics 0 --stringparam navig.graphics 0 --stringparam admon.textlabel 1 --stringparam admon.graphics 0  --stringparam base.dir "FAQ.chunked/" "/home/erik/nut/docs/chunked.xsl" "/home/erik/nut/configure-test.docbuild.143257.F9258D7/FAQ.xml" returned non-zero exit status 9

no
configure: WARNING: Unable to build html-chunked documentation which you requested
checking desire and ability to build pdf documentation... yes
checking if dblatex version can build pdf (minimum required 0.2.5)... a2x: WARNING: --destination-dir option is only applicable to HTML and manpage based outputs

<unknown>:1: SyntaxWarning: invalid escape sequence '\S'
<unknown>:1: SyntaxWarning: invalid escape sequence '\S'
yes
checking if dblatex can process non-ASCII section titles for PDF... a2x: WARNING: --destination-dir option is only applicable to HTML and manpage based outputs

<unknown>:1: SyntaxWarning: invalid escape sequence '\S'
<unknown>:1: SyntaxWarning: invalid escape sequence '\S'
yes
configure: error: Unable to build html-chunked documentation (check for 'no' results above)
```

But this is Chinese to me...

jimklimov, 2024-11-10:
I saw some warnings about "invalid escape sequence" in Windows or MacOS builds, I think, but they were not immediately fatal there :\ (and generated docs did not cursorily seem broken either)

As a shortcut for now, you can `configure --without-docs` or `--with-docs=man` to avoid this hassle while investigating if the driver is okay.

ErikDB87, 2024-11-10:
> As a shortcut for now, you can `configure --without-docs` or `--with-docs=man` to avoid this hassle while investigating if the driver is okay.

This looks successful!
```
erik@MinipcLG2:~/nut$ sudo ./configure --with-docs=man
checking for CONFIG_FLAGS... --with-docs=man
Network UPS Tools version 2.8.2.1151 (2.8.2.1151-1151-g6941fb88d) development iteration
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... x86_64-pc-linux-gnu
checking for host system name... mint
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for autoconf macro to enable system extensions... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
checking for unistd.h... yes
checking for wchar.h... yes
checking for minix/config.h... no
checking for stdbool.h... yes
checking for sys/time.h... yes
checking for time.h... yes
checking for fcntl.h... yes
checking for sys/socket.h... yes
checking for netdb.h... yes
checking for ifaddrs.h... yes
checking for netinet/in.h... yes
checking for net/if.h... yes
checking whether it is safe to define __EXTENSIONS__... yes
checking whether _XOPEN_SOURCE should be defined... no
checking how to run the C preprocessor... gcc -E
checking for g++... g++
checking whether the compiler supports GNU C++... yes
checking whether g++ accepts -g... yes
checking for g++ option to enable C++11 features... none needed
checking dependency style of g++... gcc3
checking whether g++ understands -c and -o together... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking whether this make implementation supports export VAR=VAL syntax... yes
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking whether usable PKG_CONFIG was already detected by autoconf... yes: /usr/bin/pkg-config
configure: checking for autoconf macro support of pkg-config (/usr/bin/pkg-config)
configure: checking for autoconf macro support of pkg-config module checker
checking for pkg-config... yes
checking for pkg-config envvar PKG_CONFIG_PATH (if passed to configure script)...
checking for pkg-config reported pc_path... /usr/local/lib/x86_64-linux-gnu/pkgconfig:/usr/local/lib/pkgconfig:/usr/local/share/pkgconfig:/usr/lib/x86_64-linux-gnu/pkgconfig:/usr/lib/pkgconfig:/usr/share/pkgconfig
[... 748 lines cut ...]
-----------------------------

* Compact version of C compiler:        gcc (Ubuntu 13.2.0-23ubuntu4) 13.2.0
* Compact version of C++ compiler:      g++ (Ubuntu 13.2.0-23ubuntu4) 13.2.0
* Compact version of C preprocessor:    gcc (Ubuntu 13.2.0-23ubuntu4) 13.2.0
* host env spec we run on:      x86_64-pc-linux-gnu
* host env spec we built on:    x86_64-pc-linux-gnu
* host env spec we built for:   x86_64-pc-linux-gnu
* host OS short spec we run on: x86_64-linux-gnu
* host OS short spec we built on:       x86_64-linux-gnu
* host OS short spec we built for:      x86_64-linux-gnu
* host multiarch spec we build for (as suggested by compiler being used):       x86_64-linux-gnu
* ccache namespace tag (if ccache is used and new enough):      nut:x86_64-linux-gnu

NUT Compiler settings:
----------------------

* CC            : gcc
* CFLAGS        : -isystem /usr/local/include -g -O2 -Wno-reserved-identifier -Wno-unknown-warning-option -std=gnu99 -Wno-system-headers -Wall -Wextra -Wsign-compare -pedantic -Wno-error
* CXX           : g++
* CXXFLAGS      : -isystem /usr/local/include -g -O2 -Wno-reserved-identifier -Wno-unknown-warning-option -std=gnu++11 -Wno-system-headers -Wall -Wextra -Wno-error
* CPP           : gcc -E
* CPPFLAGS      :
* CONFIG_FLAGS  : --with-docs=man
```

ErikDB87, 2024-11-10:
But...

```
erik@MinipcLG2:~/nut$ sudo systemctl stop nut-driver@apcupskelder
erik@MinipcLG2:~/nut$ sudo ./drivers/apc_modbus -s tempups  -d1 -DDDDDD -x port=auto
sudo: ./drivers/apc_modbus: opdracht niet gevonden
```
“command not found”...

jimklimov, 2024-11-10:
Actually, before installing, simple `make` is better so your unprivileged user can clean it up and remake when needed.

```
* User to run as:       nobody
* Group of user to run as:      nogroup
...
* Config file path:     /usr/local/ups/etc
```

These lines indicate you did not configure for an in-place build after all, so when testing the new driver pass it also the `-u root` argument so it does not drop privileges (otherwise it would run as `nobody` trying to access `nut`-owned devfs nodes as prepared by package `udev` configuration files).

ErikDB87, 2024-11-10:
> And now `sudo make`, of course... :)

For the sake of completeness:
```
erik@MinipcLG2:~/nut$ sudo make
Making all in include
make[1]: Map '/home/erik/nut/include' wordt binnengegaan
SEMVER=2.8.2; TRUNK='master'; BASE='6941fb88dc18490213834fe6345b373d0c29b7d7'; DESC='v2.8.2-1151-g6941fb88d' => TAG='v2.8.2' + SUFFIX='-1151-g6941fb88d' => VER5='2.8.2.1151.0' => VER50='2.8.2.1151' => DESC50='2.8.2.1151-1151-g6941fb88d'
NUT_VERSION: "2.8.2.1151-1151-g6941fb88d"  NUT_VERSION_IS_RELEASE:false  NUT_VERSION_SEMVER: "2.8.2"
test -f "nut_version.h" || cp "nut_version.h.tmp" "nut_version.h"
cmp -s "nut_version.h.tmp" "nut_version.h" || cp "nut_version.h.tmp" "nut_version.h"
rm -f "nut_version.h.tmp"
make  all-am
make[2]: Map '/home/erik/nut/include' wordt binnengegaan
make[2]: Map '/home/erik/nut/include' wordt verlaten
make[1]: Map '/home/erik/nut/include' wordt verlaten
Making all in common
make[1]: Map '/home/erik/nut/common' wordt binnengegaan
make  all-am
make[2]: Map '/home/erik/nut/common' wordt binnengegaan
  CC       parseconf.lo
  CCLD     libparseconf.la
  CC       libcommon_la-state.lo
  CC       libcommon_la-str.lo
  CC       libcommon_la-upsconf.lo
  CC       libcommon_la-common.lo
  CCLD     libcommon.la
  CC       libcommonclient_la-state.lo
  CC       libcommonclient_la-str.lo
  CC       libcommonclient_la-common.lo
  CCLD     libcommonclient.la
  CXX      libnutconf_la-nutconf.lo
  CXX      libnutconf_la-nutstream.lo
  CXX      libnutconf_la-nutwriter.lo
  CXX      libnutconf_la-nutipc.lo
  CXXLD    libnutconf.la
  CC       libcommonstr_la-str.lo
  CC       libcommonstr_la-common.lo
  CCLD     libcommonstr.la
make[2]: Map '/home/erik/nut/common' wordt verlaten
make[1]: Map '/home/erik/nut/common' wordt verlaten
Making all in clients
make[1]: Map '/home/erik/nut/clients' wordt binnengegaan
  CC       upsc.o
make[2]: Map '/home/erik/nut/common' wordt binnengegaan
make[2]: 'libcommonclient.la' is up-to-date.
make[2]: Map '/home/erik/nut/common' wordt verlaten
  CC       upsclient.lo
  CCLD     libupsclient.la
  CCLD     upsc
  CC       upslog.o
  CCLD     upslog
  CC       upsrw.o
  CCLD     upsrw
  CC       upscmd.o
  CCLD     upscmd
  CC       upsmon.o
make[2]: Map '/home/erik/nut/common' wordt binnengegaan
make[2]: 'libcommon.la' is up-to-date.
make[2]: Map '/home/erik/nut/common' wordt verlaten
  CCLD     upsmon
  CC       upssched.o
make[2]: Map '/home/erik/nut/common' wordt binnengegaan
make[2]: 'libparseconf.la' is up-to-date.
make[2]: Map '/home/erik/nut/common' wordt verlaten
  CCLD     upssched
  CXX      nutclient.lo
  CXXLD    libnutclient.la
  CXX      nutclientmem.lo
  CXXLD    libnutclientstub.la
[... 664 lines cut ...]
make[1]: Map '/home/erik/nut' wordt verlaten
make[1]: Map '/home/erik/nut' wordt binnengegaan
cd /home/erik/nut/docs && make  -s /home/erik/nut/docs/.prep-src-docs
cd /home/erik/nut/docs/man && make  -s /home/erik/nut/docs/man/.prep-src-docs
cd /home/erik/nut/docs && make  doc
make[2]: Map '/home/erik/nut/docs' wordt binnengegaan
cd /home/erik/nut/docs/man/ && make  -f Makefile all
make[3]: Map '/home/erik/nut/docs/man' wordt binnengegaan
make[3]: Er hoeft niets gedaan te worden voor 'all'.
make[3]: Map '/home/erik/nut/docs/man' wordt verlaten
make[2]: Map '/home/erik/nut/docs' wordt verlaten
make[1]: Map '/home/erik/nut' wordt verlaten
make[1]: Map '/home/erik/nut' wordt binnengegaan
cd /home/erik/nut/docs && make  -s /home/erik/nut/docs/.prep-src-docs
cd /home/erik/nut/docs/man && make  -s /home/erik/nut/docs/man/.prep-src-docs
cd /home/erik/nut/docs && make  doc
make[2]: Map '/home/erik/nut/docs' wordt binnengegaan
cd /home/erik/nut/docs/man/ && make  -f Makefile all
make[3]: Map '/home/erik/nut/docs/man' wordt binnengegaan
make[3]: Er hoeft niets gedaan te worden voor 'all'.
make[3]: Map '/home/erik/nut/docs/man' wordt verlaten
make[2]: Map '/home/erik/nut/docs' wordt verlaten
make[1]: Map '/home/erik/nut' wordt verlaten
```
(Is there no way to make these fields collapsable?)

ErikDB87, 2024-11-10:
> Actually, before installing, simple `make` is better so your unprivileged user can clean it up and remake when needed.
>
> ```
> * User to run as:       nobody
> * Group of user to run as:      nogroup
> ... [...]

I already went ahead and did `sudo make`... Is that reversible?

Anyway:
```
erik@MinipcLG2:~/nut$ sudo ./drivers/apc_modbus -u root -s tempups  -d1 -DDDDDD -x port=auto
   0.000000     [D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools 2.8.2.1151-1151-g6941fb88d (development iteration after 2.8.2) - NUT APC Modbus driver 0.10
   0.000054     [D1] Built-in default or configured user for drivers 'nobody' was ignored due to 'root' specified on command line
   0.000063     [D3] main_arg: var='port' val='auto'
   0.000069     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000075     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000086     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000091     [D1] Network UPS Tools version 2.8.2.1151-1151-g6941fb88d (development iteration after 2.8.2) built with gcc (Ubuntu 13.2.0-23ubuntu4) 13.2.0 and configured with flags: --with-docs=man
   0.000103     [D1] debug level is '6'
   0.000110     [D5] send_to_all: SETINFO driver.debug "6"
   0.000117     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.000637     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.000645     [D1] Signalling UPS [tempups]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.000654     Can't open /var/state/ups/apc_modbus-tempups: No such file or directory
   0.000658     [D1] Request for other driver to exit returned code -1
   0.000663     [D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
   0.000668     [D5] send_to_all: SETINFO device.type "ups"
   0.000675     [D5] send_to_all: SETINFO driver.state "init.device"
Opening auto at 9600 bauds (N, 8, 1)
ERROR Can't open the device auto (No such file or directory)
   0.000698     modbus_connect: unable to connect: No such file or directory
   0.000705     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
```

jimklimov, 2024-11-10:
Hm, seems it got built without libusb support; did you use ONLY `--with-docs=man` argument for `configure`? Others for libmodbus and USB combo were needed too :)

ErikDB87, 2024-11-10:
> Hm, seems it got built without libusb support; did you use ONLY `--with-docs=man` argument for `configure`? Others for libmodbus and USB combo were needed too :)

Ah, I did not read your answer that way 😉 So this:
```
sudo ./configure --with-drivers=apc_modbus --with-usb --with-modbus \ --with-modbus-includes=-I/usr/local/include/modbus \ --with-modbus-libs="-L/usr/local/lib -lmodbus" --without-docs
```
?

jimklimov, 2024-11-10:
To clean up, try `make distclean` before re-running `configure`.

jimklimov, 2024-11-10:
Alternately you can restore ownership of the workspace to yourself:
````
sudo chown -R `id -u`
````
and then let Git remove all the files that are not tracked:
```
git clean -fffdddxxx
```

ErikDB87, 2024-11-10:
> and then follow up with autogen.sh && ./configure \u003Cyour args> && make all without sudo :)

This time, the arguments look good? :)

jimklimov, 2024-11-11:
yes, and seems good :)

ErikDB87, 2024-11-11:
Okay, I went through all the suggested steps. The results:
```
        libmodbus 3.1.10
        ===============

        prefix:                 /usr/local
        sysconfdir:             ${prefix}/etc
        libdir:                 ${exec_prefix}/lib
        includedir:             ${prefix}/include

        compiler:               gcc
        cflags:                 -g -O2
        ldflags:

        tests:                  yes
```
&
```
NUT Configuration summary:
==========================

* configured version:   2.8.2.1151 (2.8.2.1151-1151-g6941fb88d) development iteration
* only build specific drivers:  apc_modbus
* enable SSL support:   yes (OpenSSL)
* build serial drivers: yes
* build USB drivers:    yes (libusb-1.0)
* build neon based XML driver:  no
* enable Avahi support: no
* build Powerman PDU client driver:     no
* build Modbus driver:  yes
* build IPMI driver:    no
* build GPIO driver:    no
* build Mac OS X meta-driver:   no
* build i2c based drivers:      no
* enable libwrap (tcp-wrappers) support:        no
* enable libltdl (Libtool dlopen abstraction) support:  yes
* build nut-scanner:    yes
* build CGI programs:   no
* install NUT-Monitor desktop application:      no
* install PyNUT binding module: yes
* use default  Python  interpreter:     /home/erik/.pyenv/shims/python
* use specific Python2 interpreter:
* use specific Python3 interpreter:     /usr/bin/python3.12
* build and install documentation:      no
* build specific documentation format(s):       no
* build and install the development files:      no
* consider basic SMF support:   no
* consider basic systemd support:       yes
* build with tighter systemd support:   no
* build C++11 codebase (client library, etc.):  yes
* build C++ tests with CPPUNIT: no
* build and install the nutconf tool (experimental, may lack support for recent NUT options):   yes
* build SNMP drivers:   no
* build SNMP drivers with statically linked lib(net)snmp:       no
* User to run as:       nut
* Group of user to run as:      nut

NUT Paths:
----------

* Default installation prefix path:     /usr
* State file path:      /var/state/ups
* Unprivileged PID file path:   /var/state/ups
* Privileged PID file path:     /run
* Driver program path:  /usr/bin
* CGI program path:     /usr/cgi-bin
* HTML file path:       /usr/html
* Config file path:     /etc/nut
* Data file path:       /usr/share
* Tool program path:    /usr/bin
* System program path:  /usr/sbin
* System library path:  /usr/lib
* System exec-library path:     /usr/libexec

NUT Paths for third-party integrations:
---------------------------------------

* Default  Python  interpreter site-packages:   /usr/local/lib/python3.12/dist-packages
* Specific Python2 interpreter site-packages:
* Specific Python3 interpreter site-packages:   /usr/local/lib/python3.12/dist-packages
* pkg-config *.pc directory:    ${libdir}/pkgconfig => /usr/lib/pkgconfig
* Service units for systemd:    /usr/lib/systemd/system
* Shutdown hooks for systemd:   /usr/lib/systemd/system-shutdown
* Systemd-tmpfiles configs:     /usr/lib/tmpfiles.d
* Augeas lenses directory:      ${datarootdir}/augeas/lenses
* Udev rules directory: /lib/udev

NUT Build/Target system info:
-----------------------------

* Compact version of C compiler:        gcc (Ubuntu 13.2.0-23ubuntu4) 13.2.0
* Compact version of C++ compiler:      g++ (Ubuntu 13.2.0-23ubuntu4) 13.2.0
* Compact version of C preprocessor:    gcc (Ubuntu 13.2.0-23ubuntu4) 13.2.0
* host env spec we run on:      x86_64-pc-linux-gnu
* host env spec we built on:    x86_64-pc-linux-gnu
* host env spec we built for:   x86_64-pc-linux-gnu
* host OS short spec we run on: x86_64-linux-gnu
* host OS short spec we built on:       x86_64-linux-gnu
* host OS short spec we built for:      x86_64-linux-gnu
* host multiarch spec we build for (as suggested by compiler being used):       x86_64-linux-gnu
* ccache namespace tag (if ccache is used and new enough):      nut:x86_64-linux-gnu

NUT Compiler settings:
----------------------

* CC            : gcc
* CFLAGS        : -isystem /usr/local/include -g -O2 -Wno-reserved-identifier -Wno-unknown-warning-option -std=gnu99 -Wno-system-headers -Wall -Wextra -Wsign-compare -pedantic -Wno-error
* CXX           : g++
* CXXFLAGS      : -isystem /usr/local/include -g -O2 -Wno-reserved-identifier -Wno-unknown-warning-option -std=gnu++11 -Wno-system-headers -Wall -Wextra -Wno-error
* CPP           : gcc -E
* CPPFLAGS      :
* CONFIG_FLAGS  : --prefix=/usr --sbindir=/usr/sbin --bindir=/usr/bin --sysconfdir=/etc/nut --with-group=nut --with-user=nut --with-drivers=apc_modbus --with-usb --with-modbus --with-modbus-includes=-I/usr/local/include/modbus --with-modbus-libs='-L/usr/local/lib -lmodbus' --without-docs
```

ErikDB87, 2024-11-11:
Alas...
```
erik@MinipcLG2:~/nut$ ./drivers/apc_modbus -u root -s tempups  -d1 -DDDDDD -x port=auto
   0.000000     [D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools 2.8.2.1151-1151-g6941fb88d (development iteration after 2.8.2) - NUT APC Modbus driver 0.10
   0.000194     [D1] Built-in default or configured user for drivers 'nut' was ignored due to 'root' specified on command line
   0.000228     [D3] main_arg: var='port' val='auto'
   0.000252     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000273     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000340     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000362     [D1] Network UPS Tools version 2.8.2.1151-1151-g6941fb88d (development iteration after 2.8.2) built with gcc (Ubuntu 13.2.0-23ubuntu4) 13.2.0 and configured with flags: --prefix=/usr --sbindir=/usr/sbin --bindir=/usr/bin --sysconfdir=/etc/nut --with-group=nut --with-user=nut --with-drivers=apc_modbus --with-usb --with-modbus --with-modbus-includes=-I/usr/local/include/modbus --with-modbus-libs='-L/usr/local/lib -lmodbus' --without-docs
   0.000464     [D1] debug level is '6'
   0.000492     [D5] send_to_all: SETINFO driver.debug "6"
   0.000534     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.001018     [D1] Can not become_user(root): not root initially, remaining UID=1000 GID=1000
   0.001044     [D1] Signalling UPS [tempups]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.001082     Can't open /var/state/ups/apc_modbus-tempups: No such file or directory
   0.001105     [D1] Request for other driver to exit returned code -1
   0.001142     [D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
   0.001168     [D5] send_to_all: SETINFO device.type "ups"
   0.001191     [D5] send_to_all: SETINFO driver.state "init.device"
Opening auto at 9600 bauds (N, 8, 1)
ERROR Can't open the device auto (No such file or directory)
   0.001281     modbus_connect: unable to connect: No such file or directory
   0.001304     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
```
Maybe I did something wrong in the `libmodbus` part... I think I included that `--enable-inplace-runtime` argument there... Is there a way to reverse the `make install`?

jimklimov, 2024-11-11:
Can you double-check that the custom `libmodbus` did build with USB support? Perhaps its `config.log` in the build directory can shed a light on this. If you built it before installing `libusb*-dev`, it might have skipped that part, I suppose.

ErikDB87, 2024-11-11:
> Can you double-check that the custom `libmodbus` did build with USB support? Perhaps its `config.log` in the build directory can shed a light on this. If you built it before installing `libusb*-dev`, it might have skipped that part, I suppose.

```
erik@MinipcLG2:~/libmodbus$ cat config.log | grep usb
  $ ./configure --with-libusb --enable-static --disable-shared --prefix=/usr/local/
configure:18372: checking for libusb_init in -lusb-1.0
configure:18395: gcc -o conftest -g -O2   conftest.c -lusb-1.0   >&5
ac_cv_lib_usb_1_0_libusb_init=yes
LIBUSB_LIBS='-lusb-1.0'
```

Does that shed light? :)

Can I "redo" the "lobmodbus" process? I don't know how to "reverse" the `make install` I did there...

---------------------
For the sake of completeness:
```
erik@MinipcLG2:~/libmodbus$ cat config.log
This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by libmodbus configure 3.1.10, which was
generated by GNU Autoconf 2.71.  Invocation command line was

  $ ./configure --with-libusb --enable-static --disable-shared --prefix=/usr/local/

## --------- ##
## Platform. ##
## --------- ##

hostname = MinipcLG2
uname -m = x86_64
uname -r = 6.8.0-48-generic
uname -s = Linux
uname -v = #48-Ubuntu SMP PREEMPT_DYNAMIC Fri Sep 27 14:04:52 UTC 2024

/usr/bin/uname -p = x86_64
/bin/uname -X     = unknown

/bin/arch              = x86_64
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /home/erik/.pyenv/shims/
PATH: /home/erik/.pyenv/bin/
PATH: /usr/local/sbin/
PATH: /usr/local/bin/
PATH: /usr/sbin/
PATH: /usr/bin/
PATH: /sbin/
PATH: /bin/
PATH: /usr/games/
PATH: /usr/local/games/
PATH: /snap/bin/
PATH: /usr/lib/jvm/java-17-oracle/bin/
PATH: /usr/lib/jvm/java-17-oracle/db/bin/

## ----------- ##
## Core tests. ##
## ----------- ##

configure:2980: looking for aux files: ltmain.sh config.guess config.sub compile missing install-sh
configure:2993:  trying ./build-aux/
configure:3022:   ./build-aux/ltmain.sh found
configure:3022:   ./build-aux/config.guess found
[... 1851 lines cut ...]
#define HAVE_NETINET_TCP_H 1
#define HAVE_SYS_IOCTL_H 1
#define HAVE_SYS_SOCKET_H 1
#define HAVE_SYS_TIME_H 1
#define HAVE_SYS_TYPES_H 1
#define HAVE_TERMIOS_H 1
#define HAVE_TIME_H 1
#define HAVE_UNISTD_H 1
#define HAVE_DECL___CYGWIN__ 0
#define HAVE_ACCEPT4 1
#define HAVE_GETADDRINFO 1
#define HAVE_GETTIMEOFDAY 1
#define HAVE_INET_PTON 1
#define HAVE_INET_NTOP 1
#define HAVE_SELECT 1
#define HAVE_SOCKET 1
#define HAVE_STRERROR 1
#define HAVE_STRLCPY 1
#define WINVER 0x0501
#define HAVE_DECL_TIOCSRS485 1
#define HAVE_DECL_TIOCM_RTS 1
#define HAVE_LIBUSB 1

configure: exit 0
```

ErikDB87, 2024-11-11:
<details>
<summary>Click to expand</summary>

This is the content of the collapsible section. You can include any Markdown-formatted text, lists, or code here.

</details>

jimklimov, 2024-11-11:
For reversing, there's `make uninstall` from the original build, or another `make install` to place newly built files into presumably same locations.

The `./configure --with-libusb --enable-static --disable-shared --prefix=/usr/local/` and further log messages do seem right.

Looking at NUT driver sources, it relies on `NUT_MODBUS_HAS_USB` value that should be defined in `include/config.h` (generated by `configure` script); may also be mentioned in `config.log` (both in NUT build directory) -- I wonder if it somehow did not pick up the ability of custom libmodbus to work with USB?..

ErikDB87, 2024-11-11:
> For reversing, there's `make uninstall` from the original build, or another `make install` to place newly built files into presumably same locations.

```
erik@MinipcLG2:~/libmodbus$ make uninstall
Making uninstall in src
 /bin/bash ../libtool   --mode=uninstall rm -f '/usr/local/lib/libmodbus.la'
libtool: uninstall: rm -f /usr/local/lib/libmodbus.la /usr/local/lib/libmodbus.a
rm: cannot remove '/usr/local/lib/libmodbus.la': Permission denied
rm: cannot remove '/usr/local/lib/libmodbus.a': Permission denied
make[1]: *** [Makefile:455: uninstall-libLTLIBRARIES] Fout 1
make: *** [Makefile:502: uninstall-recursive] Fout 1
erik@MinipcLG2:~/libmodbus$ sudo make uninstall
[sudo] wachtwoord voor erik:
Making uninstall in src
 /bin/bash ../libtool   --mode=uninstall rm -f '/usr/local/lib/libmodbus.la'
libtool: uninstall: rm -f /usr/local/lib/libmodbus.la /usr/local/lib/libmodbus.a
 ( cd '/usr/local/include/modbus' && rm -f modbus.h modbus-config.h modbus-version.h modbus-rtu.h modbus-tcp.h modbus-rtu-usb.h )
Making uninstall in tests
make[1]: Er hoeft niets gedaan te worden voor 'uninstall'.
 ( cd '/usr/local/share/doc/libmodbus' && rm -f AUTHORS NEWS README.md )
 ( cd '/usr/local/lib/pkgconfig' && rm -f libmodbus.pc )
```
I tried a new `make install` afterwards, in case some stuff did get uninstalled. But that failed:
```
erik@MinipcLG2:~/libmodbus$ make install
Making install in src
 /usr/bin/mkdir -p '/usr/local/lib'
 /bin/bash ../libtool   --mode=install /usr/bin/install -c   libmodbus.la '/usr/local/lib'
libtool: install: /usr/bin/install -c .libs/libmodbus.lai /usr/local/lib/libmodbus.la
/usr/bin/install: cannot create regular file '/usr/local/lib/libmodbus.la': Permission denied
make[2]: *** [Makefile:440: install-libLTLIBRARIES] Fout 1
make[1]: *** [Makefile:642: install-am] Fout 2
make: *** [Makefile:502: install-recursive] Fout 1
```

Which makes me think maybe there is a permissions problem with `/usr/local`? Then again, that is the directory the readme suggested...

> Looking at NUT driver sources, it relies on `NUT_MODBUS_HAS_USB` value that should be defined in `include/config.h` (generated by `configure` script); may also be mentioned in `config.log` (both in [...]

```
erik@MinipcLG2:~/nut$ cat include/config.h | grep NUT_MODBUS_HAS_USB
/* #undef NUT_MODBUS_HAS_USB */
```
<details>
<summary>erik@MinipcLG2:~/nut$ cat include/config.h</summary>
```
erik@MinipcLG2:~/nut$ cat include/config.h
/* include/config.h.  Generated from config.h.in by configure.  */
/* include/config.h.in.  Generated from configure.ac by autoheader.  */

/* NUT build directory when the build was configured */
#define ABS_TOP_BUILDDIR "/home/erik/nut"

/* NUT source directory when the build was configured */
#define ABS_TOP_SRCDIR "/home/erik/nut"

/* Define if building universal (internal helper macro) */
/* #undef AC_APPLE_UNIVERSAL_BUILD */

/* Path for pid files of processes not running as root */
#define ALTPIDPATH "/var/state/ups"

/* host env spec we built on */
#define AUTOTOOLS_BUILD_ALIAS "x86_64-pc-linux-gnu"

/* host OS short spec we built on */
#define AUTOTOOLS_BUILD_SHORT_ALIAS "x86_64-linux-gnu"

/* host env spec we run on */
#define AUTOTOOLS_HOST_ALIAS "x86_64-pc-linux-gnu"
[... 1614 lines cut ...]
# endif
#else
# ifndef WORDS_BIGENDIAN
/* #  undef WORDS_BIGENDIAN */
# endif
#endif

/* Define to 1 if C does not support variable-length arrays, and if the
   compiler does not already define this. */
/* #undef __STDC_NO_VLA__ */

/* Replace missing __func__ declaration */
/* #undef __func__ */

/* Define to `__inline__' or `__inline' if that's what the C compiler
   calls it, or to nothing if 'inline' is not supported under any name.  */
#ifndef __cplusplus
/* #undef inline */
#endif

/* type to use in place of socklen_t if not defined */
/* #undef socklen_t */
```

</details>

jimklimov, 2024-11-11:
Note also `*sudo* make install` for the system-owned locations (like `/usr/local`).

And ultimately, for some reason `NUT_MODBUS_HAS_USB` was not defined (as `1`) in the latest build...

ErikDB87, 2024-11-11:
> Note also `*sudo* make install` for the system-owned locations (like `/usr/local`).

```
erik@MinipcLG2:~/libmodbus$ sudo make install
Making install in src
 /usr/bin/mkdir -p '/usr/local/lib'
 /bin/bash ../libtool   --mode=install /usr/bin/install -c   libmodbus.la '/usr/local/lib'
libtool: install: /usr/bin/install -c .libs/libmodbus.lai /usr/local/lib/libmodbus.la
libtool: install: /usr/bin/install -c .libs/libmodbus.a /usr/local/lib/libmodbus.a
libtool: install: chmod 644 /usr/local/lib/libmodbus.a
libtool: install: ranlib /usr/local/lib/libmodbus.a
libtool: finish: PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin:/sbin" ldconfig -n /usr/local/lib
----------------------------------------------------------------------
Libraries have been installed in:
   /usr/local/lib

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the '-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the 'LD_LIBRARY_PATH' environment variable
     during execution
   - add LIBDIR to the 'LD_RUN_PATH' environment variable
     during linking
   - use the '-Wl,-rpath -Wl,LIBDIR' linker flag
   - have your system administrator add LIBDIR to '/etc/ld.so.conf'

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.
----------------------------------------------------------------------
 /usr/bin/mkdir -p '/usr/local/include/modbus'
 /usr/bin/install -c -m 644 modbus.h modbus-config.h modbus-version.h modbus-rtu.h modbus-tcp.h modbus-rtu-usb.h '/usr/local/include/modbus'
Making install in tests
make[2]: Er hoeft niets gedaan te worden voor 'install-exec-am'.
make[2]: Er hoeft niets gedaan te worden voor 'install-data-am'.
make[2]: Er hoeft niets gedaan te worden voor 'install-exec-am'.
 /usr/bin/mkdir -p '/usr/local/share/doc/libmodbus'
 /usr/bin/install -c -m 644 AUTHORS NEWS README.md '/usr/local/share/doc/libmodbus'
 /usr/bin/mkdir -p '/usr/local/lib/pkgconfig'
 /usr/bin/install -c -m 644 libmodbus.pc '/usr/local/lib/pkgconfig'
erik@MinipcLG2:~/libmodbus$ cd ..
erik@MinipcLG2:~$ cd nut
erik@MinipcLG2:~/nut$ cat include/config.h | grep NUT_MODBUS_HAS_USB
/* #undef NUT_MODBUS_HAS_USB */
```

So the new `sudo make install` of libmodbus didn't change anything...

ErikDB87, 2024-11-11:
> And ultimately, for some reason NUT_MODBUS_HAS_USB was not defined (as 1) in the latest build...

Can it be manually defined as `1` in `~/nut/include/config.h`?

jimklimov, 2024-11-11:
Well, after you build and `sudo make install` the dependency, such as libmodbus, you need to again `configure` NUT so it detects the build requirements and circumstances and configures what can be built for them.

[148 more replies]

PR #2750 (2025-01-08): When starting a driver program to force UPS shutdown, kill off running siblings
Closes: #2666