networkupstools/nut #2667: NUT service randomly stops connecting to UPS despite service active, restarting service workaround (Cyberpower CP1500PFCLCD)

chocmake, 2024-10-31:
Twice in the last couple months I happened to notice that the NUT server wasn't resolving info about the UPS when queried, that would be resolved when either the `nut-server` service was restarted or the server rebooted. Below is the most recent experience.

> Note that this may have occurred months earlier too but since nothing occured (no powerouts, nothing set up to push to me as logs) I didn't notice any potential prior events.

SSH'ing into the server I checked the `nut-server` and `nut-monitor` logs and they were reporting that the connection to the UPS was 'unavaiable' / timing out. Both services were 'active' status.

> `journalctl -u nut-server -n 20 -f` output (these messages repeated and latest are from a week ago, for some reason)

    Oct 22 03:48:45 pi upsd[467]: Connected to UPS [cyberpower]: usbhid-ups-cyberpower
    Oct 22 03:49:11 pi upsd[467]: Data for UPS [cyberpower] is stale - check driver
    Oct 22 04:35:04 pi upsd[467]: Send ping to UPS [cyberpower] failed: Resource temporarily unavailable

> `journalctl -u nut-monitor -n 20 -f` output (this message just repeats):

    Oct 30 14:45:59 pi upsmon[478]: UPS cyberpower@localhost is unavailable
    Oct 30 14:48:14 pi upsmon[478]: UPS [cyberpower@localhost]: connect failed: Connection failure: Connection timed out

When I tried to restart `nut-server` via `sudo systemctl restart nut-server` it failed:

    Job for nut-server.service failed because a timeout was exceeded.
    See "systemctl status nut-server.service" and "journalctl -xe" for details.  

> `systemctl status nut-server.service` output:

    ● nut-server.service - Network UPS Tools - power devices information server
         Loaded: loaded (/lib/systemd/system/nut-server.service; enabled; vendor preset: enabled)
         Active: failed (Result: timeout) since <date>; 1min 3s ago
        Process: 115904 ExecStart=/sbin/upsd (code=exited, status=0/SUCCESS)
            CPU: 49ms

    Oct 30 15:22:15 pi upsd[115904]: fopen /run/nut/upsd.pid: No such file or directory
    Oct 30 15:22:15 pi upsd[115904]: listening on 0.0.0.0 port 3493
    Oct 30 15:22:15 pi upsd[115904]: listening on 0.0.0.0 port 3493
    Oct 30 15:23:45 pi systemd[1]: nut-server.service: start operation timed out. Terminating.
    Oct 30 15:23:45 pi upsd[115904]: Can't connect to UPS [cyberpower] (usbhid-ups-cyberpower): Interrupted system call
    Oct 30 15:23:45 pi upsd[115904]: Can't connect to UPS [cyberpower] (usbhid-ups-cyberpower): Interrupted system call
    Oct 30 15:23:45 pi upsd[115918]: Startup successful
    Oct 30 15:23:45 pi upsd[115918]: Signal 15: exiting
    Oct 30 15:23:45 pi systemd[1]: nut-server.service: Failed with result 'timeout'.
    Oct 30 15:23:45 pi systemd[1]: Failed to start Network UPS Tools - power devices information server.

However, immediately upon starting the service using `sudo systemctl start nut-server` the connection to the UPS was logged as being established again (both in the server and client terminal SSH sessions) and everything went back to normal.

---

Environment:

- OS: Raspberry Pi OS (Debian 11/Bullseye)
- NUT: v2.7.4-13 (via apt package install)
- Server HW: Raspberry Pi Zero 2 W
- UPS: Cyberpower CP1500PFCLCD

> Should be noted nothing else is running on the Pi server beside NUT

---

`/etc/nut/ups.conf`

```
maxretry = 3
pollinterval = 2

[cyberpower]
    desc = "Cyberpower CP1500PFCLCD"
    driver = "usbhid-ups"
    port = "auto"
    vendorid = "0764"
    productid = "0501"
    product = "CRJB103.551"
    serial = "CPS"
    vendor = "CP1500EPFCLCD"
    bus = "001"
    offdelay = 120
    ondelay = 0
```

desertwitch, 2024-10-31:
NUT 2.7.4 is a couple of years old now, do you have any chance to try a newer version?
There have been numerous improvements in all areas since then, so it might be worth a shot.

jimklimov, 2024-11-01:
Well, 2.7.4 is actually close to 8.5 years old now (Mar 2016).

As for building, check also https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests - it lists dependencies/tools as well as the methodology; current recipes have a good chance to inherit build settings detected from the older installation to become a sort of in-place ad-hoc replacement.

With CPS, it may also help to increase the polling rate - their controllers apparently go into power-saving or something, if poked only every half a minute (default).

* https://github.com/networkupstools/nut/wiki/Troubleshooting-eventual-disconnections-(Data-stale)
* https://github.com/networkupstools/nut/wiki/CyberPower-Systems-(CPS)-know-how

chocmake, 2024-11-02:
Thanks. My `pollinterval` is already rather low (2 seconds), so upped my `maxretry` to `5` per the Arch wiki suggestion. I've also implemented the usb_resetter suggestion and I'll see how it fares over the coming month.

Otherwise I'll probably have to try the source route.

MilhouseVH, 2024-12-08:
I have this same UPS and also have been running NUT 2.7.4 on RPi4 and can confirm the same behaviour - it's an issue with the Cyberpower UPS resetting periodically (randomly) and no amount of tweaking with NUT config will resolve the issue.

The only solution with 2.7.4 is to restart the `nut-driver` service whenever a `COMMBAD` notification is received.

I did do some testing with a self-build of NUT 2.8.0 maybe a year or so ago, and that seemed to have resolved the issue - I believe the updated `libusb` driver is more robust and tolerant of this kind of situation. 👍

I've just installed a self-build of NUR 2.8.2 so will see how that goes - unfortunately the UPS disappearing could take anything from a few days to a couple of months...

chocmake, 2025-02-07:
So for the past month I've been testing with Pi OS based on Debian Bookworm instead, which has the NUT 2.8.0-7 package available (was more straightforward to swap out an SD card to test than attempting to compile NUT on the old OS).

Using 2.8.0 and the exact same config restored from my 2.7.4 Debian Bullseye install I haven't experienced any dropouts so far (or fwiw any issues with apt that I'd read about for Bullseye on a Pi Zero 2 W). Whereas I was previously getting dropouts almost exactly monthly on 2.7.4 once I began noticing it.

So yeah, looking positive.

chocmake, 2025-04-09:
Been a few more months now without connection drops on the v2.8.0-7 Debian package so figured I'd close the issue given it seems to only be an issue on 2.7.4 packages.