networkupstools/nut #2668: PowerWalker VI 1000 LCD fails to show load when connected to Home Assistant NUT Add-On

oywino, 2024-11-02:
It shows everything else, except load.
But when the same UPS is connected to a QNAP NAS, then it also shows load, so obviously the UPS itself delivers that value.
Here are my details:
[image]

```
users:
  - username: xxxxxxxx
    password: yyyyyyyy
    instcmds:
      - all
    actions: []
devices:
  - name: myups
    driver: nutdrv_qx
    port: auto
    config:
      - vendorid = "0665"
      - productid = "5161"
mode: netserver
shutdown_host: false
log_level: debug
i_like_to_be_pwned: false
leave_front_door_open: false
list_usb_devices: true
```
And here's the log:

```
-----------------------------------------------------------
 Add-on: Network UPS Tools
 Manage battery backup (UPS) devices
-----------------------------------------------------------
 Add-on version: 0.13.2
 You are running the latest version of this add-on.
 System: Home Assistant OS 13.2  (aarch64 / yellow)
 Home Assistant Core: 2024.10.4
 Home Assistant Supervisor: 2024.10.3
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[22:17:36] INFO: Configuring timezone (Europe/Oslo)...
Log level is set to DEBUG
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nut.sh
[22:17:37] INFO: Setting mode to netserver...
[22:17:38] INFO: Connected USB devices:
Bus 001 Device 100: ID 0665:5161 Cypress Semiconductor USB to Serial
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
[22:17:38] INFO: Generating /etc/nut/upsd.users...
[22:17:38] INFO: Configuring user: admin
[22:17:40] INFO: Configuring Device named myups...
[22:17:40] INFO: Starting the UPS drivers...
   0.000000	[D1] Starting UPS: myups
Using protocol: Mustek 0.07
No values for battery high/low voltages
Using 'guesstimation' (low: 20.800000, high: 26.000000)!
Battery runtime will not be calculated (runtimecal not set)
Network UPS Tools - UPS driver controller 2.8.0
cont-init: info: /etc/cont-init.d/nut.sh exited 0
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun upsd (no readiness notification)
services-up: info: copying legacy longrun upsmon (no readiness notification)
s6-rc: info: service legacy-services successfully started
[22:17:43] INFO: Starting the UPS information server...
   0.000000	fopen /run/nut/upsd.pid: No such file or directory
   0.000065	Could not find PID file '/run/nut/upsd.pid' to see if previous upsd instance is already running!

   0.000616	[D1] debug level is '1'
   0.000823	listening on 0.0.0.0 port 3493
   0.001367	Connected to UPS [myups]: nutdrv_qx-myups
   0.002925	Running as foreground process, not saving a PID file
[22:17:44] INFO: Starting the UPS monitor and shutdown controller...
   0.000000	fopen /run/nut/upsmon.pid: No such file or directory
   0.000062	Could not find PID file to see if previous upsmon instance is already running!

   0.000403	Using power down flag file /etc/killpower
   0.000595	UPS: myups@localhost (primary) (power value 1)
   0.000638	[D1] debug level is '1'
   0.000720	[D1] debug level is '1'
   0.000731	Warning: running as one big root process by request (upsmon -p)
   0.000933	[D1] Saving PID 455 into /run/nut/upsmon.pid
Init SSL without certificate database
   0.005550	[D1] Trying to connect to UPS [myups@localhost]
   0.523901	User upsmonmaster@127.0.0.1 logged into UPS [myups]
   0.007280	[D1] Logged into UPS myups@localhost
```

I have tried the usbhid-ups and blazer_usb drivers. With the usbhid-ups the NUT Add-On crashes and the blazer_usb driver gives exactly the same result as the nutdrv_qx.
I have also tried adding a pollinterval but any value result in a never ending connection loop.

Why is my QNAP NAS built in OS NUT Driver able to fetch load data while the Home Assistant NUT Add-on isn't?
I can't think of any more parameters to tweak. Please help.
My HW is Home Assistant Yellow running HAOS 2024-10-4 with the latest NUT add-on and integration.

jimklimov, 2024-11-03:
I am not sure which version of NUY is used by QNAP, but here "2.8.0" is mentioned. There was some development since, in nutdev_qx and blazer drivers, IIRC, that could have them show more data in 2.8.2 (latest NUT release).

Wiki here has a community article about building a custom HA plugin (stock one takes a package from debian-ish distro, and those lag behind - especially stable repos, by design).

jimklimov, 2024-11-04:
Can't speak to that. Maybe @aquette knows more, might have tinkered with HA?..